arendst/Tasmota #14264: External URL call Websend failed

xheen908, 2022-01-04:
### PROBLEM DESCRIPTION
I want to call an external url via Websend
http://192.168.1.7/web/powerstat?newstat=0
If i CMD: Websend [192.168.1.7:80] /web/powerstat?newstat=0

Nothing Happens

A Websend to another tasmota working without any Problem

### REQUESTED INFORMATION
Is it a Bug or a Feature?

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): wemos d1 mini
- [ x] Tasmota binary firmware version number used: tasmota-sensors.bin
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

05:07:31.843 MQT: stat/tasmota_451FC2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
05:07:32.078 MQT: stat/tasmota_451FC2/RESULT = {"Module":{"0":"Generic"}}
05:07:32.332 MQT: stat/tasmota_451FC2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"224":"Relay1"},"GPIO2":{"0":"None"},"GPIO3":{"225":"Relay2"},"GPIO4":{"0":"None"},"GPIO5":{"1120":"RFSend"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"227":"Relay4"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"226":"Relay3"},"GPIO17":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
05:09:41.762 MQT: stat/tasmota_451FC2/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":356,"Free":155,"Rules":"on Power1#State=1 do RFsend {\"Data\":\"0x154551\",\"Bits\":24,\"Protocol\":1,\"Pulse\":309} endon on Power1#State=0 do RFsend {\"Data\":\"0x154554\",\"Bits\":24,\"Protocol\":1,\"Pulse\":309} endon on Power4#State=1 do RFsend {\"Data\":\"12976816\",\"Bits\":24,\"Protocol\":5,\"Pulse\":508} endon on Power4#State=0 do RFsend {\"Data\":\"13117328\",\"Bits\":24,\"Protocol\":5,\"Pulse\":508} endon "}}
05:09:42.000 MQT: stat/tasmota_451FC2/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":177,"Free":334,"Rules":"on Power2#State=1 do RFsend {\"Data\":\"0x155151\",\"Bits\":24,\"Protocol\":1,\"Pulse\":309} endon on Power2#State=0 do RFsend {\"Data\":\"0x155154\",\"Bits\":24,\"Protocol\":1,\"Pulse\":309} endon"}}
05:09:42.249 MQT: stat/tasmota_451FC2/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":177,"Free":334,"Rules":"on Power3#State=1 do RFsend {\"Data\":\"0x155451\",\"Bits\":24,\"Protocol\":1,\"Pulse\":309} endon on Power3#State=0 do RFsend {\"Data\":\"0x155454\",\"Bits\":24,\"Protocol\":1,\"Pulse\":309} endon"}}

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
05:11:09.029 MQT: stat/tasmota_451FC2/STATUS = {"Status":{"Module":0,"DeviceName":"RF Bridge","FriendlyName":["Stehlampe","Globus","Schrank","Kirche"],"Topic":"tasmota_451FC2","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
05:11:09.061 MQT: stat/tasmota_451FC2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:42:35","StartupUTC":"2022-01-04T01:28:34","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2022-01-03T04:22:03","SaveCount":953,"SaveAddress":"FB000"}}
05:11:09.093 MQT: stat/tasmota_451FC2/STATUS2 = {"StatusFWR":{"Version":"10.1.0(sensors)","BuildDateTime":"2021-12-08T14:47:41","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
05:11:09.124 MQT: stat/tasmota_451FC2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FB 7362 SL",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00000000"]}}
05:11:09.166 MQT: stat/tasmota_451FC2/STATUS4 = {"StatusMEM":{"ProgramSize":668,"Free":332,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
05:11:09.202 MQT: stat/tasmota_451FC2/STATUS5 = {"StatusNET":{"Hostname":"RF-Bridge","IPAddress":"192.168.1.19","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"B4:E6:2D:45:1F:C2","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
05:11:09.233 MQT: stat/tasmota_451FC2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.23","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_451FC2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
05:11:09.266 MQT: stat/tasmota_451FC2/STATUS7 = {"StatusTIM":{"UTC":"2022-01-04T04:11:09","Local":"2022-01-04T05:11:09","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:06"}}
05:11:09.298 MQT: stat/tasmota_451FC2/STATUS10 = {"StatusSNS":{"Time":"2022-01-04T05:11:09"}}
05:11:09.329 MQT: stat/tasmota_451FC2/STATUS11 = {"StatusSTS":{"Time":"2022-01-04T05:11:09","Uptime":"0T02:42:35","UptimeSec":9755,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"FB 7362 SL","BSSId":"2C:3A:FD:22:C2:5E","Channel":9,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Look description

### EXPECTED BEHAVIOUR
Seccussfull API call to external API 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I testet around Note then 6 hours.
I ASK Dome Buddy. Same Problem there

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-04:
The log suggest the WebSend has worked as expected without error
As we don't know what is behind this request, it is difficult to suggest what could go wrong
What is the receiving device ?
Are you sure that the receiving device is ok with a GET request ? (doesn't requires a POST)
Are you sure it doesn't requires https ?

xheen908, 2022-01-04:
there is no bug my bad  :D had authentication problem ... closed

xheen908, 2022-01-04:
but btw? how to cal post request from websend ?

closed 2022-01-04 by xheen908