arendst/Tasmota #14265: LilyGO T-Journal Camera fails initialization with development branch

PinkFreud, 2022-01-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
LilyGO T-Journal camera init fails on 10.1.0 with:
```
CAM: PSRAM not found
CAM: Init failed with error 0xffffffff
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)   <-- No discord account, no interest in creating one
- [x] Device used (e.g., Sonoff Basic): LilyGO T-Journal
- [x] Tasmota binary firmware version number used: 10.1.0
  - [ ] Pre-compiled
  - [x] Self-compiled: Using https://github.com/benzino77/tasmocompiler
- [x] Flashing tools used: OTA upgrade from 9.2.0.2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:47:47.503 CMD: Backlog Template; Module; GPIO 255
23:47:47.505 SRC: WebConsole from 192.168.1.73
23:47:47.507 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
23:47:47.539 SRC: Backlog
23:47:47.541 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
23:47:47.544 RSL: RESULT = {"NAME":"T-Journal v1.8","GPIO":[4928,0,1,0,1,5091,0,0,0,608,640,4960,0,5088,5093,5095,0,5184,5120,5056,0,5024,5152,4992,0,0,0,0,32,1,5090,5089,5094,0,0,5092],"FLAG":0,"BASE":1}
23:47:47.767 SRC: Backlog
23:47:47.768 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
23:47:47.770 RSL: RESULT = {"Module":{"0":"T-Journal v1.8"}}
23:47:48.017 SRC: Backlog
23:47:48.018 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
23:47:48.029 RSL: RESULT = {"GPIO0":{"4928":"CAM_PWDN"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"5091":"CAM_DATA4"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"608":"I2C SCL1"},"GPIO14":{"640":"I2C SDA1"},"GPIO15":{"4960":"CAM_RESET"},"GPIO16":{"0":"None"},"GPIO17":{"5088":"CAM_DATA1"},"GPIO18":{"5093":"CAM_DATA6"},"GPIO19":{"5095":"CAM_DATA8"},"GPIO20":{"0":"None"},"GPIO21":{"5184":"CAM_PCLK"},"GPIO22":{"5120":"CAM_VSYNC"},"GPIO23":{"5056":"CAM_SIOC"},"GPIO24":{"0":"None"},"GPIO25":{"5024":"CAM_SIOD"},"GPIO26":{"5152":"CAM_HREF"},"GPIO27":{"4992":"CAM_XCLK"},"GPIO32":{"32":"Button1"},"GPIO33":{"544":"LedLink"},"GPIO34":{"5090":"CAM_DATA3"},"GPIO35":{"5089":"CAM_DATA2"},"GPIO36":{"5094":"CAM_DATA7"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5092":"CAM_DATA5"}}
```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
* No rules in use

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:45:31.330 CMD: status 0
23:45:31.332 SRC: WebConsole from 192.168.1.73
23:45:31.334 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
23:45:31.337 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"espcam05","FriendlyName":["espcam05"],"Topic":"tasmota_2F99A8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:45:31.375 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:22","StartupUTC":"2022-01-04T04:43:09","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-12-27T22:23:46","SaveCount":16}}
23:45:31.402 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(TasmoCompiler-esp32webcam)","BuildDateTime":"2022-01-02T19:01:48","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"484/699"}}
23:45:31.427 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mirkwood iDevNN 2","HomeDev"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","000000C0","00006000","00004000"]}}
23:45:31.454 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1375,"Free":1984,"Heap":146,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FF0C58E","00041FA1","680A37C3","182C0000","503A0981","800938E8","003E8000","00000048"],"Drivers":"1,2,4,7,8,9,10,12,13,24,26,27,29,38,50,52,81","Sensors":"1,2,5,6,7,8,9,10,11,14,16,24,26,27,38,39,41,46,47,55,57,58,59,63,65,67,70,71,72,73,74,88,127"}}
23:45:31.491 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-2F99A8-6568","IPAddress":"172.18.4.25","Gateway":"172.18.4.1","Subnetmask":"255.255.252.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"24:0A:C4:2F:99:A8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:45:31.515 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2F99A8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:45:31.540 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-04T04:45:31","Local":"2022-01-03T23:45:31","StartDST":"2022-03-13T02:00:00","EndDST":"2022-11-06T03:00:00","Timezone":99,"Sunrise":"07:21","Sunset":"16:45"}}
23:45:31.559 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-03T23:45:31","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
23:45:31.563 BRY: GC from 6693 to 3497 bytes, objects freed 11/42 (in 0 ms)
23:45:31.567 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-03T23:45:31","Uptime":"0T00:02:22","UptimeSec":142,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":42},"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mirkwood iDevNN 2","BSSId":"12:02:8E:94:72:56","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
23:43:43.460 CMD: weblog
23:43:43.462 SRC: WebConsole from 192.168.1.73
23:43:43.463 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 0, Data ''
23:43:43.466 RSL: RESULT = {"WebLog":4}
23:43:52.989 HTP: Main Menu
23:43:53.057 CAM: Stream init
23:43:53.060 CAM: User template
23:43:53.062 CAM: PSRAM not found
23:43:53.210 CAM: Init failed with error 0xffffffff
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile 10.1.0 from development branch, using ESP32 Webcam as base and Display as option module (T-Journal includes SSD1306 display).  Flash to LilyGO T-Journal board, and configure template. (I also have a LilyGO T-Camera, which works properly with the same compiled 10.1.0 image and a proper template for this particular board).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Camera initializes as normal with gpios configured correctly.  The same template (see: https://github.com/LilyGO/ESP32-Camera/blob/master/images/TTGOcam.jpg - D2 == Cam Data 1, D3 == Cam Data 2, ... D9 == Cam Data 8) works with 9.2.0.2, compiled last year for this same board.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
As mentioned above, this previously worked on 9.2.0.2.  I haven't tried any releases between then and the development branch showing version 10.1.0, so I can't pinpoint when the camera module stopped working on here.  Re-flashing 9.2.0.2 lets the camera initialize again, which tells me that neither the camera nor board are faulty, and the camera doesn't need to be reseated (though I did try this!).

**(Please, remember to close the issue when the problem has been addressed)**

PinkFreud, 2022-01-04:
It looks like this board may not have PSRAM, so it seems doubtful that message, above, is part of the error.  Setting weblog / seriallog 4 on 9.2.0.2 and initializing the camera works:
```
00:11:56 CMD: weblog 4
00:11:56 SRC: WebConsole from 192.168.1.73
00:11:56 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
00:11:56 RSL: stat/tasmota_2F99A8/RESULT = {"WebLog":4}
00:11:59 CMD: seriallog 4
00:11:59 SRC: WebConsole from 192.168.1.73
00:11:59 CMD: Group 0, Index 1, Command "SERIALLOG", Data "4"
00:11:59 RSL: stat/tasmota_2F99A8/RESULT = {"SerialLog":{"4":{"Active":"4"}}}
00:12:03 CMD: wcinit
00:12:03 SRC: WebConsole from 192.168.1.73
00:12:03 CMD: Group 0, Index 1, Command "WCINIT", Data ""
00:12:03 CAM: Stream init
00:12:03 CAM: User template
00:12:03 CAM: PSRAM not found
00:12:03 CAM: Initialized
00:12:03 RSL: stat/tasmota_2F99A8/RESULT = {"WCInit":"Done"}
00:12:12 WIF: Checking connection...
00:12:18 HTP: Main Menu
00:12:19 CAM: Handle camserver
00:12:19 CAM: Create client
00:12:19 CAM: Start stream
```

Jason2866, 2022-01-05:
The message
```
CAM: PSRAM not found
CAM: Init failed with error 0xffffffff
```
will not prevent the camera to run. It is a message that no PSRAM is found and therefore  can not be used. The performance of the camera is degraded and higher resolutions are not possible. It is a cheap esp32cam hardware.

PinkFreud, 2022-01-05:
> The message
> 
> ```
> CAM: PSRAM not found
> CAM: Init failed with error 0xffffffff
> ```
> 
> will not prevent the camera to run. It is a message that no PSRAM is found and therefore can not be used. The performance of the camera is degraded and higher resolutions are not possible. It is a cheap esp32cam hardware. The info is added in lately versions of Tasmota.

As I noted in my above response, `PSRAM not found` was merely an indication that the board lacks PSRAM:
> It looks like this board may not have PSRAM, so it seems doubtful that message, above, is part of the error. Setting weblog / seriallog 4 on 9.2.0.2 and initializing the camera works - and indeed, the camera streams successfully:

However, you managed to close this issue without addressing the root cause:
`CAM: Init failed with error 0xffffffff`

That *is* a problem, as it indicates camera initialization failed - and indeed, the camera does not work on this board with Tasmota 10.1.0.  Again, I'll note that the camera initializes successfully with Tasmota 9.2.0.2, and can stream just fine.  With 10.1.0, camera initialization fails, and streaming (or even single frames!) fails to work.  Are you really telling me that's by design?

Jason2866, 2022-01-05:
Ahh, misunderstood this. Thought it is working.

PinkFreud, 2022-01-05:
No problem - I probably could have been clearer.  I initially thought the lack of PSRAM might be part of the error.

Also, note that I previously mentioned I have a different LilyGO board (the T-camera: https://templates.blakadder.com/lilygo_T-Camera.html ) which 10.1.0 can initialize the camera just fine on.  I'm not certain why 10.1.0 works on the T-camera, but fails to initialize the camera on the T-journal.  Both boards are using properly configured templates for the camera pins.

Jason2866, 2022-01-05:
The camera is a bad design. A espcam without PSRAM is crippled -> Design fail

Jason2866, 2022-01-24:
Reason https://github.com/espressif/esp32-camera/issues/334

closed 2022-01-26 by Jason2866

PinkFreud, 2022-01-26:
Ouch, that sucks.  I guess this cam needs to stick w/ an older Tasmota release, then.

Thanks for looking into this.