arendst/Tasmota #14267: time#initialized flag not triggered when DS3231 attached

pkkrusty, 2022-01-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When DS3231 is attached and functioning, the 8266 gets its time from the DS3231 before hitting the NTP server, so force_sync flag is never activated (I think) and the time_init rules flag is never set to 1, so my rule never fires. Line 147 of the DS3231 code maybe should be > vs <? Rule doesn't fire whether I have a valid NTP server or not. Also I think if a user is setting time manually, with the time command, that should trigger a write for a DS3231, if attached. Should be possible to set a RTC module like DS3231 manually, in the absence of an available NTP server, IMO.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Wemos Mini D1 knockoff
- [ x] Tasmota binary firmware version number used: 10-0-0-3
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: esptool.py then OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:43:20.974 CMD: Backlog Template; Module; GPIO 255
08:43:20.976 SRC: WebConsole from 192.168.1.100
08:43:20.979 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
08:43:20.994 SRC: Backlog
08:43:20.997 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
08:43:21.002 MQT: freezer-energy-monitor/stat/RESULT = {"NAME":"HW-655 PZEM","GPIO":[0,1536,0,1472,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
08:43:21.195 SRC: Backlog
08:43:21.198 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
08:43:21.206 MQT: freezer-energy-monitor/stat/RESULT = {"Module":{"18":"Generic"}}
08:43:21.400 SRC: Backlog
08:43:21.403 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
08:43:21.413 MQT: freezer-energy-monitor/stat/RESULT = {"GPIO0":{"258":"Relay_i3"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"259":"Relay_i4"},"GPIO3":{"1536":"PZEM016 Rx"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"162":"Switch3"},"GPIO13":{"257":"Relay_i2"},"GPIO14":{"163":"Switch4"},"GPIO15":{"256":"Relay_i1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:43:59.580 CMD: Backlog Rule1; Rule2; Rule3
08:43:59.582 SRC: WebConsole from 192.168.1.100
08:43:59.585 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
08:43:59.599 SRC: Backlog
08:43:59.601 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
08:43:59.605 RUL: Rule1 On system#init do backlog power1 1;power2 1;var1 1;var2 1 endon On time#initialized>10 do backlog status 10 endon On statussns#switch3=1 do backlog var1 2;event Hr1=%time% endon On statussns#switch4=1 do backlog var2 2;event Hr2=%time% endon On switch3#state=1 do backlog var1 2;event Hr1=%time% endon On switch3#state=0 do backlog var1 1;power1 1 endon On switch4#state=1 do backlog var2 2;event Hr2=%time% endon On switch4#state=0 do backlog var2 1;power2 1 endon On event#Hr1<1140 do backlog var1 1;var2 1
08:43:59.607 RUL: Rule1 +endon On event#Hr2<1140 do backlog var1 1;var2 1 endon On event#Hr1<480 do backlog delay 100;var1 2 endon On event#Hr2<480 do backlog delay 100;var2 2 endon On time#minute=479 do backlog power1 1;power2 1;var1 1;var2 1 endon On time#minute=1200 do backlog status 10 endon On time#minute|60 do backlog power1 %var1%;ruletimer1 3595 endon On rules#timer=1 do power2 %var2% endon
08:43:59.616 MQT: freezer-energy-monitor/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":885,"Free":206,"Rules":"On system#init do backlog power1 1;power2 1;var1 1;var2 1 endon On time#initialized>10 do backlog status 10 endon On statussns#switch3=1 do backlog var1 2;event Hr1=%time% endon On statussns#switch4=1 do backlog var2 2;event Hr2=%time% endon On switch3#state=1 do backlog var1 2;event Hr1=%time% endon On switch3#state=0 do backlog var1 1;power1 1 endon On switch4#state=1 do backlog var2 2;event Hr2=%time% endon On switch4#state=0 do backlog var2 1;power2 1 endon On event#Hr1<1140 do backlog var1 1;var2 1 endon On event#Hr2<1140 do backlog var1 1;var2 1 endon On event#Hr1<480 do backlog delay 100;var1 2 endon On event#Hr2<480 do backlog delay 100;var2 2 endon On time#minute=479 do backlog power1 1;power2 1;var1 1;var2 1 endon On time#minute=1200 do backlog status 10 endon On time#minute|60 do backlog power1 %var1%;ruletimer1 3595 endon On rules#timer=1 do power2 %var2% endon"}}
08:43:59.803 SRC: Backlog
08:43:59.806 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
08:43:59.813 MQT: freezer-energy-monitor/stat/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":123,"Free":388,"Rules":"on time#minute|5 do ruletimer2 10 endon on rules#timer=2 do publish topic/stat/test {\"Var1\":\"%var1%\",\"Var2\":\"%var2%\"} endon"}}
08:44:00.007 SRC: Backlog
08:44:00.010 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
08:44:00.017 MQT: freezer-energy-monitor/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:44:53.950 MQT: freezer-energy-monitor/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Freezer Energy Monitor","FriendlyName":["Switch 1","Switch 2","",""],"Topic":"freezer-energy-monitor","ButtonTopic":"0","Power":11,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,1,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:44:54.012 MQT: freezer-energy-monitor/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:11:32","StartupUTC":"2022-01-04T08:33:22","Sleep":50,"CfgHolder":4642,"BootCount":21,"BCResetTime":"2021-11-18T16:45:08","SaveCount":67,"SaveAddress":"FD000"}}
08:44:54.073 MQT: freezer-energy-monitor/stat/STATUS2 = {"StatusFWR":{"Version":"10.0.0.3(tasmota)","BuildDateTime":"2022-01-03T20:24:29","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"486/699"}}
08:44:54.131 MQT: freezer-energy-monitor/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":1,"SysLog":3,"LogHost":"192.168.1.14","LogPort":514,"SSId":["beersheba","Roatan"],"TelePeriod":900,"Resolution":"55CB88C0","SetOption":["00008008","2805C80001000600003C5A00000000000000","00408089","08006000","00004000"]}}
08:44:54.195 MQT: freezer-energy-monitor/stat/STATUS4 = {"StatusMEM":{"ProgramSize":639,"Free":1408,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FE0078E","04100001","0003904F","000016C0","40400810","020040A0","40200000","00000A00"],"Drivers":"1,2,3,4,8,9,10,12,13,20,21,29,38,50,57,59","Sensors":"1,2,3,5,10,12,13,14,33,54,60,77"}}
08:44:54.261 MQT: freezer-energy-monitor/stat/STATUS5 = {"StatusNET":{"Hostname":"freezer-energy-monitor","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:AF:9F:0E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:44:54.318 MQT: freezer-energy-monitor/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.14","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF9F0E","MqttUser":"pkkrusty","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:44:54.378 MQT: freezer-energy-monitor/stat/STATUS7 = {"StatusTIM":{"UTC":"2022-01-04T08:44:54","Local":"2022-01-04T08:44:54","StartDST":"2022-03-27T00:00:00","EndDST":"2022-10-30T00:00:00","Timezone":"+00:00","Sunrise":"07:32","Sunset":"18: [... cut ...]
_Steps to reproduce the behavior:_
Add a DS3231 and try to get time#initialized triggered

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
time#initialized triggered once rtc.utc_time is set by either ntp or ds3231 (or ideally the time command)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pkkrusty, 2022-01-05:
Added `TasmotaGlobal.rules_flag.time_init = 1;` after line 492 in support_rtc.ino to trigger time#initialized when time is entered manually. Not sure if that's acceptable to the broader community but it makes sense to me, if someone is purposely not using NTP, that time is initialized once it's entered from an alternative source (in this case, my fingers on the keyboard). Once I get some feedback from higher-ups, I can submit a PR.

pkkrusty, 2022-01-05:
Changed line 139 in xsns_33_ds3231.ino from if `(Rtc.utc_time < START_VALID_TIME ) {` to `if (Rtc.utc_time > START_VALID_TIME ) {` and line 147 from `if (Rtc.local_time < START_VALID_TIME) {` to `if (Rtc.utc_time > START_VALID_TIME) {`
Changed line 407 in support_rtc.ino from `if (Rtc.local_time < START_VALID_TIME) {` to `if (Rtc.utc_time > START_VALID_TIME) {`
and now things work as they should. Not sure if there are further effects that I'm not seeing, but I'll submit a PR and let the pros figure it out.

pkkrusty, 2022-01-06:
Looking into the code a little more, seems like there's a conflict between the logic of DS3231 driver and rtc_support driver. The DS3231 sets rtc.utc_time before wifi starts its actions, and somehow rtc.local_time gets set in the process. (Haven't figured out where yet) Then the very first call of rtcsync() happens if NTP server is available, which calls the first instance of rtcsecond(), which without a DS3231, would have rtc.local_time still at 0, which allows the initialized flag to fire. Subsequent function calls would only fire the #set flag on the hour.  But with a DS3231, rtc.local_time is already valid, so neither the DS3231 code or rtc_support code ever fire the initialized flag.

pkkrusty, 2022-01-07:
Latest work: If I flash a lite build, without tasmesh, influxdb, sunrise, ping, expression, if_statement, and most sensors disabled, time#initialized>10 triggers. so does >50. And time#initialized fires at around 130ms.

However, I note that 120 does not equal 120 seconds as indicated by documentation. Makes no difference what number I put in, it still triggers about 20ms after the initial trigger.

```
00:00:00.001 HDW: ESP8266EX
00:00:00.020 UFS: FlashFS mounted with 1992 kB free
00:00:00.095 CFG: Loaded from File, Count 26
00:00:00.101 QPC: Count 1
00:00:00.103 CFG: CR 458/699, Busy 0
00:00:00.107 SRC: Restart
00:00:00.109 Project pkkrusty - Tasmota Version 10.0.0.3(tasmota)-2_7_4_9(2022-01-07T15:03:18)
00:00:00.111 I2C: DS3231 found at 0x68
00:00:00.131 Set time from DS3231 to RTC (UTC) 2022-01-07T15:33:15, (DST) 2022-03-27T00:00:00, (STD) 2022-10-30T00:00:00
00:00:00.131 RTC: Rtc.utc_time set correctly by DS3231
00:00:00.132 Rtc.utc_time 1641569595
00:00:00.132 Rtc.local_time 0
00:00:00.218 WIF: Attempting connection...
00:00:00.555 WIF: Connecting to AP2 beersheba Channel 6 BSSId D8:0D:17:B3:A6:E6 in mode 11n as pkkrusty-AF8D12-3346...
00:00:00.001 RTC: Not synced
15:33:16.010 RUL: TIME#INITIALIZED performs "power1 1"
15:33:16.011 SRC: Rule
15:33:16.012 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 1, Data '1'
15:33:16.015 RSL: RESULT = {"POWER":"ON"}
15:33:16.016 RSL: POWER = ON
15:33:16.027 RUL: TIME#INITIALIZED>111 performs "power1 1"
15:33:16.028 SRC: Rule
15:33:16.029 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 1, Data '1'
15:33:16.032 RSL: RESULT = {"POWER":"ON"}
15:33:16.033 RSL: POWER = ON
15:33:16.044 RUL: TIME#INITIALIZED>120 performs "power1 1"
15:33:16.045 SRC: Rule
15:33:16.046 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 1, Data '1'
15:33:16.049 RSL: RESULT = {"POWER":"ON"}
15:33:16.050 RSL: POWER = ON
```

If I remove the DS3231 and just use NTP servers, same issue.
```
00:00:00.001 HDW: ESP8266EX
00:00:00.022 UFS: FlashFS mounted with 1992 kB free
00:00:00.100 CFG: Loaded from File, Count 29
00:00:00.105 QPC: Count 1
00:00:00.107 CFG: CR 458/699, Busy 0
00:00:00.111 SRC: Restart
00:00:00.113 Project pkkrusty - Tasmota Version 10.0.0.3(tasmota)-2_7_4_9(2022-01-07T15:03:18)
00:00:00.221 WIF: Attempting connection...
00:00:00.558 WIF: Connecting to AP2 beersheba Channel 6 BSSId D8:0D:17:B3:A6:E6 in mode 11n as pkkrusty-AF8D12-3346...
00:00:01.753 WIF: Connected
00:00:02.005 HTP: Web server active on pkkrusty-AF8D12-3346 with IP address 192.168.1.50
00:00:02.006 UPP: Multicast (re)joined
00:00:02.312 HTP: Console
00:00:02.438 NTP: Synch time...
00:00:02.551 NTP: Runtime 0
00:00:02.551 Rtc.utc_time 1641569930
00:00:02.552 Rtc.local_time 2
00:00:02.554 RTC: UTC 2022-01-07T15:38:50, DST 2022-03-27T00:00:00, STD 2022-10-30T00:00:00
00:00:02.554 support_rtc function running correctly, rtc.utc_time set
00:00:02.555 Rtc.utc_time 1641569930
00:00:02.556 Rtc.local_time 2
15:38:50.000 NTP: Synched
15:38:50.035 RUL: TIME#INITIALIZED performs "power1 1"
15:38:50.036 SRC: Rule
15:38:50.037 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 1, Data '1'
15:38:50.040 RSL: RESULT = {"POWER":"ON"}
15:38:50.041 RSL: POWER = ON
15:38:50.054 RUL: TIME#INITIALIZED>10 performs "power1 1"
15:38:50.055 SRC: Rule
15:38:50.056 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 1, Data '1'
15:38:50.059 RSL: RESULT = {"POWER":"ON"}
15:38:50.061 RSL: POWER = ON
15:38:50.070 RUL: TIME#INITIALIZED>120 performs "power1 1"
15:38:50.071 SRC: Rule
15:38:50.072 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 1, Data '1'
15:38:50.075 RSL: RESULT = {"POWER":"ON"}
15:38:50.077 RSL: POWER = ON
```

Jason2866, 2022-01-07:
If `sleep` is activated it is normal that the MCU reacts not so fast.
Reducing the drivers lowers the CPU load. So the behaviour you see is normal and as expected.

pkkrusty, 2022-01-07:
So documentation is not accurate? The > symbol doesn't seem to do anything. Load average for even the big firmware is only 19, leaving plenty of overhead in the sleep cycle. Seems like > symbol in that trigger is just getting executed on the next loop.

Jason2866, 2022-01-07:
A low LoadAVG does not mean the device will react as fast as when sleep is switched off.

pkkrusty, 2022-01-07:
Makes sense. But > modifier still doesn't have an effect even on a lightly loaded processor.

sfromis, 2022-01-08:
Do not assume that LoadAvg is a simple measure of how busy it is. It can even get above 100, depending on circumstances.

pkkrusty, 2022-01-09:
@sfromis any way to get any kind of real-time feedback on load? I'm curious what levels it hits with various builds. For example, things definitely feel slower on the WebUI when I have Tasmesh included.

sfromis, 2022-01-09:
Not really. And even having the LoadAvg every TelePeriod (not available faster, unless you fire off lots of commands to query state all the time), you may be assuming too much importance of that number.

While I've not at all looked into the Tasmesh driver, on a general level timing-depend things doing blocking delays can really mess with how well Tasmota operates, as techniques quite viable for the MCU only having one thing to do may just not work when it has to relinquish control to other functions having urgent needs of being allowed to run.

I'm not surprised if mixing types of network communication at the same time may not be snappy, as there is no "real multitasking".

stefanbode, 2022-01-10:
Looks for me also as a bug. Maybe you can output the value of time initialized when the rule gets executed. It looks like this value is much higher than the 10 or 120. this would explain why all the triggers get executed immediately.

pkkrusty, 2022-01-10:
I added some debugging when I thought it was a problem with the ds3231 driver, and the two key values are rtc.local_time and rtc.utc_time being less than and greater than 2016-01-01 in epoch time, respectively. That lets an if statement trigger the flag. But I can't figure out the rules code that deals with comparators to see what the >10 or >120 is comparing against. Epoch time? Milliseconds since boot?

pkkrusty, 2022-01-10:
Ah i didn't initially understand your suggestion, but now I do. I published %value% and sure enough, it looks like milliseconds from boot. I can work with that, but either documentation or implementation should get fixed. Votes?

arendst, 2022-01-10:
Never the implementation as it has been working for many people for many years.

I haven't read this whole thread but from the beginning I understand the supposed rule ``time#initialized>10`` was never meant to be used like this. Only ``time#initialized`` is supposed to be used and occurs when the RTC code is initialised with some kind of time but not a time after 2016.

Looking at the code you'll see what happens: 
```
    if (Rtc.local_time < START_VALID_TIME) {  // 2016-01-01
      TasmotaGlobal.rules_flag.time_init = 1;
    } else {
      TasmotaGlobal.rules_flag.time_set = 1;
    }
```

The goal of this code is twofold:
- Initial call to master time interrupt handler has been done (time#initialized).
- Sync calls to master time handler have been done (time#set).

Again see code below:
```
            case 2: snprintf_P(json_event, sizeof(json_event), PSTR("{\"Time\":{\"Initialized\":%d}}"), MinutesPastMidnight()); break;
            case 3: snprintf_P(json_event, sizeof(json_event), PSTR("{\"Time\":{\"Set\":%d}}"), MinutesPastMidnight()); break;
 ```

I suggest you rather learn to read the code than keep suggesting solutions based on guess work to eleviate long discussions like this.

pkkrusty, 2022-01-10:
Based on my last test, time#initialized value is milliseconds since boot. Thus the comparator will work, but not as described in documentation. I'm happy to fix that. 

I'm also happy to leave my issues threads just stating the problem, but wanted to fill in readers with steps I took to narrow it down, since often problems are user error rather than coding or documentation. Reading code is like learning a language, it takes time. I'm currently at the 2-year-old copycat stage.

arendst, 2022-01-10:
I totally understand but...

From the code you can clearly see it's NOT milliseconds after a reboot BUT it's minutes past midnight.

Feel free to update any documentation with valid information.

stefanbode, 2022-01-10:
To overcome your problem just use time#initialized and fire a 10 second timer. The the timer will finally execute your script. I think this should work

Jason2866, 2022-01-10:
Closing since it is answered. As designed

closed 2022-01-10 by Jason2866

pkkrusty, 2022-01-10:
Question: Has anyone ever had success with Time#Initialized>10 (or any number) trigger? Even when I evaluate it as MinutesPastMidnight, it doesn't trigger when the threshhold is reached. Maybe doesn't belong in the Rules documentation.

sfromis, 2022-01-10:
Did a small test. 
`on Time#Initialized do var1 %value% endon on Time#Initialized>100 do var2 %value% endon`
Seems to work as already suggested here.....
```
18:34:17.020 RUL: TIME#INITIALIZED performs "var1 1114"
18:34:17.022 SRC: Rule
18:34:17.023 CMD: Grp 0, Cmnd 'VAR', Idx 1, Len 4, Data '1114'
18:34:17.028 RSL: VAR = {"Var1":"1114"}
18:34:17.039 RUL: TIME#INITIALIZED>100 performs "var2 1114"
18:34:17.040 SRC: Rule
18:34:17.042 CMD: Grp 0, Cmnd 'VAR', Idx 2, Len 4, Data '1114'
18:34:17.046 RSL: VAR = {"Var2":"1114"}
```
Your phrase `it doesn't trigger when the threshhold is reached` does not seem to suggest you being aware that it is only being evaluated once, and not continuously. For "normal" timers, you may want `Time#Minute>10`