arendst/Tasmota #14281: SONOFF Smart Stackable Power Meter issue with more than 2 relays

fegauthier, 2022-01-04:
### PROBLEM DESCRIPTION
I have a SPM Main with 4 relays connected to it. Tasmota detect the 4 relays (16 lines) fine. I can power toggle and see energy stats of L1 to L8 only. Above L8, there is no stats and power toggle is toggling the second relay (L5, L6, L7, L8)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): SPM Main & SPM Relays
- [ ] Tasmota binary firmware version number used: Latest development build
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: Tasmota Web GUI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

20:58:11.205 CMD: Backlog Template; Module; GPIO 255
20:58:11.208 SRC: WebConsole from 192.168.10.19
20:58:11.211 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
20:58:11.266 SRC: Backlog
20:58:11.269 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
20:58:11.277 RSL: RESULT = {"NAME":"Sonoff SPM (POC2)","GPIO":[1,0,1,0,3200,5536,0,0,1,1,1,0,3232,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,544,1,1,32,1,0,0,1],"FLAG":0,"BASE":1}
20:58:11.502 SRC: Backlog
20:58:11.505 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
20:58:11.510 RSL: RESULT = {"Module":{"0":"Sonoff SPM (POC2)"}}
20:58:11.753 SRC: Backlog
20:58:11.755 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
20:58:11.773 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"3200":"Serial Tx"},"GPIO5":{"5536":"ETH POWER"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"3232":"Serial Rx"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"544":"LedLink"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"32":"Button1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

20:58:40.463 CMD: Backlog Rule1; Rule2; Rule3
20:58:40.466 SRC: WebConsole from 192.168.10.19
20:58:40.469 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
20:58:40.523 SRC: Backlog
20:58:40.527 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
20:58:40.532 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:58:40.770 SRC: Backlog
20:58:40.773 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
20:58:40.778 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:58:41.020 SRC: Backlog
20:58:41.024 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
20:58:41.029 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

21:00:31.058 CMD: Status 0
21:00:31.061 SRC: WebConsole from 192.168.10.19
21:00:31.065 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
21:00:31.073 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","","","","","","",""],"Topic":"tasmota_C781CC","ButtonTopic":"0","Power":256,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:00:31.117 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota32-ethernet.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:04","StartupUTC":"2022-01-04T20:57:27","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2021-12-18T21:29:03","SaveCount":117}}
21:00:31.160 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0.3(tasmota)","BuildDateTime":"2021-12-29T11:51:05","Core":"2_0_2","SDK":"v4.4-2.0.2","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"411/699"}}
21:00:31.179 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garage",""],"TelePeriod":300,"Resolution":"548B00C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
21:00:31.222 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1360,"Free":1856,"Heap":139,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","0015A001","B7F7BFCF","05DA9BC0","60360DC7","400840D2","20200000","0004482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,38,50,52,59,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
21:00:31.273 BRY: GC from 7443 to 4151 bytes, objects freed 11/35 (in 1 ms)
21:00:31.280 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-C781CC-0460","IPAddress":"192.168.14.87","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer1":"1.1.1.1","DNSServer2":"8.8.8.8","Mac":"94:3C:C6:C7:81:CC","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:00:31.310 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C781CC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:00:31.331 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-04T21:00:31","Local":"2022-01-04T21:00:31","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:06"}}
21:00:31.384 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-04T21:00:31","ESP32":{"Temperature":53.3},"SPM":{"Energy":[0.0,0.1,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Yesterday":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Today":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"ActivePower":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"ApparentPower":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"ReactivePower":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Factor":[0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00],"Voltage":[0.0,0.0,0.0,0.0,237.7,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Current":[0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00]},"TempUnit":"C"}}
21:00:31.475 BRY: GC from 5694 to 3652 bytes, objects freed 5/35 (in 0 ms)
[... 38 lines cut ...]
You need to have more than 2 relays connected to the SPM Main (more than 8 lines). Go to the tasmota web gui. Energy stats doesn't seems to be good for line above L8 (L9, L10, ...).

If you power toggle L9, it will toggle L5
L10 will toggle L6
L11 will toggle L7
L12 will toggle L8

L13 will toggle L5
L14 will toggle L6
L15 will toggle L7
L16 will toggle L8

Etc...

### EXPECTED BEHAVIOUR

When clicking on L9, we expected that L9 power up or down. We expect that energy stats of lines above L8 show good ones.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-01-04:
Pls try latest development version and see if the energy display inthe GUI is updated correctly.

Pls also try to execute command ``power9 2`` in the console and see if releay 9 is toggled or another relay. Same for ``power14 2``.

fegauthier, 2022-01-04:
With the latest Development branch build, the issue is still the same.

`power9 2` toggles Line5

`power14 2` toggles Line6

arendst, 2022-01-04:
I guess you downloaded an older version. The latest is available in half an hour. Pls try that one again.

fegauthier, 2022-01-04:
I git pulled your latest commit (Fix SPM power switching over 8 relays ) and building .bin on my own. Still the same issue.

arendst, 2022-01-04:
In that case I need the full restart ``weblog 4`` logging of the first minute.

Expect a lot of data like this:
```
00:00:00.003 HDW: ESP32-D0WD-V3 
00:00:00.004 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.074 UFS: FlashFS mounted with 304 kB free
00:00:00.183 CFG: Loaded from File, Count 218
00:00:00.199 QPC: Count 1
00:00:00.207 ROT: Mode 1
00:00:00.210 BRY: GC from 1435 to 1435 bytes, objects freed 0/1 (in 0 ms)
00:00:00.223 BRY: GC from 3601 to 3033 bytes, objects freed 5/20 (in 0 ms)
00:00:00.292 BRY: GC from 4359 to 3813 bytes, objects freed 3/35 (in 1 ms)
00:00:00.293 BRY: Berry initialized, RAM used=3813
00:00:00.318 BRY: no 'preinit.be'
00:00:00.321 SRC: Restart
00:00:00.326 Project tasmota - Sspm2 Version 2022.01.1(theo)-2_0_2(2022-01-04T23:33:42)
00:00:04.155 ETH: Attempting connection...
00:00:04.410 BRY: no 'autoexec.be'
00:00:05.023 CFG: Saved, Count 219, Bytes 4096
00:00:05.492 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:05.495 WIF: Checking connection...
00:00:05.495 WIF: Attempting connection...
00:00:06.012 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as sspm2...
00:00:06.290 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:06.336 SPM: ARM AA5501000000000000000000000000800000010000FC73
00:00:06.438 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:06.486 SPM: ARM AA5501000000000000000000000000800000010000FC73
00:00:06.492 SPM: ESP AA550100000000000000000000000000150000013094
00:00:06.536 SPM: ARM AA550100000000000000000000000080150004000100000181B1
00:00:06.638 SPM: ESP AA5501FFFFFFFFFFFFFFFFFFFFFFFF0010000002CDF0
00:00:06.640 SPM: Start relay scan...
00:00:06.686 SPM: ARM AA5501FFFFFFFFFFFFFFFFFFFFFFFF801000010002E503
00:00:06.086 WIF: Checking connection...
00:00:06.087 WIF: Attempting connection...
00:00:07.554 SPM: ARM AA5501000000000000000000000000000F000102019DF8
00:00:07.555 SPM: ESP AA5501000000000000000000000000800F000100013DE6
00:00:08.086 WIF: Checking connection...
00:00:08.088 WIF: Connected
00:00:08.091 NTP: Synch time...
00:00:08.120 RTC: UTC 2022-01-04T22:36:48, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
23:36:48.000 NTP: Synched
23:36:48.228 HTP: Web server active on sspm2 with IP address 192.168.2.159
23:36:49.310 MQT: Attempting connection...
23:36:49.343 MQT: Connected
23:36:49.352 MQT: tele/sspm2/LWT = Online (retained)
23:36:49.368 MQT: cmnd/sspm2/POWER = 
23:36:49.370 MQT: Subscribe to cmnd/sspm2/#
23:36:49.377 MQT: Subscribe to cmnd/tasmotas/#
23:36:49.380 MQT: Subscribe to cmnd/DVES_0BAE94_fb/#
23:36:49.395 MQT: tele/sspm2/INFO1 = {"Info1":{"Module":"Sonoff SPM (POC2)","Version":"2022.01.1(theo)","FallbackTopic":"cmnd/DVES_0BAE94_fb/","GroupTopic":"cmnd/tasmotas/"}}
23:36:49.413 MQT: tele/sspm2/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sspm2","IPAddress":"192.168.2.159"}}
23:36:49.421 MQT: tele/sspm2/INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
23:36:49.427 TFS: File 'autoexec.bat' not found
23:36:52.068 QPC: Reset
23:36:54.072 APP: Boot Count 42
23:36:54.078 BRY: GC from 5121 to 3652 bytes, objects freed 17/35 (in 3 ms)
23:36:54.092 MQT: tele/sspm2/STATE = {"Time":"2022-01-04T23:36:54","Uptime":"0T00:00:10","UptimeSec":10,"Heap":143,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":35},"Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-6,"LinkCount":1,"Downtime":"0T00:00:03"}}
23:36:54.139 MQT: tele/sspm2/SENSOR = {"Time":"2022-01-04T23:36:54","ESP32":{"Temperature":47.8},"TempUnit":"C"}
23:36:54.002 CFG: Saved, Count 220, Bytes 4096
23:36:59.104 MQT: tasmota/discovery/E0E2E60BAE94/config = {"ip":"192.168.2.159","dn":"Sspm2","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"sspm2","mac":"E0E2E60BAE94","md":"Sonoff SPM (POC2)","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"2022.01.1","t":"sspm2","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
23:36:59.184 MQT: tasmota/discovery/E0E2E60BAE94/sensors = {"sn":{"Time":"2022-01-04T23:36:59","ESP32":{"Temperature":53.3},"TempUnit":"C"},"ver":1} (retained)
23:37:07.698 SPM: ARM AA5501000000000000000000000000001300248B343237393734134B35363704000000820100001400000A00F00000000A11300000000A02A06F
23:37:07.710 SPM: ESP AA5501000000000000000000000000801300010002FE77
23:37:07.717 SPM: ARM AA5501000000000000000000000000001300246B7E3237393734134B35363704000000820100001400000A00F00000000A11300000000A034F0C
23:37:07.729 SPM: ESP AA55010000000000000000000000008013000100033EB6
23:37:07.733 SPM: ARM AA550100000000000000000000000000190000046357
23:37:07.735 SPM: ESP AA5501000000000000000000000000801900010004FD6F
23:37:07.748 SPM: ESP AA5501000000000000000000000000000C000B07E60104162507000001000398EC
23:37:07.763 SPM: ARM AA5501000000000000000000000000800C00010003FC23
[... 36 lines cut ...]
23:37:23.155 SPM: ESP AA5501000000000000000000000000001800106B7E3237393734134B3536370101003C180E50
[... cut ...]
23:37:29.140 SPM: ESP AA5501000000000000000000000000001800106B7E3237393734134B3536370101003C1B0F10
23:37:29.165 SPM: ARM AA550100000000000000000000000080180001001BF513
23:37:29.419 WIF: Checking connection...
23:37:30.839 CMD: weblog 3
23:37:30.842 SRC: WebConsole from 192.168.2.1
23:37:30.847 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '3'
23:37:30.858 MQT: stat/sspm2/RESULT = {"WebLog":3}
23:37:31.370 CFG: Saved, Count 222, Bytes 4096
```

It contains the names of the four SPM-4Relay modules needed to select the correct relay.

I will continue investigating tomorrow.

fegauthier, 2022-01-04:
This the logs I got after a restart.

```
00:00:00.003 HDW: ESP32-D0WD-V3 
00:00:00.003 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.057 UFS: FlashFS mounted with 304 kB free
00:00:00.158 CFG: Loaded from File, Count 149
00:00:00.175 QPC: Count 1
00:00:00.177 CFG: CR 411/699, Busy 0
00:00:00.185 ROT: Mode 1
00:00:00.187 BRY: GC from 1435 to 1435 bytes, objects freed 0/1 (in 1 ms)
00:00:00.202 BRY: GC from 3601 to 3033 bytes, objects freed 5/20 (in 0 ms)
00:00:00.265 BRY: GC from 4359 to 3813 bytes, objects freed 3/35 (in 1 ms)
00:00:00.265 BRY: Berry initialized, RAM used=3813
00:00:00.289 BRY: no 'preinit.be'
00:00:00.293 SRC: Restart
00:00:00.297 Project tasmota - Tasmota Version 2022.01.1(tasmota)-2_0_2(2022-01-04T17:14:03)
00:00:04.150 ETH: Attempting connection...
00:00:04.369 BRY: no 'autoexec.be'
00:00:04.513 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:04.625 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:04.627 WIF: Checking connection...
00:00:04.628 WIF: Attempting connection...
00:00:05.043 WIF: Connecting to AP1 Garage Channel 1 BSSId 24:5A:4C:11:15:11 in mode 11n as tasmota-C781CC-0460...
00:00:05.299 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:05.346 SPM: ARM AA5501000000000000000000000000800000010000FC73
00:00:05.399 SPM: ESP AA55010000000000000000000000000000000000FC51
00:00:05.447 SPM: ARM AA5501000000000000000000000000800000010000FC73
00:00:05.499 SPM: ESP AA550100000000000000000000000000150000013094
00:00:05.547 SPM: ARM AA550100000000000000000000000080150004000100000181B1
00:00:05.599 SPM: ESP AA5501FFFFFFFFFFFFFFFFFFFFFFFF0010000002CDF0
00:00:05.601 SPM: Start relay scan...
00:00:05.647 SPM: ARM AA5501FFFFFFFFFFFFFFFFFFFFFFFF801000010002E503
00:00:05.847 WIF: Checking connection...
00:00:05.849 WIF: Attempting connection...
00:00:06.492 SPM: ARM AA5501000000000000000000000000000F000102019DF8
00:00:06.493 SPM: ESP AA5501000000000000000000000000800F000100013DE6
00:00:06.643 WIF: Checking connection...
00:00:06.644 WIF: Connected
00:00:07.181 HTP: Web server active on tasmota-C781CC-0460 with IP address 192.168.14.87
00:00:07.421 NTP: Synch time...
00:00:08.485 NTP: No reply
00:00:08.557 NTP: Synch time...
00:00:09.808 RTC: UTC 2022-01-04T22:48:51, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
22:48:51.000 NTP: Synched
22:48:51.014 HTP: Main Menu
22:48:52.285 RSL: INFO1 = {"Info1":{"Module":"Sonoff SPM (POC2)","Version":"2022.01.1(tasmota)","FallbackTopic":"cmnd/DVES_C781CC_fb/","GroupTopic":"cmnd/tasmotas/"}}
22:48:52.301 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-C781CC-0460","IPAddress":"192.168.14.87"}}
22:48:52.306 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
22:48:52.329 TFS: File 'autoexec.bat' not found
22:48:53.439 QPC: Reset
22:48:55.430 APP: Boot Count 18
22:48:56.013 CFG: Saved, Count 150, Bytes 4096
22:48:56.328 BRY: GC from 5138 to 3652 bytes, objects freed 17/35 (in 2 ms)
22:48:56.334 RSL: STATE = {"Time":"2022-01-04T22:48:56","Uptime":"0T00:00:10","UptimeSec":10,"Heap":143,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":53,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":35},"Wifi":{"AP":1,"SSId":"Garage","BSSId":"24:5A:4C:11:15:11","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:48:56.365 RSL: SENSOR = {"Time":"2022-01-04T22:48:56","ESP32":{"Temperature":43.9},"TempUnit":"C"}
22:49:09.457 SPM: ARM AA55015B64464E3635361735363236800B000200000DAA0D
22:49:09.459 SPM: ESP AA55015B64464E3635361735363236000B00000E2F90
22:49:09.507 SPM: ARM AA55015B64464E3635361735363236800B000200000EAB4D
22:49:09.510 SPM: ESP AA55015B64464E3635361735363236000B00000FEF51
22:49:09.556 SPM: ARM AA55015B64464E3635361735363236800B000200000F6B8C
22:49:09.562 SPM: Relay scan done
22:49:09.711 SPM: ESP AA55010000000000000000000000000016000D5164464E363536173536323600105F48
22:49:09.785 SPM: ARM AA55010000000000000000000000008016017E005164464E36353617353632360007E6010407E5070800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010CE77
22:49:09.812 SPM: ESP AA55010000000000000000000000000016000D5164464E363536173536323601110F88
22:49:09.899 SPM: ARM AA55010000000000000000000000008016017E005164464E36353617353632360107E6010407E5070800000000000000000000000000000000000500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001F4B1F
22:49:13.989 WIF: Checking connection...
```

arendst, 2022-01-05:
Thx for the input.

Latest commit should solve the issue and will also show what modules it finds during the relay scan.

To see this information I suggest you execute the following commands in the console:
```
weblog 3
restart 1
```
So use ``weblog 3`` to lower the amount of messages received.

Watch for messages like:
```
10:42:57.957 SPM: Module 1 type 130 version 1.0.0 found with id 8B343237393734134B353637
10:42:57.985 SPM: Module 2 type 130 version 1.0.0 found with id 6B7E3237393734134B353637
10:42:58.376 SPM: Relay scan done
```

Pls report these messages so I have an idea of what module ids are being found in the wild. I currently only test for the first two bytes of the module id to discriminate them.

fegauthier, 2022-01-05:
Thanks for your time! Very appreciated!

This is the log for each module
```

11:39:25.841 HTP: Console
11:39:27.930 SPM: Module 1 type 130 version 1.0.0 found with id 2018464E3635361435363236
11:39:27.937 SPM: Module 2 type 130 version 1.0.0 found with id 2218464E3635361435363236
11:39:27.943 SPM: Module 3 type 130 version 1.0.0 found with id 5B64464E3635361735363236
11:39:27.950 SPM: Module 4 type 130 version 1.0.0 found with id 5164464E3635361735363236
11:39:28.577 SPM: Relay scan done
11:39:49.761 HTP: Console
```

arendst, 2022-01-05:
Ah that's under control of the ARM I guess.

What happens if you execute command ``SspmScan`` several times with a minute in between. Does the order of the module id's change?

arendst, 2022-01-05:
Are you sure you cabled the RS-485 bus correctly like 

[image]

So left IN, right OUT and terminated module 4 with the switch in the correct position.

fegauthier, 2022-01-05:
Yes I cabled it correctly. Only the switch is not on the correct position. I will try to change the switch to the good position to see if it changes something.

```
==== #1 ====

14:36:00.699 CMD: SspmScan
14:36:00.702 SRC: WebConsole from 192.168.10.19
14:36:00.704 CMD: Grp 0, Cmnd 'SSPMSCAN', Idx 1, Len 0, Data ''
14:36:00.711 RSL: RESULT = {"SSPMScan":"Started"}
14:36:00.774 SPM: Start relay scan...
14:36:21.255 SPM: Module 1 type 130 version 1.0.0 found with id 2018464E3635361435363236
14:36:21.262 SPM: Module 2 type 130 version 1.0.0 found with id 2218464E3635361435363236
14:36:21.270 SPM: Module 3 type 130 version 1.0.0 found with id 5B64464E3635361735363236
14:36:21.277 SPM: Module 4 type 130 version 1.0.0 found with id 5164464E3635361735363236
14:36:22.590 HTP: Console

==== #2 ====

14:37:55.507 CMD: SspmScan
14:37:55.510 SRC: WebConsole from 192.168.10.19
14:37:55.512 CMD: Grp 0, Cmnd 'SSPMSCAN', Idx 1, Len 0, Data ''
14:37:55.521 RSL: RESULT = {"SSPMScan":"Started"}
14:37:55.583 SPM: Start relay scan...
14:38:10.443 HTP: Console
14:38:16.819 SPM: Module 1 type 130 version 1.0.0 found with id 2018464E3635361435363236
14:38:16.829 SPM: Module 2 type 130 version 1.0.0 found with id 2218464E3635361435363236
14:38:16.838 SPM: Module 3 type 130 version 1.0.0 found with id 5B64464E3635361735363236
14:38:16.849 SPM: Module 4 type 130 version 1.0.0 found with id 5164464E3635361735363236
14:38:17.707 SPM: Relay scan done

==== #3 ====

14:40:08.216 CMD: SspmScan
14:40:08.219 SRC: WebConsole from 192.168.10.19
14:40:08.221 CMD: Grp 0, Cmnd 'SSPMSCAN', Idx 1, Len 0, Data ''
14:40:08.229 RSL: RESULT = {"SSPMScan":"Started"}
14:40:08.294 SPM: Start relay scan...
14:40:28.820 SPM: Module 1 type 130 version 1.0.0 found with id 2018464E3635361435363236
14:40:28.827 SPM: Module 2 type 130 version 1.0.0 found with id 2218464E3635361435363236
14:40:28.835 SPM: Module 3 type 130 version 1.0.0 found with id 5B64464E3635361735363236
14:40:28.842 SPM: Module 4 type 130 version 1.0.0 found with id 5164464E3635361735363236
14:40:29.566 SPM: Relay scan done
14:40:30.214 HTP: Console
```

There is no change in order, but Module 3 ID is greater than Module 4 ID. I don't know if it can be a problem of ordering?

arendst, 2022-01-05:
No I don't think it's a problem of ordering as mine is also the other way round:
```
15:40:44.801 SPM: Start relay scan...
15:41:05.769 SPM: Module 1 type 130 version 1.0.0 found with id 8B343237393734134B353637
15:41:05.776 SPM: Module 2 type 130 version 1.0.0 found with id 6B7E3237393734134B353637
15:41:06.115 SPM: Relay scan done
```

What will more confuse you (but is hidden behind the scene) is that the shown module numbers are actually reversed; Tasmota receives the responses from the ARM starting with physical module 4. At that time Tasmota doesn't know how many modules are present so pushes the id on an internal module stack. Any following module is pushed below it.

In the worst case I could make a command allowing the user to shuffle this module stack in the correct order.

fegauthier, 2022-01-05:
> No I don't think it's a problem of ordering as mine is also the other way round:
>
> ```
> 15:40:44.801 SPM: Start relay scan...
> 15:41:05.769 SPM: Module 1 type 130 version 1.0.0 found with id [...]

Even with the switch on the correct position, order doesn't change.

I did a mistake on my previous post.

L9 power Module 4
L13 power Module 2
L5 power Module 3
L1 power Module 1

If there is nothing to do to change this order. It's not a big deal. But, if there's a way that we can keep the order. It would be very nice.

arendst, 2022-01-05:
Let me think about a clean solution....

arendst, 2022-01-06:
Try latest dev.

In your case execute command ``SspmMap 1,3,4,2`` once after the restart scan is done. The mapping is stored in the filesystem.

closed 2022-01-08 by ascillato2

fegauthier, 2022-01-08:
> Try latest dev.
> 
> In your case execute command `SspmMap 4,2,1,3` once after the restart scan is done. The mapping is stored in the filesystem.

I didn't have the time to test it but I will in the coming days. Thanks!