arendst/Tasmota #14295: Wakeup command changes Sleep and LoadAvg goes very high and Sleep is changed to 10

kreutpet, 2022-01-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have a H801 which i like to wakeup and smothly change the dimmer2 to a value of 67
I have set SetOption60 1 and Sleep 70.
When the wakeup is triggered the LoadAvg goes very high  on 99. Before the wakeup the LoadAvg is <20
The sleep is then changed to 10

The Device seems to be overloaded and is not able to perform the task of smoothly dimming the dimmer2

I use the 10.1.0 2021.12.08 14:47:39

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:14:18.494 MQT: stat/aquarium/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:14:18.696 MQT: stat/aquarium/RESULT = {"Module":{"20":"H801"}}
11:14:18.908 MQT: stat/aquarium/RESULT = {"GPIO0":{"1312":"DS18x20"},"GPIO1":{"288":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"420":"PWM5"},"GPIO5":{"321":"Led_i2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"418":"PWM3"},"GPIO13":{"417":"PWM2"},"GPIO14":{"419":"PWM4"},"GPIO15":{"416":"PWM1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
stat/aquarium/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":119,"Free":392,"Rules":"ON clock#timer=1 DO Backlog Wakeup 67;POWER2 ON;POWER1 OFF BREAK ON rules#timer=2 DO Backlog Speed2 40;POWER2 OFF ENDON"}}
stat/aquarium/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":129,"Free":382,"Rules":"ON clock#timer=3 DO Backlog Color 3;POWER1 ON;POWER2 OFF;Dimmer1 90 ENDON ON rules#timer=4 DO Backlog POWER1 OFF;POWER2 OFF ENDON"}}
stat/aquarium/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
stat/aquarium/STATUS0 = {"Status":{"Module":20,"DeviceName":"Aquarium","FriendlyName":["Aquarium",""],"Topic":"aquarium","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:19:54","StartupUTC":"2022-01-05T09:39:19","Sleep":70,"CfgHolder":4617,"BootCount":33,"BCResetTime":"2021-12-13T10:23:11","SaveCount":182,"SaveAddress":"F6000"},"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021.12.08 14:47:39","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"343/699"},"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["kiot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001800600003C5A0A000000000000","00000481","00006000","00004000"]},"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"aquarium-0407","IPAddress":"192.168.1.95","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"BC:FF:4D:4D:C1:97","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.1.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4DC197","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-01-05T09:59:13","Local":"2022-01-05T10:59:13","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:07"},"StatusSNS":{"Time":"2022-01-05T10:59:13"},"StatusSTS":{"Time":"2022-01-05T10:59:13","Uptime":"0T00:19:54","UptimeSec":1194,"Heap":22,"SleepMode":"Normal","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"OFF","Dimmer1":98,"POWER2":"ON","Dimmer2":98,"Color":"5252F9F900","HSBColor":"240,67,98","White":98,"CT":153,"Channel":[32,32,98,98,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Here the log if the rule is triggered
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: aus Flash geladen von F6, Anzahl 182
00:00:00.055 QPC: Count 1
00:00:00.068 Projekt tasmota - Aquarium Version 10.1.0(tasmota)-2_7_4_9(2021.12.08 14:47:39)
00:00:00.531 WIF: verbinden mit AP1 kiot Channel 9 BSSId 60:A4:4C:DC:B2:23 in Modus 11n als aquarium-0407...
00:00:01.507 WIF: verbunden
00:00:01.753 HTP: Web-Server aktiv bei aquarium-0407 mit IP-Adresse 192.168.1.95
11:21:02.313 MQT: Verbindungsversuch...
11:21:02.331 MQT: verbunden
11:21:02.334 MQT: tele/aquarium/LWT = Online (beibehalten)
11:21:02.337 MQT: cmnd/aquarium/POWER = 
11:21:02.355 MQT: tele/aquarium/INFO1 = {"Info1":{"Module":"H801","Version":"10.1.0(tasmota)","FallbackTopic":"cmnd/DVES_4DC197_fb/","GroupTopic":"cmnd/tasmotas/"}}
11:21:02.385 MQT: tele/aquarium/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"aquarium-0407","IPAddress":"192.168.1.95"}}
11:21:02.404 MQT: tele/aquarium/INFO3 = {"Info3":{"RestartReason":"External System"}}
11:21:02.419 MQT: stat/aquarium/RESULT = {"POWER1":"OFF"}
11:21:02.421 MQT: stat/aquarium/POWER1 = OFF
11:21:02.423 MQT: stat/aquarium/RESULT = {"POWER2":"ON"}
11:21:02.424 MQT: stat/aquarium/POWER2 = ON
11:21:06.417 QPC: Reset
11:21:07.395 MQT: tele/aquarium/STATE = {"Time":"2022-01-05T11:21:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Normal","Sleep":10,"LoadAvg":67,"MqttCount":1,"POWER1":"OFF","Dimmer1":100,"POWER2":"ON","Dimmer2":62,"Color":"5454FF9E00","HSBColor":"240,67,100","White":62,"CT":153,"Channel":[33,33,100,62,0],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}
[... 45 lines cut ...]
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set the sleep mode tho normal and change the sleep to an acceptable value.
Start the wakup 
verify the sleep configuration which has changed and the LoadAvg goes 99 

SetOption60 1
Sleep70
Wakup
-> Sleep has changed to 10 and LoadAvg becomes 99

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

When the wakeup is triggered the Sleep value should not set to 10 as this will impact the LoadAvg.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-01-05:
With software PWM in operation, Tasmota **needs** to lower Sleep to 10, and run at a high LoadAvg. This is not a problem, Tasmota can run just fine.

kreutpet, 2022-01-05:
Well actually the device ends up in a status after the wakeup that the it does not wakeup properly in the next run.
Running now for 1 week the wakeup never completed or even did not start.
As the LoadAvg after first run modifies sleep with the LoaadAvg of 99 my assumption is that the following wakeup and triggers are impacted in their functionality.
Even the triggers for rule 2 are not triggered properly.

You also state in the documentation that LoadAvg  should be kept below 75.
How can I verify that the other functions are not impacted by sleep 10?

I am also  working on the integration to openhab and could offload the smooth dimming to the openhab. I would prefer to run it on tasmota .

Jason2866, 2022-01-05:
Does dimming works smoothly when you set `sleep 0`?
Loadavg is not valid in this case and will end at 999.

sfromis, 2022-01-05:
If you think that there is an issue with how it works (or not), better drop the focus on LoadAvg, this number is not a simple measure of "CPU busy". High LoadAvg is simply how it works with software PWM running. This supersedes suggestions about managing dynamic sleep yourself, which is mainly focused on a situation where you add many sensors.

kreutpet, 2022-01-05:
Many thnkas for looking into this.

I created another test run and it seems the smooth dimming is working. 
Here you can see the log. After that run the LoadAvg is on 999 and it actually stays there.
I now went through 2 cycles and both were successfully completed. See the both logs. 
I will need to check now with the Sleep 0 setting if the triggers run under normal conditions.
Will give feedback in the next days.

First run
`14:23:35.378 MQT: stat/aquarium/RESULT = {"Sleep":{"0":{"Active":"0"}}}
14:23:39.304 CMD: state
14:23:39.313 MQT: stat/aquarium/RESULT = {"Time":"2022-01-05T14:23:39","Uptime":"0T00:58:40","UptimeSec":3520,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":116,"MqttCount":1,"POWER1":"OFF","Dimmer1":100,"POWER2":"ON","Dimmer2":62,"Color":"5454FF9E00","HSBColor":"240,67,100","White":62,"CT":153,"Channel":[33,33,100,62,0],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:24:14.940 CMD: Timer1 {"Enable":1,"Mode":0,"Time":"14:25","Window":0,"Days":"1111111","Repeat":1,"Action":3}
14:24:14.949 MQT: stat/aquarium/RESULT = {"Timer1":{"Enable":1,"Mode":0,"Time":"14:25","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":3}}
14:24:20.835 CMD: power2 off
14:24:20.843 MQT: stat/aquarium/RESULT = {"POWER2":"OFF"}
14:24:20.848 MQT: stat/aquarium/POWER2 = OFF
14:24:23.441 CMD: state
14:24:23.451 MQT: stat/aquarium/RESULT = {"Time":"2022-01-05T14:24:23","Uptime":"0T00:59:24","UptimeSec":3564,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":239,"MqttCount":1,"POWER1":"OFF","Dimmer1":100,"POWER2":"OFF","Dimmer2":62,"Color":"5454FF9E00","HSBColor":"240,67,100","White":62,"CT":153,"Channel":[33,33,100,62,0],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:25:00.410 RUL: CLOCK#TIMER=1 performs "Backlog Wakeup 67;POWER2 ON;POWER1 OFF"
14:25:00.446 MQT: stat/aquarium/RESULT = {"POWER1":"ON"}
14:25:00.449 MQT: stat/aquarium/POWER1 = ON
14:25:00.451 MQT: stat/aquarium/RESULT = {"Wakeup":"Started"}
14:25:00.645 MQT: stat/aquarium/RESULT = {"POWER2":"ON"}
14:25:00.649 MQT: stat/aquarium/POWER2 = ON
14:25:00.846 MQT: stat/aquarium/RESULT = {"POWER1":"OFF"}
14:25:00.851 MQT: stat/aquarium/POWER1 = OFF
14:25:10.406 MQT: tele/aquarium/STATE = {"Time":"2022-01-05T14:25:10","Uptime":"0T01:00:11","UptimeSec":3611,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":311,"MqttCount":1,"POWER1":"OFF","Dimmer1":1,"POWER2":"ON","Dimmer2":1,"Color":"0000010100","HSBColor":"240,67,0","White":1,"CT":153,"Channel":[0,0,1,1,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:26:15.778 CMD: state
14:26:15.787 MQT: stat/aquarium/RESULT = {"Time":"2022-01-05T14:26:15","Uptime":"0T01:01:16","UptimeSec":3676,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":401,"MqttCount":1,"POWER1":"OFF","Dimmer1":5,"POWER2":"ON","Dimmer2":5,"Color":"04040E0E00","HSBColor":"240,67,5","White":5,"CT":153,"Channel":[1,1,5,5,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:30:11.406 MQT: tele/aquarium/STATE = {"Time":"2022-01-05T14:30:11","Uptime":"0T01:05:12","UptimeSec":3912,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":738,"MqttCount":1,"POWER1":"OFF","Dimmer1":23,"POWER2":"ON","Dimmer2":23,"Color":"13133B3B00","HSBColor":"240,67,23","White":23,"CT":153,"Channel":[7,7,23,23,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:30:45.530 CMD: state
14:30:45.540 MQT: stat/aquarium/RESULT = {"Time":"2022-01-05T14:30:45","Uptime":"0T01:05:46","UptimeSec":3946,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":779,"MqttCount":1,"POWER1":"OFF","Dimmer1":25,"POWER2":"ON","Dimmer2":25,"Color":"1515414100","HSBColor":"240,67,25","White":25,"CT":153,"Channel":[8,8,25,25,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:31:55.634 CMD: state
14:31:55.644 MQT: stat/aquarium/RESULT = {"Time":"2022-01-05T14:31:55","Uptime":"0T01:06:56","UptimeSec":4016,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":877,"MqttCount":1,"POWER1":"OFF","Dimmer1":31,"POWER2":"ON","Dimmer2":31,"Color":"1A1A4F4F00","HSBColor":"240,67,31","White":31,"CT":153,"Channel":[10,10,31,31,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:DC:B2:23","Channel":9,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:32:45.680 CMD: state
14:32:45.690 MQT: stat/aquarium/RESULT = {"Time":"2022-01-05T14:32:45","Uptime":"0T01:07:46","UptimeSec":4066,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":945,"MqttCount":1,"POWER1":"OFF","Dimmer1":34,"POWER2":"ON","Dimmer2":34,"Color":"1D1D585800","HSBColor":"240,67,34","White":34,"CT":153,"Channel":[11,11,34,34,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot [... cut ...] ,59","White":59,"CT":153,"Channel":[19,19,59,59,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:68:82:0B","Channel":13,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":2,"Downtime":"0T00:00:05"}}
15:06:59.857 MQT: stat/aquarium/RESULT = {"Wakeup":"Done","POWER1":"OFF","Dimmer1":67,"POWER2":"ON","Dimmer2":67,"Color":"3838ABAB00","HSBColor":"240,67,67","White":67,"CT":153,"Channel":[22,22,67,67,0],"Scheme":1,"Fade":"ON","Speed":1,"LedTable":"ON"}
`

ascillato2, 2022-01-06:
> I created another test run and it seems the smooth dimming is working.
> Here you can see the log. After that run the LoadAvg is on 999 and it actually stays there.
> I now went through 2 cycles and both were successfully completed. See the both logs.

Great to know that now it is working fine for you. Thanks for sharing your logs.

closed 2022-01-06 by ascillato2