arendst/Tasmota #14303: tasmota 32 solo  Thermostat +BLE

bschatzow, 2022-01-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After installing this firmware on the LinKind Switch the relay setting are not handled correctly.  No matter what setting are used the device powers up with the relay off.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- Reported in Tasmota discussion and confirmed by 2+ users
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____Linkind Switch ESP32Solo WS240010008
- [ ] Tasmota binary firmware version number used: Tasmota 2022.01.1 tasmota32 solo Thermostat +ble
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmota from the web
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
```
23:58:46.526 MQT: stat/Bath_rm_Fan_Control_651300/STATUS = {"Status":{"Module":0,"DeviceName":"Bath_rm_Fan_Control","FriendlyName":["Bath_rm_Fan_Control"],"Topic":"Bath_rm_Fan_Control_651300","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:58:46.623 MQT: stat/Bath_rm_Fan_Control_651300/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota32-fullrules.bin","RestartReason":"Vbat power on reset","Uptime":"1T05:51:59","StartupUTC":"2022-01-04T17:06:47","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2022-01-02T15:24:55","SaveCount":82}}
23:58:46.712 MQT: stat/Bath_rm_Fan_Control_651300/STATUS2 = {"StatusFWR":{"Version":"2022.01.1(thermostat)","BuildDateTime":"2022-01-01T18:48:49","Core":"2_0_2","SDK":"v4.4-2.0.2","CpuFrequency":80,"Hardware":"ESP32-U4WDH","CR":"480/699"}}
23:58:46.758 MQT: stat/Bath_rm_Fan_Control_651300/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bill_Charlotte",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004002"]}}
23:58:46.850 MQT: stat/Bath_rm_Fan_Control_651300/STATUS4 = {"StatusMEM":{"ProgramSize":1631,"Free":1856,"Heap":71,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9F9A0786","0011A001","B7F7BFCF","05DA9BC0","70360D47","000844D2","20200000","0008482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,27,29,34,38,39,50,52,59,60,79,85","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,63,64,66,67,74,85,92,127"}}
23:58:46.942 MQT: stat/Bath_rm_Fan_Control_651300/STATUS5 = {"StatusNET":{"Hostname":"Bath-rm-Fan-Control-651300-4864","IPAddress":"192.168.1.75","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"192.168.1.1","Mac":"40:F5:20:65:13:00","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:58:47.026 MQT: stat/Bath_rm_Fan_Control_651300/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.48","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_651300","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:58:47.084 MQT: stat/Bath_rm_Fan_Control_651300/STATUS7 = {"StatusTIM":{"UTC":"2022-01-05T22:58:47","Local":"2022-01-05T23:58:47","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:07"}}
23:58:47.148 MQT: stat/Bath_rm_Fan_Control_651300/STATUS10 = {"StatusSNS":{"Time":"2022-01-05T23:58:47","ESP32":{"Temperature":53.3},"Thermostat0":{"ThermostatModeSet":0,"TempTargetSet":18.00,"CtrDutyCycleRead":0},"TempUnit":"C"}}
23:58:47.224 MQT: stat/Bath_rm_Fan_Control_651300/STATUS11 = {"StatusSTS":{"Time":"2022-01-05T23:58:47","Uptime":"1T05:52:00","UptimeSec":107520,"Heap":75,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"MqttCount":1,"Berry":{"HeapUsed":12,"Objects":197},"POWER":"ON","Wifi":{"AP":1,"SSId":"Bill_Charlotte","BSSId":"B0:6E:BF:DE:DB:48","Channel":1,"Mode":"11n","RSSI":100,"Signal":-24,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
No errors in any logs.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should remember state on reboot or be able to be set to on.
From Tony in the forum:
The thermostat driver init at line 287 checks if "enable_output" is on and if so turns off power to the primary relay, BUT it doesn't check that the thermostat driver is even enabled. This leads to the unexpected behavior in a "Thermostat + BLE" build from special (used for bluetooth but not thermostat) to power off the relay on boot rather than obey the poweronstate.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-05:
```
 {"StatusFWR":{"Version":"2022.01.1(thermostat)"
```
You haven't followed the instructions from the template page to flash the appropriate firmware tasmota32solo1.bin
https://templates.blakadder.com/linkind_WS240010008.html

bschatzow, 2022-01-06:
This device has ble.  The solo1.bin does not.

digiblur, 2022-01-06:
> ```
>  {"StatusFWR":{"Version":"2022.01.1(thermostat)"
> ```
> 
> You haven't followed the instructions from the template page to flash the appropriate firmware tasmota32solo1.bin https://templates.blakadder.com/linkind_WS240010008.html

As with a lot of things there, those instructions are incorrect.  I updated the guide to detail this bug in the OG guide.  https://www.digiblur.com/p/linkind-esp32-smart-switch-how-to-flash.html

I can confirm this bug is the thermostat bin file.

barbudor, 2022-01-06:
@digiblur if the Tasmota official templates are incorrect why not editing the official Doc instead of just publishing on your site?

digiblur, 2022-01-06:
I haven't seen an official doc for it.

bschatzow, 2022-01-06:
Is there a place to request an official tasmota 32 solo1 +BLE (no thermostat)?

Jason2866, 2022-01-07:
@bschatzow  No, there will be no official version with BLE for the single core ESP32.
As discussed yesterday in Discord I will add a unofficial version.

closed 2022-01-07 by Jason2866