arendst/Tasmota #14304: Can not set static IP on wired LAN (eth) interface

budulinek, 2022-01-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Can not set static IP on wired LAN (eth) interface. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ZB-GW03 - see https://github.com/arendst/Tasmota/discussions/12764
- [x] Tasmota binary firmware version number used: 10.1.0(tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled ( see build options here https://templates.blakadder.com/ewelink_ZB-GW03 )
- [x] Flashing tools used:  ESP-Flasher-Windows-x64
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
01:31:42.117 RSL: RESULT = {"NAME":"ZB-GW03-V1.3","GPIO":[0,0,3552,0,3584,0,0,0,5793,5792,320,544,5536,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,608,640,32,0,0,0,0,0],"FLAG":0,"BASE":1}
01:31:42.345 RSL: RESULT = {"Module":{"0":"ZB-GW03-V1.3"}}
01:31:42.600 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"3552":"Zigbee Tx"},"GPIO3":{"0":"None"},"GPIO4":{"3584":"Zigbee Rx"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"5793":"Zigbee Rst2"},"GPIO13":{"5792":"Zigbee Rst1"},"GPIO14":{"320":"Led_i1"},"GPIO15":{"544":"LedLink"},"GPIO16":{"5536":"ETH POWER"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"608":"I2C SCL1"},"GPIO33":{"640":"I2C SDA1"},"GPIO34":{"32":"Button1"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
01:32:44.699 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D2FC6C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
01:32:44.703 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T02:18:25","StartupUTC":"2022-01-05T22:14:19","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2021-12-10T23:27:07","SaveCount":70}}
01:32:44.709 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-10T23:58:30","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6-V3","CR":"396/699"}}
01:32:44.713 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"10.10.10.10","LogPort":514,"SSId":["XXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008289","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
01:32:44.720 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1299,"Free":1856,"Heap":146,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FCAC787","00540001","000000CF","010013C0","40000981","40004080","00200000","0000002C"],"Drivers":"1,2,3,4,5,8,9,10,12,20,21,23,24,26,27,29,38,50,52,82","Sensors":"1,2,3,5,6,127"}}
01:32:44.724 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D2FC6C-7276-eth","IPAddress":"10.10.10.226","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"XXXX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
01:32:44.727 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D2FC6C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:32:44.730 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-06T00:32:44","Local":"2022-01-06T01:32:44","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:08"}}
01:32:44.737 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-06T01:32:44","ESP32":{"Temperature":63.9},"TempUnit":"C"}}
01:32:44.743 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-06T01:32:44","Uptime":"0T02:18:25","UptimeSec":8305,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":47},"Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"XXXXX","Channel":8,"Mode":"11n","RSSI":30,"Signal":-85,"LinkCount":12,"Downtime":"0T00:20:42"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

connected via eth interface, IP from DHCP

```
01:37:17.372 CMD: Status 5
01:37:17.375 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D2FC6C-7276-eth","IPAddress":"10.10.10.226","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"XXXX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
```

try to change IP, restart

```
01:38:52.727 CMD: IPAddress1 10.10.10.14
01:38:52.730 RSL: RESULT = {"IPAddress1":"10.10.10.14 (10.10.10.226)"}
01:39:30.678 CMD: restart 1
01:39:30.680 RSL: RESULT = {"Restart":"Restarting"}
```

check IP after boot, ethernet IP remains the same

```
01:41:19.050 CMD: Status 5
01:41:19.054 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D2FC6C-7276-eth","IPAddress":"10.10.10.226","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"XXXX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
```

but Wifi IP is changed

[image]

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

IPAddress1 command should change the IP address of currently active interface (the one where webserver is running).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-06:
Setting the Wifi Address will not change the Ethernet address
Use the appropriate commands
https://tasmota.github.io/docs/Commands/#ethernet

budulinek, 2022-01-06:
related discussion:

https://github.com/arendst/Tasmota/discussions/14168

ascillato2, 2022-01-06:
Closing in favor of the previous feature request #14168

closed 2022-01-06 by ascillato2