arendst/Tasmota #14306: websend always delays receiving a http reply to itself for 5 seconds

pilaGit, 2022-01-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After a websend executes, that particular device will not receive/handle a http reply to that same device immediately, but will always add 5 sec delay. Only then will it falsely report {"WebSend":"Connect failed"} and continue with normal and correct handling of a http reply.
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): any, using Sonoff ZBbridge for a report 
- [x] Tasmota binary firmware version number used: tried 9.2 and 10.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:41:36.857 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Zigbee","FriendlyName":["Zigbee"],"Topic":"tasmota_9DC6F0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:41:36.915 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"5T15:42:57","StartupUTC":"2021-12-31T18:58:39","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2021-12-31T10:19:51","SaveCount":209,"SaveAddress":"1FB000"}}
11:41:36.971 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(zbbridge)","BuildDateTime":"2021-12-08T14:47:37","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"544/699"}}
11:41:37.023 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["db1","d"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A00000000000000","00000080","00046002","00004000"]}}
11:41:37.078 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":671,"Free":1120,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"}}
11:41:37.137 RSL: STATUS5 = {"StatusNET":{"Hostname":"Zigbee","IPAddress":"192.168.2.60","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:9D:C6:F0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:41:37.190 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-06T10:41:37","Local":"2022-01-06T11:41:37","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:36","Sunset":"16:25"}}
11:41:37.243 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-06T11:41:37"}}
11:41:37.293 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-06T11:41:37","Uptime":"5T15:42:58","UptimeSec":488578,"Vcc":3.499,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"db1","BSSId":"18:31:BF:E4:9B:E8","Channel":11,"Mode":"11n","RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:12"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

_from the console at 2.60 directly back to the same 2.60 - pointless, but easiest to demonstrate the 5 seconds lag_

`websend [192.168.2.60] Power 1`

`17:42:20.548 CMD: websend [192.168.2.60] Power 1`
`17:42:25.695 RSL: RESULT = {"WebSend":"Connect failed"}`

_from 2.60 To the web server 2.10 with http reply returning back to the same 2.60_

`websend [192.168.2.10] /?x=z60`

```
17:05:37.443 CMD: websend [192.168.2.10] /?x=z60
17:05:42.784 RSL: RESULT = {"WebSend":"Connect failed"}
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

After issuing a websend, command must be correctly confirmed as done and a http reply must be executed immediately (with no 5 seconds delay added). It should work like here, when the command does not return to the same device 2.60 but to the 2.71

_from 2.60 Zigbee Bridge directly to the 2.71_

`websend [192.168.2.71] Power 1`

```
12:42:11.609 CMD: websend [192.168.2.71] Power 1
12:42:12.300 RSL: RESULT = {"WebSend":"Done"}

```
_from 2.60 Zigbee Bridge to the web server sending http reply to 2.71_

`websend [192.168.2.10] /?x=z71`

```
17:09:01.923 CMD: websend [192.168.2.10] /?x=z71
17:09:03.677 RSL: RESULT = {"WebSend":"Done"}
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Problem was noticed at a Zigbee bridge when testing sending Zigbee button clicks to my SmartHome program to execute appropriate commands by sending http reply to the Tasmota devices. 

When a resulting reply command had to be sent back to **that same** Zigbeee bridge, it would always result in 5 seconds wait on the bridge and a false "Connect failed". Then it would be executed normally, as that did not occur. 

If the same websend leads to http reply being sent to any other device, all is perfectly well and quick (1.5-2 seconds).

In my tests, Zigbee keypresses are sent to a Webserver which just forwards them to my SmartHome program. SmartHome program then sends needed http commands back to the Tasmota devices directly.

Same behavior on: several very different devices I tested with current and older Tasmota. Tested: Sonof ZBbridge, Shelly Bulb Duo, Athom POW.

**(Please, remember to close the issue when the problem has been addressed)**

pilaGit, 2022-01-06:
Just to prove that everything works when websend is not followed by a http reply to the same device. Sending the same On command from 2.71 to the Zigbee bulb at 2.60

_from the console at 2.71 to the 2.60 Zigbee bridge_

`WebSend [192.168.2.60] ZbSend { "Device":"IkeaSijalica", "Send":{"Power":"on"} }
`
```
13:01:02.142 CMD: WebSend [192.168.2.60] ZbSend { "Device":"IkeaSijalica", "Send":{"Power":"on"} }
13:01:03.778 RSL: RESULT = {"WebSend":"Done"}

```
_and at 2.60 we get:_

```
13:01:03.298 RSL: RESULT = {"ZbSend":"Done"}
13:01:04.025 RSL: SENSOR = {"IkeaSijalica":{"Device":"0xF3D0","Name":"IkeaSijalica","Power":1,"Endpoint":1,"LinkQuality":60}}

```

_from 2.71 to the web server 2.10 with http reply returning to the 2.60_

`websend [192.168.2.10] /?x=z60`

```
13:10:51.507 CMD: websend [192.168.2.10] /?x=zb 60
13:10:52.860 RSL: RESULT = {"WebSend":"Done"}

```
_at 2.60 we see:_

```
13:10:53.326 RSL: RESULT = {"ZbSend":"Done"}
13:10:54.039 RSL: SENSOR = {"IkeaSijalica":{"Device":"0xF3D0","Name":"IkeaSijalica","Power":1,"Endpoint":1,"LinkQuality":60}}

```

sfromis, 2022-01-06:
Unclear what you really mean by **always delays receiving a http reply** and then go on to provide examples of having no such delay.

Are you expecting a confirmation/error while the other server is still processing? WebSend has a timeout of 5 seconds if waiting for a response, as tested by calling a server set to have a longer response time. As Tasmota cannot know if the server has an intention of responding or not, it is not possible to immediately detect the bad server behavior of just not responding. Arguably, 5 second is a bit quick for a timeout, but that's a separate question.

Your test case with a bad request to the Tasmota device itself merely exhibits that Tasmota handling of this senseless scenario is not great, nothing more.

pilaGit, 2022-01-06:
As the title says. When a Tasmota Zigbee bridge 2.60 sends out webrequest and the http reply is immediately returned back to that same 2.60: 

a) delay of 5 seconds is inevitable - problem one
b) web send is falsely calimed as failed - problem two

sfromis, 2022-01-06:
Nothing about Zigbee in the title.

Is the short version that if you, from a Zigbee bridge, sends a request to a server providing a quick well-formed response, that you will get a delay of 5 seconds, and then a failure.

Or is it that you have a badly behaving server which does not bother to respond to requests, and you just want Tasmota to not detect if the WebSend was successful or not?

pilaGit, 2022-01-06:
Zigbee is where I encountered the problem. Problem exists in all devices. Zigbee bridge and Zigbeee itself is not the problem at all.

This first thing thing you write is the problem. Tasmota does insert delay. 

As for my programming, you can see in addendum that it took 1,8 seconds to execute everything correctly when one device (2.71) is sending a web request to my program that will be terminated on another device (2.60) which is the Zigbeee bridge.

sfromis, 2022-01-06:
You are using the words "when websend is not followed by http replay to the same device". This is **not** my first point, where I specifically asked about a scenario with "server providing a quick well-formed response". 

When using a ZbBridge to do a WebSend to a "server providing a quick well-formed response", I saw no delay. When I made the server response time longer than 5 seconds, Tasmota reported failure after 5 seconds. This makes sense, as long as you accept such a brief timeout.

pilaGit, 2022-01-06:
My server response time is always the same, under 0,5 sec for this issue. My additional example shows that the entire cycle is performed in 1,8 seconds when websend goes from one Tasmota and reply to another.

When that perfectly same chain is started with websend from the one Tasmota and generates http reply to that same Tasmota, the 5 sec delay is introduced.

pilaGit, 2022-01-06:
I wrote that directly sending to itself was pointless! Just shows 5 seconds lag.

The problem is: the device sends websend to another device and then from another device this same device (the first one recives http reply. 

Use Zigbee as an example. It has a Zigbee button and a Zigbee bulb. Zigbee button sends command pressed. Tasmota forwards that to the main external program. That external program reads that input, and returns the http command: turn on the Zigbee light at that same Zigbee Bridge.

pilaGit, 2022-01-06:
OK, here is the complete test setup that shows a problem. Everything needs to be executable, adjust folders as needed

To your web server add: zbtest.php

```
<?php
$inputArgs = $_GET['x'] ;
$output = shell_exec("bash /root/ztest.sh $inputArgs 2>&1 ");
echo "$output";
?>
```

Then, create a bash script ztest.sh
```

#!/bin/bash

case "$1" in
(z60) curl -s http://192.168.2.60/cm?cmnd=ZbSend%20%7B%20%22Device%22%3A%22IkeaSijalica%22%2c%20%22Send%22%3A%7B%22Power%22%3A%22on%22%7D%20%7D
	# SendHttp "60" 'ZbSend { "Device":"IkeaSijalica", "Send":{"Power":"on"} }'
;;
(z71)
	curl -s http://192.168.2.71/cm?cmnd=Power%201
	# SendHttp "71" "Power 1"
;;
esac
```
at 2.60 there is a Zigbee bridge. There is a bulb (IkeaSijalica) on it. At 2.71 there is a Tasmota bulb.

Connect to the Zigbee bridge (and device will do as long as the reply goes back to it). Issue the fllowing two commands in the console of 2.60.

`websend [192.168.2.10] /zbtest.php?x=z60`
`websend [192.168.2.10] /zbtest.php?x=z71`

I get the result:

```
14:58:40.514 CMD: websend [192.168.2.10] /zbtest.php?x=z71
14:58:41.496 RSL: RESULT = {"WebSend":"Done"}
14:58:48.906 CMD: websend [192.168.2.10] /zbtest.php?x=z60
14:58:53.920 RSL: RESULT = {"WebSend":"Connect failed"}
14:58:54.275 RSL: RESULT = {"ZbSend":"Done"}
14:58:54.973 RSL: SENSOR = {"IkeaSijalica":{"Device":"0xF3D0","Name":"IkeaSijalica","Power":1,"Endpoint":1,"LinkQuality":63}}
```

When http reply goes back to the oroginal device, a 5 sec delay is added and {"WebSend":"Connect failed"} is false alert.

pilaGit, 2022-01-06:
If I issue the same two commands, but this time from the console of 2.71, the same thing, but now the problem is at 2.71. Zigbee is irrelevant here.

`websend [192.168.2.10] /zbtest.php?x=z71`
`websend [192.168.2.10] /zbtest.php?x=z60`

```
15:21:25.615 CMD: websend [192.168.2.10] /zbtest.php?x=z71
15:21:30.926 RSL: RESULT = {"WebSend":"Connect failed"}
15:21:31.241 RSL: RESULT = {"POWER":"ON"}
15:21:31.244 RSL: POWER = ON
15:21:43.141 CMD: websend [192.168.2.10] /zbtest.php?x=z60
15:21:43.854 RSL: RESULT = {"WebSend":"Done"}

```

s-hadinger, 2022-01-06:
Got it. This is a race condition. WebSend is a blocking command. It stops everything in Tasmota until the request returns a response or the timeout happens.

In your example, the WebSend blocks and prevents from responding to 2.71. When the timeout occurs, 2.71 keeps retrying and eventually the request goes through. But the connection from 2.60 is already broken so no answer gets back.

There is no plan to move WebSend to async. Without this, this kind of usage is not possible.

I would strongly suggest to move to MQTT which is async by nature.

sfromis, 2022-01-06:
While you indeed wrote "pointless", you still used that example it to make the point of a 5 second delay, and also used the word "reply" a lot without meaning a reply to a http request.

Seems that the misunderstanding is that when you used the word "reply" (and "http reply") you did not mean a reply to the http request, but instead the unusual behavior of creating a new independent http request from the server, in addition to the server doing the regular protocol reply to the original http request. Such things happen when it is not multi-threading.

Anyway, the http part of Tasmota id not meant as a a robust server part, and is not well-tested or tuned. It indeed seems that Tasmota receiving a http *request* while WebSend is still waiting for a *response* will make it fail. 

There are reasons why MQTT is in the name of Tasmota, as this is a much better communication protocol instead of the mess of assuming WebSend to be a good API. MQTT is what Tasmota is designed for.

pilaGit, 2022-01-06:
Just pick any two Tasmota devices, and send them Power 1 or any other command.

```
#!/bin/bash

curl -s http://192.168.2.$1/cm?cmnd=Power%201
	# SendHttp "$1" "Power 1"
```
Issue commands:

`websend [192.168.2.10] /zbtest.php?x=71`
`websend [192.168.2.10] /zbtest.php?x=60`

When you websend and receive to and from the same device: delay and false reporting.

websend [192.168.2.10] /zbtest.php?x=71
websend [192.168.2.10] /zbtest.php?x=72

_from 2.71_

```
15:37:44.766 CMD: websend [192.168.2.10] /zbtest.php?x=71
15:37:49.783 RSL: RESULT = {"WebSend":"Connect failed"}
15:37:50.021 RSL: RESULT = {"POWER":"ON"}
15:37:50.023 RSL: POWER = ON
15:37:54.158 CMD: websend [192.168.2.10] /zbtest.php?x=72
15:37:54.866 RSL: RESULT = {"WebSend":"Done"}

```

pilaGit, 2022-01-06:
OK, I understand this will not be fixed.

As for MQTT being better than http: I have a very complex SmartHome systems running perfectly using only bidirectional http communication. In multiple locations, running everything for several years now. Just one single SmartHome program and mostly Tasmotas. I bought bulbs just now to test Zigbee. I needed Zibee to add switches.

There are ways around this, as in most cases. 

I guess I should close this issue.

sfromis, 2022-01-06:
Not using MQTT, you are losing out on a number of Tasmota features, in addition to http generally being slower, and with more risk of not being quite reliable.

pilaGit, 2022-01-06:
Speed is not a problem. reliability is perfect, my program has no bugs. Nothing to maintain. I never found a feature I needed I could not use via http. My SmartHome system is very complex.

The only problem with the Tasmota that I have, is not related to this but inability to corrcetly re/set power meters. Everything else I fixed in my program.

That being said, Tasmota works amazingly good! Perfectly stable, no problems with it once you identify probablems and fix them. Tasmota is truly amazing piece of software!

And yes, Tasmota works perfectly with just a poor http :) I would only never use websend to send critical data directly from one to another Tasmota device - it may not be received.

The problem I reported here was found in my testing of Zigbee. In reality, it can be easily avoided.

pilaGit, 2022-01-06:
Speed is not a problem. reliability is perfect, my program has no bugs. Nothing to maintain. I never found a feature I needed I could not use via http. My SmartHome systems are very complex.

The only problem with the Tasmota that I have, is not related to this but inability to corrcetly re/set power meters. Everything else I fixed in my program.

That being said, Tasmota works amazingly good! Perfectly stable, no problems with it once you identify problems and fix them. Tasmota is truly amazing piece of software!

And yes, Tasmota works perfectly with just a poor http :) I would only never use websend to send critical data directly from one to another Tasmota device only once - it may not be received. Handshake or a repeat could solve this if needed.

The problem I reported here was found in my testing of Zigbee. In reality, it can be easily avoided.

ascillato2, 2022-01-08:
Moving to Ideas as this is now a feature request. Thanks.