arendst/Tasmota #14331: Can't upgrade ZBBridge to Zigbee2Mqtt

SergioRius, 2022-01-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After following digiblur tutorial on how to flash a Sonoff ZBridge, successfully flashed with Tasmota, I try to upgrade to ncp-uart-nsw_6.7.8_115200.ota, but always receive a `Upload Failed   Upload Aborted` result. I've tried every fw version available, even the archived ones. The files are downloaded through the Download button and the device is connected to a 3.0 power supply.
```
00:00:00.001 HDW: ESP8266EX
00:00:00.003 UFS: FlashFS mounted with 220 kB free
00:00:00.036 CFG: Loaded from File, Count 22
00:00:00.040 CFG: CR 412/699, Busy 0
00:00:00.043 ZIG: ZBBridge EEPROM found at address 0x50
00:00:00.044 SNS: Hardware Serial
00:00:00.044 SRC: Restart
00:00:00.045 Project tasmota - Tasmota Version 10.1.0(zbbridge)-2_7_4_9(2021-12-08T14:47:37)
00:00:00.046 ZIG: Resetting EZSP device
00:00:00.197 WIF: Attempting connection...
00:00:00.532 WIF: Connecting to AP1 rainland-IoT Channel 11 BSSId B6:FB:E4:1B:D7:06 in mode 11n as tasmota-9E151B-5403...
00:00:01.751 WIF: Connected
00:00:02.004 HTP: Web server active on tasmota-9E151B-5403 with IP address 10.1.5.142
00:00:02.636 RTC: UTC 2022-01-08T13:59:55, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
14:59:56.061 RSL: INFO1 = {"Info1":{"Module":"Sonoff ZbBridge","Version":"10.1.0(zbbridge)","FallbackTopic":"cmnd/DVES_9E151B_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:59:56.062 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-9E151B-5403","IPAddress":"10.1.5.142"}}
14:59:56.063 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
14:59:56.064 TFS: File 'autoexec.bat' not found
14:59:58.645 HTP: Firmware Upgrade
15:00:00.446 RSL: STATE = {"Time":"2022-01-08T15:00:00","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.442,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"rainland-IoT","BSSId":"B6:FB:E4:1B:D7:06","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:00:01.442 APP: Boot Count 10
15:00:01.648 CFG: Saved to flash at 1FB, Count 23, Bytes 4096
15:00:07.327 CFG: Saved to flash at 1FB, Count 24, Bytes 4096
15:00:07.327 UPL: File ncp-uart-nsw_6.7.9_115200.ota
15:00:07.343 UPL: File type 5
15:00:09.632 UPL: Progress 100 kB
15:00:11.586 UPP: Multicast (re)joined
15:00:11.587 UPL: Transfer 192240 bytes
15:00:11.588 UPL: Successful 192240 bytes
15:00:11.599 HTP: Upload transfer
15:00:11.634 XMD: Init bootloader
15:00:11.635 ZIG: Bootloader pin 5
15:00:41.824 XMD: Bootloader timeout
15:00:41.874 XMD: Failed
15:00:41.875 ZIG: Bootloader pin 5
15:00:42.231 HTP: Upload done
```
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff ZB Bridge
- [X] Tasmota binary firmware version number used: 10.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Serial converter / Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

15:13:09.169 CMD: Backlog Template; Module; GPIO 255
15:13:09.170 SRC: WebConsole from 10.1.2.90
15:13:09.171 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
15:13:09.186 SRC: Backlog
15:13:09.187 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
15:13:09.190 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:13:09.414 SRC: Backlog
15:13:09.416 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
15:13:09.418 RSL: RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
15:13:09.665 SRC: Backlog
15:13:09.666 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
15:13:09.671 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
[... 58 lines cut ...]
15:11:12.160 ZIG: Bootloader pin 5
15:11:42.346 XMD: Bootloader timeout
15:11:42.396 XMD: Failed
15:11:42.397 ZIG: Bootloader pin 5
15:11:42.753 HTP: Upload done

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try to flash the aforementioned fw

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
To correctly flash.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-08:
```
15:13:09.187 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
15:13:09.190 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
```

Start by applying the right template
Flashing the firwmare which include the ZbBridge feature is not enough. You also need to Tell tasmota that that your hardware is a ZbBridge
https://templates.blakadder.com/sonoff_ZBBridge.html

barbudor, 2022-01-08:
What is the length in bytes of you firmware file ?

SergioRius, 2022-01-08:
> Every time I have seen this problem, it was due to a cut trace on the PCB. The damage often happens when desoldering. Can yous send pictures of your board (both sides)

You were absolutely right. It seems that the RX track is so thin near the pad and inspection with a microscope revealed that it was cut just after the pad. After resoldering I was successful upgrading it:
```
16:48:45.970 HTP: Firmware Upgrade
16:49:09.105 CFG: Saved to flash at 1FB, Count 28, Bytes 4096
16:49:09.105 UPL: File ncp-uart-sw_6.7.8_115200.ota
16:49:09.121 UPL: File type 5
16:49:11.450 UPL: Progress 100 kB
16:49:13.397 UPP: Multicast (re)joined
16:49:13.398 UPL: Transfer 190756 bytes
16:49:13.399 UPL: Successful 190756 bytes
16:49:13.410 HTP: Upload transfer
16:49:13.439 XMD: Init bootloader
16:49:13.439 ZIG: Bootloader pin 5
16:49:14.655 XMD: Init sync
16:49:15.235 XMD: Init packet send
16:49:16.495 XMD: Progress 10 kB
16:49:17.619 XMD: Progress 20 kB
16:49:18.774 XMD: Progress 30 kB
16:49:19.895 XMD: Progress 40 kB
16:49:21.045 XMD: Progress 50 kB
16:49:22.191 XMD: Progress 60 kB
16:49:23.344 XMD: Progress 70 kB
16:49:24.475 XMD: Progress 80 kB
16:49:25.620 XMD: Progress 90 kB
16:49:26.751 XMD: Progress 100 kB
16:49:27.912 XMD: Progress 110 kB
16:49:29.046 XMD: Progress 120 kB
16:49:30.172 XMD: Progress 130 kB
16:49:31.321 XMD: Progress 140 kB
16:49:32.450 XMD: Progress 150 kB
16:49:33.628 XMD: Progress 160 kB
16:49:34.757 XMD: Progress 170 kB
16:49:35.898 XMD: Progress 180 kB
16:49:36.632 XMD: Transferred 190756 bytes
16:49:39.402 XMD: Successful
```

closed 2022-01-08 by SergioRius

eoghan2t9, 2022-01-18:
> > Every time I have seen this problem, it was due to a cut trace on the PCB. The damage often happens when desoldering. Can yous send pictures of your board (both sides)
> 
> You were absolutely right. It seems that the RX track is so thin near the pad and inspection with a microscope revealed that it was cut just where it joins the pad. After resoldering I was successful upgrading it:
> 
> ```
> 16:48:45.970 HTP: Firmware Upgrade
> 16:49:09.105 CFG: Saved to flash at 1FB, Count 28, Bytes 4096
> 16:49:09.105 UPL: File ncp-uart-sw_6.7.8_115200.ota
> 16:49:09.121 UPL: File type 5
> 16:49:11.450 UPL: Progress 100 kB
> 16:49:13.397 UPP: Multicast (re)joined
> 16:49:13.398 UPL: Transfer 190756 bytes
> 16:49:13.399 UPL: Successful 190756 bytes
> 16:49:13.410 HTP: Upload transfer
> 16:49:13.439 XMD: Init bootloader
> 16:49:13.439 ZIG: Bootloader pin 5
> 16:49:14.655 XMD: Init sync
> 16:49:15.235 XMD: Init packet send
> 16:49:16.495 XMD: Progress 10 kB
> 16:49:17.619 XMD: Progress 20 kB
> 16:49:18.774 XMD: Progress 30 kB
> 16:49:19.895 XMD: Progress 40 kB
> 16:49:21.045 XMD: Progress 50 kB
> 16:49:22.191 XMD: Progress 60 kB
> 16:49:23.344 XMD: Progress 70 kB
> 16:49:24.475 XMD: Progress 80 kB
> 16:49:25.620 XMD: Progress 90 kB
> 16:49:26.751 XMD: Progress 100 kB
> 16:49:27.912 XMD: Progress 110 kB
> 16:49:29.046 XMD: Progress 120 kB
> 16:49:30.172 XMD: Progress 130 kB
> 16:49:31.321 XMD: Progress 140 kB
> 16:49:32.450 XMD: Progress 150 kB
> 16:49:33.628 XMD: Progress 160 kB
> 16:49:34.757 XMD: Progress 170 kB
> 16:49:35.898 XMD: Progress 180 kB
> 16:49:36.632 XMD: Transferred 190756 bytes
> 16:49:39.402 XMD: Successful
> ```

Hi

Could you post a picture of where the cut pad is as i was able to flash tasmota without soldering but can get the zigbee firmware installed and im getting the same errors as you did. I did run a bit of wire from the zigbee chips rx to the z_rx hole on the main pcb in case the track is cut somewhere in between however flashing the firmware still fails.

Thanks

kallisti5, 2022-11-02:
The cut trace is intense lol.  I had to re-repair mine after needing to reflash the tasmota firmware to recover my Sonoff ZBBridge.

[image]

Sorry for the quality of the picture, this trace is *tiny* and seems to break for everyone in roughly the same way.  That bodge wire is a few strands of copper twisted together.   I'm not even brave enough to wipe the flux off lol.

Symptoms when this trace breaks:

* Zigbee firmware upgrades will fail
* Zigbee modem won't respond to commands (aka via TCP 8888)

The fix is a bit difficult given how thin the trace is, and how hard it is to access.

s-hadinger, 2022-11-02:
I strongly suggest to upgrade to zbbridge pro which has proper headers for flashing. And an esp32 for extensibility

kallisti5, 2022-11-02:
> I strongly suggest to upgrade to zbbridge pro which has proper headers for flashing. And an esp32 for extensibility

Not sure what the " zbbridge pro " is.    Not seeing any references online, or on Amazon

NoGrout, 2022-12-07:
Hello, I have a zbbridge pro and I seem to have the same problem. Where to look on the pcb? I am confused. Thanks
[image]
[image]

sfromis, 2022-12-07:
Look where you soldered, very carefully. The different hardware build of the "pro" version makes the risk of that type of cracks much less, as you can just solder in that 5 pin header in the nicely spaced 2.54 mm holes. However, your soldering around TX0 looks questionable, as the big glob of solder appears to possibly be connecting to the ground plane too.

Also some risk of cold soldering, you may not have heated the pin enough before coming in with the solder wire.

NoGrout, 2022-12-07:
Thank you for that answer.  The Tasmota flash went well though.  And I had already done a zigbee firmware update which was operational, but I had an error with the tcpstart 8888 for home assistant, and by dint of tweaking, I had to crash the zbbridge, and therefore start all over again.  I'm going to take the time to redo my welds in the hope that this solves my problems.  Thanks

sfromis, 2022-12-07:
[image]