arendst/Tasmota #14337: Sonoff S31 MQTT authentication failure with fresh 10.1.0 flash

kocherjj, 2022-01-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I recently received my first three Sonoff S31 smart plugs. I erased them and flashed them with Tasmota 10.1.0. I filled in all of the WiFi and MQTT information and from the device webpage they appeared to function just like the 20 or so other ESP based Tasmota devices I have around my house, but I could not control them via MQTT commands.

Upon checking my Mosquitto broker logs I see a series of <New connection from [device IP] on port 1883.><Socket error on client <unknown>, disconnecting.> messages. When I check the "Console" from the web page 

This message is vague but in my experience this usually means the user or password is invalid. I double and triple checked those items and they were correct. I flashed one of them with the latest Dev firmware, 2022.1.2, I believe it was. This did not help. I downgraded it to 9.1.0 and was able to get it working. I confirmed that this worked with all three plugs. Once it was registering correctly to the broker I updated it to 10.1.0 via the device webpage and it continues to work. I followed this process with both the others and this also worked.

TL:DR, A Sonoff S31 does connect to my MQTT Mosquitto broker with freshly flashed with >9.10 firmware. It does connect correctly when flashed with 9.1.0, and will then continue to work after updating to 10.1.0.

These devices are working fine for me at this point and I don't particularly care about the extra steps now that I know what to do, but I am filing this issue to assist others who might have this issue in the future and in hopes that it can be resolved in future updates. 

These devices are the first ones I have started from 10.1.0 so I have no idea if this issue is present on other ESP devices as well. Since this is not an issue I "need resolved" I apologize if this is not the appropriate way to bring it to the developers attention.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff S31____
- [ ] Tasmota binary firmware version number used: 9.1.0, 10.1.0, 2022.1.2_____
  - [ ] Pre-compiled Yes
  - [ ] Self-compiled No
- [ ] Flashing tools used: esptool.py v3.2_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kocherjj, 2022-01-08:
FWIW, I also just discovered that I am unable to change the MQTT Topic of a Shelly 1 PM that was previously upgraded to 10.1.0. When I try to it does not retain the old topic or set the new desired topic, but instead resets the topic to the default value and will not change it. Rolling back to 9.1.0 allowed me to make the change.

barbudor, 2022-01-08:
There use to be a historical restriction to prevent Topic and MqttClient to be identical. Trying to set Topic to the same value as MqttClient was forbidden and forced the return to the default `tasmota_xxxx` topic (not the previous one). => exactly your last message.
While this restriction was in tasmota from before v6, in some versions, there was a loophole in which, depending on the order you were changing MqttClient and Topic, this was feasible.
Now, this restriction has been officially removed in v10.1.0.1.
Upgrade to latest dev version 2022.01.02 and you will be able to set MqttClient and Topic without any restriction.

Regarding the original question, please double check that your MqttClient are different on each and every device (and preferably Topic too).

kocherjj, 2022-01-08:
On all my Tasmota devices the Topic and Client match. Each device is unique from every other device.
Whatever the cause of the issue I experienced with the Sonoff S31 plugs, it is still present in 2022.01.02. I tested that before raising the issue here. I did not attempt to update the Shelly 1 PM to 2022.01.02 when I experienced the Topic name issue, but rolled it back to 9.1.0 since that seemed the best approach considering the issues I'd had with the S31s. I did just now attempt to update it to 2022.01.02 to test again but it says there is not enough space for the upgrade.

If so, the MQTT Topic change and the MQTT broker connection issue might be completely unrelated.

Is/was there a good reason to keep the client and topic unique on a particular device? Since that is how I differentiate between devices it seems inefficient to have two unique strings for every device rather than just one but if there is a best practice I am not aware of I would like to know.

barbudor, 2022-01-08:
Originaly the fallback topic was equal to the MqttClient. So the main Topic was to be different from MqttClient.
But later on the fallback topic was changed to <MqttClient>_fb so the restriction was no more needed anymore but the code preventing them to be equal wasn't removed until 10.1.0.1.

Regarding your update, if you update using OTA (== http directly from the device) on url `http://ota.tasmota.com/tasmota/tasmota.bin.gz` it should work seamlessly with possibly intermediate flashing with `tasmota-minimal.bin.gz` behind the scene without you needing to take care.
If you use File Upload from your computer :
- You should use `tasmota.bin.gz` and not the `.bin` which is only for serial flashing
- If you still have the size error, upload first the intermediate `tasmota-minimal.bin.gz` then upgra to full version

As a general rule:
- `.bin` are for initial serial flashing
- `.bin.gz` are for upgrade

ascillato2, 2022-01-09:
Topic must be unique for each device. That is the identifier and the way to control each device.
MQTTCLIENT should be different on each device too. If MqttClient is the same, the broker will disconnect the duplicated one. This is enforced by the Mqtt broker and the Mqtt standard.

ascillato2, 2022-01-10:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2022-01-10 by ascillato2

kocherjj, 2022-01-14:
I had time to get back to this again and confirmed that, as I originally described, if I flash a SonoffS31 smart plug with 10.1.0 it will not connect to my Mosquitto MQTT broker. Each S31 has a single identifying string that I use as the device name, the MQTT topic, and MQTT client identifier. This string is unique to this device. It is not present on or associated with any other MQTT device on this network.

If I flash 9.1.0 with all of the same information, it connects just fine and can then be updated to 10.1.0 without losing any functionality.

I did confirm that changing the Topic and Client to different strings has no effect on the connection failure but the device does save the client string instead of resetting. I'm not sure I understand the explanation that the code preventing this was removed in 10.1.0.1. I have no problem setting them duplicate on 9.1 so whatever is preventing it in 10.1 was not present in 9.1.

Leaving this note here in case anyone else comes upon this with the same issue. It has not been answered or resolved yet, but this workaround works for me.

jlr2000, 2022-03-11:
Came here to say Thanks to @kocherjj for posting about issues connecting with higher versions of the tasmota firmware.  I'm completly new to HA and have been trying it out before moving everything over to it.  I got a Sonoff31 and followed all the guides on how to update the firmware and get MQTT setup.  I used v11 of the tasmota firmware, the most recent.  Everything went smooth but I couldn't get the device to connect to the MQTT broker.  After re-trying multiple steps and reviewing other guides I couldn't get it to work until I stumbled here and read this post.  I reflashed the s31 to v9.1.0, set it up and it connected immediately!  I then did an OTA firmare update and all the settings remained.  
Thanks for this information.  Maybe something has changed and this isn't the best way now, but this helped me get this working.