arendst/Tasmota #14338: OnWire with ESP32S (DS18x20)

dknubben, 2022-01-08:
### PROBLEM DESCRIPTION
OneWire library (exactly a DS18B20 sensor) does not work with ESP32S2, (wemos S2 mini) see also
https://github.com/stickbreaker/OneWire/issues/7
and
https://github.com/pstolarz/OneWireNg

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py, platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
22:20:53.305 MQT: stat/tasmota_D841A0/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1312":"DS18x20"},"GPIO14":{"0":"None"},"GPIO15":{"288":"Led1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO33":{"0":"None"},"GPIO34":{"608":"I2C SCL1"},"GPIO35":{"224":"Relay1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"}}

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [x] Provide the output of this command: `Status 0`:

22:09:45.946 MQT: stat/tasmota_D841A0/STATUS = {"Status":{"Module":1,"DeviceName":"","FriendlyName":[""],"Topic":"tasmota_D841A0","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:09:45.967 MQT: stat/tasmota_D841A0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/install/blob/main/firmware/unofficial/tasmota32s2.bin","RestartReason":"Software reset digital core","Uptime":"0T00:43:15","StartupUTC":"2022-01-08T20:26:30","Sleep":50,"CfgHolder":4617,"BootCount":62,"BCResetTime":"2021-12-17T11:35:29","SaveCount":175}}
22:09:45.991 MQT: stat/tasmota_D841A0/STATUS2 = {"StatusFWR":{"Version":"2022.01.2(tasmota)","BuildDateTime":"2022.01.08 19:16:31","Core":"2_0_2","SDK":"v4.4-2.0.2","CpuFrequency":240,"Hardware":"ESP32-S2","CR":"455/699"}}
22:09:46.011 MQT: stat/tasmota_D841A0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxx","xxxxxxxxxx"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
22:09:46.048 MQT: stat/tasmota_D841A0/STATUS4 = {"StatusMEM":{"ProgramSize":1540,"Free":1856,"Heap":111,"StackLowMark":4,"PsrMax":2044,"PsrFree":2032,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000407","9FDAC787","0015A001","FFFFBFCF","05DB9BC0","E0BE0DD7","000A40D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,32,34,35,38,41,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,28,31,34,37,39,40,42,43,45,51,52,54,55,56,57,58,59,64,66,67,74,85,92,127"}}
22:09:46.067 MQT: stat/tasmota_D841A0/STATUS5 = {"StatusNET":{"Hostname":"tasmota-D841A0-0416","IPAddress":"192.168.0.33","Gateway":"192.168.0.11","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.10","DNSServer2":"0.0.0.0","Mac":"84:F7:03:D8:41:A0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:09:46.093 MQT: stat/tasmota_D841A0/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.72","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D841A0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:09:46.116 MQT: stat/tasmota_D841A0/STATUS7 = {"StatusTIM":{"UTC":"2022-01-08T21:09:46","Local":"2022-01-08T22:09:46","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"08:09","Sunset":"16:48"}}
22:09:46.144 MQT: stat/tasmota_D841A0/STATUS10 = {"StatusSNS":{"Time":"2022-01-08T22:09:46","ESP32":{"Temperature":38.4},"TempUnit":"C"}}
22:09:46.171 MQT: stat/tasmota_D841A0/STATUS11 = {"StatusSTS":{"Time":"2022-01-08T22:09:46","Uptime":"0T00:43:16","UptimeSec":2596,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":43,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"POWER":"OFF","Wifi":{"AP":2,"SSId":"xxxxxxxxxx","BSSId":"38:D5:47:XX:XX:XX","Channel":9,"Mode":"11n","RSSI":100,"Signal":-23,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect a DS18B20 with resistor to a ESP32S, ex. GPIO 13

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Show temp...

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
[_Add any other context about the problem here._
See my closed issue here
https://github.com/arendst/Tasmota/issues/14163

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-01-09:
Confirmed DS18B20 not working with the ESP32-S2. It does work with the ESP32.
@dknubben Tasmota does NOT use the orig. stickbreaker/OneWire library. It is enhanced to work with the Core 2.0.x and ESP32 / C3. Seems it needs a another change for the S2.

Jason2866, 2022-01-17:
Fix confirmed. Gpio19 used.

closed 2022-01-17 by Jason2866