arendst/Tasmota #14374: Invalid file signature

flattery103, 2022-01-10:
When attempting to update a brand new Sonoff ZHABridge pre-flashed with Tasmota from CloudFree.shop I get the error Upload Failed Invalid file signature.

I found a previous report saying they forgot to put Backlog Weblog 3; so65 1; Module 75 in consol, but I have done that multiple times and still not working. I followed the steps from https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

Program Version | 10.1.0(tasmota)
-- | --
Build Date & Time | 2021-12-08T14:47:33
Core/SDK Version | 2_7_4_9/2.2.2-dev(38a443e)
Uptime | 0T00:39:32
Flash write Count | 25 at 0xFB000

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-01-10:
Better tell what you did, instead of claiming to have followed some guide. A very common thing is that details are not understood exactly the same way. And do not just ignore filling out the template of information asked for.

One key point is how exactly you downloaded the file you were trying to upgrade to, and where (URL) it came from.

flattery103, 2022-01-10:
-I connected to the device and setup wifi.
- I went to consol and entered Weblog 3; so65 1; Module 75
- I downloaded the latest firmware ncp-uart-nsw_6.7.9_115200.ota from https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp
- After reboot I went to firmware upgrade and added that file.
- Clicked upgrade and got the error

sfromis, 2022-01-10:
How did you download? Not a trick question, it can happen that .ota files are not processed correctly during download, you should generally click the "Download" button on the next page after clicking on the file, and not use right-click (or other browser shortcuts) to download. The file should be 192240 bytes with a MD5 checksum of `6f86ea9c34fe0eff7d75f490d98d0fd9`

barbudor, 2022-01-10:
One of the reason to fill up the template is also to insure that you have looked for similar issues and their solution
[image]

Some little effort would have pointed you also to this opened and solved no latter than 2 days ago
https://github.com/arendst/Tasmota/issues/14331

closed 2022-01-10 by ascillato2

flattery103, 2022-01-11:
I did search for it and found one similar as noted in my very first post, but did not find the one you linked. Apparently, github search isn't very good. Also that form with checkboxes didn't show up until after I posted. It only shows up as very ugly unformatted text when making the initial post so there was no way to click those boxes.

flattery103, 2022-01-11:
I downloaded it exactly as you described and checked the MD5 and it matches 6f86ea9c34fe0eff7d75f490d98d0fd9 but it still gives the same error.

ascillato, 2022-01-11:
Please, what is the output of `module 75` command?

flattery103, 2022-01-11:
03:06:08 CMD: Backlog Weblog 3; so65 1; Module 75
03:06:08 SRC: WebConsole from 192.168.1.190
03:06:08 CMD: Group 0, Index 1, Command "BACKLOG", Data "Weblog 3; so65 1; Module 75"
03:06:08 SRC: Backlog
03:06:08 CMD: Group 0, Index 1, Command "WEBLOG", Data "3"
03:06:08 RSL: stat/tasmota_94CF31/RESULT = {"WebLog":3}
03:06:08 SRC: Backlog
03:06:08 CMD: Group 0, Index 65, Command "SO", Data "1"
03:06:08 RSL: stat/tasmota_94CF31/RESULT = {"SetOption65":"ON"}
03:06:08 SRC: Backlog
03:06:08 CMD: Group 0, Index 1, Command "MODULE", Data "75"
03:06:08 RSL: stat/tasmota_94CF31/RESULT = {"Module":{"0":"Sonoff ZHABridge"}}