arendst/Tasmota #14398: TuyaMCU: Dimmable light dont turn off

Daddel15, 2022-01-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When the dimmer is turned of in Tasmota, light is still on in full brightness. In addition the dimming range is reversed aswell (should be very bright but is pretty dark in reality).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Maxcio Rotary Dimmer (EDM-1WAA-EU)____
- [x] Tasmota binary firmware version number used: 10.1.0_____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: serial to usb_____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:24:42.649 CMD: console Backlog
14:24:42.656 MQT: stat/DimmerBett/RESULT = {"Command":"Unknown"}
14:25:06.636 CMD: Backlog Template
14:25:06.659 MQT: stat/DimmerBett/RESULT = {"NAME":"Maxico Smart Dimmer Switch","GPIO":[1,2272,1,2304,1,1,1,1,1,1,1,1,1,0],"FLAG":0,"BASE":54}
14:25:49.174 CMD: Module
14:25:49.180 MQT: stat/DimmerBett/RESULT = {"Module":{"0":"Maxico Smart Dimmer Switch"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:15:02.351 MQT: stat/DimmerBett/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"DimmerBett","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:15:02.358 MQT: stat/DimmerBett/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:11:18","StartupUTC":"2022-01-12T13:03:44","Sleep":50,"CfgHolder":4617,"BootCount":94,"BCResetTime":"2020-11-12T12:40:17","SaveCount":2487,"SaveAddress":"F4000"}}
14:15:02.366 MQT: stat/DimmerBett/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
14:15:02.372 MQT: stat/DimmerBett/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["It hurts when IP","Guadeloupe"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["001A8009","2805C80001000600003C5A0A000000000000","00000290","0000E000","00004000"]}}
14:15:02.386 MQT: stat/DimmerBett/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:15:02.394 MQT: stat/DimmerBett/STATUS5 = {"StatusNET":{"Hostname":"DimmerBett-7820","IPAddress":"192.168.188.66","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.188.1","DNSServer2":"0.0.0.0","Mac":"40:F5:20:1A:5E:8C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:15:02.401 MQT: stat/DimmerBett/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.59","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1A5E8C","MqttUser":"Leon","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:15:02.409 MQT: stat/DimmerBett/STATUS7 = {"StatusTIM":{"UTC":"2022-01-12T13:15:02","Local":"2022-01-12T14:15:02","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:40","Sunset":"17:16"}}
14:15:02.417 MQT: stat/DimmerBett/STATUS10 = {"StatusSNS":{"Time":"2022-01-12T14:15:02"}}
14:15:02.425 MQT: stat/DimmerBett/STATUS11 = {"StatusSTS":{"Time":"2022-01-12T14:15:02","Uptime":"0T00:11:18","UptimeSec":678,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"It hurts when IP","BSSId":"7C:FF:4D:F4:5D:3B","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:15:42.193 CMD: weblog4
14:15:42.199 MQT: stat/DimmerBett/RESULT = {"WebLog":2}
14:15:47.577 MQT: tele/DimmerBett/STATE = {"Time":"2022-01-12T14:15:47","Uptime":"0T00:12:03","UptimeSec":723,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":21,"MqttCount":1,"POWER":"ON","Dimmer":20,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"It hurts when IP","BSSId":"7C:FF:4D:F4:5D:3B","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:15:47.582 MQT: stat/DimmerBett/RESULT = {"POWER":"ON"}
14:15:47.586 MQT: stat/DimmerBett/POWER = ON
14:15:47.609 MQT: stat/DimmerBett/DPTYPE1ID1 = 1
14:15:51.228 MQT: tele/DimmerBett/STATE = {"Time":"2022-01-12T14:15:51","Uptime":"0T00:12:07","UptimeSec":727,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"It hurts when IP","BSSId":"7C:FF:4D:F4:5D:3B","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:15:51.234 MQT: stat/DimmerBett/RESULT = {"POWER":"OFF"}
14:15:51.238 MQT: stat/DimmerBett/POWER = OFF
14:15:51.259 MQT: stat/DimmerBett/DPTYPE1ID1 = 0

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
applied template
{"NAME":"EDM-1WAA-EU","GPIO":[1,1,1,1,1,1,0,0,1,1,1,1,1,0],"FLAG":0,"BASE":54}

SetOption97 1
TuyaMCU 11,1
TuyaMCU 21,2
DimmerRange 10,1000
setoption54 1
setoption20 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Light off in tasmota
Lights off in reality

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I use parts from to different dimmers. I fried the actual dimmer of an old Maxcio dimmer (since then they updated the daughterboard with the controler so it cant be flashed easily). Since my daughterboard with the tasmota flashed chip survived I tried pluging this into the new dimmer and got this result.

**(Please, remember to close the issue when the problem has been addressed)**

mvanlijden, 2022-01-17:
Since my topic was closed by issue still persists, I'll share some updates from my side.
I am using a WF-DS01 (https://templates.blakadder.com/WF-DS01.html) running Tasmota dev build 2022.01.1 
Template:
`{"NAME":"Dimmer WF-DS01","GPIO":[255,255,255,255,255,255,0,0,255,255,54,255,255],"FLAG":0,"BASE":54}`

My config is setup with following changes to Tasmota default: 
- TuyaMCU 11,1 - Switch on DataPoint 1
- TuyaMCU 21,2 - Dimmer op DataPoint 2
- SetOption20 Off (default) - Update of Dimmer/Color/CT without turning power on 
- SetOption54 Off (default) - Apply SetOption20 settings to commands from Tuya device
- SetOption131 Off (default) - (Tuya) Allow save dimmer = 0 received by MCU
   - Have had little success with the new option
- TuyaSend2 3,100 - Min value is on DataPoint3, setting to 10%
- DimmerRange 10,255
- Ledtable 0
- Fade 0

Behavior is still erratic and I have yet to find a pattern in the logging. I'm not a dev, just a power user :)
I have the feeling that depending on how the light was turned on, the off switch responses are also handled differently. 
Maybe an update of the dimmer value is or isn't send when using MQTT/Web of Physical switch.
It still happens to me that the dimmer turns itself back on (to previous value, I think stored by MCU) when I turn it off.
Below are three log trails, one for successful turning off with MQTT, one for successful turning off with physical button and the unsuccessful turning off using physical button.

Unsuccesful turning off with physical button:
```
10:03:52.360 TYA: Send "55aa00000000ff"
10:03:52.363 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
10:03:52.365 TYA: Heartbeat
10:03:58.625 TYA: fnId=11 is set for dpId=1
10:03:58.627 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:58.637 {"TuyaReceived":{"Data":"55AA0007000802020004000000A5BB","Cmnd":7,"CmndData":"02020004000000A5","DpType2Id2":165,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000A5"}}}
10:03:58.640 TYA: fnId=21 is set for dpId=2
10:03:58.641 TYA: RX value 165 from dpId 2 
10:03:58.702 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
10:03:58.704 TYA: fnId=11 is set for dpId=1
10:03:58.706 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:58.718 {"TuyaReceived":{"Data":"55AA0007000802020004000000A2B8","Cmnd":7,"CmndData":"02020004000000A2","DpType2Id2":162,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000A2"}}}
10:03:58.720 TYA: fnId=21 is set for dpId=2
10:03:58.721 TYA: RX value 162 from dpId 2 
10:03:58.781 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
10:03:58.783 TYA: fnId=11 is set for dpId=1
10:03:58.785 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:58.797 {"TuyaReceived":{"Data":"55AA00070008020200040000009FB5","Cmnd":7,"CmndData":"020200040000009F","DpType2Id2":159,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000009F"}}}
10:03:58.800 TYA: fnId=21 is set for dpId=2
10:03:58.801 TYA: RX value 159 from dpId 2 
10:03:58.862 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
10:03:58.864 TYA: fnId=11 is set for dpId=1
10:03:58.866 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:58.877 {"TuyaReceived":{"Data":"55AA00070008020200040000009CB2","Cmnd":7,"CmndData":"020200040000009C","DpType2Id2":156,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000009C"}}}
10:03:58.879 TYA: fnId=21 is set for dpId=2
10:03:58.881 TYA: RX value 156 from dpId 2 
10:03:58.941 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
10:03:58.943 TYA: fnId=11 is set for dpId=1
10:03:58.945 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:58.956 {"TuyaReceived":{"Data":"55AA00070008020200040000009AB0","Cmnd":7,"CmndData":"020200040000009A","DpType2Id2":154,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000009A"}}}
10:03:58.958 TYA: fnId=21 is set for dpId=2
10:03:58.960 TYA: RX value 154 from dpId 2 
10:03:59.020 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
10:03:59.022 TYA: fnId=11 is set for dpId=1
10:03:59.024 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:59.034 {"TuyaReceived":{"Data":"55AA000700080202000400000097AD","Cmnd":7,"CmndData":"0202000400000097","DpType2Id2":151,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000097"}}}
10:03:59.036 TYA: fnId=21 is set for dpId=2
10:03:59.038 TYA: RX value 151 from dpId 2 
10:03:59.100 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
10:03:59.102 TYA: fnId=11 is set for dpId=1
10:03:59.103 TYA: RX Relay-1 --> MCU State: Off Current State:Off
10:03:59.115 {"TuyaReceived":{"Data":"55AA000700080202000400000094AA","Cmnd":7,"CmndData":"0202000400000094","DpType2Id2":148,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000094"}}}
10:03:59.117 TYA: fnId=21 is set for dpId=2
10:03:59.119 TYA: RX value 148 from dpId 2 
10:03:59.178 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
[... 395 lines cut ...]
10:06:16.425 TYA: fnId=21 is set for dpId=2
[... cut ...]
10:06:31.384 MQT: tele/light-werkkamer-achter_003DFE/STATE = {"Time":"2022-01-17T10:06:31","Uptime":"10T19:35:34","UptimeSec":934534,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":405,"POWER":"OFF","Dimmer":83,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"--","Channel":13,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":6,"Downtime":"0T00:01:18"}}
10:06:37.393 TYA: Send "55aa00000000ff"
10:06:37.396 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
10:06:37.398 TYA: Heartbeat
```

Tortue95, 2022-01-17:
Hi, if the range of the device is 0 to 255 please try:
```
TuyaSend2 3,10
DimmerRange 10,255
SetOption54 1
SetOption131 1
```

mvanlijden, 2022-01-17:
> Hi, if the range of the device is 0 to 255 please try:
> 
> ```
> TuyaSend2 3,10
> DimmerRange 10,255
> SetOption54 1
> SetOption131 1
> ```

This enable the behavior of @Daddel15 not being able to shut down via webinterface. 
First issues a dimmer command to see current dimmer status and then pressed Toggle button to turn off. See below log. 

```
15:31:55.761 CMD: dimmer
15:31:55.763 SRC: WebConsole from 192.168.142.156
15:31:55.766 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 0, Data ''
15:31:55.771 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"Dimmer":78}
15:31:57.738 TYA: Send "55aa00000000ff"
15:32:02.117 SRC: WebGui from 192.168.142.156
15:32:02.136 TYA: Send "55aa0006000802020004000000c9de"
15:32:02.151 TYA: Send "55aa0006000501010001000d"
15:32:02.176 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"OFF"}
15:32:02.181 MQT: stat/light-werkkamer-achter_003DFE/POWER = OFF
15:32:02.493 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
15:32:02.495 TYA: Heartbeat
15:32:02.500 {"TuyaReceived":{"Data":"55AA0007000802020004000000C9DF","Cmnd":7,"CmndData":"02020004000000C9","DpType2Id2":201,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C9"}}}
15:32:02.503 TYA: fnId=21 is set for dpId=2
15:32:02.504 TYA: RX value 201 from dpId 2 
15:32:02.507 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
15:32:02.510 TYA: fnId=11 is set for dpId=1
15:32:02.511 TYA: RX Relay-1 --> MCU State: On Current State:Off
15:32:02.513 SRC: Switch
15:32:02.519 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"ON"}
15:32:02.524 MQT: stat/light-werkkamer-achter_003DFE/POWER = ON
15:32:02.528 {"TuyaReceived":{"Data":"55AA0007000802020004000000C6DC","Cmnd":7,"CmndData":"02020004000000C6","DpType2Id2":198,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C6"}}}
15:32:02.532 TYA: fnId=21 is set for dpId=2
15:32:02.534 TYA: RX value 198 from dpId 2 
15:32:03.083 WIF: Checking connection...
15:32:08.472 TYA: fnId=11 is set for dpId=1
15:32:08.474 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:32:08.477 {"TuyaReceived":{"Data":"55AA0007000802020004000000C6DC","Cmnd":7,"CmndData":"02020004000000C6","DpType2Id2":198,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C6"}}}
15:32:08.480 TYA: fnId=21 is set for dpId=2
15:32:08.482 TYA: RX value 198 from dpId 2 
15:32:08.486 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
15:32:08.488 TYA: Heartbeat
15:32:08.529 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:32:08.530 TYA: fnId=11 is set for dpId=1
15:32:08.532 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:32:08.544 {"TuyaReceived":{"Data":"55AA0007000802020004000000C4DA","Cmnd":7,"CmndData":"02020004000000C4","DpType2Id2":196,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C4"}}}
15:32:08.547 TYA: fnId=21 is set for dpId=2
15:32:08.549 TYA: RX value 196 from dpId 2 
15:32:08.608 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:32:08.611 TYA: fnId=11 is set for dpId=1
15:32:08.613 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:32:08.623 {"TuyaReceived":{"Data":"55AA0007000802020004000000C1D7","Cmnd":7,"CmndData":"02020004000000C1","DpType2Id2":193,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C1"}}}
15:32:08.626 TYA: fnId=21 is set for dpId=2
15:32:08.628 TYA: RX value 193 from dpId 2 
15:32:08.688 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:32:08.691 TYA: fnId=11 is set for dpId=1
15:32:08.693 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:32:08.703 {"TuyaReceived":{"Data":"55AA0007000802020004000000BED4","Cmnd":7,"CmndData":"02020004000000BE","DpType2Id2":190,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000BE"}}}
15:32:08.706 TYA: fnId=21 is set for dpId=2
15:32:08.708 TYA: RX value 190 from dpId 2 
15:32:08.767 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:32:08.770 TYA: fnId=11 is set for dpId=1
15:32:08.772 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:32:08.783 {"TuyaReceived":{"Data":"55AA0007000802020004000000BBD1","Cmnd":7,"CmndData":"02020004000000BB","DpType2Id2":187,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000BB"}}}
15:32:08.785 TYA: fnId=21 is set for dpId=2
15:32:08.787 TYA: RX value 187 from dpId 2 
15:32:08.788 SRC: Switch
15:32:08.791 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '72'
15:32:08.794 SRC: Light
15:32:08.810 TYA: Send "55aa0006000501010001010e"
15:32:08.836 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"ON","Dimmer":72}
15:32:08.842 {"TuyaReceived":{"Data":"55AA0007000802020004000000C9DF","Cmnd":7,"CmndData":"02020004000000C9","DpType2Id2":201,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C9"}}}
15:32:08.844 TYA: fnId=21 is set for dpId=2
15:32:08.846 TYA: RX value 201 from dpId 2 
15:32:08.848 SRC: Switch
15:32:08.850 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '78'
15:32:08.858 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"ON","Dimmer":78}
15:32:08.863 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
[... cut ...]
15:32:19.417 TYA: Send "55aa00000000ff"
15:32:19.425 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
15:32:19.427 TYA: fnId=11 is set for dpId=1
15:32:19.429 TYA: RX Relay-1 --> MCU State: On Current State:On
15:32:23.334 WIF: Checking connection...
15:32:30.421 TYA: Send "55aa00000000ff"
15:32:30.424 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
15:32:30.426 TYA: Heartbeat
```

mvanlijden, 2022-01-17:
> Hi, if the range of the device is 0 to 255 please try:
> 
> ```
> TuyaSend2 3,10
> DimmerRange 10,255
> SetOption54 1
> SetOption131 1
> ```
Even when having below config and the correct lower limit values of 10, the result is still not working.
```
SetOption20 1 - Update of Dimmer/Color/CT without turning power on
SetOption54 1 - Apply SetOption20 settings to commands from Tuya device
SetOption131 1 - (Tuya) Allow save dimmer = 0 received by MCU
```
Setting the dimmer to 80% and then pressing off button does not work. The dimmer goes down and shuts off for as a second, but after that he send the original value before turning off. Even though SetOption20 should prevent this from triggering the power, it does and the light goes back on at 80%...

```
17:42:33.051 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:33.054 TYA: fnId=11 is set for dpId=1
17:42:33.055 TYA: RX Relay-1 --> MCU State: Off Current State:On
17:42:33.057 SRC: Switch
17:42:33.075 TYA: Send "55aa0006000802020004000000cee3"
17:42:33.080 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"OFF"}
17:42:33.086 MQT: stat/light-werkkamer-achter_003DFE/POWER = OFF
17:42:33.089 {"TuyaReceived":{"Data":"55AA0007000802020004000000CCE2","Cmnd":7,"CmndData":"02020004000000CC","DpType2Id2":204,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000CC"}}}
17:42:33.091 TYA: fnId=21 is set for dpId=2
17:42:33.093 TYA: RX value 204 from dpId 2 
17:42:33.096 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:42:33.098 TYA: Heartbeat
17:42:33.138 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:33.140 TYA: fnId=11 is set for dpId=1
17:42:33.142 TYA: RX Relay-1 --> MCU State: Off Current State:Off
17:42:33.153 {"TuyaReceived":{"Data":"55AA0007000802020004000000C9DF","Cmnd":7,"CmndData":"02020004000000C9","DpType2Id2":201,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C9"}}}
17:42:33.156 TYA: fnId=21 is set for dpId=2
17:42:33.157 TYA: RX value 201 from dpId 2 
17:42:33.217 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:33.220 TYA: fnId=11 is set for dpId=1
17:42:33.221 TYA: RX Relay-1 --> MCU State: Off Current State:Off
17:42:33.232 {"TuyaReceived":{"Data":"55AA0007000802020004000000C6DC","Cmnd":7,"CmndData":"02020004000000C6","DpType2Id2":198,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C6"}}}
17:42:33.234 TYA: fnId=21 is set for dpId=2
17:42:33.236 TYA: RX value 198 from dpId 2 
17:42:33.297 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:33.300 TYA: fnId=11 is set for dpId=1
17:42:33.302 TYA: RX Relay-1 --> MCU State: Off Current State:Off
17:42:33.313 {"TuyaReceived":{"Data":"55AA0007000802020004000000C4DA","Cmnd":7,"CmndData":"02020004000000C4","DpType2Id2":196,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C4"}}}
17:42:33.315 TYA: fnId=21 is set for dpId=2
17:42:33.317 TYA: RX value 196 from dpId 2 
17:42:33.430 CFG: Saved to flash at F5, Count 654, Bytes 4096
17:42:33.438 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:33.441 TYA: fnId=11 is set for dpId=1
17:42:33.443 TYA: RX Relay-1 --> MCU State: Off Current State:Off
17:42:33.448 {"TuyaReceived":{"Data":"55AA0007000802020004000000C1D7","Cmnd":7,"CmndData":"02020004000000C1","DpType2Id2":193,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C1"}}}
17:42:33.450 TYA: fnId=21 is set for dpId=2
17:42:33.452 TYA: RX value 193 from dpId 2 
17:42:33.454 SRC: Switch
17:42:33.456 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '75'
17:42:33.465 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"OFF","Dimmer":75}
17:42:33.471 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:33.473 TYA: fnId=11 is set for dpId=1
17:42:33.474 TYA: RX Relay-1 --> MCU State: Off Current State:Off
17:42:33.479 {"TuyaReceived":{"Data":"55AA0007000802020004000000BED4","Cmnd":7,"CmndData":"02020004000000BE","DpType2Id2":190,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000BE"}}}
17:42:33.481 TYA: fnId=21 is set for dpId=2
17:42:33.483 TYA: RX value 190 from dpId 2 
17:42:35.460 TYA: RX value 120 from dpId 2 
17:42:35.462 SRC: Switch
17:42:35.464 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '45'
17:42:35.472 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"OFF","Dimmer":45}
17:42:35.521 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
17:42:35.524 TYA: fnId=11 is set for dpId=1
17:42:35.526 TYA: RX Relay-1 --> MCU State: Off Current State:Off
[... 118 lines cut ...]
17:42:38.881 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
17:42:38.884 TYA: fnId=21 is set for dpId=2
17:42:38.886 TYA: RX value 0 from dpId 2 
17:42:39.393 CFG: Saved to flash at F7, Count 660, Bytes 4096
17:42:40.765 TYA: Send "55aa00000000ff"
[... cut ...]
17:42:40.812 MQT: stat/light-werkkamer-achter_003DFE/POWER = ON
17:42:40.815 {"TuyaReceived":{"Data":"55AA0007000802020004000000CCE2","Cmnd":7,"CmndData":"02020004000000CC","DpType2Id2":204,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000CC"}}}
17:42:40.817 TYA: fnId=21 is set for dpId=2
17:42:40.820 TYA: RX value 204 from dpId 2 
17:42:41.389 CFG: Saved to flash at F6, Count 661, Bytes 4096
17:42:46.773 TYA: Send "55aa001c000801160111112a2e01b6"
17:42:46.776 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:42:46.778 TYA: Heartbeat
```

Tortue95, 2022-01-18:
Hello, I admit that I don't understand what is happening, because nothing is sent from tasmota except the heartbeat, and at the end of the dimming as you say "Suddenly dimmer update to previous value before turning off" but it comes from the device no tasmota...  (you can see the sended command form tasmota in log with the cmd: "TYA: Send ...")

Just for information, if i look the code, SetOption20 is not used with Tuya but SetOption54: allows to save the value of the dimmer from MCU in tasmota even when the light is OFF (the current documentation is not clear about it)

Tortue95, 2022-01-18:
In fact, this comes from the beginning:
17:42:33.075 TYA: Send "55aa0006000802020004000000cee3"

your device performs its dimming routine when it goes off and only after it executes the received command
it would be necessary to log before this line to see what happened

mvanlijden, 2022-01-18:
> Hello, I admit that I don't understand what is happening, because nothing is sent from tasmota except the heartbeat, and at the end of the dimming as you say "Suddenly dimmer update to previous value before turning off" but it comes from the device not from tasmota... (you can see the sended command form tasmota in log with: "TYA: Send ...")
No that's also what I see. This is not something that tasmota triggers, but it is something it is not processing correctly when received. The MCU seems to sent the previous dim value after shutting down when using the physical button to turn off. Tasmota receives this and sets the dimmer value, but incorrectly reactivates the power as well.

> Just for information, if i look the code, SetOption20 is not used with Tuya but SetOption54: allows to save the value of the dimmer from MCU in tasmota even when the light is OFF (the current documentation is not clear about it)
That is really strange, than maybe this is where the error is. Because when I do not use SetOption20 (but do use 54 and 131), the light does not even dim to 0 before turning back on, it just flickers and stays on. Unable to shut down even when using web or mqtt.

Hearing this I would believe that SetOption54 is not working as it should. Because like you see in the logs, I receive the dimmer update back to 80% after shutting down from the MCU, but this also triggers a power switch action on the side of tasmota it seems. Or am I reading the sequence of the last actions wrong?

mvanlijden, 2022-01-18:
> In fact, this comes from the beginning: 17:42:33.075 TYA: Send "55aa0006000802020004000000cee3"
> 
> your device performs its dimming routine when it goes off and only after it executes the received command it would be necessary to log before this line to see what happened

Here is a new log, same settings (20,54,131 on). Dimmer was at 1% so it to 80% waited a bit to shot the heartbeats and then turned it off

```
11:10:16.591 TYA: Send "55aa00000000ff"
11:10:16.595 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:10:16.597 TYA: Heartbeat
11:10:25.387 WIF: Checking connection...
11:10:27.591 TYA: Send "55aa00000000ff"
11:10:27.594 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:10:27.597 TYA: Heartbeat
11:10:38.588 TYA: Send "55aa00000000ff"
11:10:38.591 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:10:38.594 TYA: Heartbeat
11:10:45.388 WIF: Checking connection...
11:10:49.595 TYA: Send "55aa00000000ff"
11:10:49.612 TYA: Send "55aa001c0008011601120b0a310295"
11:10:49.615 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:10:49.617 TYA: Heartbeat
11:10:49.619 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:10:49.620 TYA: Heartbeat
11:10:50.581 WIF: Sending Gratuitous ARP
11:11:00.591 TYA: Send "55aa00000000ff"
11:11:05.390 WIF: Checking connection...
11:11:11.591 TYA: Send "55aa00000000ff"
11:11:11.595 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:11:11.597 TYA: Heartbeat
11:11:22.589 TYA: Send "55aa00000000ff"
11:11:22.592 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:11:22.595 TYA: Heartbeat
11:11:25.385 WIF: Checking connection...
11:11:33.593 TYA: Send "55aa00000000ff"
11:11:33.597 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:11:33.599 TYA: Heartbeat
11:11:44.588 TYA: Send "55aa00000000ff"
11:11:44.591 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:11:44.594 TYA: Heartbeat
11:11:45.382 WIF: Checking connection...
11:11:49.597 TYA: Send "55aa001c0008011601120b0b310296"
11:11:49.601 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:11:49.603 TYA: Heartbeat
11:11:50.580 WIF: Sending Gratuitous ARP
11:11:55.597 TYA: Send "55aa00000000ff"
11:12:05.389 WIF: Checking connection...
11:12:06.598 TYA: Send "55aa00000000ff"
11:12:06.601 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:12:06.604 TYA: Heartbeat
11:12:10.594 MQT: tele/light-werkkamer-achter_003DFE/STATE = {"Time":"2022-01-18T11:12:10","Uptime":"0T17:15:23","UptimeSec":62123,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":3,"POWER":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":2,"Downtime":"0T00:00:04"}}
11:12:17.592 TYA: Send "55aa00000000ff"
11:12:17.595 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:12:17.597 TYA: Heartbeat
11:12:25.390 WIF: Checking connection...
11:12:28.594 TYA: Send "55aa00000000ff"
11:12:28.597 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:12:28.600 TYA: Heartbeat
11:12:39.588 TYA: Send "55aa00000000ff"
11:12:39.591 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:12:39.593 TYA: Heartbeat
11:12:45.386 WIF: Checking connection...
11:12:49.605 TYA: Send "55aa001c0008011601120b0c310297"
11:12:49.608 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:12:49.610 TYA: Heartbeat
11:12:50.582 WIF: Sending Gratuitous ARP
11:12:50.592 TYA: Send "55aa00000000ff"
11:13:01.595 TYA: Send "55aa00000000ff"
11:13:01.598 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:13:01.600 TYA: Heartbeat
[... 196 lines cut ...]
11:14:29.650 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"OFF","Dimmer":80}
11:14:29.655 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
11:14:29.658 TYA: fnId=11 is set for dpId=1
11:14:29.660 TYA: RX Relay-1 --> MCU State: On Current State:Off
11:14:29.662 SRC: Switch
11:14:29.669 MQT: stat/light-werkkamer-achter_003DFE/RESULT = {"POWER":"ON"}
11:14:29.674 MQT: stat/light-werkkamer-achter_003DFE/POWER = ON
11:14:29.678 {"TuyaReceived":{"Data":"55AA0007000802020004000000CCE2","Cmnd":7,"CmndData":"02020004000000CC","DpType2Id2":204,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000CC"}}}
11:14:29.680 TYA: fnId=21 is set for dpId=2
11:14:29.683 TYA: RX value 204 from dpId 2 
11:14:30.200 CFG: Saved to flash at FA, Count 697, Bytes 4096
11:14:40.592 TYA: Send "55aa00000000ff"
11:14:40.595 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:14:40.597 TYA: Heartbeat
11:14:45.384 WIF: Checking connection...
11:14:49.599 TYA: Send "55aa001c0008011601120b0e310299"
11:14:49.602 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:14:49.604 TYA: Heartbeat
11:14:50.581 WIF: Sending Gratuitous ARP
11:14:51.589 TYA: Send "55aa00000000ff"
11:15:02.591 TYA: Send "55aa00000000ff"
11:15:02.594 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:15:02.597 TYA: Heartbeat
11:15:05.386 WIF: Checking connection...
```

Tortue95, 2022-01-18:
in fact it would seem that it is your MCU which turns the lamp back ON when it receives a heartbeat... it's curious.

mvanlijden, 2022-01-18:
> in fact it would seem that it is your MCU which turns the lamp back ON when it receives a heartbeat... it's curious.

Nice "feature" tuya .... :(

Tortue95, 2022-01-18:
yes each manufacturer does a little what he wants... and does not necessarily respect the tuya protocol...

mvanlijden, 2022-01-18:
I have no idea why.... I just updated to the lasted dev build 2022.01.3 and my dimmer work perfectly now....
I also tested this on a seperate device I was using to test espHome (as alternative but not working as well) and did a reset 5 to be sure and started from scratch. I also see the dimmer value of the fade out is not being stored on turning off. 

The config that works for me is really just this, without any SetOption20,54,131 turned on:
TuyaMCU 21,2
DimmerRange 10,255

I also tested this on previous builds and then it failed, so no idea what they updated? There was a rewrite of the tuyaMCU to support ESP32, but I thought that was already in the previous build as well...

ascillato2, 2022-03-16:
Moving this issue to Discussions as it contains valuable information for anyone interested on the Tuya Driver enhance.

Thanks.