arendst/Tasmota #14399: TuyaMCU cant set FnId to 97

MadDoct, 2022-01-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): tuya curtain motor
- [x ] Tasmota binary firmware version number used: 10.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: nodemcupyflasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

14:47:33.123 MQT: stat/CortinaQMeninas/STATUS = {"Status":{"Module":54,"DeviceName":"Cortina Quarto Meninas","FriendlyName":["Cortina Quarto Meninas"],"Topic":"CortinaQMeninas","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:47:33.147 MQT: stat/CortinaQMeninas/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.2.3:9541/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:57:04","StartupUTC":"2022-01-12T12:50:29","Sleep":50,"CfgHolder":4617,"BootCount":74,"BCResetTime":"2020-06-07T15:29:32","SaveCount":1096,"SaveAddress":"FA000"}}
14:47:33.167 MQT: stat/CortinaQMeninas/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"452/699"}}
14:47:33.191 MQT: stat/CortinaQMeninas/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SnowMania","NOT4iPHONE"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["80008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
14:47:33.220 MQT: stat/CortinaQMeninas/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:47:33.245 MQT: stat/CortinaQMeninas/STATUS5 = {"StatusNET":{"Hostname":"CortinaQMeninas-1536","IPAddress":"192.168.2.164","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"2C:F4:32:23:86:00","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:47:33.265 MQT: stat/CortinaQMeninas/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_238600","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:47:33.290 MQT: stat/CortinaQMeninas/STATUS7 = {"StatusTIM":{"UTC":"2022-01-12T14:47:33","Local":"2022-01-12T14:47:33","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:40","Sunset":"16:16"}}
14:47:33.314 MQT: stat/CortinaQMeninas/STATUS10 = {"StatusSNS":{"Time":"2022-01-12T14:47:33"}}
14:47:33.337 MQT: stat/CortinaQMeninas/STATUS11 = {"StatusSTS":{"Time":"2022-01-12T14:47:33","Uptime":"0T01:57:04","UptimeSec":7024,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SnowMania","BSSId":"50:C7:BF:16:92:90","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:26"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
TuyaMCU comes with DpId 1 defines as FnId 11
When I try to define DpId 1 (it is a motor direction indicator - 0 closing, 1 stopped, 2 opening) as FnId 97, it reboots and resets do FnId 11.
The only thing I can do is define it as FnId 99 and, in that case, it accepts the change

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
To be able to set DpId 1 as FnId 97

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-01-12:
Adding the label **Help Needed** due to there is no active development on tuya drivers for Tasmota.

PhilipTS, 2022-01-15:
@MadDoct could you post the output of TuyaSend0?
It will give us all the DpIds and their Type

MadDoct, 2022-01-15:
Here you go:

```
13:08:43.001 CMD: Tuyasend0
13:08:43.003 SRC: WebConsole from 192.168.x.x
13:08:43.005 CMD: Grp 0, Cmnd 'TUYASEND', Idx 0, Len 0, Data ''
13:08:43.008 TYA: Read MCU state
13:08:43.017 TYA: Send "55aa0008000007"
13:08:43.022 MQT: stat/CortinaQMeninas/RESULT = {"TuyaSend":"Done"}
13:08:43.064 {"TuyaReceived":{"Data":"55AA030700170104000101020200040000000005010001010A0500010047","Cmnd":7,"CmndData":"0104000101020200040000000005010001010A05000100","DpType4Id1":1,"1":{"DpId":1,"DpIdType":4,"DpIdData":"01"},"DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"},"DpType1Id5":1,"5":{"DpId":5,"DpIdType":1,"DpIdData":"01"},"DpType5Id10":"0x00","10":{"DpId":10,"DpIdType":5,"DpIdData":"00"}}}
13:08:43.067 TYA: fnId=99 is set for dpId=1
13:08:43.069 TYA: fnId=21 is set for dpId=2
13:08:43.070 TYA: RX value 0 from dpId 2 
13:08:43.072 TYA: fnId=0 is set for dpId=5
13:08:43.074 TYA: fnId=0 is set for dpId=10
13:08:44.794 TYA: Send "55aa00000000ff"
13:08:44.808 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
13:08:44.810 TYA: Heartbeat
13:08:49.794 TYA: Send "55aa001c00080116010f0d08310696"
13:08:55.793 TYA: Send "55aa00000000ff"
13:08:55.806 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
13:08:55.808 TYA: Heartbeat
13:08:57.858 WIF: Checking connection.
```

PhilipTS, 2022-01-15:
@MadDoct should have also mentioned that it would help if you put the log in debug mode :)
weblog 4
TuyaSend0

PhilipTS, 2022-01-15:
@MadDoct but all the needed info is there ... did not notice ...

DpType4Id1":1
DpType2Id2":0
DpType1Id5":1
DpType5Id10":"0x00"

So I see that DdId 1 is type 4 = enum

When you set with TuyaSend4 1,x does it work as expected?

PhilipTS, 2022-01-15:
@MadDoct so in the code it checks to see if we have a relay set. If not it sets TuyaMCU 11,1
So you need to go around this check. So try

TuyaMCU 18,200 (assign relay 18 to DpId 200, there is no relay 8 and you will never receive a DpId 200)
TuyaMCU 11,0     (clear the relay1 assignment)
TuyaMCU 97,1

I assume that TuyaSend4 1,x works for you ... (in other words that DpId is settable and not read only)

PhilipTS, 2022-01-15:
And this goes to our super tasmota gurus ...

In file - xdrv_16_tuyamcu.ino
In function - bool TuyaModuleSelected(void)
we have the below code
It would be nice to have an option to diasble this default assingment of relay1 to DpId1

````
  bool relaySet = false;

  for (uint8_t i = 0 ; i < MAX_TUYA_FUNCTIONS; i++) {
    if ((Settings->tuya_fnid_map[i].fnid >= TUYA_MCU_FUNC_REL1 && Settings->tuya_fnid_map[i].fnid <= TUYA_MCU_FUNC_REL8 ) ||
    (Settings->tuya_fnid_map[i].fnid >= TUYA_MCU_FUNC_REL1_INV && Settings->tuya_fnid_map[i].fnid <= TUYA_MCU_FUNC_REL8_INV )) {
      relaySet = true;
      TasmotaGlobal.devices_present++;
    }
  }

  if (!relaySet && TuyaGetDpId(TUYA_MCU_FUNC_DUMMY) == 0) { //by default the first relay is created automatically the dummy let remove it if not needed
    TuyaAddMcuFunc(TUYA_MCU_FUNC_REL1, 1);
    TasmotaGlobal.devices_present++;
    SettingsSaveAll();
  }
````

PhilipTS, 2022-01-15:
``
16:51:13.733 CMD: TuyaMCU
16:51:13.746 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1}]}

16:51:53.945 CMD: TuyaMCU 18,200
16:51:53.959 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1},{"fnId":18,"dpId":200}]}
``
It will auto reset
``
16:52:26.339 CMD: TuyaMCU
16:52:26.357 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1},{"fnId":18,"dpId":200}]}

16:53:01.592 CMD: TuyaMCU 11,0
16:53:01.605 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":18,"dpId":200}]}
``
It will auto reset
``
16:53:30.298 CMD: TuyaMCU
16:53:30.311 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":18,"dpId":200}]}

16:54:07.016 CMD: TuyaMCU 97,1
16:54:07.029 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":97,"dpId":1},{"fnId":18,"dpId":200}]}
``
It will auto reset
``
16:54:34.554 CMD: TuyaMCU
16:54:34.567 MQT: ESP32_ME102H/stat/RESULT = {"TuyaMCU":[{"fnId":97,"dpId":1},{"fnId":18,"dpId":200}]}
``

MadDoct, 2022-01-15:
I still can't do it, but maybe I misinterpreted the usage of fnid 97. I thought it was a "sensor", but it may be a "switch"

PhilipTS, 2022-01-15:
@MadDoct your MCU report the below DpIds ... try using the relevant TuyaSendX commands to send diferent values to figure out what each DpId is.

DpType4Id1":1     -     TuyaSend4 X,1
DpType2Id2":0     -     TuyaSend2 X,1
DpType1Id5":1     -     TuyaSend1 X,5
DpType5Id10":"0x00"

Change one every time and then run TuyaSend0 to query the MCU and see if it changed.
Some might be read only to report a status and not change it

MadDoct, 2022-01-15:
I can confirm that it indeed is writable, so I can't understand why I can't set the FnId...

PhilipTS, 2022-01-15:
did you try the above sequence?

TuyaMCU 18,200
TuyaMCU 11,0
TuyaMCU 97,1

The code looks for a relay assignment and if there is none it does TuyaMCU 11,1
So we have to trick/fool it.

We assign a fake relay 8 (fnid 18) to a fake DpId 200. This will keep tasmota happy.
We can then unassigned relay 1 (fnid 1)  from DpId 1
Now we can assign TuyaMCU 97,1

MadDoct, 2022-01-15:
Ok, It didn't work previosuly, but now worked...

It shouldn't be becessary, but ok.

Thanks!

closed 2022-01-15 by MadDoct