arendst/Tasmota #14404: sendmail cont connect to smtp.gmail.com

mrpatate76, 2022-01-12:
Hello, i try to send an email on high température into a script

all that is in my sonoff zigbee bridge, flashed with tasmota of course

so far, manage to print on high température and link it to a bulb, to be sure that i am going to the right place in the code

19:24:08.916 CMD: sendmail [*:*:*:*:*:mrpatate76@gmail.com:test sendmail] juste un test
19:24:08.918 mailsize: 44
19:24:08.919 smtp.gmail.com - 465 - mrpatate76@gmail.com - password
19:24:08.920 mrpatate76@gmail.com - mrpatate76@gmail.com - test sendmail -  juste un test
19:24:08.921 Connecting: smtp.gmail.com on port 465
19:24:09.021 Connection failed
19:24:09.024 RSL: RESULT = {"Sendmail":"Wrong parameters"}

i test my password to be sure it is working ( in app paswaord) and it is ok, so i guess it is something in code or related to authentification on gmail side

i compile using tasmotiser including  this
#ifndef USE_SCRIPT
#define USE_SCRIPT  // adds about 17k flash size, variable ram size
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif 

#define DEBUG_EMAIL_PORT Serial
#define USE_BUTTON_EVENT
#define SCRIPT_POWER_SECTION
#define USE_SENDMAIL
#define USE_SCRIPT_WEB_DISPLAY
#define SUPPORT_MQTT_EVENT
#define SUPPORT_IF_STATEMENT
#define USE_SCRIPT_SUB_COMMAND

#define EMAIL_USER "mrpatate76@gmail.com"
#define EMAIL_PASSWORD "password"
#define EMAIL_FROM "<mrpatate76@gmail.com>"
#define EMAIL_SERVER "smtp.gmail.com"
#define EMAIL_PORT 465
#define MAIL_TIMEOUT 2000

barbudor, 2022-01-12:
> test my password to be sure it is working ( in app paswaord) and it is ok,

Are you using your Google account password ?
This will not work
You must create an "application specific password"
https://support.google.com/mail/answer/185833

barbudor, 2022-01-12:
And you need to set the EMAIL_FROM with `<` and `>`
```
    #define USE_SENDMAIL
    #define EMAIL_USER "username"
    #define EMAIL_PASSWORD "xxxxxxxxxx"
    #define EMAIL_FROM "<username@gmail.com>"
    #define EMAIL_SERVER "smtp.gmail.com"
    #define EMAIL_PORT 465
 ```

mrpatate76, 2022-01-12:
i use 

sendmail [smtp.gmail.com:25:mrpatate76@gmail.com:16caracter password for in app:<mrpatate76@gmail.com>:<mrpatate76@gmail.com>:alarme temperature frigo] Important, Verifier le frigo, la temperature est elevee

always same result : 

21:05:53.518 CMD: sendmail [smtp.gmail.com:25:mrpatate76@gmail.com:**password**:<mrpatate76@gmail.com>:<mrpatate76@gmail.com>:alarme temperature frigo] Important, Verifier le frigo, la temperature est elevee
21:05:53.521 mailsize: 126
21:05:53.522 smtp.gmail.com - 25 - mrpatate76@gmail.com - **password**
21:05:53.523 <mrpatate76@gmail.com> - <mrpatate76@gmail.com> - alarme temperature frigo -  Important, Verifier le frigo, la temperature est elevee
21:05:53.524 Connecting: smtp.gmail.com on port 25
21:05:55.581 Connection failed
21:05:55.584 RSL: RESULT = {"Sendmail":"Wrong parameters"}

mrpatate76, 2022-01-12:
i think it is a authentification or  Secured Connection matter from gmail server

mrpatate76, 2022-01-12:
still the same with or without the <>

21:10:56.987 CMD: sendmail [smtp.gmail.com:465:mrpatate76@gmail.com:**password**:<mrpatate76@gmail.com>:<mrpatate76@gmail.com>:alarme temperature frigo] Important, Verifier le frigo, la temperature est elevee
21:10:56.990 mailsize: 127
21:10:56.991 smtp.gmail.com - 465 - mrpatate76@gmail.com - **password**
21:10:56.992 <mrpatate76@gmail.com> - <mrpatate76@gmail.com> - alarme temperature frigo -  Important, Verifier le frigo, la temperature est elevee
21:10:56.993 Connecting: smtp.gmail.com on port 465
21:10:57.079 Connection failed
21:10:57.082 RSL: RESULT = {"Sendmail":"Wrong parameters"}

barbudor, 2022-01-12:
Did you followed the instruction for application specific password ?

It will NOT work with your Google Account password

mrpatate76, 2022-01-12:
yes got the 16 letters for application specific password. try it on my phone and worked, but same one in tasmota get wrong parameter, and send nothing since not activity is shown on my google account  page

barbudor, 2022-01-12:
Don't send to yourself, I think gmail is not supporting that
I'm recompiling a version with sendmail to test again as I haven't used that for months

mrpatate76, 2022-01-12:
oh,  good to know, i will try  to send it to my other email, juste to see if it work.

dont look like it change anything, let me know if you got it to work on your side

thanks for the help

barbudor, 2022-01-12:
Can't get it to work, neither with gmail nor my local smtp server
Looks like something might be broken
will need more investigation

Alternatively an easy way to send an simple email is to use a WebHook with IFTTT or Integromat or Zappier that you can call from Tasmota using WebQuery command
You can also use Discord with a WebHook too

mrpatate76, 2022-01-13:
hello,
I set-up for WebQuery 
i compile adding
#define USE_WEBSEND_RESPONSE
on ifttt, i manage to get it working for webhook, and my email, i can trigger online using 
https://maker.ifttt.com/trigger/alarme/with/key/d1kz82kwUI0Tr8SThMsx1X
and this work, since i get email 

but in tasmota console, when i try to do the WebQuery unsing: 

WebQuery Https://maker.ifttt.com/trigger/alarme/with/key/d1kz82kwUI0Tr8SThMsx1X GET
i received : Host not found. As you can see below:

is there anything else that need to be compile to get it working?

thanks again for your help

17:12:02.846 CMD: WebQuery Https://maker.ifttt.com/trigger/alarme/with/key/d1kz82kwUI0Tr8SThMsx1X GET
17:12:02.848 SRC: WebConsole from 10.0.0.157
17:12:02.849 CMD: Grp 0, Cmnd 'WEBQUERY', Idx 1, Len 74, Data 'Https://maker.ifttt.com/trigger/alarme/with/key/d1kz82kwUI0Tr8SThMsx1X GET'
17:12:02.852 RSL: RESULT = {"WebQuery":"Host not found"}

mrpatate76, 2022-01-13:
got two alarm mail from ifttt

barbudor, 2022-01-13:
You just got mail another mail :)
I suggest you delete you IFTTT action and recreate it with another token if you don't want to receive alerts from jokers of my kind or worst

gemu2015, 2022-01-14:
ok tested email on esp32 and esp8266. (both use very different approaches)

esp32 works without any problems. 
keep in mind that emails must not be enclosed in sharp brackets here while in esp8266 it has to be

esp8266 indeed does no longer connect to gmail in the latest dev version. however at least in an older version (9.4) it works well. so it cannot be due to change at gmail.
since nothing has changed within email driver itself some changes in light_bear_ssl or core must cause this failure.
will take some time to evaluate this issue.

gemu2015, 2022-01-14:
it turns out that light_bear is the cause. taking the ssl lib from 9.4 into latest dev solves the problem.
@s-hadinger may be you can determine what causes failure in SMPT  for ESP8266

gemu2015, 2022-01-14:
found the reason and fixed the issue. will do a pr later today

s-hadinger, 2022-01-14:
Thanks @gemu2015 I'm curious what was the problem?

gemu2015, 2022-01-14:
simple esp8266 smtp needs insecure TLS.  the new default was insecure set to false.
i simply introduced a set insecure call like in standard TLS.

barbudor, 2022-01-14:
Thanks @gemu2015

barbudor, 2022-01-14:
Which tasmota are you compiling ? "tasmota.bin" ?
When customizing you must compile "tasmota" and not any variant (such as tasmota-sensors or other)

mrpatate76, 2022-01-14:
i use tasmotizer for sonoff zigbee bridge, and passing custum parameters

mrpatate76, 2022-01-14:
by the way, i manage to get an email using

=>WebSend [maker.ifttt.com] /trigger/alarme/with/key/d1kz82kwUI0Tr8SThMsx1X

barbudor, 2022-01-14:
Tasmotizer is a PC app to flash firmware. It's not a compiler.

Are you talking about TasmoCompiler ?

mrpatate76, 2022-01-14:
sorry you are right
tasmocompiler

ESP8266 with onOff Zigbee Bridge
option for 
Light sensors
Temp/Hum sensors
Timers
Display Vcc
Web interface
Script

#ifndef USE_SCRIPT
#define USE_SCRIPT  // adds about 17k flash size, variable ram size
#endif

custom parameters:
#define USE_WEBSEND_RESPONSE
#define SCRIPT_POWER_SECTION
#define USE_SENDMAIL

Jason2866, 2022-01-15:
ALWAYS use Tasmota as base for self compile. Other env are specialized and restricted.

gemu2015, 2022-01-21:
@mrpatate76 
i now also get this compile error (didn't have this before)
try enabling TLS together with USE_SENDMAIL
#define USE_TLS
#define USE_SENDMAIL

Tino4, 2022-01-29:
@gemu2015 , hi sir can you tell me sendmail is working at the moment ? I've tried almost everything and still not working.I have changed the 3 files and i added the rule #define TLS in config override file ,still not working when i type the command =>sendmail.... in the console , can you give me advice / info ? Thank you very much in advance.Grtz

gemu2015, 2022-01-29:
@Tino4 

use the very last dev Version.
#define USE_SENDMAIL   should be enough now

and try the full cmd in console (with your own credentials and email addresses)

=>sendmail [smtp.gmail.com:465:user:password:<from@gmail.com>:<to@gmail.com>:test]Hello world

gemu2015, 2022-01-29:
you must use the sharp clips on esp8266 (not on esp32)

Tino4, 2022-01-29:
Thx!!!! Verzonden vanaf mijn Huawei-telefoon-------- Oorspronkelijk bericht --------Van: gemu ***@***.***>Datum: za 29 jan. 2022 11:30Aan: arendst/Tasmota ***@***.***>Cc: Tino4 ***@***.***>, Mention ***@***.***>Onderwerp: Re: [arendst/Tasmota] sendmail cont connect to smtp.gmail.com (Issue #14404)
@Tino4
use the very last dev Version.
#define USE_SENDMAIL   should be enough now
and try the full cmd in console (with your own credentials and email addresses)
=>sendmail ***@***.******@***.***:test]Hello world

Tino4, 2022-01-29:
Thx guys its working now with the latest developer firmware and the command as follows :sendmail [smtp.gmail.com:465:user:passwd:<from@gmail.com>:<to@gmail.com>:test]Hello world ,i use the 8266 and need the sharp clips !!Many thx @ gemu and Theo !!!I'm so happy now :-) Grtz

damiankuziora, 2022-02-07:
Hello
I am new to smart home. I am learning tasmota and I like it very much. Recently I learned to compile correctly. I need help setting up for sending to icloud mail.

gemu2015, 2022-02-08:
the mail option in esp8266 is very limited. it does only work with a few mail servers and only with setting the email server to reduced security. (gmail is working, icloud does not)
the esp32 mailer supports all kinds of email servers, i assume also icloud will work.

Tino4, 2022-04-16:
Hi guys , i've still a problem with sendmail esp8266 , always the first time execution troubles with the sendmail command , second ,third .....time no problem at all.it seems that if the esp8266 is inactive for a while the problem occurs.I've disabled sleep ....no solution.I've tried last firmware version ...no solution.Anyone a suggestion?P.S. there is also a bug in the calculation of the ISL28022 , the ISL28022 uses a 5 milliohm resistor instead of an 100milliohm and the isl28022 calculation is based on a ina219 thus wrong current value , also the bus voltage calculation is wrong.Maybe someone can give me info or can help me with these.Best regards , Tino

barbudor, 2022-04-16:
Open a separate discussion for the ISL28022

Tino4, 2022-04-16:
> Open a separate discussion for the ISL28022

okay , i'm sorry for that.Best Regards,Tino

lalo-uy, 2022-04-16:
Starting on May gmail will not alow user/pass login any more.
That could impact on sendmail with both cpu.

barbudor, 2022-04-16:
Where did you got this information ?
You need an application specific password for that but I can't find any info it will be ending.
https://support.google.com/mail/answer/185833?hl=en

famdejongjohan, 2023-01-07:
oh, oops this item is closed, I will open a new one.

famdejongjohan, 2023-01-07:
Fixed the problem ... 
- the "*" body can not have a <space>  in front of it .. the >m section is now evaluated  .. :-)))

famdejongjohan, 2023-01-07:
"space" in front