arendst/Tasmota #14426: Tasmota 10.x - multiple restarts and flash writes

bgolab, 2022-01-14:
### PROBLEM DESCRIPTION
I recently started monitoring number of flash writes on two existing at the moment switches.
These switches experience crashes and the same number of flash writes - probably in the night. I see this for a couple of days.
Both the SetOptions0 and SaveData are set to OFF.
09:45:10.693 CMD: setoption0
09:45:10.697 RSL: RESULT = {"SetOption0":"OFF"}
09:45:19.017 CMD: savedata
09:45:19.130 RSL: RESULT = {"SaveData":"OFF"}
-------------------------------------------------------
Switch 1 (sonoff, Tasmota 10.0.0, very simple configuration: 'PulseTime 3' ONLY ) - 24 CRASHES!
23:00: BootCount": 2129, "SaveCount": 2504,
7:42(morning): "BootCount": 2153, "SaveCount": 2528,

Switch 2 (DGM-PP-W162, Tasmota 10.1.0, Rule 1 configured ONLY) - 91 CRASHES!
23:00:  "BootCount": 1666, "SaveCount": 2135,
7:42 (morning): BootCount": 1757, "SaveCount": 2226,

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

switch 1:
09:56:03.391 CMD: Backlog Template; Module; GPIO 255
09:56:03.484 RSL: RESULT = {"NAME":"Tasmota","GPIO":[32,1,1,1,1,0,0,0,224,320,1,0,0,0],"FLAG":0,"BASE":1}
09:56:03.690 RSL: RESULT = {"Module":{"0":"Tasmota"}}
09:56:03.944 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

switch 2: 
08:57:14.886 CMD: Backlog Template; Module; GPIO 255
08:57:14.959 RSL: RESULT = {"NAME":"DGM","GPIO":[0,0,0,32,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":18}
08:57:15.174 RSL: RESULT = {"Module":{"0":"DGM"}}
08:57:15.381 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
switch 1:
09:36:42.844 CMD: Backlog Rule1; Rule2; Rule3
09:36:42.914 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:36:43.124 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:36:43.374 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

switch 2: 
08:35:52.680 CMD: Backlog Rule1; Rule2; Rule3
08:35:52.773 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":120,"Free":391,"Rules":"ON Power1#State=1 DO ruletimer1 120 ENDON ON ENERGY#Power>25 DO ruletimer1 120 ENDON ON Rules#Timer=1 DO power off ENDON"}}
08:35:52.991 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
08:35:53.190 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
switch 1: 
09:37:26.927 CMD: Status 0
09:37:26.933 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_AB8881","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:37:26.938 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:46:30","StartupUTC":"2022-01-14T04:50:56","Sleep":50,"CfgHolder":4617,"BootCount":2153,"BCResetTime":"2020-04-10T14:03:13","SaveCount":2528,"SaveAddress":"F6000"}}
09:37:26.943 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:23:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"353/699"}}
[... 20 lines cut ...]
08:37:20.693 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-14T08:37:20","ENERGY":{"TotalStartTime":"2020-12-13T15:06:02","Total":6.069,"Yesterday":0.184,"Today":0.000,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
08:37:20.709 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-14T08:37:20","Uptime":"0T02:34:27","UptimeSec":9267,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FunBox2-DB13","BSSId":"E8:D2:FF:4F:F4:6C","Channel":11,"Mode":"11n","RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:06"}}}
08:38:06.675 RSL: STATE = {"Time":"2022-01-14T08:38:06","Uptime":"0T02:35:13","UptimeSec":9313,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FunBox2-DB13","BSSId":"E8:D2:FF:4F:F4:6C","Channel":11,"Mode":"11n","RSSI":100,"Signal":-33,"LinkCount":1,"Downtime":"0T00:00:06"}}
08:38:06.687 RSL: SENSOR = {"Time":"2022-01-14T08:38:06","ENERGY":{"TotalStartTime":"2020-12-13T15:06:02","Total":6.069,"Yesterday":0.184,"Today":0.000,"Period": 0,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bgolab, 2022-01-14:
What is unusual in my home network:  the Access Point scheduler disables the WiFi  at midnight (00:00 AM) and enables it at 6:00 AM.

bgolab, 2022-01-14:
I monitor the stats during the day today and do not see any boot/writes updates. It it happens in the night I guess.

bgolab, 2022-01-14:
It is likely that the crash (and flash writes) are a result of the WiFi not being accessible at night.

bgolab, 2022-01-14:
I found the difference between switch 1 and switch 2 which may explain why there is different number of nigh crashes:

switch 1 (Tasmota 10.0.0)
11:25:46.074 CMD: Wificonfig
11:25:46.078 RSL: RESULT = {"WifiConfig":{"4":"Retry"}}

switch 2 (Tasmota 10.1.0)
11:23:35.445 CMD: wificonfig
11:23:35.450 RSL: RESULT = {"WifiConfig":{"2":"WifiManager"}}

arendst, 2022-01-14:
Congratulations!!

You managed to solve your own issue. Well done.

bgolab, 2022-01-14:
Yes, I really appreciate how this software is designed also from troubleshooting perspective.

I am an old guy who spent with computer networking my last 30 years and I believe that product without debugging is worth nothing.

bgolab, 2022-01-15:
'WiFiConfig 5'  solved the issue

closed 2022-01-15 by bgolab