arendst/Tasmota #14430: OTA update not working if firmware is located on a webserver using a non-standard port

bigcookie, 2022-01-14:
### PROBLEM DESCRIPTION
Using Tasmota 2022.01.2.
I tried to upload the tasmota firmware by providing an URL with non-standard port.
Reason: still using TasmoAdmin - I patched it to download from ota.tasmota.com, but prefer using its "update all at once" as I have 30+ Tasmota devices in action :-).
URL in question: 
```
http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1, WF-DS01 dimmer
- [x] Tasmota binary firmware version number used: 2022.01.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmota device webpage
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
10:51:15.410 CMD: Backlog Template; Module; GPIO 255
10:51:15.436 MQT: licht-az/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
10:51:15.662 MQT: licht-az/stat/RESULT = {"Module":{"46":"Shelly 1"}}
10:51:15.918 MQT: licht-az/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"192":"Switch_n1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
10:50:44.474 CMD: status 0
10:50:44.481 MQT: licht-az/stat/STATUS = {"Status":{"Module":46,"DeviceName":"licht-az","FriendlyName":["licht-az"],"Topic":"licht-az","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:50:44.489 MQT: licht-az/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:17:29","StartupUTC":"2022-01-14T09:33:15","Sleep":50,"CfgHolder":4617,"BootCount":38,"BCResetTime":"2020-04-24T11:55:18","SaveCount":5427,"SaveAddress":"FB000"}}
10:50:44.498 MQT: licht-az/stat/STATUS2 = {"StatusFWR":{"Version":"2022.01.2(tasmota)","BuildDateTime":"2022.01.14 08:29:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"439/699"}}
10:50:44.506 MQT: licht-az/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndroLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C80001000600003C5AFF000000000000","00000280","00006000","00004000"]}}
10:50:44.520 MQT: licht-az/stat/STATUS4 = {"StatusMEM":{"ProgramSize":618,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:50:44.531 MQT: licht-az/stat/STATUS5 = {"StatusNET":{"Hostname":"SH-Licht-AZ","IPAddress":"192.168.20.236","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.20.1","DNSServer2":"0.0.0.0","Mac":"3C:71:BF:2C:67:AD","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:50:44.539 MQT: licht-az/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.253","MqttPort":1883,"MqttClientMask":"licht-az","MqttClient":"licht-az","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:50:44.548 MQT: licht-az/stat/STATUS7 = {"StatusTIM":{"UTC":"2022-01-14T09:50:44","Local":"2022-01-14T10:50:44","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:39","Sunset":"17:19"}}
10:50:44.558 MQT: licht-az/stat/STATUS10 = {"StatusSNS":{"Time":"2022-01-14T10:50:44","Switch1":"ON"}}
10:50:44.566 MQT: licht-az/stat/STATUS11 = {"StatusSTS":{"Time":"2022-01-14T10:50:44","Uptime":"0T00:17:29","UptimeSec":1049,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"AndroLAN","BSSId":"7C:FF:4D:FF:AF:2C","Channel":13,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
10:49:05.436 HTP: Hauptmenü
10:49:06.426 HTP: Firmware Update
10:49:08.088 HTP: Update gestartet
10:49:08.091 SRC: WebGui from 192.168.20.38
10:49:08.093 CMD: Grp 0, Cmnd 'OTAURL', Idx 1, Len 61, Data 'http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin'
10:49:08.095 CFG: CR 439/699, Busy 0
10:49:08.100 MQT: licht-az/stat/RESULT = {"OtaUrl":"http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin"}
10:49:08.147 SRC: WebGui from 0.0.0.0
10:49:08.149 CMD: Grp 0, Cmnd 'UPGRADE', Idx 1, Len 1, Data '1'
10:49:08.155 MQT: licht-az/stat/RESULT = {"Upgrade":"Version 2022.01.2 from http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin"}
10:49:08.560 CFG: in Flash gespeichert an FB, Anzahl 5426, Bytes 4096
10:49:10.518 HTP: Web-Server angehalten
10:49:10.520 UPL: http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin 2022.01.2(tasmota)
10:49:11.255 HTP: Web-Server aktiv bei SH-Licht-AZ mit IP-Adresse 192.168.20.236
10:49:11.660 HTP: Hauptmenü
10:49:12.521 HTP: Web-Server angehalten
10:49:12.523 UPL: http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin 2022.01.2(tasmota)
10:49:13.272 HTP: Web-Server aktiv bei SH-Licht-AZ mit IP-Adresse 192.168.20.236
10:49:13.527 HTP: Konsole
[... 24 lines cut ...]
10:49:29.270 HTP: Web-Server aktiv bei SH-Licht-AZ mit IP-Adresse 192.168.20.236
10:49:30.535 MQT: licht-az/stat/UPGRADE = {"Upgrade":"Failed File Not Found (404)"}

```

### TO REPRODUCE
1.) Set update URL to a http URL with IP and non-standard port:
Go to firmware update in device web page.
Set OTA URL to: http://192.168.20.201:8180/data/firmwares/tasmota-minimal.bin
Click "update"
Check console output

### EXPECTED BEHAVIOUR
Device can download file from URL location with non-standard port, as file can be downloaded via webbrowser by entering the same URL

### SCREENSHOTS
not appliable

### ADDITIONAL CONTEXT
User tries to use the bulk update function from TasmoAdmin webpage as 30+ devices need to be updated.
TasmoAdmin is patched to read the latest binaries from ota.tasmota.com (this is verified - latest binaries have been downloaded)
TasmoAdmin runs in docker on host network on a non-standard port
The move to host network happened recently for setup reasons. When TasmoAdmin ran on a standard port in bridge mode (different IP address), the update was possible.

**(Please, remember to close the issue when the problem has been addressed)**

bigcookie, 2022-01-14:
Found whitespaces in the URL - will check again. THus closed

closed 2022-01-14 by ascillato2

barbudor, 2022-01-14:
I'm using a webserver with port 8080 so no problems

bigcookie, 2022-01-14:
Yes, it was somehow another problem. It worked when forcing to download gz files… very weird, but not a Tasmota bug. Thus I closed the ticket…