arendst/Tasmota #14454: _ in DNS Names

BobFrankston, 2022-01-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
A DNS name can't have _ but when I look at /etc/hosts on my Ubiquiti USG I see the name tasmota_0EFB57-6999.xxx.xx though the hostname is shown as tasmota-0EFB57-6999.xxx.xx but in the status information on Ubiquiti's Unifi. I do not the _ is used in the MQTT Topic.

The problem is that there is an invalid name in the hosts file though pinging works with _ and -. This makes creates a special case fo linking data.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): AHOm.Tech wall switch
- [x] Tasmota binary firmware version number used: 10.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: athom.tech w/OTA update
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:08:35.582 MQT: stat/tasmota_0EFB57/RESULT = {"NAME":"Athom SW013EU","GPIO":[576,290,0,33,225,34,0,0,32,224,289,226,288,0],"FLAG":0,"BASE":18}
19:08:35.795 MQT: stat/tasmota_0EFB57/RESULT = {"Module":{"0":"Athom SW013EU"}}
19:08:36.001 MQT: stat/tasmota_0EFB57/RESULT = {"GPIO0":{"576":"LedLink_i"},"GPIO1":{"290":"Led3"},"GPIO2":{"0":"None"},"GPIO3":{"33":"Button2"},"GPIO4":{"225":"Relay2"},"GPIO5":{"34":"Button3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"224":"Relay1"},"GPIO14":{"289":"Led2"},"GPIO15":{"226":"Relay3"},"GPIO16":{"288":"Led1"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:03:16.110 MQT: stat/tasmota_0EFB57/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota_0EFB57","ButtonTopic":"0","Power":7,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:03:16.115 MQT: stat/tasmota_0EFB57/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"1T00:57:12","StartupUTC":"2022-01-14T17:06:04","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-01-06T16:23:39","SaveCount":76,"SaveAddress":"F4000"}}
19:03:16.122 MQT: stat/tasmota_0EFB57/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"382/699"}}
19:03:16.128 MQT: stat/tasmota_0EFB57/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rmfDrayTek2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","0405C80001000600003C5A0A000000000000","08000080","00006000","00004000"]}}
19:03:16.139 MQT: stat/tasmota_0EFB57/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
19:03:16.146 MQT: stat/tasmota_0EFB57/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0EFB57-6999","IPAddress":"172.20.3.175","Gateway":"172.20.0.1","Subnetmask":"255.255.252.0","DNSServer1":"172.20.0.1","DNSServer2":"0.0.0.0","Mac":"4C:EB:D6:0E:FB:57","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:03:16.152 MQT: stat/tasmota_0EFB57/STATUS6 = {"StatusMQT":{"MqttHost":"pi4c.xxx.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0EFB57","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:03:16.158 MQT: stat/tasmota_0EFB57/STATUS7 = {"StatusTIM":{"UTC":"2022-01-15T18:03:16","Local":"2022-01-15T19:03:16","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:20"}}
19:03:16.162 MQT: stat/tasmota_0EFB57/STATUS10 = {"StatusSNS":{"Time":"2022-01-15T19:03:16"}}
19:03:16.168 MQT: stat/tasmota_0EFB57/STATUS11 = {"StatusSTS":{"Time":"2022-01-15T19:03:16","Uptime":"1T00:57:12","UptimeSec":89832,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"ON","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"rmfDrayTek2","BSSId":"6A:D7:9A:10:EB:E4","Channel":1,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
n/ since this is during setup

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Setup a new athom.tech wall switch with the standard OTA update.
https://www.athom.tech/blank-1/eu-key-button-switch-1-gangd

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect valid DNS names with - not _

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
This happens with both the US and EU version of https://www.athom.tech/blank-1/eu-key-button-switch-1-gangd
FYI - I've changed the domain name to xxx.xx for a modicum of security.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-15:
As _ is invalid in a host name, Tasmota as converted the _ to a - as you can see in the StatusNET line:
```
{"StatusNET":{"Hostname":"tasmota-0EFB57-6999"
```

I have no idea what trick your Unify is doing.
Is the hosts file on a Unify automatically updated by the software ?
On a standard Linux/Windows, the file is only manually edited file.

BobFrankston, 2022-01-15:
You inspired me to look deeper. I have a tool that allows me to watch DHCP activity and in option #12 it is indeed providing a hostname with _ not -.

```

{
        "op": "BOOTREQUEST",
        "htype": "HW_Ethernet",
        "xid": 1081490799,
        "chaddr": "4C-EB-D6-0F-00-3B",
        "magicCookie": 1669485411,
        "options": "{\"53\":1,\"55\":[1,3,28,6,42],\"57\":1500}",
        "optionsx": {
            "DHCP Message Type #53": "DHCPDISCOVER",
            "Parameter Request List #55": "[1,3,28,6,42]",
            "Maximum DHCP Message Size #57": 1500
        },
        "stamp": "2022-01-15 14:28:10"
    }

{
        "op": "BOOTREQUEST",
        "htype": "HW_Ethernet",
        "xid": 1081490799,
        "chaddr": "4C-EB-D6-0F-00-3B",
        "magicCookie": 1669485411,
        "options": "{\"12\":\"tasmota_0F003B-0059\",\"50\":\"172.20.3.76\",\"53\":3,\"54\":\"172.20.0.1\",\"55\":[1,3,28,6,42],\"57\":1500}",
        "optionsx": {
            "Host Name #12": "tasmota_0F003B-0059",
            "Requested IP Address #50": "172.20.3.76",
            "DHCP Message Type #53": "DHCPREQUEST",
            "Server Identifier #54": "172.20.0.1",
            "Parameter Request List #55": "[1,3,28,6,42]",
            "Maximum DHCP Message Size #57": 1500
        },
        "stamp": "2022-01-15 14:28:11"
    }
```

barbudor, 2022-01-15:
Let me check in the code
Does it comes again later with the hyphen or only this ?

barbudor, 2022-01-15:
Do you see that at every reboot of Tasmota ?

barbudor, 2022-01-15:
```
19:03:16.146 MQT: stat/tasmota_0EFB57/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0EFB57-6999",...."Mac":"4C:EB:D6:0E:FB:57"
```
vs
```
  "chaddr": "4C-EB-D6-0F-00-3B",
        "options": "{\"12\":\"tasmota_0F003B-0059\",\"50\":\"172.20.3.76\",\"53\":3,\"54\":\"172.20.0.1\",\"55\":[1,3,28,6,42],\"57\":1500}",
```

Not the same device, so may not be the same Tasmota version

Enforcing the proper hostname has been implemented in Tasmota in July 2021
So the device with MAC address above is using a version prior to that one.

Jason2866, 2022-01-15:
As designed following the RFC standards

closed 2022-01-15 by Jason2866

BobFrankston, 2022-01-15:
Not the same only because I purposely tried onboarding a new device but otherwise identical

[image]

barbudor, 2022-01-16:
@BobFrankston
Not sure to get your point now
Are you asking/complaining about something?

BobFrankston, 2022-01-16:
I'm saying that these devices are providing a nonDNS-compliant name to DHCP. Since this seems to be a standard Tasmota build it would seem to be a bug. Apparently, (as per the note above) this was considered a bug and fixed as of July but this build is dated December so something is off. I don't know the code base well enough to know why these names are being issued so have to assume there is still a bug that needs fixing.

barbudor, 2022-01-16:
Ok.
So from the code, I can't see how this could happen:
https://github.com/arendst/Tasmota/blob/153e509a940e259549b515750517425d344b59ed/tasmota/tasmota.ino#L420-L431
This code is executed at the boot, even before Wifi is up (in Arduino ` setup()` function):
- Read SET_HOSTNAME string from the flash settings
- Copy it to TasmotaGlobal.hostname
- Scan TasmotaGlobal.hostname for any invalid character (non alphanum, non `.`) and patch it to `-`
From there, TasmotaGlobal.hostname is compliant and this is what is used everywhere as Hostname

Nevertheless, I have a platform where I can test and use tcpdump to verify.

On your side, could you please confirm that the data you published above is fresh and not something that was remaining from a pre-July version ?
For example, can you edit the hosts file on your Unify, remove the line and restart the Tasmota device ?

Let's compare our results later today.

barbudor, 2022-01-16:
So I pickup a NodeMCU and doing a fresh serial erase and flash with v10.1.0 http://ota.tasmota.com/tasmota/release/tasmota.bin

With an AP on a RasPI, running ` tcpdump -ni 2 -s 2048 -w dhcp.pcap port 67`, I recorded 3 requests:
- Initial request after entering the SSID/Password, Tasmota restart, connect to the RasPi's AP and get IP address
- After 40 seconds, I did a `restart 1`
- After around 30 more seconds, I unplugged the NodeMCU, waited 10 seconds and replugged.

In all 3 cases, the Option12 is correct.
[image]

Full pcap for your reference: [tasmota_10-1-0_dhcp.zip](https://github.com/arendst/Tasmota/files/7877063/tasmota_10-1-0_dhcp.zip)

BobFrankston, 2022-01-16:
Thanks.

Since I first onboarded with the out-of-the-box version it is possible the devices were running an old version. But after I did the reset and had to onboard again with the new version the old name might have indeed been remembered by Unifi.

I'm going to try to work on this afternoon (Eastern TIme). I did try a quick test using a different network but Starry kept rejecting it -- maybe enforcing the name rule. I want to try to upgrade Tasmota before connecting to Unifi.

BobFrankston, 2022-01-16:
Confirmed. I first connected to a different AP and, indeed, the device comes with a 2020 build and uses _. After I update and then connect to my main system the name is correct with the "-". Since some devices already have multiple DHCP names (bulbs that have a Tasmota name as well as an ESP Name), it should be possible to ignore the _ version and force a new name with "-" may be as simple as a replaceAll("_","-") or the equivalent for option 12. A bonus would be to remove the old name -- DHCP seems to have that option but I can't say it works. 

Too bad I can't see a way to manually remove the name in Unifi but am still looking.

BobFrankston, 2022-02-06:
Just as an FY*I https://www.rfc-editor.org/rfc/rfc8552 seems to indicate that _'s are indeed acceptable.

barbudor, 2022-02-06:
No it doesn't.
It applies to DNS records where the underscore can be used to distinguish between different records for the same host name.
It's about using DNS records to store informations, not about hostnames.
DNS records store much more than hostnames

BobFrankston, 2022-02-06:
The DNS is just a database with keys. I agree that - is preferable to _ but it's not an absolute ban on all usage.

Jason2866, 2022-02-06:
Anyways it is Tasmota rfc :-)