arendst/Tasmota #14466: Sonoff ZbBridge:  Reboot loop after flashing

Floatcher, 2022-01-16:
### PROBLEM DESCRIPTION
After flashing a Sonoff ZbBridge with Tasmota and connecting to the local wifi, Tasmota is in a reboot loop. I have no chance to run the ota firmware update, becaue it always restarts before finished and so the upload need finished successfull. Maximum I was able to tansfere were 100kb before reboot stoped ota update.

Also looks like it resets the logging settings.

This happens with the latest stable and the laste dev version.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:38:19.495 CMD: status 0
15:38:19.496 SRC: WebConsole from 192.168.1.165
15:38:19.497 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
15:38:19.500 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9DBD13","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:38:19.502 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:00:08","StartupUTC":"2022-01-16T14:38:11","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2022-01-16T14:47:50","SaveCount":18,"SaveAddress":"1000"}}
15:38:19.503 RSL: STATUS2 = {"StatusFWR":{"Version":"2022.01.2(zbbridge)","BuildDateTime":"2022-01-16T11:00:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"357/699"}}
15:38:19.505 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["unserWlan",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00000000000000","00008080","00006000","00004000"]}}
15:38:19.510 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":672,"Free":1116,"Heap":28,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"}}
15:38:19.512 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9DBD13-7443","IPAddress":"192.168.1.114","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.80","DNSServer2":"192.168.1.1","Mac":"E8:DB:84:9D:BD:13","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:38:19.514 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9DBD13","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:38:19.517 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-16T14:38:19","Local":"2022-01-16T15:38:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:37","Sunset":"17:21"}}
15:38:19.520 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-16T15:38:19"}}
15:38:19.524 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-16T15:38:19","Uptime":"0T00:00:08","UptimeSec":8,"Vcc":3.475,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"unserWlan","BSSId":"C0:06:C3:35:97:96","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
15:38:19.530 RSL: STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4023dcfc","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["4023dcf8","4023dd84","40256598","40101776","4025af81","4010173f","4027628c","40253b9d","402539c3","4028075e","4028076f","40275eb3","4027bf7b","40000f49","40000f49","40000e19","40105649","4010564f","4010000d","40274d90","40274d41","40250d3a","40251671","40250d3a","40250c70","4025179c","40250d3a","4029f03f","4029f03f","4029f040","40250f24"]}}
15:38:20.409 RSL: STATE = {"Time":"2022-01-16T15:38:20","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.475,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"unserWlan","BSSId":"C0:06:C3:35:97:96","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:38:21.405 APP: Boot Count 8

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.002 HDW: ESP8266EX
00:00:00.010 UFS: FlashFS mounted with 220 kB free
00:00:00.050 CFG: Loaded from File, Count 17
00:00:00.053 CFG: CR 357/699, Busy 0
00:00:00.056 ZIG: ZBBridge EEPROM found at address 0x50
00:00:00.057 SNS: Hardware Serial
00:00:00.058 SRC: Restart
00:00:00.059 Project tasmota - Tasmota Version 2022.01.2(zbbridge)-2_7_4_9(2022-01-16T11:00:12)
00:00:00.060 ZIG: Resetting EZSP device
00:00:00.210 WIF: Attempting connection...
00:00:00.546 WIF: Connecting to AP1 unserWlan Channel 1 BSSId C0:06:C3:35:97:96 in mode 11n as tasmota-9DBD13-7443...
00:00:01.751 WIF: Connected
00:00:02.002 HTP: Web server active on tasmota-9DBD13-7443 with IP address 192.168.1.114
00:00:02.534 RTC: UTC 2022-01-16T14:32:15, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
15:32:16.040 RSL: INFO1 = {"Info1":{"Module":"Sonoff ZbBridge","Version":"2022.01.2(zbbridge)","FallbackTopic":"cmnd/DVES_9DBD13_fb/","GroupTopic":"cmnd/tasmotas/"}}
15:32:16.041 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-9DBD13-7443","IPAddress":"192.168.1.114"}}
15:32:16.046 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40101068","4026fa1f","4026f9b4","4026f95b","4026eab4","4026eadd","4026c544","4010006e","402688b1","4010061c","4026d23b","4024935c","4026ccae","402765f7","40275eb3","402491d8","40000f49","40000f49","40000e19","40105649","4010564f","4010000d","40274d90","40274d41","40250d3a","40251671","40250d3a","40250c70","4025179c","40250d3a","4029f03f"]}}}
15:32:16.049 TFS: File 'autoexec.bat' not found
15:32:20.400 RSL: STATE = {"Time":"2022-01-16T15:32:20","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.475,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"unserWlan","BSSId":"C0:06:C3:35:97:96","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:32:21.403 APP: Boot Count 8
15:32:28.028 ZIG: Resetting EZSP device
15:32:28.463 ZbEZSPReceived: time-out, discarding FE0D0180020000000001010878

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Falsh Sonoff Zigbee Bridge bin and connect to local wifi

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Don't reboot while ota firmware upgrade. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-01-16:
Hi,

I can't reproduce this, sorry. Sometimes Exception 29 is due to a bad power supply. Please, can you change the power supply of your device?

ascillato2, 2022-01-16:
Please, also complete the part:

> Provide the output of command: Backlog Template; Module; GPIO 255:
>   Configuration output here:
> 15:45:52.489 CMD: Backlog Template; Module; GPIO 255
> 15:45:52.490 SRC: WebConsole from 192.168.1.165
> 15:45:52.491 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'

After the last line you posted were the actual output of TEMPLATE, MODULE and GPIO commands. Please, complete it to see if the GPIO assignment is correct. Thanks.

Floatcher, 2022-01-16:
I had just a discussion on the tasmota discord. When i connect serial to my zbbridge i get an exception from the wifi Controller and a dump all the time before reboot. Power supplies i tried 3 different.  I guess my hardware is broken.

closed 2022-01-16 by Floatcher