arendst/Tasmota #14468: Wakeup does not dimm all dimmers on H801

kreutpet, 2022-01-16:
### PROBLEM DESCRIPTION
I am using a H801 to control the lights of an aquarium.
In the morning i want to wakeup the white lights in the smooth way.
I have been struggling to get a reliable wakeup on H801, see also here (https://github.com/arendst/Tasmota/issues/14295) 
The Wakeup is successfully triggered by a timer in the morning but the Dimmer2 (white LED) is remaining on value 0 while the Dimmer1 (color LED) is showing the intended behavior and increases the dimmer value to the target value.

Interesting is that in rare cases this seems to work. I could not see any difference in the per-conditions.  In the log you can see the last state message from the log. In the last days all wake ups failed do change dimmer2 values.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): H801
- [x] Tasmota binary firmware version number used: 10.1.0 (2021.12.08 14:47:39) from tasmota download
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esp linux
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:30:11.606 MQT: stat/aquarium/TEMPLATE = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:30:11.805 MQT: stat/aquarium/MODULE = {"Module":{"20":"H801"}}
17:30:12.011 MQT: stat/aquarium/GPIO = {"GPIO0":{"1312":"DS18x20"},"GPIO1":{"288":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"1216":"AM2301"},"GPIO4":{"420":"PWM5"},"GPIO5":{"321":"Led_i2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"418":"PWM3"},"GPIO13":{"417":"PWM2"},"GPIO14":{"419":"PWM4"},"GPIO15":{"416":"PWM1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
17:32:24.665 MQT: stat/aquarium/RULE = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":119,"Free":392,"Rules":"ON clock#timer=1 DO Backlog Wakeup 60;POWER2 ON;POWER1 OFF BREAK ON rules#timer=2 DO Backlog Speed2 40;POWER2 OFF ENDON"}}
17:32:24.865 MQT: stat/aquarium/RULE = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":129,"Free":382,"Rules":"ON clock#timer=3 DO Backlog Color 3;POWER1 ON;POWER2 OFF;Dimmer1 70 BREAK ON clock#timer=4 DO Backlog POWER1 OFF;POWER2 OFF ENDON"}}
17:32:25.066 MQT: stat/aquarium/RULE = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":33,"Free":478,"Rules":"ON clock#timer=5 DO Restart BREAK"}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:33:05.937 MQT: stat/aquarium/STATUS = {"Status":{"Module":20,"DeviceName":"Aquarium","FriendlyName":["Aquarium",""],"Topic":"aquarium","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:33:05.963 MQT: stat/aquarium/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"3T08:21:18","StartupUTC":"2022-01-13T08:11:47","Sleep":0,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2021-12-13T10:23:11","SaveCount":75,"SaveAddress":"F4000"}}
17:33:05.988 MQT: stat/aquarium/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021.12.08 14:47:39","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"343/699"}}
17:33:06.007 MQT: stat/aquarium/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["kiot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008019","2805C80001800600003C5A0A000000000000","00000481","00006000","00004000"]}}
17:33:06.036 MQT: stat/aquarium/STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:33:06.062 MQT: stat/aquarium/STATUS5 = {"StatusNET":{"Hostname":"aquarium-1279","IPAddress":"192.168.1.99","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"BC:FF:4D:4D:E4:FF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:33:06.086 MQT: stat/aquarium/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4DE4FF","MqttUser":"DVES_USER","MqttCount":44,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:33:06.113 MQT: stat/aquarium/STATUS7 = {"StatusTIM":{"UTC":"2022-01-16T16:33:06","Local":"2022-01-16T17:33:06","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:37","Sunset":"17:21"}}
17:33:06.137 MQT: stat/aquarium/STATUS10 = {"StatusSNS":{"Time":"2022-01-16T17:33:06","DS18B20-1":{"Id":"3C01F096889E","Temperature":24.9},"DS18B20-2":{"Id":"3C01F096D865","Temperature":24.6},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}}
17:33:06.164 MQT: stat/aquarium/STATUS11 = {"StatusSTS":{"Time":"2022-01-16T17:33:06","Uptime":"3T08:21:19","UptimeSec":289279,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":1000,"MqttCount":44,"POWER1":"OFF","Dimmer1":60,"POWER2":"ON","Dimmer2":61,"Color":"0000999C00","HSBColor":"240,100,60","White":61,"CT":153,"Channel":[0,0,60,61,0],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"kiot","BSSId":"60:A4:4C:68:82:0B","Channel":13,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":11,"Downtime":"0T00:00:59"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:33:05.937 MQT: stat/aquarium/STATUS = {"Status":{"Module":20,"DeviceName":"Aquarium","FriendlyName":["Aquarium",""],"Topic":"aquarium","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:33:05.963 MQT: stat/aquarium/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"3T08:21:18","StartupUTC":"2022-01-13T08:11:47","Sleep":0,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2021-12-13T10:23:11","SaveCount":75,"SaveAddress":"F4000"}}
17:33:05.988 MQT: stat/aquarium/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021.12.08 14:47:39","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"343/699"}}
17:33:06.007 MQT: stat/aquarium/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["kiot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008019","2805C80001800600003C5A0A000000000000","00000481","00006000","00004000"]}}
17:33:06.036 MQT: stat/aquarium/STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
[... cut ...]
There possible 2 way for the expected behavior ewhen the wakeup is triggeres : 
1. the all dimmer will follow the wakup command
2. there would be an option in the wakeup command to define which dimmer shall follow the wakeup : wakup Dimmer1 60

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

NA

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Issue may get additional info here https://github.com/arendst/Tasmota/issues/14295

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-01-17:
As I understand it, wakeup does not work on two dimmer channels at the same time. Changing that would be an enhancement suggestion, not reporting an issue to be fixed. As things are now, you could use rules to get close to the same effect as wakeup.

kreutpet, 2022-01-17:
i looked into https://github.com/arendst/Tasmota/blob/d6fc62e3769dc1655cce63445f08d933a6289ad2/tasmota/xdrv_04_light.ino to understand how the wakeup is working.
To my understanding the dimmer of RGB and white should be controlled both.
The only thing is see is that with rule 2 the dimmer2 have value0 before rule1 triggers the wakeup.
I am still trying to trace in the code if the dimmer2 is not included as it value 0.
i will tr to set the dimmer2 before wakeup to value 1 to see if it is then included.

kreutpet, 2022-01-17:
just to mention the SetOption37 is 128

Chiller2019, 2022-01-19:
My H801 hang up after 15minutes, idk why.

ascillato, 2022-01-19:
@Chiller2019

Hi, as your problem is different to this thread, if you need assistance, please use the discord chat or open a new issue with the template fully completed. thanks.

closed 2022-02-18 by github-actions[bot]