arendst/Tasmota #14472: add docs for I2C HT16K33 based 7 segment LED

HarHarLinks, 2022-01-16:
PRs like #8116, #8900 seem to not have made it to the docs properly. See also https://github.com/arendst/Tasmota/discussions/14451

closed 2022-01-16 by close-issue-app[bot]

HarHarLinks, 2022-01-16:
@arendst template makes no sense here

barbudor, 2022-01-16:
@HarHarLinks 
Issue makes perfectly sense when raising for qualified bug or mis behavior.
Help should be asked in Discussion or in Discord

If you have something to say about docs, here is the place : https://github.com/tasmota/docs
The docs are open source and anyone is invited to collaborate and participate. You can easily create PR ffor the docs by clicking on the pencil button on the top right of the doc page.

Thanks for your collaboration.

HarHarLinks, 2022-01-16:
thanks for pointing that out

ascillato, 2022-01-16:
Link just for reference: https://github.com/tasmota/docs/issues/257