arendst/Tasmota #14483: DeepSleepTime":28800 does not work correct

ManuelF001, 2022-01-17:
### PROBLEM DESCRIPTION
Why the DeepSleep is only to 2022-01-17T01:00:00 with "DeepSleepTime":28800   ?
I think with DeepSleepTime":28800 it must be in deepsleep until 2022-01-17T08:00:00?
`
00:00:00.001 HDW: ESP8266EX
00:00:00.049 CFG: Loaded from flash at F4, Count 324
00:00:00.064 Project tasmota - Tasmota Version 10.1.0(tasmota)-2_7_4_9(2021-12-08T14:47:33)
00:00:00.554 WIF: Connecting to AP1 Home1 Channel 5 BSSId 10:FE:ED:AF:61:83 in mode 11n as tasmota-C0A179-0377...
00:00:02.820 WIF: Connected
00:00:03.072 HTP: Web server active on tasmota-C0A179-0377 with IP address 192.168.1.110
20:59:58.101 MQT: Attempting connection...
20:59:58.125 MQT: Connected
20:59:58.129 MQT: tele/tasmota_C0A179/LWT = Online (retained)
20:59:58.130 MQT: cmnd/tasmota_C0A179/POWER = 
20:59:58.184 RUL: SYSTEM#BOOT performs "ruletimer1 65"
20:59:58.191 MQT: stat/tasmota_C0A179/RESULT = {"T1":65,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
21:00:02.446 MQT: tele/tasmota_C0A179/STATE = {"Time":"2022-01-16T21:00:02","Uptime":"0T00:00:10","UptimeSec":10,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":43,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Home1","BSSId":"10:FE:ED:AF:61:83","Channel":5,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:00:02.514 MQT: tele/tasmota_C0A179/SENSOR = {"Time":"2022-01-16T21:00:02","ANALOG":{"A0":1024},"DS18B20":{"Id":"3C01D60797D2","Temperature":16.2},"TempUnit":"C"}
21:00:53.410 RUL: TIME#MINUTE=1260 performs "deepsleeptime 28800"
21:00:53.416 MQT: stat/tasmota_C0A179/RESULT = {"DeepSleepTime":28800}
21:00:53.541 DSL: Reset wrong settings wakeup: 0, slip 10000
21:00:53.548 MQT: stat/tasmota_C0A179/RESULT = {"DeepSleep":{"Time":"2022-01-17T01:00:00","Epoch":1642377600}} 
`
I found this:
https://tasmota.github.io/docs/DeepSleep/
The ESP8266 has a limitation of a maximum of ~71 minutes DeepSleep. To overcome the limitation, a short wake-up is performed - the device will wake up every hour for <0.3 seconds until the DeepSleep time is reached. The remaining DeepSleep time is decremented, and the device is then put back in DeepSleep again. The remaining time is stored in RTC memory. As long as the device is powered (e.g., via the battery), this should work fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): AZ-Delivery D1 mini ESP8266
- [ ] Tasmota binary firmware version number used: 10.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: ESP8266Flasher.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
`11:35:32.230 CMD: Backlog Template; Module; GPIO 255
11:35:32.297 MQT: stat/tasmota_C0A179/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:35:32.545 MQT: stat/tasmota_C0A179/RESULT = {"Module":{"0":"Generic"}}
11:35:32.780 MQT: stat/tasmota_C0A179/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1312":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"4096":"DeepSleep"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4704":"ADC Input"}}`
- [ ] 
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
`11:36:43.927 CMD: Backlog Rule1
11:36:43.994 MQT: stat/tasmota_C0A179/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":473,"Free":38,"Rules":"ON system#boot DO ruletimer1 65 ENDON ON Rules#Timer=1 DO deepsleeptime 300 ENDON ON time#minute=1259 DO deepsleeptime 28800 ENDON ON time#minute=1260 DO deepsleeptime 28800 ENDON ON time#minute=1261 DO deepsleeptime 28800 ENDON ON time#minute=1262 DO deepsleeptime 28800 ENDON ON time#minute=1263 DO deepsleeptime 28800 ENDON ON time#minute=1264 DO deepsleeptime 28800 ENDON ON time#minute=1265 DO deepsleeptime 28800 ENDON ON time#minute=1266 DO deepsleeptime 28800 ENDON"}}`

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
`11:37:18.732 CMD: Status 0
11:37:18.740 MQT: stat/tasmota_C0A179/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_C0A179","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:37:18.808 MQT: stat/tasmota_C0A179/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:03","StartupUTC":"2022-01-17T10:35:15","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2021-12-18T14:46:05","SaveCount":327,"SaveAddress":"F9000"}}
11:37:18.870 MQT: stat/tasmota_C0A179/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"397/699"}}
11:37:18.919 MQT: stat/tasmota_C0A179/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home1",""],"TelePeriod":70,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
11:37:18.977 MQT: stat/tasmota_C0A179/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
11:37:19.039 MQT: stat/tasmota_C0A179/STATUS5 = {"StatusNET":{"Hostname":"tasmota-C0A179-0377","IPAddress":"192.168.1.110","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"A8:48:FA:C0:A1:79","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:37:19.091 MQT: stat/tasmota_C0A179/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.104","MqttPort":1883,"MqttClientMask":"Tasmota_Raumtemperatur_WC","MqttClient":"Tasmota_Raumtemperatur_WC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:37:19.147 MQT: stat/tasmota_C0A179/STATUS7 = {"StatusTIM":{"UTC":"2022-01-17T10:37:19","Local":"2022-01-17T11:37:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"17:23"}}
11:37:19.231 MQT: stat/tasmota_C0A179/STATUS10 = {"StatusSNS":{"Time":"2022-01-17T11:37:19","ANALOG":{"A0":1024},"DS18B20":{"Id":"3C01D60797D2","Temperature":19.1},"TempUnit":"C"}}
11:37:19.274 MQT: stat/tasmota_C0A179/STATUS11 = {"StatusSTS":{"Time":"2022-01-17T11:37:19","Uptime":"0T00:02:04","UptimeSec":124,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Home1","BSSId":"10:FE:ED:AF:61:83","Channel":9,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}}`

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
With active rule each day at 9:00pm you can see it.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2022-01-27:
I agree that the expectation with deepsleep 28800 is to get a wakeup at 8am, 4pm and 12pm. I will check what is wrong here. The timezone shift is 1h. Therefore maximum 1h shift should be ok.

stefanbode, 2022-01-27:
Ok, got it. All calculations are done in UTC_Time. Therefore wakeup is 0,8 and 16 UTC time. You start the deepsleep at 2022-01-16T21:00:02. The next wakeup will be there at midnight UTC time. In local timezone this is 1am and correctly.  This how the algorithm is working. But I agree that is much more user friendly if the 0,8,16 is in the local timezone of the user. This is more what you expect. In this case the device will wakeup at: 2022-01-17T00:00:00 and next time at 2022-01-17T08:00:00, I will do a PR for this soon

ManuelF001, 2022-01-27:
Okay I understand. There is no way to sleep from 21:00 to 08:00.
In this time there is no wifi and then it don't goes to deepleep and discharge the battery.
For few days I have changed to Espeasy. There this is no problem. (deepsleep if no connection)
But thanks.

stefanbode, 2022-01-27:
in most cases it makes no sense without connection. If you do not have wifi and get the correct time, why wakeup? Doing a wakeup once a day at 8am should be now possible with just shifting the timezone +7h. But then is is only once a day. I will take a look how espeasy is doing it. Maybe I can learn and make an enhancement

ManuelF001, 2022-01-27:
I would like transfer sensor data every hour.
Between it should be in deepsleep.
But if there is no wifi it should also go to deepsleep if there is no connection possible.
In Espeasy you can set an option "sleep on connection failure".

stefanbode, 2022-01-27:
> In Espeasy you can set an option "sleep on connection failure".
Ok, this is interesting. So the device should recognize the missing WLAN and just don't care and instead of sucking down the batteries give it a new try. Agree. let's see how we can make this possible

ManuelF001, 2022-01-28:
Thanks. This would be very nice. And this would enable more possible applications for tasmota.

stefanbode, 2022-01-28:
Hi @ManuelF001. I have implemented an automatic mechanism to send the device back to deepsleep if the start does not work as expected to avoid battery problems. Thanks for bringing this up as an enhancement. I skipped making this optional. If there is no connection there is no reason to let it stay online. The grace period to establish a connection is 15 seconds. This is 2.5x more than normally needed. Please close the ticket.

ManuelF001, 2022-01-28:
Thanks. Is this included in the next firmware?

closed 2022-01-28 by ManuelF001

stefanbode, 2022-01-28:
yes, this is now in the development branch and can be tested. The release branch is for sure not updated yet. (http://ota.tasmota.com/tasmota/)

stefanbode, 2022-01-28:
Not yet. We are finalizing some minor optimizations and flexibility. Latest tomorrow I assume

stefanbode, 2022-01-28:
It is now merged into the development branch.