arendst/Tasmota #14487: PM sensor values disappear as soon as CO2 sensor is configured.

TheBattleWolf, 2022-01-17:
### PROBLEM DESCRIPTION
_I tried to connect a CO2 sensor(MH-Z19E) and a PM2.5 sensor(PMSA003) on my [ESP32-C3](https://de.aliexpress.com/item/1005003348936965.html). The issue I'm facing is that, as soon as I configure the CO2 module the PM sensor just disappears. For the MH-Z19E it doesn't seem to matter where I configure it. It always makes the PM sensor not show up as soon as i configure it in the template, even when it's not connected. After the configuration only my BMP280 and the MH-Z19E are shown._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [LilyGO T-OI Plus ESP32-C3](https://de.aliexpress.com/item/1005003348936965.html)
- [x] Tasmota binary firmware version number used: [tasmota32c3.bin 10.1.0](http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: [esptool](https://github.com/espressif/esptool)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
18:35:27.435 MQT: stat/tasmota_B64330/RESULT = {"NAME":"LilyGO T-OI Plus","GPIO":[0,0,4704,544,1,1,1,1,1696,1,1,0,0,0,0,0,0,0,608,640,1,1],"FLAG":0,"BASE":1}
18:35:27.642 MQT: stat/tasmota_B64330/RESULT = {"Module":{"0":"LilyGO T-OI Plus"}}
18:35:27.894 MQT: stat/tasmota_B64330/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"4704":"ADC Input1"},"GPIO3":{"544":"LedLink"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"1696":"PMS5003 Rx"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO18":{"608":"I2C SCL1"},"GPIO19":{"640":"I2C SDA1"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}
```

- [x] Provide the output of this command: `Status 0`:
```lua
18:36:44.344 MQT: stat/tasmota_B64330/STATUS = {"Status":{"Module":0,"DeviceName":"LilyGO T-OI Plus Air Sensor","FriendlyName":["Tasmota"],"Topic":"tasmota_B64330","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:36:44.354 MQT: stat/tasmota_B64330/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T00:11:28","StartupUTC":"2022-01-17T17:25:16","Sleep":50,"CfgHolder":4617,"BootCount":166,"BCResetTime":"2021-12-27T16:23:25","SaveCount":232}}
18:36:44.362 MQT: stat/tasmota_B64330/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:52:31","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":160,"Hardware":"ESP32-C3","CR":"483/699"}}
18:36:44.370 MQT: stat/tasmota_B64330/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home_WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
18:36:44.381 MQT: stat/tasmota_B64330/STATUS4 = {"StatusMEM":{"ProgramSize":1312,"Free":1856,"Heap":185,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
18:36:44.391 MQT: stat/tasmota_B64330/STATUS5 = {"StatusNET":{"Hostname":"tasmota-B64330-0816","IPAddress":"192.168.0.21","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.2","DNSServer2":"0.0.0.0","Mac":"7C:DF:A1:B6:43:30","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:36:44.398 MQT: stat/tasmota_B64330/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B64330","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:36:44.406 MQT: stat/tasmota_B64330/STATUS7 = {"StatusTIM":{"UTC":"2022-01-17T17:36:44","Local":"2022-01-17T18:36:44","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"17:23"}}
18:36:44.455 MQT: stat/tasmota_B64330/STATUS10 = {"StatusSNS":{"Time":"2022-01-17T18:36:44","ANALOG":{"A1":511},"BMP280":{"Temperature":19.9,"Pressure":990.5},"PMS5003":{"CF1":12,"CF2.5":18,"CF10":18,"PM1":12,"PM2.5":18,"PM10":18,"PB0.3":1302,"PB0.5":424,"PB1":80,"PB2.5":6,"PB5":0,"PB10":0},"ESP32":{"Temperature":41.9},"PressureUnit":"hPa","TempUnit":"C"}}
18:36:44.473 MQT: stat/tasmota_B64330/STATUS11 = {"StatusSTS":{"Time":"2022-01-17T18:36:44","Uptime":"0T00:11:28","UptimeSec":688,"Heap":187,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":48},"Wifi":{"AP":1,"SSId":"Home_WiFi","BSSId":"34:8A:AE:B7:33:5A","Channel":9,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Configure PMS5003 Sensor
2. Save template
3. Configure MHZ Sensor
4. Save Template
5. PMS values are gone

### EXPECTED BEHAVIOUR
_After configuring the CO2 sensor the values from the PM sensor should still be visible_

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-17:
As of today, Tasmota on ESP32-C3 is limited to only 1 serial sensor.
ESP32-C3 has 2 hardware uarts but UART0 is locked for the console and only UART1 is available for sensors.
Software serial has not been integrated in any ESP32 so far.

We need to think about it but the solution may takes some time.
Can you move to an ESP32 in the meantime ?

TheBattleWolf, 2022-01-17:
Hello,

thanks for your quick response. Unfortunately I only have this ESP. But I guess I'll have to buy a new one. Please let me know if I can do anything to assist in you solution finding process.

sfromis, 2022-01-17:
FTR, I've verified being able to have two serial sensors on a standard ESP32.

ascillato2, 2022-01-18:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2022-01-18 by ascillato2

barbudor, 2022-01-19:
@TheBattleWolf please see this message https://github.com/arendst/Tasmota/discussions/10203#discussioncomment-1996563
Theo as enabled using uart0 for sensors at the expense of loosing serial console
Please report in the other discussion
Thanks

TheBattleWolf, 2022-01-19:
Works like a charm, thank you guys!

arendst, 2022-01-19:
Great! Thx for reporting back.