arendst/Tasmota #14501: Tasmota stop working

bananajoe86, 2022-01-18:
### PROBLEM DESCRIPTION
Tasmota stop working after a time (between one or two days). The Wemos D1 Mini isn´t working local (physical botton) or can be accessed by interface. I can solve the problem if I switch off the device several times. 6x Powerless does not reset the controller, but I can get it running again with the configuration. A Software-Restart is everytime possible (if the controller is online)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
Configuration
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-01-18:
Without a good crystal ball (we dont have found a source for), it is hard to find the reason. You did not mentioned details nor the issue template is filled.

sfromis, 2022-01-18:
With 3 relays, you have to make sure that your power source is up to the task, and that you have done appropriate wiring for relays, which can draw quite a bit of power (in the context of ESP8266).

bananajoe86, 2022-01-18:
I'm trying to provide as much information as possible to help narrow down the problem. if something is missing please let me know.
The power source should be okay (3A). As long as the device does not go into blackout, it works perfectly with all relays and sensors. For the momenten I haven´t andy idea, why it goes offline regularly and not online again with only one power off.