arendst/Tasmota #14509: Can't compile for ESP8266 with Telegram enabled

FerFemNemBem, 2022-01-19:
On compilation of the latest development (for ESP8266) I get the following compilation errors:
```
Compiling .pio\build\tasmota\lib4a0\bearssl-esp8266\ec\ec_c25519_m15.c.o
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino:54:1: error: 'WiFiClientSecure_light' in namespace 'BearSSL' does not name a type
 BearSSL::WiFiClientSecure_light *telegramClient = nullptr;
 ^
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino: In function 'bool TelegramInit()':
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino:83:10: error: 'telegramClient' was not declared in this scope
     if (!telegramClient) {
          ^
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino:84:28: error: expected type-specifier
       telegramClient = new BearSSL::WiFiClientSecure_light(tls_rx_size, tls_tx_size);
                            ^
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino:84:28: error: expected ';'
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino:89:41: error: 'GoDaddyCAG2_TA' was not declared in this scope
         telegramClient->setTrustAnchor(&GoDaddyCAG2_TA, 1);
                                         ^
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino: In function 'String TelegramConnectToTelegram(const String&)':
S:/Development/git/pearl-luminea-plugs-jan/Tasmota/tasmota/xdrv_40_telegram.ino:111:7: error: 'telegramClient' was not declared in this scope
   if (telegramClient->connect(host.c_str(), 443)) {
       ^
Compiling .pio\build\tasmota\lib4a0\bearssl-esp8266\ec\ec_curve25519.c.o
```
Anyone runs in this problem too?

Only a guess but could this be related to this commit? https://github.com/arendst/Tasmota/commit/8b8f6260b3072db3cb2beeeb335045259ea47f4d

"v10.1.0" compiles fine with the following entries inside the platformio.ini:
```
platform                    = espressif8266 @ 2.6.3
platform_packages     = tasmota/framework-arduinoespressif8266 @ ~2.7.4
                              platformio/tool-esptoolpy @ ~1.30100
```                              
"development" does not compile with the following entries inside the platformio.ini:
```
platform                          = https://github.com/tasmota/platform-espressif8266/releases/download/v2.7.4.9/platform-espressif8266-2.7.4.9.zip
platform_packages           =
```
Changing the "platform*"-entries back to the values of "v10.1.0" did **not** have solved the problem.
Any suggestions what I'm missing or what I could try in order to get the compilation working again.

closed 2022-01-19 by close-issue-app[bot]

Jason2866, 2022-01-19:
Please provide your `user_config_override.h` setup.

ascillato, 2022-01-19:
Found the bug. It is missing the define USE_TLS for TELEGRAM. Making a PR to fix that dependency.

ascillato2, 2022-01-19:
Please try again with latest codebase.

FerFemNemBem, 2022-01-19:
With the latest all compiles fine.
Thank you for the fast solution!

Jason2866, 2022-01-19:
To be clear it was not really a bug. You did undefine needed stuff in your `user_config_override.h`
It did compile successfully without this change! The PR added a safety belt.

FerFemNemBem, 2022-01-19:
@Jason2866 
You are not correct. This was indeed a bug. Take a version from this morning without the fix, leave it untouched except put:
#define USE_TELEGRAM
into the "user_config_orverride.h" => compilation fails.
If it worked for you before the fix was done than you had defined:
#define USE_MQTT_TLS
which is disabled in a default tasmota.

But this doesn't matter as it is fixed now.

Thanks again ascillato2

PR #14512 (2022-01-19): Fix compilation issue for Telegram
## Description:

Fix compilation issue for Telegram.  It was missing the define USE_TLS for TELEGRAM.

**Related issue (if applicable):** fixes #14509

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.2
  - [x] I accept the [CLA](https://gith [... cut]
PR #14514 (2022-01-19): Move Post Processing Macros after user_config_override.h
## Description:

Move Post Processing Macros after user_config_override.h
Fix compilation issue for Telegram and other features that require USE_TLS when the user enables them in user_config_override.h

**Related issue (if applicable):** fixes #14509, #14512

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP [... cut]