arendst/Tasmota #14529: switching from AP1 to AP2 is not working

dweston, 2022-01-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

All attempts I have made to switch a Tasmotarised device from one AP point to another is failing. Try as I may attempting to achieve this by applying in the console the WiFi commands listed at:
https://tasmota.github.io/docs/Commands/#wi-fi
are unsuccessful. 
The only successful method that works to switch APs that works has been to resort to using:
https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recoverych 
which seems an altogether inelegant, blunt instrument, approach to switching this critical parameter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)h
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: 10.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: 
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

04:44:56.913 CMD: Backlog Template; Module; GPIO 255
04:44:56.936 RSL: RESULT = {"NAME":"Ptech MS6107","GPIO":[0,288,0,32,224,225,0,0,227,228,226,0,0,0],"FLAG":0,"BASE":18}
04:44:57.160 RSL: RESULT = {"Module":{"0":"Ptech MS6107"}}
04:44:57.414 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"288":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"224":"Relay1"},"GPIO5":{"225":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"227":"Relay4"},"GPIO13":{"228":"Relay5"},"GPIO14":{"226":"Relay3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
04:51:48.544 CMD: Status 0
04:51:48.550 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ms6107_2","FriendlyName":["ms61072_2_1","ms6107_2_2","ms6107_2_3","ms6107_2_4","ms6107_2_USB"],"Topic":"tasmota_CC07CF","ButtonTopic":"0","Power":31,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
04:51:48.555 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"2T04:51:32","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2022-01-07T01:25:44","SaveCount":87,"SaveAddress":"F5000"}}
04:51:48.560 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"425/699"}}
04:51:48.564 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vista-mobile","Vista_TC_1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208001","2805C80001000600003C5A0A000000000000","00000088","00006000","00004000"]}}
04:51:48.573 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
04:51:48.578 RSL: STATUS5 = {"StatusNET":{"Hostname":"ms6107-2","IPAddress":"10.0.1.202","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.1.1","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:CC:07:CF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
04:51:48.583 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-03T04:51:48","Local":"1970-01-03T04:51:48","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
04:51:48.587 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-03T04:51:48"}}
04:51:48.593 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-03T04:51:48","Uptime":"2T04:51:32","UptimeSec":190292,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"ON","Wifi":{"AP":1,"SSId":"vista-mobile","BSSId":"94:83:C4:03:BD:4C","Channel":1,"Mode":"11n","RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:15"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

00:00:00.001 HDW: ESP8285
00:00:00.050 CFG: Loaded from flash at F7, Count 93
00:00:00.056 QPC: Count 1
00:00:00.062 ROT: Mode 1
00:00:00.063 SRC: Restart
00:00:00.065 Project tasmota - ms6107_2 Version 10.1.0(tasmota)-2_7_4_9(2021-12-08T14:47:33)
00:00:00.218 WIF: Checking connection...
00:00:00.219 WIF: Attempting connection...
00:00:00.555 WIF: Connecting to AP2 xxxxxxxx Channel 1 BSSId 94:83:C4:03:BD:4C in mode 11n as ms6107-2...
00:00:01.421 WIF: Checking connection...
00:00:01.422 WIF: Attempting connection...
00:00:02.421 WIF: Checking connection...
00:00:02.422 WIF: Attempting connection...
[... 53 lines cut ...]
00:00:18.449 NTP: Unable to resolve IP address
00:00:19.407 NTP: Synch time...
00:00:19.420 NTP: Unable to resolve IP address
00:00:20.410 NTP: Synch time...
00:00:20.422 NTP: Unable to resolve IP address

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Apply the console command `backlog ssid2 xxxxxxxx; password2 zzzzzzzzzzzzzzzzzzzzzzzzzzzzzz`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

It is expected that the device successfully switches to the new AP.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-20:
```
00:00:09.394 WIF: Connect failed as AP cannot be reached
```
The message is clear, AP2 doesn't seem to be around

If you have tweaked a lot the wifi settings and your routers, you may want to issue a `reset 3` and then power cycle the device.
This will reset Espressif internal RF calibration parameters (but nothing from Tasmota configuration) and force a Wifi calibration to re-occur.

```
00:00:14.422 NTP: Synch time...
00:00:14.446 NTP: Unable to resolve IP address
```
This is also very clearn your DHCP server (or your static configuration) do not provide a DNS server required to get the IPAddress of the NTP server.
Up to you to define a local in-lan NTPServer by IP if you want Tasmota to be isolated from the Internet, but Tasmota needs a NTP server (or an external RTC or a GPS receive), but it need Time.

dweston, 2022-01-20:
AP2 is definitely around as it is the primary AP for my dwelling. It is the AP which the PC I am communicating to you on now is connected to.

AP1 is a mobile AP I use when traveling and is also used for troubleshooting. I also use it to assist configuring Tasmota devices at home and transporting them to other venues.

Further with respect to AP2, if I employ the troubleshooting solution documented here:
[https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery](url)
I can reset Tasmota to the AP2 credentials and it will immediately connect. Subsequently, of course, I then have to fully reconfigure the device to regain the original functionality. So this works but it seems to be a a very blunt ended tool way of affecting a transfer to a different AP. Hence my raising this issue.

AP2 has been in use for many years, and so there definitely has been much tweaking of its settings over that time, both in terms of hardware updates, in firmware and in terms of network configuration. However in as the Tasmota device that is the subject device ("Hostname":"ms6107-2") in this instance has never previously 'seen' AP2. It is fresh to AP2.

I will follow your advice to issue a `reset 3` and power cycle the device while I still have access to it via AP1. I will then attempt to switch the device to AP2 using the console command `backlog ssid2 xxxxxxxx; password2 zzzzzzzzzzzzzzzzzzzzzzzzzzzzzz`.

As to the point you make that AP1 cannot currently access the NTP as it doesn't have access to the Internet; yes this is the case as we have just returned home from a trip where AP1 was being used to as a portable router.

If it would assist resolving this issue I will connect AP1 as a repeater to my home network.

dweston, 2022-01-20:
Hmm.. I had no difficulty connecting AP1 as a repeater of AP2, however there was a LAN IP conflict as both APs had been set-up to the same IP range. After reconfiguring the AP1 LAN IP range unfortunately the subject Tasmota device ("Hostname":"ms6107-2") has disappeared off the AP1 network and I haven't so far managed to recover access to it. I tried disconnecting AP1 from AP2 and reconfiguring AP1 back to the original IP range however that did not work. In the end I resorted to trying the `Fast Power Cycle Device Recovery` sequence, only that too was unsuccessful. The internal Tasmota device AP was not visible.

Not sure where to from here short of again unsoldering the `ESP8285 Wireless Tag` and reflashing it via serial connection.

rapi3, 2022-01-21:
You have a very big chance to have a HW defective device, this happened to me with few brand new wemos D1 mini.
In my case the TX was ok but the RX was faulty and wemos D1 did not find the AP at distance >1m.
You can find more of this problem here: https://geekenargentina.wordpress.com/2019/10/08/esp8266-connectivity-troubleshooting/

dweston, 2022-01-21:
Thanks @rapi3 but the issue I raised here is not limited in my instance just to this one specific device. It applies to all the Tasmotarised devices I've been working with that use these embedded `ESP8285 Wireless Tags`, and I'm pretty sure it extends to some of the Sonoff devices I have around as well. But I'd need to verify that again.

barbudor, 2022-01-21:
@dweston What is the brand/model of AP2 ? Are you aware that ESPs and Asus are not very friendly to each others ?

I've been using similar configuration for units that I'm preparing home to be installed at my mother's house (AP1 = mother's house, AP2 = my home). So far it worked fine, as soon as I brought the unit to my mother's house they connected.

barbudor, 2022-01-21:
If you set `SetOption56 1` and `Weblog 3`, you should see a list of detected AP at restart with RSSI?
May be that could give you some clues on what Tasmota is seeing ?

ascillato2, 2022-01-24:
Hi, Any news on this?

Jason2866, 2022-01-26:
Closing since no feedback. Issue can not be reproduced.

closed 2022-01-26 by Jason2866

dweston, 2022-01-27:
Still trying to find an hour or so spare to pull down the device, desolder the `ES8285 Wireless Tag` off the mother board and reflash it with Tasmota again so that I can get back to re-creating the symptoms referred to above.

Unfortunately during the testing process I related above I again lost compete access to the Tag using less time consuming means of recovering control of the device.

I should find time at some point this coming weekend.