arendst/Tasmota #14530: Parenthesis in IF expression not working correctly - probably

phedders, 2022-01-20:
### PROBLEM DESCRIPTION

Program Version | 2022.01.3(prhother)
Compiled with:
#define USE_EXPRESSION                         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)
#define SUPPORT_IF_STATEMENT                 // Add support for IF statement in rules (+4k2 code, -332 bytes mem)
#define SUPPORT_MQTT_EVENT                     // Support trigger event with MQTT subscriptions (+3k5 code)

Parenthesis in IF expressions in RULES are not working as I expect them to.
For example this does what I would expect:

12:53:02.274 CMD: rule3 ON Event#test DO IF (%value% > 2) publish stat/%topic%/test4 YES else publish stat/%topic%/test4 NO ENDIF ENDON
12:53:02.280 RUL: Stored uncompressed, would compress from 111 to 72 (-35%)
12:53:02.286 MQT: stat/LightsCoalStore/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":111,"Free":400,"Rules":"ON Event#test DO IF (%value% > 2) publish stat/%topic%/test4 YES else publish stat/%topic%/test4 NO ENDIF ENDON"}}
12:53:04.720 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:53:04.756 RUL: EVENT#TEST performs "IF (1 > 2) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:53:04.764 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:53:04.975 MQT: stat/LightsCoalStore/test4 = NO
12:53:05.628 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:53:05.664 RUL: EVENT#TEST performs "IF (2 > 2) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:53:05.672 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:53:05.883 MQT: stat/LightsCoalStore/test4 = NO
12:53:06.634 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:53:06.670 RUL: EVENT#TEST performs "IF (3 > 2) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:53:06.679 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:53:06.889 MQT: stat/LightsCoalStore/test4 = YES

But this expression always returns true:

12:53:58.377 CMD: rule3 ON Event#test DO IF (%value% > (1+1)) publish stat/%topic%/test4 YES else publish stat/%topic%/test4 NO ENDIF ENDON
12:53:58.383 RUL: Stored uncompressed, would compress from 115 to 77 (-33%)
12:53:58.388 MQT: stat/LightsCoalStore/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":115,"Free":396,"Rules":"ON Event#test DO IF (%value% > (1+1)) publish stat/%topic%/test4 YES else publish stat/%topic%/test4 NO ENDIF ENDON"}}
12:54:02.469 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:54:02.505 RUL: EVENT#TEST performs "IF (1 > (1+1)) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:54:02.514 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:54:02.722 MQT: stat/LightsCoalStore/test4 = YES
12:54:05.749 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:54:05.784 RUL: EVENT#TEST performs "IF (2 > (1+1)) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:54:05.792 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:54:06.000 MQT: stat/LightsCoalStore/test4 = YES
12:54:06.968 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:54:06.003 RUL: EVENT#TEST performs "IF (3 > (1+1)) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:54:07.001 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:54:07.209 MQT: stat/LightsCoalStore/test4 = YES
12:54:08.680 MQT: stat/LightsCoalStore/RESULT = {"Event":"Done"}
12:54:08.716 RUL: EVENT#TEST performs "IF (X > (1+1)) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF"
12:54:08.724 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
12:54:08.934 MQT: stat/LightsCoalStore/test4 = YES

The only difference is

IF (%value% > 2) 
IF (%value% > (1+1)) 

But that could be anything with the parentesis - this fails in the same way:
IF (%value% > (2)) 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____ Various Sonoffs and DIY modules
- [x] Tasmota binary firmware version number used: _____ 2022.01.03
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____OTA
[... 49 lines cut ...]
13:15:57.993 CMD: Grp 0, Cmnd 'IF', Idx 1, Len 97, Data '(1 > (2)) publish stat/LightsCoalStore/test4 YES else publish stat/LightsCoalStore/test4 NO ENDIF'
13:15:57.000 MQT: stat/LightsCoalStore/RESULT = {"If":"Done"}
13:15:58.202 SRC: Backlog
13:15:58.205 CMD: Grp 0, Cmnd 'PUBLISH', Idx 1, Len 30, Data 'stat/LightsCoalStore/test4 YES'
13:15:58.209 MQT: stat/LightsCoalStore/test4 = YES

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Please see above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect that sending event test=1 should result in NO because 1 is not > (2) - but I get YES.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

phedders, 2022-01-20:
I asked about this in Discord as well:
https://discord.com/channels/479389167382691863/490244847568158751/933709820073213992

Also forgot to add my thanks for Tasmota - it is *amazing* code. I have around 60 devices that use it here and around our church building.

sfromis, 2022-01-20:
Even simpler: Are arithmetic expressions supported as part of logical expressions?

phedders, 2022-01-20:
I think my last paste above shows that arithmetic **is** supported in expressions --  (%value% < 5+5) does as expected assuming + has greater priority than the logical test <.

It is only when parenthesis are added inside the logical expression that I have problems... I think!

sfromis, 2022-01-20:
As the outcome of that last test does not depend on getting to the value 10, it does not confirm arithmetic to be supported. In general, things not supported gives no error messages, just "partial execution" without clear results.

phedders, 2022-01-21:
@sfromis Could you explain what you mean by "does not depend on getting to the value 10" ?

I'm probably being a bit slow - but the test is just doing a comparison with a calculated value of 10 and works correctly or at least as I expected it to with expected affinity rules, so basic arithmetic appears to be working - unless the arithmetic is parenthesised.
I would love to be shown that I'm doing something daft (well sort of...) because it would mean there does not need to a fix in the code :)

barbudor, 2022-01-21:
As you are using values below 5 or higher than 10  you don't prove if the actual comparison is done against 5 or against 5+5
You need to test with a value in between 5 and 10

sfromis, 2022-01-21:
What I mean is that it would be possible for your test case `(1 < 5+5)` to match without the addition being performed, in case of the comparison working out as `1 < 5` and then `+5` ignored due to it not being understood. 
A better test strategy would be `(2 < 1+3)`, which will give different outcomes depending on the addition done or not.

phedders, 2022-01-21:
@sfromis Ahha! I understand now. However I think the arithmetic question is getting away from the core problem somewhat but for completeness, using equality for simplicity:

     rule3 on event#test do if (%value% == 1+3) publish %topic/test YES endif endon

does as you would expect:

11:33:33.299 CMD: event test=0
11:33:33.355 RUL: EVENT#TEST performs "if (0 == 1+3) publish %topic/test YES endif"

11:33:36.400 CMD: event test=1
11:33:36.460 RUL: EVENT#TEST performs "if (1 == 1+3) publish %topic/test YES endif"

11:33:39.589 CMD: event test=2
11:33:39.649 RUL: EVENT#TEST performs "if (2 == 1+3) publish %topic/test YES endif"

11:33:42.477 CMD: event test=3
11:33:42.537 RUL: EVENT#TEST performs "if (3 == 1+3) publish %topic/test YES endif"

11:33:45.161 CMD: event **test=4**
11:33:45.220 RUL: EVENT#TEST performs "if (4 == 1+3) publish %topic/test YES endif"
11:33:45.440 MQT: %topic/test = **YES**

11:33:47.940 CMD: event test=5
11:33:47.997 RUL: EVENT#TEST performs "if (5 == 1+3) publish %topic/test YES endif"

**Now with**
rule3 on event#test do if (%value% == **(1+3)**) publish %topic/test YES endif endon

11:35:50.649 CMD: event test=1
11:35:50.710 RUL: EVENT#TEST performs "if **(1 == (1+3))** publish %topic/test YES endif"
11:35:50.930 MQT: %topic/test = **YES**

11:35:57.235 CMD: event test=2
11:35:57.299 RUL: EVENT#TEST performs "if **(2 == (1+3))** publish %topic/test YES endif"
11:35:57.517 MQT: %topic/test = **YES**

11:36:00.122 CMD: event test=3
11:36:00.182 RUL: EVENT#TEST performs "if **(3 == (1+3))** publish %topic/test YES endif"
11:36:00.397 MQT: %topic/test = **YES**

11:36:02.553 CMD: event test=4
11:36:02.612 RUL: EVENT#TEST performs "if **(4 == (1+3))** publish %topic/test YES endif"
11:36:02.829 MQT: %topic/test = **YES**

11:36:04.933 CMD: event test=5
11:36:04.993 RUL: EVENT#TEST performs "if **(5 == (1+3))** publish %topic/test YES endif"
11:36:05.211 MQT: %topic/test = **YES**

Those should definately **not** all return YES indicating that the expression was TRUE. Only one of those was correct.

sfromis, 2022-01-21:
To try to simplify again... Your original case was:
`(%TIME% > 360 AND %TIME% < 1410 AND (%TIME% < (%SUNRISE%+%mem1%) OR %TIME% > (%SUNSET%-%mem2%)))`

My repeated response is that all comparisons should be wrapped in (), and you are trying to wrap arithmetic expressions in superfluous (), and then see it as a problem that you cannot make that work.

If my assumption about where to place () is correct, you should have used:
`((%TIME% > 360) AND (%TIME% < 1410) AND`
`((%TIME% < %SUNRISE%+%mem1%) OR (%TIME% > %SUNSET%-%mem2%)))`

() around each comparison but not around each arithmetic expression. The latter part seems confirmed by your tests, arithmetic seems to work ok without you "obscuring" it with ().

phedders, 2022-01-21:
@sfromis Wow. It seems you've hit the nail on the head for my use case! As we discussed previously I did have () around all the logic previously but also around the arithmetic evaluations 'for completeness'.

So you've got me a working solution for which I am very grateful.

I think it is still a bug that the 'extra' () are not nullified/optimised away? At least it breaks the "do whats obvious" rule IMHO. If not considered fixable or desirable to fix - I can close the ticket - but perhaps a note in the docs would be helpful for the next poor sucker! :)

sfromis, 2022-01-21:
The "do whats obvious" does not apply when it comes to Tasmota commands or rules. The parsing has been optimized to be as simple as feasible, meaning that you cannot treat it as a programming language where you have freedom to embellish the syntax within a general framework.

phedders, 2022-01-21:
Thanks @sfromis  - It seems to be that way at the moment - but not always:

16:31:07.275 CMD: var9=5+2
16:31:07.283 MQT: stat/LightsCoalStore/RESULT = {"Var9":"7.000"}
16:31:13.948 CMD: var9=(5+2)
16:31:13.956 MQT: stat/LightsCoalStore/RESULT = {"Var9":"7.000"}
16:31:18.328 CMD: var9=(5+3)
16:31:18.336 MQT: stat/LightsCoalStore/RESULT = {"Var9":"8.000"}
16:31:25.895 CMD: var9=(5+(4))
16:31:25.903 MQT: stat/LightsCoalStore/RESULT = {"Var9":"9.000"}

The parser there seems to do what I was expected - perhaps it is a different parser although for compact code I assumed it would be the same one.

So the question is - should the documentation be updated (I can give that a go) or should this ticket stay for a code tweak to fix the false TRUE?

sfromis, 2022-01-21:
> ( ) (parentheses can be used to explicitly control the order of operations)
And you are not trying to control the order of operations.

The parsing situation within a logical expression is much more complex than in a standalone expression in a `var` command, as the parser has too look for () in the meaning of controlling the logical expressions. I'm not trying to say what "should" be done, just observing the complexities of the situation, and how "whats obvious" *really* does not apply to Tasmota due to the constraints of doing as much as possible on ESP8266.

ascillato2, 2022-01-24:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2022-01-24 by ascillato2

phedders, 2022-01-24:
Thanks everyone :)