arendst/Tasmota #14537: After restart, if run Buzzer 0 then sleep cleared to 0

lexxai, 2022-01-20:
### PROBLEM DESCRIPTION
After restart with saved sleep settings, if run Buzzer 0 then sleep cleared to 0
```
[23:20:30] cmnd/pir_01/sleep 
[23:20:30] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"100"}}}
[23:20:40] cmnd/pir_01/buzzer 0
[23:20:41] stat/pir_01/RESULT {"Buzzer":"Done"}
[23:20:44] cmnd/pir_01/sleep 
[23:20:44] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"0"}}}
```

But if after restart enter sleep 100, then buzzer 1,1,1, sleep restored to 100.
And buzzer 0, restored to 100 ok.

```
[23:25:48] cmnd/pir_01/buzzer 1,1,1
[23:25:48] stat/pir_01/RESULT {"Buzzer":"Done"}
[23:25:53] cmnd/pir_01/sleep 
[23:25:53] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"0"}}}
[23:26:15] cmnd/pir_01/sleep 100
[23:26:15] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"100"}}}
[23:26:21] cmnd/pir_01/buzzer 0
[23:26:21] stat/pir_01/RESULT {"Buzzer":"Done"}
[23:26:23] cmnd/pir_01/sleep 
[23:26:24] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"0"}}}
[23:26:32] cmnd/pir_01/sleep 100
[23:26:32] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"100"}}}
[23:26:45] cmnd/pir_01/buzzer 1,1,1
[23:26:45] stat/pir_01/RESULT {"Buzzer":"Done"}
[23:26:52] cmnd/pir_01/sleep 
[23:26:52] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"100"}}}
[23:27:06] cmnd/pir_01/buzzer 0
[23:27:07] stat/pir_01/RESULT {"Buzzer":"Done"}
[23:27:08] cmnd/pir_01/sleep 
[23:27:09] stat/pir_01/RESULT {"Sleep":{"100":{"Active":"100"}}} 
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): R2
- [x] Tasmota binary firmware version number used: 10.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
[23:37:38] cmnd/pir_01/Backlog Template; Module; GPIO 255
[23:37:38] stat/pir_01/RESULT {"NAME":"Generic-pir","GPIO":[1,0,0,1,0,0,1,1,224,1,0,0,0,1],"FLAG":0,"BASE":18}
[23:37:38] stat/pir_01/RESULT {"Module":{"0":"Generic-pir"}}
[23:37:39] stat/pir_01/RESULT {"GPIO0":{"192":"Switch_n1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"162":"Switch3"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"195":"Switch_n4"},"GPIO10":{"196":"Switch_n5"},"GPIO12":{"224":"Relay1"},"GPIO13":{"480":"Buzzer"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
buzzer 0
sleep

### EXPECTED BEHAVIOUR
Must be restored sleep to saved value after command

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-01-20:
And what problem do you get?
(You are doing something nonsensical.....)

lexxai, 2022-01-20:
> And what problem do you get? (You are doing something nonsensical.....)
after "buzzer 0", sleep value must be restored. Now sleep value set to 0 after use  "buzzer 0"

lexxai, 2022-01-20:
May be it related with code of it changes:
https://github.com/arendst/Tasmota/pull/9530/commits/a59c56d4b560097e6f96a5c749fc90f8ecd77427

sfromis, 2022-01-21:
You did not respond to my question about what *problem* you get. You see a number, and create an issue without explaining why you think that it is *bad*.

lexxai, 2022-01-21:
I defect problem that LoadAvg informed as 999 in case of use Buzzer 0 on rule.
```
ON LOCK#Action=ON DO Backlog NoDelay;var3 2;RuleTimer2 0;Buzzer 0 BREAK 
```
for now I resolve local problem by 
```
ON LOCK#Action=ON DO Backlog NoDelay;var3 2;RuleTimer2 0;Buzzer 0;sleep 100 BREAK 
```
And State now show ok value of LoadAvg
"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,

sfromis, 2022-01-21:
You've still not even tried to suggest what the *harm* is. What breaks when the numbers are different?

lexxai, 2022-01-21:
if the sleep value is 0, the device hardly sleeps, energy consumption increases

arendst, 2022-02-19:
I had a look at the code and conclude there won't be a fix. It's impossible to know what the last sleep setting was if a ``buzzer 0`` command is executed BEFORE any buzzer command as it could have been changed by the light driver.

For now continue not to use buzzer 0 before any other buzzer command.

closed 2022-02-19 by arendst