arendst/Tasmota #14546: Wifi#RSSI broken when #define MAXFILT 6

paul43210, 2022-01-21:
### PROBLEM DESCRIPTION
Within Tasmota Scripting, the following "rssi=Wifi#RSSI" returns 0 when #define MAXFILT 6

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU
- [X] Tasmota binary firmware version number used: 10.0.1
  - [] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
03:18:31.238 MQT: stat/nodeMCUArtLights/TEMPLATE = {"NAME":"Generic","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
N/A
```
- [X] Provide the output of this command: `Status 0`:
```
03:20:07.721 MQT: stat/nodeMCUArtLights/STATUS = {"Status":{"Module":18,"DeviceName":"nodeMCUArtLights","FriendlyName":["Tasmota","","","",""],"Topic":"nodeMCUArtLights","ButtonTopic":"0","Power":31,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[1,1,0,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
03:20:07.734 MQT: stat/nodeMCUArtLights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://10.2.0.2/tasmota/tasmota-art2.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:36","StartupUTC":"2022-01-21T02:09:31","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-10-20T03:31:02","SaveCount":116,"SaveAddress":"FA000"}}
03:20:07.746 MQT: stat/nodeMCUArtLights/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2022-01-21T02:08:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"356/699"}}
03:20:07.755 MQT: stat/nodeMCUArtLights/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Charlie",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0410801C","2805C8000100060000005A0A000A00000000","00040280","00006000","00004001"]}}
03:20:07.771 MQT: stat/nodeMCUArtLights/STATUS4 = {"StatusMEM":{"ProgramSize":511,"Free":492,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","07000586","04080001","00001C03","00000000","80000801","00000000","00000000","00000000"],"Drivers":"1,2,4,9,10,19,24,29,35","Sensors":"1,2,9,10"}}
03:20:07.783 MQT: stat/nodeMCUArtLights/STATUS5 = {"StatusNET":{"Hostname":"nodeMCUArtLights-4297","IPAddress":"10.2.0.124","Gateway":"10.2.0.2","Subnetmask":"255.255.255.0","DNSServer1":"10.2.0.2","DNSServer2":"8.8.8.8","Mac":"AC:0B:FB:D0:70:C9","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
03:20:07.793 MQT: stat/nodeMCUArtLights/STATUS6 = {"StatusMQT":{"MqttHost":"10.2.0.177","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D070C9","MqttUser":"ohmqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:20:07.805 MQT: stat/nodeMCUArtLights/STATUS7 = {"StatusTIM":{"UTC":"2022-01-21T02:20:07","Local":"2022-01-21T03:20:07","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"17:29"}}
03:20:07.822 MQT: stat/nodeMCUArtLights/STATUS10 = {"StatusSNS":{"Time":"2022-01-21T03:20:07","Switch1":"OFF","Switch2":"OFF","BME680":{"Temperature":26.2,"Humidity":24.8,"DewPoint":4.5,"Pressure":1025.7,"Gas":109.22},"BH1750":{"Illuminance":4},"Global":{"Temperature":26.2,"Humidity":24.8,"DewPoint":4.5,"Pressure":1025.7,"SeaPressure":1026.7},"PressureUnit":"hPa","TempUnit":"C"}}
03:20:07.837 MQT: stat/nodeMCUArtLights/STATUS11 = {"StatusSTS":{"Time":"2022-01-21T03:20:07","Uptime":"0T00:10:36","UptimeSec":636,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"ON","Channel1":78,"POWER2":"ON","Channel2":100,"POWER3":"ON","Channel3":100,"POWER4":"ON","Channel4":100,"POWER5":"ON","Channel5":100,"Color":"C7FFFFFFFF","Wifi":{"AP":1,"SSId":"Charlie","BSSId":"30:23:03:18:2D:25","Channel":4,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}
03:20:08.236 RSSI=72.00
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```
N/A
```

### TO REPRODUCE
Add the following to a script:
```
>T
rssi=Wifi#RSSI
print RSSI=%rssi%
```
The output will be:
`03:20:08.639 RSSI=72.00
`
https://tasmota.github.io/docs/Scripting-Language/#features
"Number of filters (arrays) = 5 (#define MAXFILT)"
As described in the scripting guide, add the following to increase the number of Arrays:
```
#ifdef MAXFILT
#undef MAXFILT
#endif
#define MAXFILT 6
```
The output will be:
`03:25:54.906 RSSI=0.00
`

### EXPECTED BEHAVIOUR
I would expect the RSSI value to not return 0 when the # of max arrays > 5

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

**(Please, remember to close the issue when the problem has been addressed)**

paul43210, 2022-01-21:
I should add, the second I see the following in the console, the Wifi#RSSI starts working:
```
04:16:16.411 T=27.10 : H=23.50 : D=4.50 : P=1026.20 : G=106.80 : I=4.00 : **R=0.00**
04:16:17.418 MQT: tele/nodeMCUArtLights/STATE = {"Time":"2022-01-21T04:16:17","Uptime":"0T00:40:11","UptimeSec":2411,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER1":"ON","Channel1":78,"POWER2":"ON","Channel2":100,"POWER3":"ON","Channel3":100,"POWER4":"ON","Channel4":100,"POWER5":"ON","Channel5":100,"Color":"C7FFFFFFFF","Wifi":{"AP":1,"SSId":"Charlie","BSSId":"30:23:03:18:2D:25","Channel":4,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
04:16:17.438 MQT: tele/nodeMCUArtLights/SENSOR = {"Time":"2022-01-21T04:16:17","Switch1":"OFF","Switch2":"OFF","BME680":{"Temperature":27.1,"Humidity":23.5,"DewPoint":4.5,"Pressure":1026.2,"Gas":106.74},"BH1750":{"Illuminance":4},"Global":{"Temperature":27.1,"Humidity":23.5,"DewPoint":4.5,"Pressure":1026.2,"SeaPressure":1027.3},"PressureUnit":"hPa","TempUnit":"C"}
04:16:26.406 T=27.10 : H=23.50 : D=4.50 : P=1026.20 : G=107.08 : I=4.00 : **R=70.00**

```

It can take around 5 mins for these logs to show.

gemu2015, 2022-01-21:
i can not reproduce this issue.
please post the complete script

paul43210, 2022-01-22:
@gemu2015 that was the complete script, except of the >D section

```
>D
rssi=0

>T
rssi=Wifi#RSSI
print RSSI=%rssi%

```

Here is the complete user_config_override.h
I may be removing a module or two that I should not be.

```
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

// -- Setup your own Wifi settings  ---------------
#undef  STA_SSID1
#define STA_SSID1         "xxxx"             // [Ssid1] Wifi SSID

#undef  STA_PASS1
#define STA_PASS1         "xxxx"     // [Password1] Wifi password

// -- Setup your own MQTT settings  ---------------
#undef  MQTT_HOST
#define MQTT_HOST         "xxxx" // [MqttHost]

#undef  MQTT_PORT
#define MQTT_PORT         1883                   // [MqttPort] MQTT port (10123 on CloudMQTT)

#undef  MQTT_USER
#define MQTT_USER         "xxx"         // [MqttUser] Optional user

#undef  MQTT_PASS
#define MQTT_PASS         "xxxx"         // [MqttPassword] Optional password

// You might even pass some parameters from the command line ----------------------------
// Ie:  export PLATFORMIO_BUILD_FLAGS='-DUSE_CONFIG_OVERRIDE -DMY_IP="192.168.1.99" -DMY_GW="192.168.1.1" -DMY_DNS="192.168.1.1"'

#undef USE_DOMOTICZ
#undef USE_HOME_ASSISTANT
#undef USE_EMULATION_HUE
#undef USE_EMULATION_WEMO
#undef USE_SONOFF_RF
#undef USE_RF_FLASH
#undef USE_SONOFF_SC
#undef USE_TUYA_MCU
#undef USE_ARMTRONIX_DIMMERS
#undef USE_SONOFF_IFAN
#undef USE_ARILUX_RF
#undef USE_SHUTTER
#undef USE_EXS_DIMMER
#undef USE_DEVICE_GROUPS
#undef USE_SONOFF_D1
#undef USE_SHELLY_DIMMER
#undef USE_WS2812
#undef USE_MY92X1
#undef USE_SM16716
#undef USE_SM2135
#undef USE_SONOFF_L1
#undef USE_ELECTRIQ_MOODL
#undef USE_ENERGY_SENSOR
#undef USE_ENERGY_DUMMY
#undef USE_PZEM004T
#undef USE_PZEM_AC
#undef USE_PZEM_DC
#undef USE_MCP39F501
#undef USE_BL09XX
#undef USE_COUNTER
#undef USE_DS18x20
#undef USE_DHT
#undef USE_ADE7953
#undef USE_SERIAL_BRIDGE
#undef USE_IR_REMOTE
#undef USE_IR_RECEIVE

#define USE_I2C
#define USE_BMP
#define USE_BME680
#define USE_BH1750

#ifndef USE_SCRIPT
#define USE_SCRIPT  // adds about 17k flash size, variable ram size
#define USE_SCRIPT_JSON_EXPORT
#define USE_SCRIPT_WEB_DISPLAY
#define SCRIPT_FULL_WEBPAGE
//#define USE_TOUCH_BUTTONS
#define USE_ANGLE_FUNC
#define USE_SCRIPT_I2C
#define USE_SCRIPT_SUB_COMMAND
#define SUPPORT_MQTT_EVENT
#define USE_SCRIPT_GLOBVARS
#ifdef MAXFILT
#undef MAXFILT
#endif
#define MAXFILT 7
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif

#endif  // _USER_CONFIG_OVERRIDE_H_

```

paul43210, 2022-01-22:
Here is the full console output from the script above, and the custom user_config_override.h:
```
01:30:31.631 script compressed to 40 bytes = 86 %
01:30:31.634 Script: nv=1, tv=0, vns=5, vmem=34, smem=2560
01:30:31.697 RSSI=0.00
01:30:31.798 RSSI=0.00
01:30:32.101 RSSI=0.00
01:30:32.200 RSSI=0.00
...
01:30:58.503 RSSI=0.00
01:30:58.594 RSSI=0.00
01:30:58.897 RSSI=0.00
01:30:58.997 RSSI=0.00
01:30:59.301 RSSI=0.00
01:30:59.399 RSSI=0.00
01:30:59.701 RSSI=0.00
01:30:59.801 RSSI=0.00
01:31:00.104 RSSI=0.00
01:31:00.203 RSSI=0.00
01:31:00.297 CMD: status 10
01:31:00.309 MQT: stat/nodeMCUArtLights/STATUS10 = {"StatusSNS":{"Time":"2022-01-22T01:31:00","Switch1":"ON","Switch2":"ON","BME680":{"Temperature":28.0,"Humidity":20.5,"DewPoint":3.3,"Pressure":1029.6,"Gas":100.57},"BH1750":{"Illuminance":4},"Global":{"Temperature":28.0,"Humidity":20.5,"DewPoint":3.3,"Pressure":1029.6,"SeaPressure":1030.8},"PressureUnit":"hPa","TempUnit":"C"}}
01:31:00.497 RSSI=0.00
01:31:00.597 RSSI=0.00
01:31:00.899 RSSI=0.00
01:31:00.999 RSSI=0.00
01:31:01.300 RSSI=0.00
01:31:01.398 RSSI=0.00
01:31:01.702 RSSI=0.00
01:31:01.800 RSSI=0.00
01:31:02.106 RSSI=0.00
01:31:02.204 RSSI=0.00
...
01:32:05.003 RSSI=0.00
01:32:05.304 RSSI=0.00
01:32:05.403 RSSI=0.00
01:32:05.704 RSSI=0.00
01:32:05.802 RSSI=0.00
01:32:06.106 RSSI=0.00
01:32:06.205 RSSI=0.00
01:32:06.499 RSSI=0.00
01:32:06.598 RSSI=0.00
01:32:06.794 MQT: tele/nodeMCUArtLights/STATE = {"Time":"2022-01-22T01:32:06","Uptime":"0T19:55:41","UptimeSec":71741,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":6,"POWER1":"ON","Channel1":78,"POWER2":"ON","Channel2":61,"POWER3":"ON","Channel3":65,"POWER4":"ON","Channel4":80,"POWER5":"ON","Channel5":83,"Color":"C79CA6CCD4","Wifi":{"AP":1,"SSId":"Charlie","BSSId":"30:23:03:18:2D:C5","Channel":8,"Mode":"11n","RSSI":100,"Signal":-26,"LinkCount":2,"Downtime":"0T00:00:09"}}
01:32:06.807 RSSI=100.00
01:32:06.817 MQT: tele/nodeMCUArtLights/SENSOR = {"Time":"2022-01-22T01:32:06","Switch1":"ON","Switch2":"ON","BME680":{"Temperature":27.9,"Humidity":20.7,"DewPoint":3.3,"Pressure":1029.6,"Gas":99.44},"BH1750":{"Illuminance":4},"Global":{"Temperature":27.9,"Humidity":20.7,"DewPoint":3.3,"Pressure":1029.6,"SeaPressure":1030.7},"PressureUnit":"hPa","TempUnit":"C"}
01:32:06.826 RSSI=100.00
01:32:06.897 RSSI=100.00
01:32:07.004 RSSI=100.00
01:32:07.299 RSSI=100.00
01:32:07.398 RSSI=100.00
01:32:07.703 RSSI=100.00
01:32:07.802 RSSI=100.00
01:32:08.104 RSSI=100.00
01:32:08.202 RSSI=100.00
01:32:08.498 RSSI=100.00
01:32:08.598 RSSI=100.00
01:32:08.902 RSSI=100.00
01:32:08.001 RSSI=100.00

```

gemu2015, 2022-01-22:
MAXFILT does nothing if you don't use arrays or filters,
arrays are only allocated when they are defined in >D.
i guess you use a defective build. try the very latest tasmota dev version.

paul43210, 2022-01-22:
I'm running the latest version.
You are correct that MAXFILT has nothing to do with it.
I experience the behaviour when this is compiled in.
```
#define USE_I2C
#define USE_BMP
#define USE_BME680
```
It triggers the ">T" code a lot, but the "Wifi#RSSI" returns 0.
The "BME680#Temperature" works.

Happy to dump some output from this code:
```
  if (bitRead(Settings->rule_enabled, 0) && (TasmotaGlobal.uptime > 4)) {
    ResponseClear();
    uint16_t script_tele_period_save = TasmotaGlobal.tele_period;
    TasmotaGlobal.tele_period = 2;
    XsnsNextCall(FUNC_JSON_APPEND, xsns_index);
    TasmotaGlobal.tele_period = script_tele_period_save;
    if (ResponseLength()) {
      ResponseJsonStart();
      ResponseJsonEnd();
      Run_Scripter(">T", 2, ResponseData());
    }
  }

```

gemu2015, 2022-01-22:
while sensors (SENSOR) JSONs are scanned fast, rssi (STATE) JSON is only emitted on tele period time (defaults to 300 secs)  so the fastest update time for RSSI would be 10 seconds if you set tele period to 10 (minimum tele period time)

paul43210, 2022-01-22:
Thanks. So its expected behaviour. Many thanks for your hard work.

closed 2022-01-24 by ascillato2