arendst/Tasmota #14557: BMP / BME wrong readings after DeepSleep

ludwich66, 2022-01-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I get incorrect temperature values from 
a BME 280 (on ESP32 // Tasmota 10.1.0 by Theo Arends), 
a BMP280 or (on ESP 8285 // Own Firmware Tasmota 2022.01.3 by Theo Arends).
a combination breakout with AHT20 and BMP280 (on ESP 8285 // Own Firmware Tasmota 2022.01.3 by Theo Arends).
All BMP provide a "wrong" value that does not change, the AHT20 sensor provides correct values. BMP280 and AHT20 Board are on the same bus.
If DeepSleep is set to "DeepSleepTime 0" via MQTT, all BMP(and AHT20) have correct values. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: Tasmota 10.1.0 by Theo Arends // Own Firmware Tasmota 2022.01.3 by Theo Arends
  - [X ] Pre-compiled Tasmota 10.1.0 by Theo Arends
  - [ X] Self-compiled Tasmota 2022.01.3 by Theo Arends
- [ ] Flashing tools used:  NodeMCU Flasher 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:26:50.489 CMD: Backlog Template; Module; GPIO 255
19:26:50.553 MQT: stat/tasmota_F8F02C/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
19:26:50.778 MQT: stat/tasmota_F8F02C/RESULT = {"Module":{"1":"ESP32-DevKit"}}
19:26:51.040 MQT: stat/tasmota_F8F02C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"7584":"HallEffect1"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"7585":"HallEffect2"}}
----------------
19:28:23.906 MQT: stat/tasmota_CF1C9E/RESULT = {"NAME":"ESP-M3_8285","GPIO":[1,1,1,1,1,0,0,0,1,0,1,0,1,0],"FLAG":0,"BASE":18}
19:28:24.128 MQT: stat/tasmota_CF1C9E/RESULT = {"Module":{"0":"ESP-M3_8285"}}
19:28:24.383 MQT: stat/tasmota_CF1C9E/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"4096":"DeepSleep"},"GPIO4":{"224":"Relay1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"608":"I2C SCL"},"GPIO13":{"0":"None"},"GPIO14":{"640":"I2C SDA"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
--ESP-M3 ESP8285 with BMP280 / AHT20 Kombisensor--
  STATUS 0 output here:
19:29:03.936 CMD: Status 0
19:29:03.944 MQT: stat/tasmota_CF1C9E/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CF1C9E","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:29:03.950 MQT: stat/tasmota_CF1C9E/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"2022-01-20_ESP8285_2M:BMP,AHT2,MAX44009","RestartReason":"Deep-Sleep Wake","Uptime":"0T01:47:15","StartupUTC":"2022-01-21T16:41:48","Sleep":50,"CfgHolder":4619,"BootCount":11,"BCResetTime":"2022-01-20T16:55:25","SaveCount":36,"SaveAddress":"F8000"}}
19:29:03.956 MQT: stat/tasmota_CF1C9E/STATUS2 = {"StatusFWR":{"Version":"2022.01.3(tasmota)","BuildDateTime":"2022.01.20 15:42:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"373/699"}}
19:29:03.961 MQT: stat/tasmota_CF1C9E/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FritzBoxFonWLAN",""],"TelePeriod":240,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000088","00006000","00004000"]}}
19:29:03.975 MQT: stat/tasmota_CF1C9E/STATUS4 = {"StatusMEM":{"ProgramSize":625,"Free":376,"Heap":13,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F9AC787","04368001","0008044F","002013C0","D000F981","00004004","00001000","00000220"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,57","Sensors":"1,2,3,4,5,9,16,41,63"}}
19:29:03.982 MQT: stat/tasmota_CF1C9E/STATUS5 = {"StatusNET":{"Hostname":"tasmota-CF1C9E-7326","IPAddress":"192.168.178.45","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:CF:1C:9E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:29:03.989 MQT: stat/tasmota_CF1C9E/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.35","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CF1C9E","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:29:03.998 MQT: stat/tasmota_CF1C9E/STATUS7 = {"StatusTIM":{"UTC":"2022-01-21T18:29:03","Local":"2022-01-21T19:29:03","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:53","Sunset":"16:54"}}
19:29:04.002 MQT: stat/tasmota_CF1C9E/STATUS10 = {"StatusSNS":{"Time":"2022-01-21T19:29:03","BMP280":{"Temperature":0.6,"Pressure":966.7},"AHT2X":{"Temperature":-0.8,"Humidity":73.1,"DewPoint":-5.0},"PressureUnit":"hPa","TempUnit":"C"}}
19:29:04.011 MQT: stat/tasmota_CF1C9E/STATUS11 = {"StatusSTS":{"Time":"2022-01-21T19:29:04","Uptime":"0T01:47:16","UptimeSec":6436,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FritzBoxFonWLAN","BSSId":"CC:CE:1E:F4:D9:A6","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}

--ESP32  with BME280-----
19:29:51.016 CMD: status 0
19:29:51.030 MQT: stat/tasmota_F8F02C/STATUS = {"Status":{"Module":1,"DeviceName":"DIY_ESP32_Hasenhöhle","FriendlyName":["Tasmota"],"Topic":"tasmota_F8F02C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:29:51.044 MQT: stat/tasmota_F8F02C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-DE.bin","RestartReason":"Deep Sleep reset digital core","Uptime":"0T00:16:48","StartupUTC":"2022-01-21T18:13:03","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-11-05T17:42:37","SaveCount":108}}
19:29:51.056 MQT: stat/tasmota_F8F02C/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021.12.08 14:52:56","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"419/699"}}
19:29:51.070 MQT: stat/tasmota_F8F02C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FritzBoxFonWLAN",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
[... cut ...]
19:38:57.396 MQT: stat/tasmota_CF1C9E/RESULT = {"DeepSleep":{"Time":"2022-01-21T19:40:00","Epoch":1642790400}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Change Settings for DeepSleepTime

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Picture of the reading with node red Combiboard BMP280/AHT20
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-21:
I have an outdoor, solar + liion powered esp8266 with BME 280. All values provided are correct.
Never seen this.

You get the same behavior from all 3 ESPs?

ascillato2, 2022-01-21:
Just for reference:

#9300, https://github.com/arendst/Tasmota/issues/13236, https://github.com/arendst/Tasmota/commit/26e041ab6efe14da3d918154e9d4d0d87f352198

barbudor, 2022-01-21:
I confirm that I have no problems with BME280 + HDC1080 and DeepSleep
I setup an ESP8266 with a Teleperiod of 330 (5 Min and 30 sec) and a DeepSleepTime of 600 (10 minutes).
I also added a rule `on time#minute do status 10 endon `
With this configuration, the ESP wakes up for 5 minutes, publish sensor once after boot, once every minute and once at Teleperiod just befoire going to sleep. It wakes up every 10 minutes.

- Both sensors are coherent (BME280 + HDC1080) with no more than 0.2°C between the 2 sensors.
- Measure right after wake up are coherent with the other measurments during the 5 minutes of wake time

[image]