arendst/Tasmota #14559: BearSSL client is now causing error -1

ksaye, 2022-01-21:
### PROBLEM DESCRIPTION
BearSSL client is now causing error -1

In the past, I have used the httpsClient.begin(*tlsHttpsClient, dPSURL); to communicate with a HTTPS host.  It had worked back in May, but even rolling back to a prior branch does not work. 

In the following code (https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_02_9_mqtt.ino#L349):

    httpsClient.setReuse(true);

    httpsClient.begin(*tlsHttpsClient, dPSURL);
    httpsClient.addHeader("User-Agent", "Tasmota");
    httpsClient.addHeader("Content-Type", "application/json");
    httpsClient.addHeader("Content-Encoding", "utf-8");
    httpsClient.addHeader("Authorization", dPSSASToken);
    httpsClientReturn = httpsClient.PUT(dPSPutContent);

httpsClientReturn now returns -1.

If I use curl, the Authorization and content works fine:

`
curl -L -i -X PUT -H 'Content-Type: application/json' -H 'User-Agent: Tasmota' -H 'Content-Encoding:  utf-8' -H 'Authorization: SharedAccessSignature sr=0ne001***A%2Fregistrations%2FA020A605E539&sig=Ue%2BWNxr5*****REMOVED*****ScR4%3D&skn=registration&se=1642803271' -d '{"registrationId": "A020A605E539"}' https://global.azure-devices-provisioning.net/0ne001****A/registrations/A020A605E539/register?api-version=2019-03-31
`

Is there a known issue with an update to BearSSL?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: _2022.01.3(tasmota)-2_7_4_9(2022-01-21T15:12:21)____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: __PlatformIO___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

I just followed the documentation here: https://github.com/tasmota/docs/blob/development/docs/Azure-IoT-Central.md

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

In the past that command as returned content, just as the curl command does.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-01-22:
Not known issue. Please try to compile from scratch. Seems that you have some libraries with issues or you have a wrong modification in platformio.ini file

s-hadinger, 2022-01-22:
The main change recently is merging in the same code instance Fingerprint and CA validation. It's now a runtime choice and no more a compile time decision.

In you case, you might want to use `httpsClient->setInsecure();` to disable host identity validation and connect to whatever SSL/TLS host.

closed 2022-01-24 by ascillato2