arendst/Tasmota #14568: Problem parsing HSBColor JSON payload

RyanYuckx, 2022-01-22:
### PROBLEM DESCRIPTION
Tasmota seems to have problems parsing JSON payload for HDBColor. I hope this excerpt from the console log spaks for itself.

````
10:27:09.295 CMD: dimmer 100
10:27:09.320 MQT: stat/bulb02/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0,255,56,0,0","HSBColor":"133,100,100","White":0,"CT":153,"Channel":[0,100,22,0,0]}
10:28:05.732 CMD: HSBColor {"HSBColor":"153,84,100"}
10:28:05.752 MQT: stat/bulb02/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,41,41,0,0","HSBColor":"0,84,100","White":0,"CT":153,"Channel":[100,16,16,0,0]}
10:33:39.136 CMD: HSBColor {"HSBColor":"153,84,70"}
10:33:39.157 MQT: stat/bulb02/RESULT = {"POWER":"ON","Dimmer":70,"Color":"178,29,29,0,0","HSBColor":"0,84,70","White":0,"CT":153,"Channel":[70,11,11,0,0]}
14:17:40.478 CMD: HSBColor {"HSBColor":"60,70,80"}
14:17:40.553 MQT: stat/bulb02/RESULT = {"POWER":"ON","Dimmer":80,"Color":"204,62,62,0,0","HSBColor":"0,70,80","White":0,"CT":153,"Channel":[80,24,24,0,0]}
14:18:09.808 CMD: HSBColor {"HSBColor":"50,90"}
14:18:09.868 MQT: stat/bulb02/RESULT = {"POWER":"ON","Dimmer":80,"Color":"204,21,21,0,0","HSBColor":"0,90,80","White":0,"CT":153,"Channel":[80,8,8,0,0]}
14:28:54.754 CMD: HSBColor {"Dimmer":70,"HSBColor":"40,50,60"}
14:28:54.779 MQT: stat/bulb02/RESULT = {"POWER":"ON","Dimmer":50,"Color":"128,128,128,0,0","HSBColor":"0,0,50","White":0,"CT":153,"Channel":[50,50,50,0,0]}
````

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Calex A60 RGB CCT
- [x] Tasmota binary firmware version number used: 10.0.0, 10.1.0 and 2022.01.3(scripting)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

14:22:52.620 CMD: Backlog Template; Module; GPIO 255
14:22:52.647 MQT: stat/bulb02/RESULT = {"NAME":"Calex E27 RGB ","GPIO":[0,0,0,0,416,419,0,0,417,420,418,0,0,0],"FLAG":0,"BASE":18}
14:22:52.854 MQT: stat/bulb02/RESULT = {"Module":{"0":"Calex E27 RGB "}}
14:22:53.060 MQT: stat/bulb02/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"419":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"420":"PWM5"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

14:23:31.830 CMD: status 0
14:23:31.839 MQT: stat/bulb02/STATUS = {"Status":{"Module":0,"DeviceName":"Bulb02 Tasmota","FriendlyName":["Bulb02"],"Topic":"bulb02","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:23:31.871 MQT: stat/bulb02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/beta/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:06:07","StartupUTC":"2022-01-22T13:17:24","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2022-01-19T23:38:46","SaveCount":201,"SaveAddress":"FB000"}}
14:23:31.902 MQT: stat/bulb02/STATUS2 = {"StatusFWR":{"Version":"2022.01.3(scripting)","BuildDateTime":"2022-01-21T14:32:01","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"386/699"}}
14:23:31.923 MQT: stat/bulb02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"phoenix.squirrel.nl","LogPort":514,"SSId":["Oh Denneboom",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C80001000600003C5A0A000000000000","00000280","00006000","00004000"]}}
14:23:31.958 MQT: stat/bulb02/STATUS4 = {"StatusMEM":{"ProgramSize":711,"Free":292,"Heap":17,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","079AC786","043E8001","000004C3","00000000","C000F989","00000004","00200000","00000000"],"Drivers":"1,2,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,50","Sensors":"1,2,4,5,6,9,53"}}
14:23:31.992 MQT: stat/bulb02/STATUS5 = {"StatusNET":{"Hostname":"bulb02","IPAddress":"192.168.1.84","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.10","DNSServer2":"192.168.1.16","Mac":"EC:FA:BC:64:FA:28","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:23:32.002 MQT: stat/bulb02/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.squirrel.nl","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_64FA28","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:23:32.027 MQT: stat/bulb02/STATUS7 = {"StatusTIM":{"UTC":"2022-01-22T13:23:32","Local":"2022-01-22T14:23:32","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:29","Sunset":"16:57"}}
14:23:32.044 MQT: stat/bulb02/STATUS10 = {"StatusSNS":{"Time":"2022-01-22T14:23:32"}}
14:23:32.055 MQT: stat/bulb02/STATUS11 = {"StatusSTS":{"Time":"2022-01-22T14:23:32","Uptime":"0T00:06:08","UptimeSec":368,"Heap":10,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":80,"Color":"204,21,21,0,0","HSBColor":"0,90,80","White":0,"CT":153,"Channel":[80,8,8,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Oh Denneboom","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

N/A
```

### TO REPRODUCE
1. Enter the command in the command window, or send it via HTTP or MQTT
2. Observe the result

### EXPECTED BEHAVIOUR
I expect the resultant HSB to be what was specified in the JSON.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-22:
Where does in the doc do you see that the command supports a JSON parameter ?

gizmocuz, 2022-01-22:
It's specified in the value template

"hs_val_tpl": "{{value_json.HSBColor.split(',')[0:2]|join(',')}}",

barbudor, 2022-01-22:
Where in Tasmota doc do you see that ?

gizmocuz, 2022-01-22:
This is the config object

`{"name":"Bulb01","stat_t":"tele/bulb01/STATE","avty_t":"tele/bulb01/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/bulb01/POWER","pl_off":"OFF","pl_on":"ON","stat_val_tpl":"{{value_json.POWER}}","uniq_id":"F4FAA9_LI_1","dev":{"ids":["F4FAA9"]},"bri_cmd_t":"cmnd/bulb01/Dimmer","bri_stat_t":"tele/bulb01/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","hs_cmd_t":"cmnd/bulb01/HSBColor","hs_stat_t":"tele/bulb01/STATE","hs_val_tpl":"{{value_json.HSBColor.split(',')[0:2]|join(',')}}","fx_cmd_t":"cmnd/bulb01/Scheme","fx_stat_t":"tele/bulb01/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"],"clr_temp_cmd_t":"cmnd/bulb01/CT","clr_temp_stat_t":"tele/bulb01/STATE","clr_temp_val_tpl":"{{value_json.CT}}","clrm_stat_t":"tele/bulb01/STATE","clrm_val_tpl":"{%if value_json.White%}color_temp{%else%}hs{%endif %}"}`

barbudor, 2022-01-22:
This specifies the answer to the command `HSBColor` which actually is a JSON
But the command itself do not accept a JSON command, check the docs.

By the way, the old Home Assistant discovery used with `SetOption19 1` is deprecated and no maintenance/fix will be provided.

barbudor, 2022-01-22:
For reference [HsbColor](https://tasmota.github.io/docs/Commands/#hsbcolor)

[image]

gizmocuz, 2022-01-22:
@barbudor , I can not make a difference for Tasmota and other hardware (like zwavejs2mqtt and zigbee2mqtt)
The config object specifies it is JSON just as the status objects are also in JSON
To bad it won't get fixed/implemented. We just adapted the Auto Discovery (for other hardware as well), A lot of Tasmota is now supported by default.
The idea is not to read any manual but to be compatible with the protocol.
Hope the developers of Tasmota might consider implementing/correcting it, should not be to difficult
Or there should be a clear way how to distinguish between sending a value plain or in JSON

arendst, 2022-01-22:
Hi gizmocus! (love Domoticz)

I will have a look into this "protocol" and see if I can come up with it something smallish regarding code size.

Let you know once fixed.

Rgrds, Theo.

barbudor, 2022-01-22:
@gizmocuz 
My understanding is that newer HA versions are already interoperable with Tasmota discovery (SO19 0) and that you don't need that anymore

sfromis, 2022-01-22:
In the case of HsbColor, it would be feasible (albeit somewhat cumbersome) to have an alternate command name to match a build using the subscribe feature, and a set of rules to pick up the JSON values to construct the basic command. 

You could also make rules to pick up the values from the already published JSON in a similar way, and the publish an alternate format not being JSON, if the backend system is too inflexible to handle command/result data structure being different.

Doable, but maybe not very practical for saying "supported".

arendst, 2022-01-22:
The inital reason for having non JSON commands is making sure a result message does not re-trigger the same command. 

> We just adapted the Auto Discovery (for other hardware as well), A lot of Tasmota is now supported by default.

Problem is we are deviating away from the legacy discovery message which was designed for HASS. Currently we use a much shorter discovery message AND a plugin in HASS to provide the neccessary information.

In the near future the legacy HASS discovery message will be phased out.

gizmocuz, 2022-01-23:
Yes the problem is we also have 100 plugins and it would be nice getting away from that with one global discovery protocol.

Still problem remains, the status is published in JSON (which makes perfect sense), the hsb value template also mentions value_json

It would be easier to implement only one 'set' topic where you can specify on/multiple of the following

cmnd/bulb01/set
{
  "HSBColor": "360,100,100",
  "Scheme": xx,
  "brightness": xx,
  "color_temp": xx
};

arendst, 2022-01-23:
I'm working on your last suggestion. For tasmota this would be a conversion from json to backlog commands. No big deal as long as the json contains tasmota recognized commands.

arendst, 2022-01-23:
With latest development branvh you can now do this:
```
cmnd/bulb01/json {"HSBColor":"360,100,100","Scheme": xx,"dimmer": xx,"ct": xx}
```

gizmocuz, 2022-01-23:
Great!! Thank you for this!

sciurius, 2022-01-25:
@gizmocuz Note that the command is `cmnd/bulb01/json`, not `cmnd/bulb01/HSBColor`.
Mail me on rya8yu@xs4all.nl and I'll give you access to the lamp so you can see for yourself.

sciurius, 2022-01-25:
I don't do Slack, only Telegram and Signal.

gizmocuz, 2022-01-26:
Slack can also be used from a web page. Look I spent quite some time on your issue and am willing to help.
If you do not want it, please feel free to use the Tasmota Python plugin

Jason2866, 2022-01-26:
@RyanYuckx  Can you please close?

ascillato2, 2022-01-26:
Closing as this issue has been fixed. Thanks.

closed 2022-01-26 by ascillato2

sciurius, 2022-01-28:
Now that Tasmota can handle JSON payload when input via the `json` command, isn't it fairly trivial to add this to the  commands that identify themselves as taking JSON payload: `HSBColor`, `Dimmer`, `Scheme`, `CT` and `White`? 
Something like 'if the value starts with a `{` call the json handler else do what you always did'?

ascillato, 2022-01-28:
Now Tasmota supports:

`cmnd/bulb01/json {"HSBColor":"360,100,100","Scheme": xx,"dimmer": xx,"ct": xx}`

where topic is `cmnd/bulb01/json` and payload is `{"HSBColor":"360,100,100","Scheme": xx,"dimmer": xx,"ct": xx}`

How do you want it to behave? As I understood from your comments, you want also:

`cmnd/bulb01/HSBColor {"HSBColor":"360,100,100"}` ?

where topic is `cmnd/bulb01/HSBColor` and payload is `{"HSBColor":"360,100,100"}` ?

barbudor, 2022-01-28:
The above PR may solve this so that the considered commands will support a JSON payload identical their response payload
```
HSBColor {"HSBColor": "nn, nn, nn"}
CT {"CT": "nnn"}
Dimmer {"Dimmer": "nnn"}
Power {"Power": "xxx"}
Scheme {"Scheme": "nn"}
```

sciurius, 2022-01-28:
IIUC the Domoticz people want to send commands like

    HSBColor {"Dimmer":30,"HSBColor": "nn, nn, nn","STATE":"on"}

ascillato, 2022-01-29:
At this moment, it is possible to send:

```haskell
JSON {"Dimmer":30,"HSBColor": "nn, nn, nn","POWER":"on"}
``` 

And that set of commands inside a JSON don't have neither duplication nor confusion on what it does. As it is now, it is not ok for this use-case?

gizmocuz, 2022-01-29:
Well from my understanding the new JSON commands can be send to 'cmnd/bulb01/json' which is fine (would be better if cmnd/bulb01/HSBColor understood JSON as well ...)
But there is no reference to  'cmnd/bulb01/HSBColor' in the MQTT Auto discovery config object

sfromis, 2022-01-29:
The old MQTT auto discovery format is deprecated, up for removal "sometime", and not expected to be maintained further. I'd suggest to not rely on that.

barbudor, 2022-01-29:
The above PR does exactly that.
Although I have listed only the obvious case, HSBColor, CT, Dimmer, Scheme and Power command will behave the same as JSON command if the parameter starts with a `{`
Alternative is to change the HA Discovery message to state `cmnd/.../json` instead of the current specific command

gizmocuz, 2022-01-29:
@barbudor , can the end user change this ? (I am not very familiar with Tasmota myself, just implementing the Auto Discovery part)

@sfromis , I don't know what is the reason behind it, seems like a fine protocol 'to rule them all'. We have a dedicated plugin for Tasmota, but some people want to use the auto discovery protocol, or moving from HA and they used the Auto Discovery protocol. I don't know. In this case I am trying to help a user getting it to work... On/Off goes fine, setting the HSB color not
That's working fine with other middleware at the moment

I am all for one protocol that supports different middlewares (like Tasmota) and not a different implementation for each single one of them.

barbudor, 2022-01-29:
@gizmocuz if you mean the HA Discovery message ? No it's built into the code

The problem is that almost no-one in the Tasmota core devs is using Home Assistant (I'm not, neither Domoticz) so it's difficult for us to maintain a piece of code that we are not using on a daily basis.
If some-one with the dev skills and using those software wants to maintain this part of the code, he will be welcomed.
Otherwise, that will just be totally dropped soon and removed from the code base.

This is why the tasmota core devs have introduced the Tasmota discovery as an alternative. Up to the backend software to support it.

sfromis, 2022-01-29:
As I see it, the old MQTT auto discovery protocol for HA is a pretty bad protocol, as it has a lot of baggage related to the internal structure of how things are set up in HA, instead of allowing the device to just tell what it can do. People should not even anymore use that old protocol with HA, as the new Tasmota integration replaces it, with better device support. The new format was created due to the old one just not doing the job well, not "merely" a question of developer availability.

arendst, 2022-01-29:
> I am all for one protocol that supports different middlewares (like Tasmota) and not a different implementation for each single one of them.

Holy grail that won't be reached any time soon.

The current situation is that there is a basic Tasmota discovery procedure using two retained MQTT messages. For HA there is a local plug-in that decodes these cryptic messages for HA usage. TasmoManager directly uses the information from the discovery message. 

For domoticz you would need to do the same OR you could use the already available Domoticz/Tasmota plug-in.

Both plug-ins are managed by non-Tasmota programmers.

gizmocuz, 2022-01-30:
I understand and respect you opinion, it's only that your protocol is also a propriety protocol needed to be implemented by every software.
I've been looking at your MQTT documentation and many devices I found on some brokers, and the idea is similar but different.
Thanks all for your time and patience! I will tell Tasmota users to use the plugin and not use option 19