arendst/Tasmota #14583: Tasmota32 Becomes Unresponsive / Locks-Up / Needs Reboot

BFLK, 2022-01-23:
### PROBLEM DESCRIPTION
I am using the latest pre-compiled version of Tasmota32-bluetooth (http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin) on an ESP-32 D1 Mini for reading values from Xiaomi Temperature and Humidity sensors around the house (3 sensors). After around 24-48h the device becomes unresponsive (MQTT / web interface are not working, additionally the device is not reachable via ICMP) and a soft reset (via the reset button) or a power reset is needed for it to start working again. The issue started to occur after upgrading Tasmota32 from version 9.5.0 to 10.1.0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-32 D1 Mini
- [x] Tasmota binary firmware version number used: 10.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:46:42.231 CMD: Backlog Template; Module; GPIO 255
20:46:42.317 MQT: stat/BLEGateway_EDFCD0/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
20:46:42.549 MQT: stat/BLEGateway_EDFCD0/RESULT = {"Module":{"1":"ESP32-DevKit"}}
20:46:42.816 MQT: stat/BLEGateway_EDFCD0/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules are used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:53:17.064 CMD: STATUS 0
20:53:17.081 MQT: stat/BLEGateway_EDFCD0/STATUS = {"Status":{"Module":1,"DeviceName":"BLE Gateway","FriendlyName":["Wemos D1 Mini ESP32"],"Topic":"BLEGateway_EDFCD0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:53:17.132 MQT: stat/BLEGateway_EDFCD0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"0T00:49:47","StartupUTC":"2022-01-23T19:03:30","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2021-08-20T15:20:10","SaveCount":44}}
20:53:17.175 MQT: stat/BLEGateway_EDFCD0/STATUS2 = {"StatusFWR":{"Version":"10.1.0(bluetooth)","BuildDateTime":"2021-12-08T14:50:26","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"433/699"}}
20:53:17.199 MQT: stat/BLEGateway_EDFCD0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004002"]}}
20:53:17.246 MQT: stat/BLEGateway_EDFCD0/STATUS4 = {"StatusMEM":{"ProgramSize":1488,"Free":1856,"Heap":76,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","8F9AC787","00140001","000000CF","010013C0","40000981","00004080","00200000","0000082C"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,38,50,52,59,79,85","Sensors":"1,2,3,5,6,52,62,127"}}
20:53:17.294 MQT: stat/BLEGateway_EDFCD0/STATUS5 = {"StatusNET":{"Hostname":"BLEGateway-EDFCD0-7376","IPAddress":"192.168.1.116","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"24:0A:C4:AA:AA:AA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:53:17.329 MQT: stat/BLEGateway_EDFCD0/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EDFCD0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:53:17.366 MQT: stat/BLEGateway_EDFCD0/STATUS7 = {"StatusTIM":{"UTC":"2022-01-23T19:53:17","Local":"2022-01-23T20:53:17","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:31","Sunset":"17:32"}}
20:53:17.395 MQT: stat/BLEGateway_EDFCD0/STATUS10 = {"StatusSNS":{"Time":"2022-01-23T20:53:17","ESP32":{"Temperature":33.9},"TempUnit":"C"}}
20:53:17.422 MQT: stat/BLEGateway_EDFCD0/STATUS11 = {"StatusSTS":{"Time":"2022-01-23T20:53:17","Uptime":"0T00:49:47","UptimeSec":2987,"Heap":75,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":48},"Wifi":{"AP":1,"SSId":"TEST","BSSId":"D8:47:32:AA:AA:AA","Channel":2,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:12"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
I cannot provide output from the console after the issue occurs as the device is unresponsive / unreachable.

```

### TO REPRODUCE
Install Tasmota32-bluetooth on ESP32 D1 Mini and configure Wifi / MQTT / enable BLE. After somewhere between 24/48h the device hangs and it's unreachable. Seems to stop reporting sensor values via MQTT, it's unreachable via the web interface and stops responding to Ping. This did not happen on Tasmota32-bluetooth 9.5.0. A soft reset via the reset button or a power reset would solve the issue which is reoccurring again after somewhere between 24-48h.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should be stable as it was on 9.5.0.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tony-fav, 2022-01-23:
Are you able to connect the ESP to a computer to record Serial logging? That may be able to get us some post-failure logs.

BFLK, 2022-01-25:
I will try to catch that in a serial trace. After the last reset (no configuration change performed), I am trying to reproduce the issue again - for now I reached 1d 18h of uptime without anything going wrong. Keep you posted.

crhass, 2022-01-25:
I am seeing the same issue on a sonoff basic and sonoff T1 so not just the ESP32 version.
After some time the web interface does not respond and they become unavailable to mqtt.
All my devices were on 10.0, some are not affected, or not regularly.
Upgrade to 10.1.0 is the same.
Rollback to 9.5.0 is stable, web interface works and not dropping from mqtt.

mrekin, 2022-01-25:
Same problem at esp32-bluetooth build. Version":"2022.01.2(bluetooth)" on ESP32-wroom.
Added free heap history logging, but looks like there are no leaks (75-80).

BFLK, 2022-01-25:
I am using Tasmota 10.1.0 on the following ESP8266 devices without any issue: Arilux LC04 / Gosund SP111 / Wemos D1 Mini / Sonoff TH16. The only device where I face this issue is the ESP32 D1 Mini.

Jason2866, 2022-01-25:
@BFLK Please use a different device than a ESP32 D1 mini. Many ESP32 D1 mini have power issues and are not designed as espressif recommends.
In other words many of this boards have issues which are not software related.
Most Boards which are clones of the official devkit are working.

barbudor, 2022-01-25:
Last upgrade for me was 10.1.0.1 on Jan 8th. No problems since then with 8 LYWSD03 flashed with PVVX firmware.
Running on a ESP32 DevKit clone

BFLK, 2022-01-26:
I don't have any other ESP32 devices, only ESP8266 so I can't test it on a different device. The reason I got the ESP32 one is just for the BLE capabilities. However, after the last soft-reset the issue is not occurring anymore which is weird. The only thing that is different from the past is that I access the web interface 2-3 times per day to see if it happened or not. I will leave it untouched for a while to see how it goes. My supposition is that sometimes WIFI disconnects and does not reconnect causing this issue.

When you say that the ESP32 D1 mini has power issues, what do you mean? I am powering this one via USB from a beefy enough power supply. I didn't suspect a power issues because everything was nice and stable on 9.5.0 and issues started with 10.1.0.

Thanks,
Alex

Jason2866, 2022-01-26:
The D1 mini ONBOARD 3.3v vreg is (too) poor. It does not help having a good extern power supply. Wifi problems and partitial reachable device are typical symptoms for power problems.

Jason2866, 2022-01-26:
I will close this until a decent ESP32 have this problems too.
Please reopen if you have such a device and the problems are there too.

closed 2022-01-26 by Jason2866

dkwireless, 2022-01-29:
@Jason2866 I have the same problem on two different devices flashed with latest bluetooth firmware.
ESP32 CAM and ESP32S NodeMCU.
They just hang after some hours/days. 
I have specifically flashed bluetooth firmware to ESP32 CAM just to test if it has the same effect.
I think you should reopen this issue.

barbudor, 2022-01-29:
ESP32 CAM is not know to be a good board neither
I have tasmota32-bluetooth 2022.01.1 running on a ESP32DevKitC for 3 weeks without interruption.
WIth 10.1.0 I also had some lockup

Jason2866, 2022-01-29:
Webcam and Bluetooth will NEVER work reliable together.
Both need much CPU power, too much for the ESP32 to do both.
My ESP32 (decoding 2 LYWSD03) is running 4 days since i did the test setup.
Sorry i cant reproduce the issue.

dkwireless, 2022-01-29:
Just to be clear my main setup is on ESP32S NodeMCU with bluetooth firmware 2022.01.3. 
It monitors 5 Xiaomi sensors with ATC firmware and hangs sometimes in few hours, sometimes in few days. This behavior started from 10.0.0 firmware.
I have just tried reproducing the same setup on ESP32 CAM module with same bluetooth firmware (not custom, camera is off). It has shown the same behavior.

barbudor, 2022-01-29:
@dkwireless Are you running latest dev ?
With 10.1.0 I had 24 to 48 hours locks-up in December. No problems since 2022.01.1 flashed Jan 8th.

dkwireless, 2022-01-30:
@barbudor Yesterday, after reading your message, I have reflashed both devices with latest dev.
They are still running. I will let you know.

flgh, 2022-01-30:
Same problem on AZDelivery ESP32 NodeMCU 10.0.0 and 10.1.0 monitoring 2 Xiaomi sensors reflashed with pvxx and forwarding to MQTT.
Disconnects from wifi after 2-7 days and never reconnects.

`10:37:10.017 CMD: status 0
10:37:10.019 SRC: Serial
10:37:10.020 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
10:37:10.025 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_8298A0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":}
10:37:10.076 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"7T17:19:30","StartupUTC":"2022-01-22T16:17:40","Sleep":50,"CfgHolder":4617,"BootCount":59,"BCResetTime":"2021-11-14T21:31:17","SaveCount":184}}
10:37:10.123 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(bluetooth)","BuildDateTime":"2021-12-08T14:50:26","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"435/699"}}
10:37:10.146 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DrayTek","Terpentine"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A0A000000000000","000000C0","00006000","00004002"]}}
10:37:10.188 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1488,"Free":1856,"Heap":39,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","00140001","000000CF","010013C0","40000981","00004080","00200000","0000082C"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,38,50,52,59,79,85","Sensors":"1,2,3,}
10:37:10.237 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-8298A0-6304","IPAddress":"0.0.0.0","Gateway":"172.16.1.13","Subnetmask":"255.255.255.0","DNSServer1":"172.16.1.22","DNSServer2":"172.16.1.25","Mac":"24:0A:C4:82:98:A0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:37:10.272 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"172.16.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8298A0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:37:10.300 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-30T09:37:10","Local":"2022-01-30T10:37:10","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:22","Sunset":"17:43"}}
10:37:10.337 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-30T10:37:10","ESP32":{"Temperature":43.9,"HallEffect":95},"TempUnit":"C"}}
10:37:10.360 BRY: GC from 7239 to 3774 bytes, objects freed 10/48 (in 1 ms)
10:37:10.363 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-30T10:37:10","Uptime":"7T17:19:30","UptimeSec":667170,"Heap":37,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":48}}}
10:37:11.188 WIF: Checking connection...
10:37:11.189 WIF: Connect failed as AP cannot be reached
10:37:12.188 WIF: Checking connection...
10:37:12.189 WIF: Connect failed as AP cannot be reached
10:37:13.189 WIF: Checking connection...
10:37:13.190 WIF: Connect failed as AP cannot be reached
`

dkwireless, 2022-02-01:
It's been three days with latest dev firmware without restart.
[image]
I think it's good now. Sorry for confusion.

jma89, 2022-02-06:
I'm just popping in to say that I'm seeing this same problem with 10.1.0 on an ESP32 Dev Board from Adafruit.

(My workaround is an automation in Home Assistant that toggles an outlet when the BLE devices go offline.)

Is there a known date when an update will be released to ... Release ... with this fix? (Despite the irony of this issue, I tend to stay off the dev channel for "prod" devices.)

barbudor, 2022-02-06:
You can pop, but that will not change that the fix is there and you can upgrade.
If you want to wait a Release version because you like round numbers, up to you. 
2022.02 is coming this month, normally before mid Feb