arendst/Tasmota #14624: WiFiClientSecureLightBearSSL: _wait_for_handshake: failed

ksaye, 2022-01-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Making an MQTT connection to: iotc-fd8e9a3c-acee-455b-9d3d-1672cf1acbe7.azure-devices.net,8883 or any Azure IoT Hub, now fails the handshake.

With DEBUG_BSSL I see the following:

```
17:41:21.725 MQT: Starting Azure DPS registration...
17:41:21.727 MQT: DPS register for 5CCF7F892B53, scope 0ne001BB77A to https://global.azure-devices-provisioning.net/.
connect(global.azure-devices-provisioning.net,443)
connect(104.214.34.123,443)
_connectSSL: start connection
Connected! MFLNStatus = 0

17:41:26.547 MQT: Azure DPS registration success, no changes.
17:41:26.552 MQT: Attempting connection...
17:41:26.557 MQT: Authenticating with an Azure IoT Hub Token
connect(iotc-fd8e9a3c-acee-455b-9d3d-1672cf1acbe7.azure-devices.net,8883)
connect(13.89.231.149,8883)
_connectSSL: start connection
_wait_for_handshake: failed
Couldn't connect. Error = 0
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: __2022.01.3(tasmota)___
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: _esptool____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Follow the build directions here: https://github.com/tasmota/docs/blob/development/docs/Azure-IoT-Central.md
At line https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_02_9_mqtt.ino#L331 add the line
```
tlsHttpsClient->setInsecure();
```
set #define DEBUG_ESP_SSL in WiFiClientSecureLightBearSSL.cpp

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I am under the impression that the Certificate chain has changed for the Azure IoT Hubs and perhaps it is causing an error in the LightBearSSL

The following simple code will show the issue:
```
      String iotHub = "ksaye.azure-devices.net";
      String deviceId = "tasmota";
      tlsClient.setInsecure();      
      PubSubClient pubSubClient(*tlsClient);
      pubSubClient.setServer(iotHub.c_str(), 8883);
      pubSubClient.setBufferSize(1024);
      String sasToken = "INVALID, but should return Error 5 for invalid password, not -2 for can not connect";
 
      while(!pubSubClient.connected()){
        pubSubClient.connect(deviceId.c_str(), deviceId .c_str(), sasToken.c_str());
        delay(5000);
        AddLog(LOG_LEVEL_INFO, PSTR(D_LOG_MQTT "connect status, %s."), String(pubSubClient.state()).c_str());
      }     
```

And you should see the response:
```
connect(ksaye.azure-devices.net,8883)
connect(20.40.207.8,8883)
_connectSSL: start connection
_wait_for_handshake: failed
Couldn't connect. Error = 0
18:42:49.101 MQT: connect status, -2.
```

**(Please, remember to close the issue when the problem has been addressed)**

ksaye, 2022-01-31:
Open to any suggestion about how to better debug the SSL connection.

s-hadinger, 2022-01-31:
You could try setInsecure to disable any control of the server. This would confirm that the issue is with the certificate

ksaye, 2022-02-10:
@s-hadinger I am already trying that, in the following code sample:

```
      String iotHub = "ksaye.azure-devices.net";
      String deviceId = "tasmota";
      tlsClient.setInsecure();      
      PubSubClient pubSubClient(*tlsClient);
      pubSubClient.setServer(iotHub.c_str(), 8883);
      pubSubClient.setBufferSize(1024);
      String sasToken = "INVALID, but should return Error 5 for invalid password, not -2 for can not connect";
 
      while(!pubSubClient.connected()){
        pubSubClient.connect(deviceId.c_str(), deviceId .c_str(), sasToken.c_str());
        delay(5000);
        AddLog(LOG_LEVEL_INFO, PSTR(D_LOG_MQTT "connect status, %s."), String(pubSubClient.state()).c_str());
      }    
```

ascillato, 2022-03-10:
Hi,

Any news on this?

quil-cityu, 2022-03-17:
I found that the _wait_for_handshake function fail at **line 672** in `WiFiClientSecureLightBearSSL.cpp` file.

The _run_until function return -1 when connected function of the WiFiClient class return false.

redhat421, 2022-04-02:
Interesting. Looks like some complication to making two back to back SSL connections?
If you dump the traffic from the device, do you see it making a TCP connection to the second address?

I noted that I could make an SSL connection to the second IP listed in your debug output using this:
```
% openssl s_client -ciphersuites TLS_AES_128_GCM_SHA256 -connect 13.89.231.149:8883
```

The SSL connection setup looked identical to the sample when trying: `ksaye.azure-devices.net:8883`

Thanks!

maartendamen, 2022-04-14:
I run into the same problem. All Azure connections fail. 
Anything I can do to help with this ticket? Please let me know.

maartendamen, 2022-04-19:
Did some troubleshooting. Also DPS registration is failing:

```
08:50:36.676 MQT: Starting Azure DPS registration...
connect(global.azure-devices-provisioning.net,443)
connect(40.113.176.170,443)
_connectSSL: start connection
_wait_for_handshake: failed
Couldn't connect. Error = 62
```

Using openssl:

`openssl s_client -ciphersuites TLS_AES_128_GCM_SHA256 -connect global.azure-devices-provisioning.net:443`

Gives the following validation error:

`Verify return code: 20 (unable to get local issuer certificate)`

As requested I also did a packet capture.

[image]

maartendamen, 2022-04-25:
Friendly bump, anyone got ideas how to get this forward?

s-hadinger, 2022-05-31:
Nice investigation. I'm very surprised that `ECDHE_RSA_WITH_AES_128_GCM_SHA256` wouldn't be supported. It's one of the most commons ciphers (if not the most) for EC. The RSA one is largely deprecated and shouldn't be enabled again.

pwandrag, 2022-06-10:
I had the same issue with two Sonoff Basic R3's and one ESP01. Spent many iterations figuring out the correct cipher to use.

It seems like Azure dropped the GCM encryption from the ellipic curve diffie helman (ECDHE ciper) from their SSL cert; they only have CBC encryption support for ECDHE RSA AES128 which is called TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256. It is also their most preferred cipher at the moment.

I tried all the RSA CBC based ciphers but I couldn't get the handshake to work so I finally changed the cipher to BR_TLS_RSA_WITH_AES_128_GCM_SHA256 and disabled the elliptic curve as @hnguyen1223 suggested, which worked. My working assuption is because it is based on GCM encryption, so the handhake part of the cipher might be a red herring. There might be a permutation that would get elliptic curve "EC" based cipers to work, but my knowledge of the underlying logic is limited.

Hardcoding a single cipher, is problematic in my view, at least wrap it in a config to allow user settable ciphers to be defined, or at least switch between a ciper per platform e.g. AWS, Mosquito and Azure IoT; and perhaps update https://tasmota.github.io/docs/TLS/#compiling-tls-for-esp8266

Below is the list of supported ciphers for the Microsoft Azure endpoint. I scanned it with https://www.ssllabs.com/ssltest/analyze.html

Microsoft Azure IoT Cert capabilities
# TLS 1.2 (suites in server-preferred order)
TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 
TLS_DHE_RSA_WITH_AES_256_GCM_SHA384 
TLS_DHE_RSA_WITH_AES_128_GCM_SHA256 
TLS_RSA_WITH_AES_256_GCM_SHA384 
TLS_RSA_WITH_AES_128_GCM_SHA256
TLS_RSA_WITH_AES_256_CBC_SHA256
TLS_RSA_WITH_AES_128_CBC_SHA256
TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA 
TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA 
TLS_RSA_WITH_AES_256_CBC_SHA 
TLS_RSA_WITH_AES_128_CBC_SHA 
TLS_RSA_WITH_3DES_EDE_CBC_SHA
# TLS 1.1 (suites in server-preferred order)
TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA 
TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA 
TLS_RSA_WITH_AES_256_CBC_SHA 
TLS_RSA_WITH_AES_128_CBC_SHA 
TLS_RSA_WITH_3DES_EDE_CBC_SHA
# TLS 1.0 (suites in server-preferred order)
TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA 
TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA 
TLS_RSA_WITH_AES_256_CBC_SHA 
TLS_RSA_WITH_AES_128_CBC_SHA 
TLS_RSA_WITH_3DES_EDE_CBC_SHA

> Nice investigation. I'm very surprised that `ECDHE_RSA_WITH_AES_128_GCM_SHA256` wouldn't be supported. It's one of the most commons ciphers (if not the most) for EC. The RSA one is largely deprecated and shouldn't be enabled again.

s-hadinger, 2022-06-10:
Thanks for this dive deep. Hardcoding ciphers is unfortunately a necessity to maintain code size small.

I'm ok to add `ECDHE_RSA_WITH_AES_128_CBC_SHA256` as it should have minimal impact. How did you try CBC? Did you add also the required implementation?

s-hadinger, 2022-06-10:
Can you test:

Add in file `lib/libssl/tls_mini/src/WiFiClientSecureLightBearSSL.cpp`, in `br_ssl_client_base_init` (line 946):

``` C
    // AES CBC for Azure IoT
    br_ssl_engine_set_cbc(&cc->eng, &br_sslrec_in_cbc_vtable, &br_sslrec_out_cbc_vtable);
```

And change ciphers line 922-924:
``` C
  static const uint16_t suites[] = {
    BR_TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256,
    BR_TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256,
  };
```

ksaye, 2022-06-10:
I will do a quick test soon with the most current dev version and report back.

s-hadinger, 2022-06-10:
Hold on, you need to add some more lines (which increases even more the code size):

``` C
    // AES CBC for Azure IoT
    br_ssl_engine_set_cbc(&cc->eng, &br_sslrec_in_cbc_vtable, &br_sslrec_out_cbc_vtable);
    br_ssl_engine_set_aes_cbc(&cc->eng, &br_aes_small_cbcenc_vtable, &br_aes_small_cbcdec_vtable);
```

s-hadinger, 2022-06-13:
@ksaye Did you have a chance to test? I would prefer to have confirmation that the patch is working before pushing to the dev branch.

s-hadinger, 2022-06-13:
Ha ha. No sorry, the "hold on" was over. You can test with the two changes above

ksaye, 2022-06-14:
Sadly, no go.

Here were my steps:

1. Download the dev branch:   git clone -b development https://github.com/arendst/Tasmota.git
2. Modify: tasmota/user_config_override.h

        #undef  STA_SSID1
        #define STA_SSID1         "ssid"             // [Ssid1] Wifi SSID
        
        #undef  STA_PASS1
        #define STA_PASS1         "password"     // [Password1] Wifi password
        
        #ifndef USE_MQTT_TLS
        #define USE_MQTT_TLS
        #endif
        #define USE_MQTT_AZURE_IOT

3. Compile (we expect an error):
4. Per https://tasmota.github.io/docs/Azure-IoT-Hub/#4-configure-your-device-for-you-azure-iot-hub, configure it for Azure IoT Hub

5. View the following error:

        00:00:04.004 HTP: Web server active on tasmota-E16543-1347 with IP address 192.168.15.85
        14:46:58.052 MQT: Attempting connection...
        14:46:58.078 MQT: Authenticating with an Azure IoT Hub Token
        14:46:59.064 MQT: TLS connection error: 0
        14:46:59.066 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 10 sec
        14:46:59.456 QPC: Reset
        14:47:09.937 MQT: Attempting connection...
        14:47:09.997 MQT: Authenticating with an Azure IoT Hub Token
        14:47:10.058 MQT: TLS connection error: 0
        14:47:10.060 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 20 sec

6. Modify: lib/libssl/tls_mini/src/WiFiClientSecureLightBearSSL.cpp

        //static const uint16_t suites[] = {
          //BR_TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
        //};
      
        static const uint16_t suites[] = {
          BR_TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256,
          BR_TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256,
        };
    
        // we support only P256 EC curve for AWS IoT, no EC curve for Letsencrypt unless forced
        //br_ssl_engine_set_ec(&cc->eng, &br_ec_p256_m15); // TODO
        // AES CBC for Azure IoT
        br_ssl_engine_set_cbc(&cc->eng, &br_sslrec_in_cbc_vtable, &br_sslrec_out_cbc_vtable);
        br_ssl_engine_set_aes_cbc(&cc->eng, &br_aes_small_cbcenc_vtable, &br_aes_small_cbcdec_vtable);

7. Compile and upload, not the new error message:

        00:00:12.879 WIF: Connecting to AP1 saye.org in mode 11n as tasmota-E16543-1347...
        00:00:22.012 WIF: Connected
        00:00:22.265 HTP: Web server active on tasmota-E16543-1347 with IP address 192.168.15.85
        14:52:33.264 MQT: Attempting connection...
        14:52:33.317 MQT: Authenticating with an Azure IoT Hub Token
        14:52:34.002 MQT: TLS connection error: 62
        14:52:34.004 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 10 sec
        14:52:48.016 MQT: Attempting connection...
        14:52:48.055 MQT: Authenticating with an Azure IoT Hub Token
        14:52:49.001 MQT: TLS connection error: 62
        14:52:49.004 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 20 sec

s-hadinger, 2022-06-14:
That's better. Error 62 means that the server's certificate is not valid according to local CA. You should turn fingerprint on with `SetOption132 1`

ksaye, 2022-06-14:
Hmm, am I missing something?

      18:45:18.180 MQT: TLS connection error: 26
      18:45:18.182 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 20 sec
      18:45:39.030 MQT: Attempting connection...
      18:45:39.126 MQT: Authenticating with an Azure IoT Hub Token
      18:45:39.205 MQT: TLS connection error: 26
      18:45:39.207 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 30 sec
      18:45:47.440 CMD: SetOption132 1
      18:45:47.445 RSL: RESULT = {"SetOption132":"ON"}
      18:45:48.518 APP: Restarting
      
       ets Jan  8 2013,rst cause:2, boot mode:(3,6)
      
      load 0x4010f000, len 3584, room 16 
      tail 0
      chksum 0xb0
      csum 0xb0
      v3969889e
      ~ld
      
      00:00:00.001 HDW: ESP8266EX
      00:00:00.050 CFG: Loaded from flash at F9, Count 40
      00:00:00.055 QPC: Count 1
      00:00:00.065 Project tasmota - Tasmota Version 11.1.0.4(tasmota)-2_7_4_9(2022-06-14T12:43:45)
      00:00:00.554 WIF: Connecting to AP1 saye.org Channel 3 BSSId F4:9*******:1E in mode 11n as tasmota-1347...
      00:00:01.754 WIF: Connected
      00:00:02.006 HTP: Web server active on tasmota-1347 with IP address 192.168.15.85
      18:45:52.016 MQT: Attempting connection...
      18:45:52.066 MQT: Authenticating with an Azure IoT Hub Token
      18:45:52.148 MQT: TLS connection error: 26
      18:45:52.150 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 10 sec
      18:45:55.463 QPC: Reset
      18:46:03.005 MQT: Attempting connection...
      18:46:03.050 MQT: Authenticating with an Azure IoT Hub Token
      18:46:03.157 MQT: TLS connection error: 26
      18:46:03.159 MQT: Connect failed to ksaye.azure-devices.net:8883, rc -2. Retry in 20 sec

s-hadinger, 2022-06-14:
26 means Invalid Algorithm. I don't have any clue

ksaye, 2022-06-14:
Ha!  I read 26 and thought 62.  yes the error changed and need to noodle on next steps.  @pwandrag any suggestion?

pwandrag, 2022-06-15:
> Ha! I read 26 and thought 62. yes the error changed and need to noodle on next steps. @pwandrag any suggestion?

looking at the return code and where it is used in code, it seems it either had a key computation error (which I doubt), or it couldn't find the ciper in the allowed curves list possibly around line `288` in `ssl_hs_client.c`. Line `1296` in `WifiClientSecureBearSSL.cpp` also hints at that. 

TLDR
The cipher suite consists of four parts, 
- a key echange algorithm; (how the client/server agrees session keys will be safely exchanged)
- an authentication algorithm; (how the keys will be authenticated)
- an bulk encryption algorithm (how the actual data transmission will be encrypted)
- and a hashing algorithm (how the hashing will work (which is the basis of tamper detection)
the 'SSL handshake' tries to figure out which algorithms the client and server both agree on to use in order of preference.

`t_bearssl.h` line `2941` has a nice writeup of the suite logic.

TLS 1.3 drops support for RSA entirely as it is bulky in how it uses large keys, and bloats the handshake process, so 1.3 promises much faster connection setup and teardown, and 2 less algorithms in the suite to worry about; only the encryption and hashing algorithm is decided in the handshake, as key exchange now only occurs over Diffie-Hellman.

closed 2022-07-15 by github-actions[bot]

patricktownley, 2022-08-26:
Not a fix, but a workaround: if you enable "Preview" features (I say enable but I think you can only set it at creation time, and only in very limited regions: Central US, West Europe, Southeast Asia) on the IoT Hub itself, it uses a different cert with better cipher suite support, including the suite Tasmota is using:

**TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256**
TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256
TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384
TLS_DHE_RSA_WITH_AES_256_GCM_SHA384
TLS_DHE_RSA_WITH_AES_128_GCM_SHA256
TLS_RSA_WITH_AES_256_GCM_SHA384
TLS_RSA_WITH_AES_128_GCM_SHA256
TLS_RSA_WITH_AES_256_CBC_SHA256
TLS_RSA_WITH_AES_128_CBC_SHA256
TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA
TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA
TLS_RSA_WITH_AES_256_CBC_SHA
TLS_RSA_WITH_AES_128_CBC_SHA
TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256
TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256
TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384

ksaye, 2022-08-26:
> Not a fix, but a workaround: if you enable "Preview" features (I say enable but I think you can only set it at creation time, and only in very limited regions: Central US, West Europe, Southeast [...]

@patricktownley, have you confirmed that stock Tasmota work if using IoT Hub with Preview enabled?  

I see that Microsoft did add more ciphers: https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-tls-support#elliptic-curve-cryptography-ecc-server-tls-certificate-preview but was also wondering if the TLS fragment lenght: https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-tls-support#tls-maximum-fragment-length-negotiation-preview could be part.

s-hadinger, 2022-08-26:
MFLN is better, but not mandatory unless you send big messages to Tasmota.

patricktownley, 2022-08-26:
@ksaye Yes, confirmed.  Still had to compile with the Azure IOT option of course, and SetOption132,  but but worked from master with no other changes.

paulobriennz, 2022-11-25:
any luck on this issue?  really keen to get IOT Central working with Tasmota, but cannot find a working example..

paulobriennz, 2022-11-26:
i can confirm if i create an IOTHub in US Central region where i can turn on preview function, upload custom firmware with just MQTTS and IOTHub enabled, setoption 132 1, then i get a connection to IOTHub..  now if anyone knows how to make that work with IOT Central !!! :)

01:06:09.646 MQT: Attempting connection...
01:06:09.715 MQT: Authenticating with an Azure IoT Hub Token
01:06:12.013 MQT: TLS connected in 2297 ms, stack low mark 1140
01:06:12.015 MQT: Fingerprint learned: A6 B9 F5 E9 2D 5E 10 CE FA F4 43 DF 05 64 A2 98 64 50 CD 89
01:06:12.450 MQT: Connected