arendst/Tasmota #14626: Channel switching does not work in ZNP

fpalarminiscenario, 2022-01-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Ebyte E18-MS1PA1-IPX
- [x] Tasmota binary firmware version number used: Tasmota Version 10.1.0(tasmota)-2_7_4_9(2022-01-09T14:56:42)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP8266 as CCLoader
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:55:00.681 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:55:00.887 RSL: RESULT = {"Module":{"18":"Generic"}}
20:55:01.140 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"5312":"Zigbee Rst1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"3584":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"3552":"Zigbee Tx"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:56:17.961 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B331C8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:56:17.965 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:29:17","StartupUTC":"2022-01-27T19:27:00","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2022-01-18T12:22:29","SaveCount":28,"SaveAddress":"3FB000"}}
20:56:17.968 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2022-01-09T14:56:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"360/699"}}
20:56:17.970 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EB-WIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
20:56:17.977 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":749,"Free":1296,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04768001","000000CF","010013C0","C000F981","00004084","00201000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37,38,45,46,50","Sensors":"1,2,3,4,5,6"}}
20:56:17.981 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-B331C8-4552","IPAddress":"10.0.0.116","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:B3:31:C8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:56:17.983 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B331C8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:56:17.987 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-27T19:56:17","Local":"2022-01-27T20:56:17","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:26","Sunset":"17:38"}}
20:56:17.990 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-27T20:56:17"}}
20:56:17.993 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-27T20:56:17","Uptime":"0T00:29:17","UptimeSec":1757,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"EB-WIFI","BSSId":"B0:4E:26:08:C5:9E","Channel":3,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
20:59:48.935 CMD: ZbConfig
20:59:48.937 SRC: WebConsole from 10.0.0.111
20:59:48.939 CMD: Grp 0, Cmnd 'ZBCONFIG', Idx 1, Len 0, Data ''
20:59:48.942 RSL: RESULT = {"ZbConfig":{"Channel":18,"PanID":"0x1A63","ExtPanID":"0xCCCCCCCCCCCCCCCC","KeyL":"0x0F0D0B0907050301","KeyH":"0x0D0C0A0806040200","TxRadio":20}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Change the channel by the ZbConfig command and then check in the sniffer if the change really happened

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Channel change, confirming in sniffer

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-01-27:
If you power cycle your device, It is stills on channel 11 instead of 17 ?

fpalarminiscenario, 2022-01-27:
Yes @ascillato2 , I tried turning the device off and on and also closing ZBOSS and opening it again to see if it was really on channel 11

s-hadinger, 2022-01-27:
What is your device? What is the ZNP version?

Can you share the log at `weblog 4` of the znp start?

fpalarminiscenario, 2022-01-27:
Device:
CC2530+CC2592 (Ebyte E18-MS1-PA1-IPX)
Log of the ZNP start:
00:00:00.001 HDW: ESP8266EX
00:00:00.013 UFS: FlashFS mounted with 1992 kB free
00:00:00.063 CFG: Loaded from File, Count 33
00:00:00.067 QPC: Count 1
00:00:00.068 CFG: CR 360/699, Busy 0
00:00:00.072 ROT: Mode 1
00:00:00.584 CFG: Saved to flash at 3FB, Count 34, Bytes 4096
00:00:00.584 ZIG: GPIOs Rx:13 Tx:15
00:00:00.586 SRC: Restart
00:00:00.587 Project tasmota - Tasmota Version 10.1.0(tasmota)-2_7_4_9(2022-01-09T14:56:42)
00:00:00.839 WIF: Checking connection...
00:00:00.840 WIF: Attempting connection...
00:00:01.108 WIF: Connecting to AP1 EB-WIFI Channel 3 BSSId B0:4E:26:08:C5:9E in mode 11n as tasmota-B331C8-4552...
00:00:02.802 WIF: Checking connection...
00:00:02.803 WIF: Connected
00:00:02.804 WIF: Sending Gratuitous ARP
00:00:02.805 NTP: Synch time...
00:00:02.852 RTC: UTC 2022-01-27T20:59:25, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
21:59:25.001 NTP: Synched
21:59:26.007 HTP: Web server active on tasmota-B331C8-4552 with IP address 10.0.0.116
21:59:27.023 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"10.1.0(tasmota)","FallbackTopic":"cmnd/DVES_B331C8_fb/","GroupTopic":"cmnd/tasmotas/"}}
21:59:27.027 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-B331C8-4552","IPAddress":"10.0.0.116"}}
21:59:27.037 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
21:59:27.048 TFS: File 'autoexec.bat' not found
21:59:29.835 QPC: Reset
21:59:31.793 APP: Boot Count 20
21:59:31.796 RSL: STATE = {"Time":"2022-01-27T21:59:31","Uptime":"0T00:00:10","UptimeSec":10,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"EB-WIFI","BSSId":"B0:4E:26:08:C5:9E","Channel":3,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:59:32.702 CFG: Saved to flash at FD, Count 35, Bytes 4096
21:59:38.916 ZIG: rebooting CC2530 device
21:59:40.989 ZIG: {"ZbZNPReceived":"4180010202020702"}
21:59:40.991 RSL: RESULT = {"ZbState":{"Status":1,"Message":"CCxxxx booted","RestartReason":"External","MajorRel":2,"MinorRel":7}}
21:59:41.137 ZIG: checking device configuration
21:59:41.139 ZIG: ZbZNPSent 2102
21:59:41.189 ZIG: {"ZbZNPReceived":"610202020207023B1534010200000000"}
21:59:41.190 RSL: RESULT = {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":7,"MaintRel":2,"Revision":20190523}}
21:59:41.192 ZIG: ZbZNPSent 2108600000
21:59:41.239 ZIG: {"ZbZNPReceived":"6108000155"}
21:59:41.241 ZIG: ZbZNPSent 2108870000
21:59:41.289 ZIG: {"ZbZNPReceived":"6108000100"}
21:59:41.291 ZIG: ZbZNPSent 2108830000
21:59:41.339 ZIG: {"ZbZNPReceived":"61080002FFFF"}
21:59:41.341 RSL: RESULT = {"ZbState":{"Status":2,"Message":"Resetting configuration"}}
21:59:41.343 ZIG: ZbZNPSent 21090300000103
21:59:41.389 ZIG: {"ZbZNPReceived":"610900"}
21:59:41.390 ZIG: ZbZNPSent 410000
21:59:43.516 ZIG: {"ZbZNPReceived":"4180020202020702"}
21:59:43.518 ZIG: ZbZNPSent 210983000002631A
21:59:43.567 ZIG: {"ZbZNPReceived":"610900"}
21:59:43.569 ZIG: ZbZNPSent 21092D000008CCCCCCCCCCCCCCCC
21:59:43.617 ZIG: {"ZbZNPReceived":"610900"}
21:59:43.619 ZIG: ZbZNPSent 21098400000400000400
21:59:43.668 ZIG: {"ZbZNPReceived":"610900"}
21:59:43.670 ZIG: ZbZNPSent 21098700000100
21:59:43.717 ZIG: {"ZbZNPReceived":"610900"}
21:59:43.720 ZIG: ZbZNPSent 21096200001001030507090B0D0F00020406080A0C0D
21:59:43.972 ZIG: {"ZbZNPReceived":"610900"}
21:59:43.974 ZIG: ZbZNPSent 21096300000100
21:59:44.020 ZIG: {"ZbZNPReceived":"610900"}
21:59:44.025 ZIG: ZbZNPSent 210901010020FFFFFFFFFFFFFFFF5A6967426565416C6C69616E636530390000000000000000
21:59:44.070 ZIG: {"ZbZNPReceived":"610902"}
21:59:44.072 ZIG: ZbZNPSent 21098F00000101
21:59:44.121 ZIG: {"ZbZNPReceived":"610900"}
21:59:44.123 ZIG: ZbZNPSent 2107600001000100
21:59:44.170 ZIG: {"ZbZNPReceived":"610700"}
21:59:44.172 ZIG: ZbZNPSent 21096000000155
21:59:44.221 ZIG: {"ZbZNPReceived":"610900"}
21:59:44.224 RSL: RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
21:59:44.226 ZIG: ZbZNPSent 25406400
21:59:45.373 ZIG: {"ZbZNPReceived":"654000"}
21:59:45.423 ZIG: {"ZbZNPReceived":"45C008"}
21:59:45.473 ZIG: {"ZbZNPReceived":"4F80010204"}
21:59:45.794 WIF: Checking connection...
21:59:45.945 ZIG: {"ZbZNPReceived":"45C009"}
21:59:45.948 RSL: RESULT = {"ZbState":{"Status":40,"NewState":9,"Message":"Started as coordinator"}}
21:59:45.950 ZIG: ZbZNPSent 2700
21:59:45.995 ZIG: {"ZbZNPReceived":"4F80000200"}
21:59:46.046 ZIG: {"ZbZNPReceived":"67000082355007004B12000000070900"}
21:59:46.049 RSL: RESULT = {"ZbState":{"Status":51,"IEEEAddr":"0x00124B0007503582","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":0}}
21:59:46.054 ZIG: ZbZNPSent 250200000000
21:59:46.096 ZIG: {"ZbZNPReceived":"650200"}
21:59:46.147 ZIG: {"ZbZNPReceived":"4582000000000000408F000050A0000100A00000"}
21:59:46.149 ZIG: ZbZNPSent 250500000000
21:59:46.195 ZIG: {"ZbZNPReceived":"650500"}
21:59:46.246 ZIG: {"ZbZNPReceived":"4585000000000000"}
21:59:46.249 ZIG: ZbZNPSent 2400010401050000000000
21:59:46.296 ZIG: {"ZbZNPReceived":"640000"}
21:59:46.298 ZIG: ZbZNPSent 24000B0401050000000000
21:59:46.345 ZIG: {"ZbZNPReceived":"640000"}
21:59:46.348 ZIG: ZbZNPSent 21098400000400000400
21:59:46.396 ZIG: {"ZbZNPReceived":"610900"}
21:59:46.398 ZIG: ZbZNPSent 250500000000
21:59:46.447 ZIG: {"ZbZNPReceived":"650500"}
21:59:46.497 ZIG: {"ZbZNPReceived":"45850000000000020B01"}
21:59:46.499 ZIG: ZbZNPSent 25360200000000
21:59:46.547 ZIG: {"ZbZNPReceived":"653600"}
21:59:46.597 ZIG: {"ZbZNPReceived":"45B6000000"}
21:59:46.599 RSL: RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
21:59:46.600 ZIG: Zigbee started
21:59:46.605 ZIG: No Zigbee device information
21:59:46.609 ZIG: No Zigbee device data
22:00:05.773 WIF: Checking connection...
22:00:23.785 WIF: Sending Gratuitous ARP
22:00:25.787 WIF: Checking connection...

fpalarminiscenario, 2022-01-27:
The above log is after a ZbReset 1

And this is after a powercycle:
00:00:00.001 HDW: ESP8266EX
00:00:00.013 UFS: FlashFS mounted with 1992 kB free
00:00:00.053 CFG: Loaded from File, Count 35
00:00:00.056 QPC: Count 1
00:00:00.057 CFG: CR 360/699, Busy 0
00:00:00.061 ROT: Mode 1
00:00:00.062 ZIG: GPIOs Rx:13 Tx:15
00:00:00.063 SRC: Restart
00:00:00.065 Project tasmota - Tasmota Version 10.1.0(tasmota)-2_7_4_9(2022-01-09T14:56:42)
00:00:00.216 WIF: Checking connection...
00:00:00.216 WIF: Attempting connection...
00:00:00.554 WIF: Connecting to AP1 EB-WIFI Channel 3 BSSId B0:4E:26:08:C5:9E in mode 11n as tasmota-B331C8-4552...
00:00:01.553 ZIG: {"ZbZNPReceived":"4180010202020702"}
00:00:01.802 WIF: Checking connection...
00:00:01.802 WIF: Connected
00:00:02.003 HTP: Web server active on tasmota-B331C8-4552 with IP address 10.0.0.116
00:00:02.413 WIF: Sending Gratuitous ARP
00:00:02.414 NTP: Synch time...
00:00:02.458 RTC: UTC 2022-01-27T21:09:18, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
22:09:18.000 NTP: Synched
22:09:19.015 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"10.1.0(tasmota)","FallbackTopic":"cmnd/DVES_B331C8_fb/","GroupTopic":"cmnd/tasmotas/"}}
22:09:19.019 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-B331C8-4552","IPAddress":"10.0.0.116"}}
22:09:19.030 RSL: INFO3 = {"Info3":{"RestartReason":"External System"}}
22:09:19.039 TFS: File 'autoexec.bat' not found
22:09:22.474 QPC: Reset
22:09:23.397 RSL: STATE = {"Time":"2022-01-27T22:09:23","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"EB-WIFI","BSSId":"B0:4E:26:08:C5:9E","Channel":3,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:09:24.396 APP: Boot Count 21
22:09:24.705 CFG: Saved to flash at 3FB, Count 36, Bytes 4096
22:09:31.033 ZIG: rebooting CC2530 device
22:09:34.086 ZIG: {"ZbZNPReceived":"4180010202020702"}
22:09:34.088 RSL: RESULT = {"ZbState":{"Status":1,"Message":"CCxxxx booted","RestartReason":"External","MajorRel":2,"MinorRel":7}}
22:09:34.236 ZIG: checking device configuration
22:09:34.237 ZIG: ZbZNPSent 2102
22:09:34.286 ZIG: {"ZbZNPReceived":"610202020207023B1534010200000000"}
22:09:34.288 RSL: RESULT = {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":7,"MaintRel":2,"Revision":20190523}}
22:09:34.290 ZIG: ZbZNPSent 2108600000
22:09:34.336 ZIG: {"ZbZNPReceived":"6108000155"}
22:09:34.337 ZIG: ZbZNPSent 2108870000
22:09:34.386 ZIG: {"ZbZNPReceived":"6108000100"}
22:09:34.387 ZIG: ZbZNPSent 2108830000
22:09:34.436 ZIG: {"ZbZNPReceived":"61080002631A"}
22:09:34.438 ZIG: ZbZNPSent 21082D0000
22:09:34.486 ZIG: {"ZbZNPReceived":"61080008CCCCCCCCCCCCCCCC"}
22:09:34.487 ZIG: ZbZNPSent 2108840000
22:09:34.536 ZIG: {"ZbZNPReceived":"6108000400000400"}
22:09:34.538 ZIG: ZbZNPSent 2108630000
22:09:34.585 ZIG: {"ZbZNPReceived":"6108000100"}
22:09:34.587 ZIG: ZbZNPSent 2108620000
22:09:34.637 ZIG: {"ZbZNPReceived":"6108001001030507090B0D0F00020406080A0C0D"}
22:09:34.639 RSL: RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
22:09:34.641 ZIG: ZbZNPSent 25406400
22:09:36.390 ZIG: {"ZbZNPReceived":"654000"}
22:09:36.440 ZIG: {"ZbZNPReceived":"45C009"}
22:09:36.442 RSL: RESULT = {"ZbState":{"Status":40,"NewState":9,"Message":"Started as coordinator"}}
22:09:36.444 ZIG: ZbZNPSent 2700
22:09:36.490 ZIG: {"ZbZNPReceived":"4F800D0004"}
22:09:36.540 ZIG: {"ZbZNPReceived":"67000082355007004B12000000070900"}
22:09:36.542 RSL: RESULT = {"ZbState":{"Status":51,"IEEEAddr":"0x00124B0007503582","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":0}}
22:09:36.547 ZIG: ZbZNPSent 250200000000
22:09:36.590 ZIG: {"ZbZNPReceived":"650200"}
22:09:36.640 ZIG: {"ZbZNPReceived":"4582000000000000408F000050A0000100A00000"}
22:09:36.642 ZIG: ZbZNPSent 250500000000
22:09:36.690 ZIG: {"ZbZNPReceived":"4F80080200"}
22:09:36.740 ZIG: {"ZbZNPReceived":"650500"}
22:09:36.790 ZIG: {"ZbZNPReceived":"4585000000000000"}
22:09:36.792 ZIG: ZbZNPSent 2400010401050000000000
22:09:36.839 ZIG: {"ZbZNPReceived":"640000"}
22:09:36.842 ZIG: ZbZNPSent 24000B0401050000000000
22:09:36.890 ZIG: {"ZbZNPReceived":"640000"}
22:09:36.892 ZIG: ZbZNPSent 21098400000400000400
22:09:36.939 ZIG: {"ZbZNPReceived":"610900"}
22:09:36.941 ZIG: ZbZNPSent 250500000000
22:09:36.991 ZIG: {"ZbZNPReceived":"650500"}
22:09:37.039 ZIG: {"ZbZNPReceived":"45850000000000020B01"}
22:09:37.041 ZIG: ZbZNPSent 25360200000000
22:09:37.089 ZIG: {"ZbZNPReceived":"653600"}
22:09:37.139 ZIG: {"ZbZNPReceived":"45B6000000"}
22:09:37.141 RSL: RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
22:09:37.142 ZIG: Zigbee started
22:09:37.147 ZIG: No Zigbee device information
22:09:37.152 ZIG: No Zigbee device data

s-hadinger, 2022-01-27:
This is strange. The channel is set with `ZbZNPSent 21098400000400000400` which is indeed channel 18. 

When reading back the channel information after powercycle it's `{"ZbZNPReceived":"6108000400000400"}` which confirms that the channel configured is 18. 

The only explanation is that the channel information is no more in register 0x0084

s-hadinger, 2022-01-27:
Also be aware that the logs above contains your network encryption key. You'll need to change it in production

s-hadinger, 2022-01-28:
I confirm the bug on a CC2652P with ZNP 2.7.1. It uses channel 11 whatever the configured channel.

ZNP 2.6 works fine, but ZNP 2.7 seems to ignore the channel indication in the original format and probably needs additional configuration.

Stay tuned

fpalarminiscenario, 2022-01-28:
Ok @s-hadinger, about the network encryption key I will change it, this is just for testing. I also tested channel switching on sonoff zbbridge EFR32MG21 and it worked.

PR #14855 (2022-02-15): Zigbee ZStack3 fix to changing channel
## Description:

Fix a bug that would prevent channel change on ZStack 3 (it would always take channel 11). ZStack 3 needs to use the new BDB APIs.
Also allowed EP to be already registered by firmware (#14819)

**Related issue (if applicable):** fixes #14626 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
   [... cut]
closed 2022-02-15 by s-hadinger