arendst/Tasmota #14634: Setoption21 1 does not work with Shelly Plug S

FerFemNemBem, 2022-01-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Shelly Plug S does not show the current "Power" is switched "OFF". Even not, if Setoption21 is set to 1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Shelly Plug S
- [x ] Tasmota binary firmware version number used: 10.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
08:42:48.400 CMD: Backlog Template; Module; GPIO 255
08:42:48.466 RSL: RESULT = {"NAME":"Shelly Plug S","GPIO":[320,1,576,1,1,2720,0,0,2624,32,2656,224,1,4736],"FLAG":0,"BASE":45}
08:42:48.680 RSL: RESULT = {"Module":{"0":"Shelly Plug S"}}
08:42:48.937 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No Rules used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:43:49.788 CMD: Status 0
08:43:49.794 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_42EAA6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:43:49.820 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:21","StartupUTC":"2022-01-28T07:37:28","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2022-01-28T08:36:34","SaveCount":15,"SaveAddress":"F5000"}}
08:43:49.855 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"416/699"}}
08:43:49.873 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FFNB-WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
08:43:49.904 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:43:49.933 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-42EAA6-2726","IPAddress":"192.168.178.115","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"BC:FF:4D:42:EA:A6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:43:49.958 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_42EAA6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:43:49.978 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-28T07:43:49","Local":"2022-01-28T08:43:49","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"17:40"}}
08:43:49.995 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:43:49.015 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-28T08:43:49","ANALOG":{"Temperature":36.2},"ENERGY":{"TotalStartTime":"2022-01-28T08:36:34","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000},"TempUnit":"C"}}
08:43:49.039 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-28T08:43:49","Uptime":"0T00:06:21","UptimeSec":381,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FFNB-WLAN","BSSId":"2C:91:AB:6D:1E:6C","Channel":6,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash a Sehlly Plug S with Tasmota 10.1.0, use the Template from Blackadder: https://templates.blakadder.com/shelly_plug_S.html , Setoption21 to 1, switch the Plug S "OFF" and observe the Power Value. It is "0". After Switch "ON" the Power is shown.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
After Setoption21 to 1 the current Power should be shown even if the device is switched "OFF"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-28:
When switched OFF, no power flows through the device. So only Voltage should display a non 0 Value
However I can see Voltage at 0 in your case:
`08:43:49.015 RSL: STATUS10 = {"StatusSNS":... "ENERGY":{...,"Voltage": 0,...}}`
Does it change anything when you are switching Power ON
- without a load ?
- with a load ?

Did you tested it with original firmware before flashing Tasmota, was power monitoring working ?

FerFemNemBem, 2022-01-28:
Sorry, my fault! I will correct this in my initial Post. Of cause I meant "Voltage" not "Power"!

barbudor, 2022-01-28:
What about the behavior with/without a load ?

FerFemNemBem, 2022-01-28:
[image]
[image]

barbudor, 2022-01-28:
to show voltage in state off, you need `Setoption21 1`
Do you have proper voltage when switch is ON but no load is connected (or with SetOption21 1) ?

FerFemNemBem, 2022-01-28:
Exactly this is the point. See the topic of this issue. Setoption21 1 does not work on Shelly Plug S. On my other devices it works like expected.

sfromis, 2022-01-28:
Don't know details of the hardware, but possibly the energy monitoring is done on the output side of the relay. As a longshot, I'd consider twisting the plug to swap which prongs are connected to live/neutral.

FerFemNemBem, 2022-01-29:
Changing the L/N pongs by twisting the plug 180° does not change anything.

FerFemNemBem, 2022-01-31:
I have doublechecked with the original Shelly Firmware -  Icould not find any place to check the Voltage. Even not using the endpoints described here: https://shelly-api-docs.shelly.cloud/gen1/#shelly-plug-plugs
So maybe this is really a hardware limitation?
Should I close this issue here?
Should this be documented anywhere? (Maybe at: https://templates.blakadder.com/shelly_plug_S.html )

closed 2022-02-09 by FerFemNemBem