arendst/Tasmota #14639: TuyaMCU TYA: RX unknown command / Need Help!

BiOs2k, 2022-01-28:
### PROBLEM DESCRIPTION
Tuya MCU 54 
Dimmer switch everything work physically and via WebUI, but console don't get result of the state of the dimmer so not updating state in WebUI

Tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
TYA: RX unknown command

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Tuya MCU 54 

TuyaMCU 11,1; 
TuyaMCU 21,2;
Ledtable 0;

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:53:52.275 MQT: stat/tasmota_6BEE56/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_6BEE56","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:53:52.282 MQT: stat/tasmota_6BEE56/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:40:47","StartupUTC":"2022-01-28T14:13:05","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2022-01-28T04:07:25","SaveCount":107,"SaveAddress":"F9000"}}
15:53:52.287 MQT: stat/tasmota_6BEE56/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"382/699"}}
15:53:52.292 MQT: stat/tasmota_6BEE56/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["-=BiOs2k-2.4GHz=-",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00108009","2805C80001000600003C5A0A000000000000","00010290","00006000","00004000"]}}
15:53:52.305 MQT: stat/tasmota_6BEE56/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:53:52.316 MQT: stat/tasmota_6BEE56/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6BEE56-3670","IPAddress":"192.168.2.132","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"50:02:91:6B:EE:56","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
15:53:52.322 MQT: stat/tasmota_6BEE56/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6BEE56","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:53:52.329 MQT: stat/tasmota_6BEE56/STATUS7 = {"StatusTIM":{"UTC":"2022-01-28T14:53:52","Local":"2022-01-28T15:53:52","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"17:40"}}
15:53:52.335 MQT: stat/tasmota_6BEE56/STATUS10 = {"StatusSNS":{"Time":"2022-01-28T15:53:52"}}
15:53:52.341 MQT: stat/tasmota_6BEE56/STATUS11 = {"StatusSTS":{"Time":"2022-01-28T15:53:52","Uptime":"0T00:40:47","UptimeSec":2447,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":87,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"-=BiOs2k-2.4GHz=-","BSSId":"60:38:E0:C7:4C:A3","Channel":9,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:54:16.449 MQT: tele/tasmota_6BEE56/STATE = {"Time":"2022-01-28T15:54:16","Uptime":"0T00:41:11","UptimeSec":2471,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":87,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"-=BiOs2k-2.4GHz=-","BSSId":"60:38:E0:C7:4C:A3","Channel":9,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:54:16.497 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:16.499 TYA: RX unknown command
15:54:17.502 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:17.505 TYA: RX unknown command
15:54:18.508 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:18.510 TYA: RX unknown command
15:54:19.513 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:19.515 TYA: RX unknown command
15:54:20.418 TYA: Send "55aa00000000ff"
15:54:20.430 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
15:54:20.432 TYA: Heartbeat
15:54:20.516 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:20.519 TYA: RX unknown command
15:54:21.523 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:21.526 TYA: RX unknown command
15:54:22.527 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
15:54:22.530 TYA: RX unknown command
15:54:23.531 MQT: tele/tasmota_6BEE56/RESULT = {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-01-28:
This issue is similar to https://github.com/arendst/Tasmota/issues/14398. For Tuya issues we add the label **_Help Needed_** due to there is no active development for Tuya drivers in Tasmota at this moment. Sorry.

ascillato2, 2022-02-14:
Closing in favor of https://github.com/arendst/Tasmota/issues/14398