arendst/Tasmota #14681: Shelly 2 power measurement differs by about 10%

andre1974, 2022-01-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: Tasmota 10.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tasmotizer 1.2
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:36:20.093 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:36:20.528 RSL: RESULT = {"Module":{"47":"Shelly 2"}}
00:36:20.779 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2752":"MCP39F5 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2784":"MCP39F5 Rx"},"GPIO4":{"224":"Relay1"},"GPIO5":{"225":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"160":"Switch1"},"GPIO13":{"0":"None"},"GPIO14":{"161":"Switch2"},"GPIO15":{"2816":"MCP39F5 Rst"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:37:02.231 RSL: STATUS = {"Status":{"Module":47,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_5A4305","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:37:02.236 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:36:47","StartupUTC":"","Sleep":250,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2022-01-30T15:06:11","SaveCount":40,"SaveAddress":"F4000"}}
00:37:02.240 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021.12.08 14:47:39","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"422/699"}}
00:37:02.245 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndreSeins_IoT","Garten_IoT"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
00:37:02.254 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:37:02.258 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-5A4305-0773","IPAddress":"10.0.0.15","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"B4:E6:2D:5A:43:05","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:37:02.261 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A4305","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:37:02.266 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:37:02","Local":"1970-01-01T00:37:02","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:37:02.269 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:37:02.277 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:37:02","Switch1":"OFF","Switch2":"OFF","ENERGY":{"TotalStartTime":"2022-01-30T15:06:11","Total":0.042,"Yesterday":0.000,"Today":0.042,"Power":36,"ApparentPower":36,"ReactivePower": 0,"Factor":1.00,"Frequency":50,"Voltage":231,"Current":0.157}}}
00:37:02.282 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:37:02","Uptime":"0T00:36:47","UptimeSec":2207,"Heap":27,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"MqttCount":0,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"AndreSeins_IoT","BSSId":"7A:45:58:19:39:95","Channel":6,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I have calibrated the power measurement on switch1 and everything works fine.
If i switch the same load to switch2 the measurement is nearly 10% lower than on switch1.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-30:
Have you tested the same with original firmware?

andre1974, 2022-01-30:
No.
I flashed two Shelly 2 and it happens on both of them.

barbudor, 2022-01-30:
I don't own any Shelly 2 but in case of doubt I would revert one to the original firmware to compare.
If you want to do it, you can do that using https://github.com/yaourdt/tasmota-to-mg os but you need to strictly follow the procedure

andre1974, 2022-01-30:
Your link leads to a 404 page.

barbudor, 2022-01-30:
Correct, Github seems to have corrupted my link

arendst, 2022-01-31:
As designed. Only one calibration set is available.

andre1974, 2022-02-01:
> As designed. Only one calibration set is available.

Too bad. Thanks for your reply.

Jason2866, 2022-02-01:
Closing since answered

closed 2022-02-01 by Jason2866