arendst/Tasmota #14688: When trying #14385, EthDNSServer1 and  EthDNSServer2 not correctly shown in Tasmota Web UI "Information"

labmaster, 2022-02-01:
### PROBLEM DESCRIPTION
tried to use the additions (fixed IP settings for Ethernet) from #14385 
It seem this is working so far.

For checking the settings i used the "Information" menu from the WebUI, 
but there, the DNS Server1 and DNS Server2 in the Ethernet Section always stays at 0.0.0.0 independent of the settings done on the command line ( e.g. EthDNSServer1 192.168.2.1 or EthDNSServer2 8.8.8.8  ..... )

Beside this i do not even know if the manual DNS Settings recognized at all.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Olimex ESP32-POE
- [x] Tasmota binary firmware version number used: 2022.01.3(tasmota)  / Core/SDK Version	2_0_2_1/v4.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: WebUI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
use command "EthDNSServer1" or "EthDNSServer2" to set Ethernet DNS Servers and check in WebUI if "DNS Server1" and/or "DNSServer 2" is show accordingly 

### EXPECTED BEHAVIOUR
use command "EthDNSServer1" or "EthDNSServer2" to set Ethernet DNS Servers 
Settings in WebUI if "DNS Server1" and/or "DNSServer 2" should be shown accordingly 

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-02-01:
I saw the same while implementing this ethernet driver work. i guess because the ethernet driver is a subclass of the wifi driver the DNS code just keeps using the wifi configured dns serers.

Solving this will probably need to be done ate core or idf level.

barbudor, 2022-02-01:
It's very strange
It looks like the EthDNS are overwritten
And if wifi is disabled, you get no DNS at all.
I'm trying to understand ...

barbudor, 2022-02-01:
Fixed by PR above
The problem was not only a display problem as DNS settings where actually really overwritten to 0.0.0.0

Thanks for reporting @labmaster 
You should be notified when the fix is merged

ascillato2, 2022-02-01:
PR Merged !

Thanks !!!

closed 2022-02-01 by ascillato2

labmaster, 2022-02-02:
Thanks a lot !!!
You guys are great.