arendst/Tasmota #14704: Everytime Color<x> set Dimmer back to 100

Senfruecken, 2022-02-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Everytime Color<x> set Dimmer back to 100.
Because of that I have to correct the dimmer and now it is so that the leds flash short, because of the 100.
The command are send over mqtt.
I set Dimmer to 20. Later I have to change the color of my PixelRing and with the changing the dimmer goes to 100. I can send also Dimmer 20, but than I have a short flashing to 100.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wemos d1 + Pixel Ring 16
- [x] Tasmota binary firmware version number used: 2022.01.3
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

Generic:

D3 GPIO0 : WS2812
TX GPIO1 : None
D4 GPIO2 : LedLink
RX GPIO3 : None
D2 GPIO4 : PN532 Tx
D1 GPIO5 : PN532 Rx
D6 GPIO12 : I2C SDA
D7 GPIO13 : I2C SCL
D5 GPIO14 : None
D8 GPIO15 : None
D0 GPIO16 : None
A0 GPIO17 : None

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

CMD: Color 255,0,55
CMD: Dimmer 20
CMD: Color 40,0,255

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected, that the dimmer has the old value and that not overwritten by the RGBxx value

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-02-02:
As designed. Use `color2` command instead.

Explained in the docs at https://tasmota.github.io/docs/Commands/#color

closed 2022-02-02 by ascillato2

Senfruecken, 2022-02-02:
Sorry, I read and tried the commandos all evening, but overlooked that.  I'll try again tomorrow... Thank you