arendst/Tasmota #14711: Mixed timer programming fixed time / sunrise goes wrong

dwarning, 2022-02-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If I program a time slot with fixed time ON, e.g. 6:00 and OFF with sunrise the device doesn't work. Same for the other slot: Switch ON with sunset and OFF with fixed time, e.g. 23:00.
The timer works correct with fixed times for ON and OFF.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Luminea wall box
- [x] Tasmota binary firmware version number used: 10.1.0 (same for version 10.0.0)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Want allow mixed programming capability with fixed time and sunrise/sunset time.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-03:
You may think that providing details are optional and that we can use a crystal ball to infere your configuration.
But that's unfortunately wrong.

It doesn't work means nothing and doesn't provide any clue of how you have configured your Tasmota
Is that the sunrise/sunset event that doesn't work ?
Or the fixed time ?
Why not providing logs at the expected times to give more details ?
Have you a proper time ? NTP ?
Have you configure latitude and longitude correctly to your location (or approximate) ? Otherwise Tasmorta has no way to calculate sunrise/sunset time.

ascillato2, 2022-02-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2022-02-04 by ascillato2

dwarning, 2022-02-04:
> You may think that providing details are optional and that we can use a crystal ball to infere your configuration. But that's unfortunately wrong.
>
> It doesn't work means nothing and doesn't provide [...]

Wow, what a friendly atmosphere. And ticket closed so quick.
To your questions:
For timer programmable wall plug, _doesn't work_ means that is not switching at the expected time.
Failing is the sunset/sunrise switch action - not the time programmable event.
As I said, pure time programmed slots are working correct, so your NTP question is needless.
Longitude/Latitude seems OK because in the programming menu the correct time (e.g. 8:16) is shown:
[image]

barbudor, 2022-02-04:
You are asking us to spend time to help so spend a little time to explain.
Issue as been close because this is most likely a help request

> For timer programmable wall plug, doesn't work means that is not switching at the expected time.
> Failing is the sunset/sunrise switch action - not the time programmable event.

Thanks for clarifying, that could have been done in the 1st post
As requested in the Issue template, please provide logs with `Weblog 4` around the time you were expecting a sunrise/sunset action.

> As I said, pure time programmed slots are working correct, so your NTP question is needless.

Thank you for trusting Tasmota collaborators to ask for the right question.
Please confirm your NTP is working fine

> Longitude/Latitude seems OK because in the programming menu the correct time (e.g. 8:16) is shown:

Can't you just answer the question and confirm you have entered proper Latitude and Longitude ?

dwarning, 2022-02-06:
OK, I spend little too - but w/o success, so far. This is my console weblog today morning with noticed sunrise 7:37:
`07:35:55.418 WIF: Sending Gratuitous ARP
07:36:02.432 MQT: tele/tasmota_D27C0A/STATE = {"Time":"2022-02-06T07:36:02","Uptime":"2T21:01:10","UptimeSec":248470,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"on","Wifi":{"AP":1,"SSId":"Woodstock","BSSId":"DC:39:6F:42:07:D9","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}
07:36:02.463 MQT: tele/tasmota_D27C0A/SENSOR = {"Time":"2022-02-06T07:36:02","ENERGY":{"TotalStartTime":"2021-04-20T11:13:58","Total":54.202,"Yesterday":0.043,"Today":0.016,"Period": 0,"Power": 8,"ApparentPower": 8,"ReactivePower": 0,"Factor":1.00,"Voltage":241,"Current":0.035}}
07:36:09.546 WIF: Checking connection...
07:36:29.572 WIF: Checking connection...
07:36:33.429 NTP: Synch time...
07:36:33.480 RTC: UTC 2022-02-06T06:36:33, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
07:36:33.000 NTP: Synched
07:36:49.556 WIF: Checking connection...
07:36:55.461 WIF: Sending Gratuitous ARP
07:37:09.584 WIF: Checking connection...
07:37:29.568 WIF: Checking connection...
07:37:49.545 WIF: Checking connection...
07:37:55.457 WIF: Sending Gratuitous ARP

08:54:59.510 CMD: NTPserver1
08:54:59.517 MQT: stat/tasmota_D27C0A/RESULT = {"NtpServer1":"pool.ntp.org"}

07:38:31.667 CMD: longitude
07:38:31.670 SRC: WebConsole from XXX.XXX.XXX.XXX
07:38:31.672 CMD: Grp 0, Cmnd 'LONGITUDE', Idx 1, Len 0, Data ''
07:38:31.678 MQT: stat/tasmota_D27C0A/RESULT = {"Longitude":5Y.GB0000}

07:39:11.222 CMD: latitude
07:39:11.224 SRC: WebConsole from XXX.XXX.XXX.XXX
07:39:11.227 CMD: Grp 0, Cmnd 'LATITUDE', Idx 1, Len 0, Data ''
07:39:11.234 MQT: stat/tasmota_D27C0A/RESULT = {"Latitude":1Z.DW0001}`

I faked my IP and the coordinates for this forum. But you can trust me, they are correct for Berlin.
I was not able to rise the weblog level. I checked level 4 in the scroll-box and then SAVE, but it stays still in info level. Sorry. I attached my config, perhaps you are able to analyze.

Thanks for your attention.

[Config_tasmota_D27C0A_7178_10.1.0.zip](https://github.com/arendst/Tasmota/files/8009692/Config_tasmota_D27C0A_7178_10.1.0.zip)

barbudor, 2022-02-06:
I you want confidentiality, don't post the configuration file on the forum.

`WebLog` is set by a command of the same name through the console, not from a menu : https://tasmota.github.io/docs/Commands/#weblog

When I enter the following coordinates for Berlin (From Google map, close to the Dom)
```
09:50:02.238 CMD: backlog latitude 52.518617128288916; longitude 13.40432468475509
09:50:02.329 MQT: stat/nodemcu/LATITUDE = {"Latitude":52.518616}
09:50:02.539 MQT: stat/nodemcu/LONGITUDE = {"Longitude":13.404325}
```
I have the following time listed for sunset and sunrise
[image]
Which is very coherent with values I see on various websites (+/- 1 or 2 minutes)
But very far from the 8:16 and 17:51 that you are listing which would position you more closely to Paris than Berlin.
It appears that the default coordinate in Tasmota is those of the Eiffel Tower (I need to ask Theo for the story bahind that). So I will assume your initial screenshot was taken before setting the coordinates to Berlin.

So, borrowing Elon's jet, I jump in the middle of the Altlantic just for you (N45 W30) to have a sunset time of 10:22 and onfigured a relay toogle:
[image]

Et voilà:
```
10:21:39.487 MQT: tele/nodemcu/STATE = {"Time":"2022-02-06T10:21:39+01:00","Uptime":"0T00:05:13","UptimeSec":313,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}
10:21:50.121 HTP: Console
10:21:54.251 WIF: Checking connection...
10:22:00.506 SRC: Timer
10:22:00.514 MQT: stat/nodemcu/POWER = {"POWER":"ON"}
10:22:00.518 MQT: stat/nodemcu/POWER = ON (retained)
10:22:01.098 CFG: Saved to flash at F4, Count 20, Bytes 4096
```

There seems to be something wrong on your side, although I can't point what it is at that time.

barbudor, 2022-02-06:
Fomr your config I see that you are using a fixed timezone of 1
You may want to try with the auto mode that support switching automatically between winter (STD) and summer time (DST).
You need to use `Timezone 99` and set `TimeSTD` and `TimeDST` as needed althouh I believe that Tasmota's default are ok for all across EU.

My config is:
```haskell
10:47:50.992 CMD: timezone
10:47:51.001 MQT: stat/nodemcu/TIMEZONE = {"Timezone":99}
10:47:56.797 CMD: timedst
10:47:56.808 MQT: stat/nodemcu/TIMEDST = {"TimeDst":{"Hemisphere":0,"Week":0,"Month":3,"Day":1,"Hour":2,"Offset":120}}
10:48:00.592 CMD: timestd
10:48:00.603 MQT: stat/nodemcu/TIMESTD = {"TimeStd":{"Hemisphere":0,"Week":0,"Month":10,"Day":1,"Hour":3,"Offset":60}}
```

dwarning, 2022-02-06:
My first post has have wrong sunrise/sunset times (8:16 and 17:51) because the Tasmota default coordinates (Netherland?).
Yesterday I corrected Latitudes/Longitudes.

Yes - today with similar coordinates like Berlin Dom I have had 7:37 for sunrise. But it has not switched, also not 5 hours later.

Timezone, DST and STD I have just corrected. And I will start debug log before next event.

Life would be easier to left out the menu entry for debug levels if they not are functional.

arendst, 2022-02-06:
> Life would be easier to left out the menu entry for debug levels if they not are functional.

Works just fine on my devices. No issues at all. Might be a sign you're doing something not quite right...

barbudor, 2022-02-06:
The menu IS functionnal
Sorry if my comment was mis interpreted
It's just like when you are in the Console pageto looking at the log, it is way more reasy to type the command than go back to the configuration menu.
And I must admit that I'm so used to commands `Weblog` `SerialLog` and `SysLog` that I totally forgot ther eis a menu about that.

dwarning, 2022-02-07:
Just greped the weblog 4 and attached, IP and coordinates are faked.
The timeslot stay at 6:00 fixed time ON, and sunrise (7:36 Berlin) OFF.
Nothing happen - the lamp still burning.
What could I do at next? Timer source code looked complex. Can I provide other info from console?

Regarding Menu/Configuration I am not sure. The difference in verbosity is small. So it can be that I am wrong. Will test it again.

[neu 2.txt](https://github.com/arendst/Tasmota/files/8013000/neu.2.txt)
.

dwarning, 2022-02-08:
Ok, my fault.

At first I used this timer for fixed time ON/OFF. Then I switched to the mixed timer configuration, means fixed time / sunrise. But I take no care about +/- entries below in the menu. The formerly fixed times are now interpreted as offsets. And they were extreme.

Preferable would be to change the behavior of the GUI if I switch from time to sunrise/sunset to set these entries automatic to 0.

Sorry for my alarm and thanks for your help.

barbudor, 2022-02-08:
Good to know you found the right config

Changing the values would requires additional (javascript) code which we prefer to save for actual feature code 😂
Eevry byte counts on ESP8266

barbudor, 2022-02-08:
I could also have noticed the +8:00 shift in your initial screenshot 😎