arendst/Tasmota #14819: TI cc2538 cc2592 won't start as coordinator with latest firmware from jet-home.ru

m0m4x, 2022-02-12:
### PROBLEM DESCRIPTION
Hi! 
I have Tasmota 11.0.0.1 running on board d1_mini_pro connected via UART with ti cc2538+cc2592 flashed with [firmware revision 20211222 from jet-home.ru](https://github.com/jethome-ru/zigbee-firmware/blob/master/ti/coordinator/cc2538_cc2592/) 
I get Abort after "Started as coordinator" (see below).  
All works fine with fimware revision 20201010 from same author jet-home. 
Same problem if i flash older version of tasmota and/or on different boards (example. nodemcu_v2 ) 
Hardware cc2538 bought from here [Aliexpress](https://it.aliexpress.com/item/4000588326203.html?spm=a2g0s.9042311.0.0.27424c4dvSqjNq) , flashed with j-link from segger.
Thank you!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nodemcu ESP8266, wemos D1 mini pro, ti cc2538+cc2592
- [x] Tasmota binary firmware version number used: 11.0.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VS Code + PlatformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:31:28.977 CMD: Backlog Template; Module; GPIO 255
00:31:28.978 SRC: WebConsole from 192.168.1.150
00:31:28.981 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
00:31:29.301 SRC: Backlog
00:31:29.303 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
00:31:29.310 MQT: stat/tasmota_03/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:31:29.522 SRC: Backlog
00:31:29.524 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
00:31:29.529 MQT: stat/tasmota_03/RESULT = {"Module":{"18":"Generic"}}
00:31:29.773 SRC: Backlog
00:31:29.775 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
00:31:29.785 MQT: stat/tasmota_03/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"3584":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"3552":"Zigbee Tx"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
00:32:03.772 CMD: Backlog Rule1; Rule2; Rule3
00:32:03.775 SRC: WebConsole from 192.168.1.150
00:32:03.778 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
00:32:04.088 SRC: Backlog
00:32:04.090 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
00:32:04.096 MQT: stat/tasmota_03/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
00:32:04.304 SRC: Backlog
00:32:04.306 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
00:32:04.311 MQT: stat/tasmota_03/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
00:32:04.554 SRC: Backlog
00:32:04.557 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
00:32:04.563 MQT: stat/tasmota_03/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:32:29.288 CMD: Status 0
00:32:29.290 SRC: WebConsole from 192.168.1.150
00:32:29.293 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
00:32:29.300 MQT: stat/tasmota_03/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota3","FriendlyName":["Tasmota3"],"Topic":"tasmota_03","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:32:29.307 MQT: stat/tasmota_03/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.1.15/tasmota/release/tasmota3.bin.gz","RestartReason":"External System","Uptime":"0T00:28:48","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-02-12T22:43:30","SaveCount":29,"SaveAddress":"FD000"}}
00:32:29.315 MQT: stat/tasmota_03/STATUS2 = {"StatusFWR":{"Version":"11.0.0.1(tasmota)","BuildDateTime":"2022-02-12T23:42:41","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"407/699"}}
00:32:29.321 MQT: stat/tasmota_03/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["max.net","FASTWEB-F2L9SZ"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A0A000000000000","00000080","00006080","00004001"]}}
00:32:29.334 MQT: stat/tasmota_03/STATUS4 = {"StatusMEM":{"ProgramSize":750,"Free":2320,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDAC787","04768001","000000CF","010013C0","C000F981","00004084","00201000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37,38,45,50","Sensors":"1,2,3,4,5,6"}}
00:32:29.341 MQT: stat/tasmota_03/STATUS5 = {"StatusNET":{"Hostname":"tasmota-03-8166","IPAddress":"192.168.1.39","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.254","DNSServer2":"192.168.1.1","Mac":"EC:FA:BC:05:7F:E6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:32:29.346 MQT: stat/tasmota_03/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":3883,"MqttClientMask":"DVES_T03","MqttClient":"DVES_T03","MqttUser":"DVES_T03","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:32:29.353 MQT: stat/tasmota_03/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:32:29","Local":"1970-01-01T00:32:29","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":99,"Sunrise":"20:13","Sunset":"05:47"}}
[... 49 lines cut ...]
00:00:19.276 ZIG: {"ZbZNPReceived":"4585000000000001F2"}
00:00:20.233 ZIG: timeout, goto label 99
00:00:20.238 MQT: tele/tasmota_03/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
00:00:20.240 ZIG: Abort
00:00:20.241 ZIG: Stopping (99)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash latest firmware from jet-home.ru on cc2538 antenna.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Coordinator started.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Zigbee started with previous version of firmware: 
https://github.com/jethome-ru/zigbee-firmware/blob/master/ti/coordinator/cc2538_cc2592/JH_2538_2592_ZNP_UART_20201010.zip

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-02-13:
I looked at the code, Tasmota sends `Active EP Request` but the MCU returns an error. There's nothing we can do, the firmware does not act as expected.

ascillato2, 2022-02-14:
Closing as jet-home firmware is not supported. Sorry.

closed 2022-02-14 by ascillato2

s-hadinger, 2022-02-15:
I just revisited this, it seems that it should be a non-blocking error. It only indicates that this firmware automatically declares EP 0xF2 locally, which is not a problem.
I will change to allow this value.