arendst/Tasmota #14821: Option E1 appears to not work on SD01 with Tasmota 11

Tediore, 2022-02-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Option E1 on the ESP32 on Tasmota 11.0.0.1 does not appear to enable the PWM dimmer functionality on the SD01 dimmer. I transplanted an ESP32-C3F (4MB) onto an SD01 dimmer, set up the template with the pin layout of the SD01, assigned Option E1 to an unused GPIO, and applied the template with `module 0`. Confirmed the pin layout I used is correct (all peripherals including the dimmer LEDs are functional)

- Tasmota web UI shows a button for the relay and a button for the PWM
- Button1 (power) toggles the relay
- Button2 (dim down) toggles the PWM
- Button3 (dim up) toggles the relay
- `dimmer` commands do dim the light up/down if the relay is on, but `dimmer 0` does not turn the relay off, and `dimmer` with a value greater than 0 does not turn the relay on

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
06:02:48.686 RSL: RESULT = {"NAME":"ESP32 MJ Dimmer","GPIO":[8448,1,256,320,321,416,1,1,96,34,0,0,0,0,0,0,0,0,576,322,323,33],"FLAG":0,"BASE":1}
06:02:48.912 RSL: RESULT = {"Module":{"0":"ESP32 MJ Dimmer"}}
06:02:49.114 RSL: RESULT = {"GPIO0":{"8448":"Option E1"},"GPIO1":{"0":"None"},"GPIO2":{"256":"Relay_i1"},"GPIO3":{"320":"Led_i1"},"GPIO4":{"321":"Led_i2"},"GPIO5":{"416":"PWM1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"96":"Button_i1"},"GPIO9":{"34":"Button3"},"GPIO10":{"0":"None"},"GPIO18":{"576":"LedLink_i"},"GPIO19":{"322":"Led_i3"},"GPIO20":{"323":"Led_i4"},"GPIO21":{"33":"Button2"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
[No rules]
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
06:03:52.337 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_A56F0C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
06:03:52.344 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T00:10:57","StartupUTC":"2022-02-13T04:52:55","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2022-02-13T05:07:13","SaveCount":27}}
06:03:52.348 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0.1(tasmota)","BuildDateTime":"2022-02-12T23:08:37","Core":"2_0_2_2","SDK":"v4.4-3-g6afb23d90a","CpuFrequency":160,"Hardware":"ESP32-C3","CR":"417/699"}}
06:03:52.352 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tasmota",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
06:03:52.360 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1370,"Free":1856,"Heap":189,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":80,"FlashMode":3,"Features":["00000809","9FDAC7C7","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","000840D2","20200000","0030482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
06:03:52.365 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-A56F0C-3852","IPAddress":"10.0.4.69","Gateway":"10.0.4.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"84:F7:03:A5:6F:0C","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
06:03:52.371 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A56F0C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
06:03:52.377 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-02-13T05:03:52","Local":"2022-02-13T06:03:52","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:01","Sunset":"18:06"}}
06:03:52.385 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-02-13T06:03:52","ESP32":{"Temperature":41.1},"TempUnit":"C"}}
06:03:52.393 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-02-13T06:03:52","Uptime":"0T00:10:57","UptimeSec":657,"Heap":188,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":39},"POWER1":"OFF","POWER2":"OFF","Dimmer":0,"Fade":"OFF","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Tasmota","BSSId":"E8:48:B8:D6:09:24","Channel":6,"Mode":"11n","RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Button1 pressed:
06:07:57.987 RSL: RESULT = {"POWER1":"ON"}
06:07:57.988 RSL: POWER1 = ON

Button2 pressed:
06:08:00.343 RSL: RESULT = {"POWER2":"ON"}
06:08:00.344 RSL: POWER2 = ON

Button3 pressed:
06:08:02.258 RSL: RESULT = {"POWER1":"OFF"}
06:08:02.259 RSL: POWER1 = OFF

pwmdimmerpwms command:
06:09:05.834 CMD: pwmdimmerpwms
06:09:05.838 RSL: RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Assign the appropriate pin layout in the template, assign Option E1 to a free GPIO, enable with module 0, and press buttons on the dimmer and enter commands in the console

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected the dimmer to function like the PWM Dimmer module on the ESP8266

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
pwmdimmerpwms command returns unknown

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-13:
@arendst I'm not on top of this feaure but I feel the difference below is type and should be `mpin - AGPIO(GPIO_OPTION_E)`:
https://github.com/arendst/Tasmota/blob/5dabb623bf026ee60f4a652056ca7a04527bdf44/tasmota/support_tasmota.ino#L1763-L1768

arendst, 2022-02-13:
YOU'RE right!! Will fix now.

Tediore, 2022-02-13:
Wow, that was fast. Thank you!

closed 2022-02-13 by arendst

AloneHUANG, 2024-12-22:
I am using the latest tasmota v.13, but I have the exact same issue. has this bug been fixed?

AloneHUANG, 2024-12-23:
> Wow, that was fast. Thank you!
Have you tested the new firmware? I have the same issue with the latest firmware.

AloneHUANG, 2024-12-24:
I have two questions.
1. How do I know this OptionE1 is necessary if I don't see this post? Is there any instructions about this part, like OptionA and OptionE?
2. I set up all template as this post and only have one bug. When I use 1st button to turn off the lamp, it will turn on immediately. Is there anything I forget?
Thanks!