arendst/Tasmota #14826: ZbPermitJoin 1 Error

nickwild-999, 2022-02-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Although appearing to work I cannot use the ZbPermitJoin 1 command which produces the `command is unkown` I am initially flashed  **ncp-uart-nsw_6.7.9_115200.ota**  which reported no errors
I tried to reflash with **ncp-uart-sw_6.7.8_115200.ota**  and also reflash with 6.7.9 in case it was corrupted but I now get `Upload Failed Invalid file signature`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used Sonoff ZBBRidge
- [ ] Tasmota binary firmware version number used: 11.0.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: NODE MCU PYFLASHER
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```18:22:03.976 RSL: RESULT = {"NAME":"Sonoff ZHABridge","GPIO":[320,5088,0,5120,323,322,0,0,0,0,0,0,32,0],"FLAG":0,"BASE":18}
18:22:04.186 RSL: RESULT = {"Module":{"0":"Sonoff ZHABridge"}}
18:22:04.439 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"5088":"TCP Tx"},"GPIO2":{"0":"None"},"GPIO3":{"5120":"TCP Rx"},"GPIO4":{"323":"Led_i4"},"GPIO5":{"322":"Led_i3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
```
- [ ] Provide the output of this command: `Status 0`:
```l18:11:00.780 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_04D2CC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:11:00.784 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:08:17","StartupUTC":"2022-02-13T17:02:43","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2022-02-13T14:16:57","SaveCount":30,"SaveAddress":"1FB000"}}
18:11:00.787 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0(zbbridge)","BuildDateTime":"2022-02-12T14:13:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"370/699"}}
18:11:00.790 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["BT-8WAFN2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00190000000000","00000080","00006000","00004000"]}}
18:11:00.795 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":673,"Free":1116,"Heap":26,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"}}
18:11:00.799 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-04D2CC-4812","IPAddress":"192.168.1.139","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.254","DNSServer2":"0.0.0.0","Mac":"C8:2B:96:04:D2:CC","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:11:00.802 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_04D2CC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:11:00.807 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-02-13T17:11:00","Local":"2022-02-13T18:11:00","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:01","Sunset":"18:06"}}
18:11:00.811 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-02-13T18:11:00"}}
18:11:00.816 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-02-13T18:11:00","Uptime":"0T00:08:17","UptimeSec":497,"Vcc":3.499,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"BT-8WAFN2","BSSId":"8C:19:B5:73:78:F3","Channel":11,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```18:23:56.795 CMD: ZbPermitJoin 1
18:23:56.796 SRC: WebConsole from 192.168.1.130
18:23:56.797 CMD: Grp 0, Cmnd 'ZBPERMITJOIN', Idx 1, Len 1, Data '1'
18:23:56.800 RSL: RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Type in ZbPermitJoin 1 or download and try to reupload ota software from repo

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Entering pairing mode and successful relfashing of fw ota

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-02-13:
It looks like you are using ZHA. In this mode zigbee is managed by zha and Tasmota has no control.

nickwild-999, 2022-02-13:
@s-hadinger thanks for the quick reply. I am, now trying ZHA as I couldn't get any success with Z2M, but I would rather use Tasmota and Z2M. These logs are after I installed ZHA -  I didn't realise they would be any different sorry

s-hadinger, 2022-02-13:
Either way, ZHA or Zigbee2Mqtt, tasmota is just connecting serial to tcp and has no control on zigbee. 
To use Tasmota mode you need `module 75` (or if you want to flash a Zigbee firmware)

nickwild-999, 2022-02-13:
@s-hadinger Sorry I don't understand (noob on this as you can tell) ]
What do I need to do to allow ZbPermitJoin to work and do I need to try an earlier version of the ota fw?

sfromis, 2022-02-13:
If you want Tasmota to control the Zigbee coordinator, you'll have to be in Z2T mode (Zigbee2Tasmota), `Module 75`. This includes `ZbPermitJoin` and so on.

nickwild-999, 2022-02-13:
@sfromis @s-hadinger I understand now (I think) this is what comes from following 2 different tutorials.  🤪
I was having difficulty trying to understand how to pair devices under Z2M as nothing was working so I thought that it had to be permitted to allow connections thru ZbPermitJoin

sfromis, 2022-02-13:
Well, if you use Z2M instead of Z2T, you're asking in the wrong forum. When it comes to uploading the Zigbee firmware, you need to temporarily switch to Z2T mode, to give Tasmota control over the Zigbee chip.

nickwild-999, 2022-02-13:
@sfromis OK Thanks

closed 2022-02-13 by nickwild-999