arendst/Tasmota #14829: Smart plugs with a BL0942 power metering chip (and probably also BL0939 and BL0940) only processes messages from the BL0942 chip alternately.

Dam5, 2022-02-13:
### PROBLEM DESCRIPTION

In my Elivco smart plugs I replaced the Tuya CB2S module with a drop-in ESP-2S module, and flashed them with Tasmota 10.1.0 and calibrated them.

I observed 2 problems:
1) regularly checksum errors
2) ca 0.8 kWh daily energy consumption, relay ON, while no load is attached

(Third problem is that calibrating the current for a BL0942 is in Amps, and the calibrating should be in mAmps, or visa versa, but that's a minor problem)

While investigation this strange energy consumption behaviour of my smartplugs with BL0942 chip, I discovered that half of the messages received from the BL0942 chip are disgarded.

It boils down to reading 24 bytes from the BL0942 chip before processing, while the packet is only 23 bytes long.
This 24th byte is the packet-header of the next (23 byte) packet, but is not saved and never handled, so the complete next packet is disgarded waiting for the subsequent BL09XX_PACKET_HEADER byte.

Unless there's a BL09XX_PACKET_HEADER byte (0x55) in the actual data of the (otherwise disgarded) packet, in which case the receive-buffer is filled with 24 bytes from that point on.
In that case, most of the time, this results in a checksum error. 
But sometimes (once every few  minutes) the (simple) checksum turns out to be correct, and the (shifted) bytes lead to processing of incorrect data.
For example resulting in huge (1 second) power spikes which adds up to the daily total energy.
This is the source of the power consumption I was observing.
These power spikes are so short that they are not displayed (as Watts) in the main webpage of the device (I've never observed one), and are not logged in the sensor-data via MQTT.
So the power is always 0 Watt, and the energy is slowly adding up.
(Btw, 0.8 kWh daily corresponds to ca 30 Watt continuously)

In Tasmota every second a poll request is sent out to the BL0942, and there's no synchronisation between sending and receiving data (which in this case is good, the communication is not blocked waiting for eachother)

Enabling debugging (DEBUG_BL09XX) reveals that indeed the data is processed only once per two seconds, while the poll is send out every second.
Another symptom: In a device like this, with ultra short communication lines, there shouldn't be checksum errors.
Third symptom: The time between the (2nd) poll and the processing of the received message (10 ms) is way too fast for 23 bytes at 4800 baud (should be ca 48 ms)

The bug seems a classic index/byte-count error.

I also expect memory corruption (store 24 bytes in a 23 byte malloced buffer), but I've not looked into that.

IMO the cullpit is in line 221 of xnrg_14_bl09xx.ino

```lua

211  void Bl09XXSerialInput(void) {
212    while (Bl09XXSerial->available()) {
213      yield();
214      uint8_t serial_in_byte = Bl09XXSerial->read();
215      if (!Bl09XX.received && (BL09XX_PACKET_HEADER == serial_in_byte)) {
216        Bl09XX.received = true;
217        Bl09XX.byte_counter = 0;
218      }
219      if (Bl09XX.received) {
220        Bl09XX.rx_buffer[Bl09XX.byte_counter++] = serial_in_byte;    //  Possible memory corruption
221        if (Bl09XX.buffer_size == Bl09XX.byte_counter -1) {      //   byte_counter is 24, while buffer_size is 23
```

replacing line 221 with

```lua
221      if (Bl09XX.buffer_size == Bl09XX.byte_counter) {
```

solves the alternating problem.(and memory corruption, and checksum errors, and energy creep) 

I also checked the datasheets of the BL0939 and the BL0940 chips and I expect these have the same problem (packet is 35 bytes, while 36 bytes are read before processing).
(https://www.belling.com.cn/media/file_object/bel_product/BL0939/datasheet/BL0939_V1.2_cn.pdf, https://www.belling.com.cn/media/file_object/bel_product/BL0940/datasheet/BL0940_V1.1_en.pdf)
The chinese one was the only I could find.

Am I the only one observing this behaviour, or am I barking at the wrong tree (and making a fool of myself)?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
[... 134 lines cut ...]
BTW a number of AddLog()'s are in print-float format "%f" which are not by default resolved when building.

In my debugging I have replaced these with dtostrf() calls, but in the above output the stock "%f"'s are not resolved, resulting is empty strings for Voltage en Temp, or skipping these Voltage-strings alltogether.

### TO REPRODUCE

- Relay On
- No load plugged in
- Wait a couple of hours 
- Watch the energy consumption rising slowly

It greatly depends if 0x55 is part of the actual data. In my case it is.

### EXPECTED BEHAVIOUR

- No energy creep when no load is attached.
- No checksum errors

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-13:
@Dam5
I do agree and see 2 problems
- The comparison you pointed
- The loop for calculating the checksum should be from 0 to buffer_size-2 because rx_buffer[buffer_size-1] is the incoming checksum

Do you want to make the change and test, then PR ?
I could make the change and PR but I have not device to test.

Dam5, 2022-02-13:
@barbudor 
The only problem is the comparison in line 221.
Then there are 

The checksum calculation is (and was) correct.
First the complete message is received and stored, including checksum in rx_buffer[buffer_size-1].
Your proposal for changing the for-loop is IMHO incorrect.
Calculate checksum for index 0 to 21, compare with index 22 (= checksum-byte).
In your solution the for-loop is one byte short.
If tested your solution, and it accidentally works, but only because the last data-byte is (always) 0x00, so it has no impact in the checksum calculation.

I've run my solution for the last 2 days and seemed OK.
But I might be wrong.

barbudor, 2022-02-13:
Yes, I change my original message
The loop is correct as it is

Will you PR ?

arendst, 2022-02-13:
Let me do it. I need to do some checks first.

Dam5, 2022-02-13:
Dispite almost 40 years of (proffessional) software development I've got only basic git skills. Never needed it.
It's on my TODO list, but not now.

In short; could your PR?

PVerburg, 2022-02-14:
Hi, I have 2 device here with similar problems no code expert but can assist with testing ?

Dam5, 2022-02-14:
Probably superfluous, but just to be clear.
These incorrect data-packets (as a result of wrong counting bytes) are always possible, also when load is attached.
But then they're not so obvious.
I've smart plugs with constant load where the energy-totals didn't match with the power.
(Which triggered the investigation)

arendst, 2022-02-14:
Started working on it and first impressions are you're right regarding the buffer size checks.

I was also surprised to find ``%f`` as we never supported this due to code size increase of several ks. Searching the code for ``%f`` revealed more inconsistencies in the BLE drivers. I will need to change this as we support ``%*_f``.

Will continue now first with the BL09xx driver.

arendst, 2022-02-14:
Thx for the thorough investigation.

The suspected buffer overflow did not occur as the byte_counter is incremented AFTER it is used as index into the buffer.

Anyway, I noticed my BL0940 device also skipped a beat and I was also wondering how come the daily result was offtrack too.

Again. Keep em coming this way as it always reveals nice black spots.

Dam5, 2022-02-14:
I have to disagree.
I know boundery checks and indexing are tricky so I should be carefull.

Indeed byte_counter is incremented after used as an index, bute byte_couter is at that point already out-of-bounds.

buffe_size = 23; rx_buffer[0..22] ; But byte_counter = 23, incrementing to 24.
rx_buffer[23] is out-of-bounds.

Love your project.

closed 2022-02-15 by arendst

ferbulous, 2022-03-21:
hi @Dam5 
do you mind sharing your templates for your power plug with BL0942?

Dam5, 2022-03-21:
@ferbulous 
no problem, it's also in the OP, but there it is somewhat obfuscated.
Template: {"NAME":"Power Monitoring Plug (BL0942)","GPIO":[0,3200,0,7520,0,32,0,0,0,544,224,0,0,0],"FLAG":0,"BASE":18}

ferbulous, 2022-03-21:
@Dam5  Thanks! Just another question, how can I determine the corrent gpio pin is for HLWBL CF1/HLWBL SELi? I tried googling around but I can't find clear answer on this

Dam5, 2022-03-21:
@ferbulous 
I'm not sure what you mean.

The BL0942-based power plug uses a serial (4800 baud) protocol (to communicate whith the BL0942 chip).
Tasmota template config: Transmit is default TX (there is no BL0942-TX), Receive is BL0942-RX.
HLWBL CF1and HLWBL SELi, which are pulse-based protocols, are not used.

HLWBL CF1and HLWBL SELi  (among others) are used in e.g. BL0937-based power plugs.

Are you sure you're dealing whith a BL0942-based power plug?

Beware that there are a lot of different, almost identically looking, smart plugs for sale.
(Most of them don't use ESP8266 anymore, but use TUYA-based modules, which are not Tasmota compatible)
I had to reverse engineer mine, to track down the pin/gpio numbers (and replace the module with a ESP-02S or simular)
Yours could use different GPIO-numbers.

See attached images for a  BL0937-based power plug (OctoPlug) and a BL0942-based power plug (Smartplug05) that I keep for reference.

[image]
[image]

ferbulous, 2022-03-21:
@Dam5 
Thanks for clearing that up, 
I'm just starting to delve into power monitoring plugs so I'm not too familiar with all the technical details yet. Currently I'm looking at other existing power monitoring templates, so I assumed HLWBL CF1 and HLWBL SELi are required.
The one I recently bought is this which I plan to swap the CB2S tuya module with ESP-02

[image]
[image]

Dam5, 2022-03-21:
@ferbulous 
Alltough your PCB is very different, the pin-layout is probably simular to mine.
The BL0942 traces (upper-left 2 of the BL0942 chip) are going to the same 2 pins on the Tuya-module as mine.
The Tuya-module looks very simular too, so when you use a ESP-02S drop-in replacement, the 3.3V, GND, TX and BL0942-RX are most likely the same.

For Relay (probably the same, P26 on the Tuya), Pushbutton (trace is not visible)  and blue LED (trace not visible) you have to trace the PCB.
(The red LED is hard-wired, and is not addressable/controlable)

PVerburg, 2022-03-22:
That looks like the same board I have modified works well now they fixed a small bug in the driver. :-)

ferbulous, 2022-03-22:
@Dam5 Hi, just swapped with ESP-02S and your template matches the exact pin with my power plug. I'll proceed with power calibration as per the tasmota guideline page which I've never done before so hopefully I'll get it right
As for the red led hardwired, it because it's not tied to any gpio for that specific led red color?
With power monitoring feature, could we add some rules say if low voltage within 15-20 minutes, then turn off the plug?

@PVerburg do you by any chance got it from aliexpress?

Dam5, 2022-03-22:
@ferbulous 
The red LED is hard-wired to (the state of) the relay.
I 've never used (Tasmota) rules, so I can't help you with that.
Beware that the BL0942 bug-fix, discribed in this issue, is scheduled for Tasmota 11.1, which is not released yet.

PVerburg, 2022-03-23:
@Dam5 yep did get it from ali.
And yes you could make a rule you must be more precise as your needs and maybe use discord so more people can benefit and assist?

myfingerhurt, 2022-09-15:
I have a gosund wf6-2,  it is a BL0937-based power plugs, and the pin configuration is same as OctoPlug that @Dam5 has posted.
I have successfully downloaded tasmota 12.1.1 into gosund wf6-2.
Thanks to @Dam5