arendst/Tasmota #14832: Sendmail causes soft WDT if used from script.

PHPJJ, 2022-02-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini Pro
- [x ] Tasmota binary firmware version number used: 11.0.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: platform IO comm port
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
state
15:23:59.008 CMD: state
15:23:59.014 RSL: RESULT = {"Time":"2022-02-13T15:23:59","Uptime":"0T00:01:32","UptimeSec":92,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"jongen","BSSId":"F4:92:BF:A0:C5:20","Channel":6,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:17"}}
**sendmail [smtp.gmail.com:465:nnnnnnnnnn:password:<tasmota2@gmail.com>:<mygmail@gmail.com>:Water gedetecteerd]Water gedetecteerd in stookkelder
15:24:13.866 CMD: sendmail [smtp.gmail.com:465:nnnnnnnnn:password:<tasmota2@gmail.com>:<mygmail.peter@gmail.com>:Water gedetecteerd]Water gedetecteerd in stookkelder
15:24:17.128 RSL: RESULT = {"Sendmail":"Done"}**
15:24:27.168 Script: performs "power 1 "
15:24:27.173 RSL: RESULT = {"POWER":"ON"}
15:24:27.175 RSL: POWER = ON
**15:24:27.178  report send Water gedetecteerd
15:24:27.180 Script: performs "sendmail [smtp.gmail.com:465:nnnnnnnnnn:password:<tasmota2@gmail.com>:<mygmail@gmail.com>:Stookkelder]Water gedetecteerd in stookkelder"**

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffff5e0 end: c314c060 offset: 01a0
3ffff780:  0000006b 00000000 00000001 4025a4be  
3ffff790:  00000000 00000000 3fff7f4c 4024b987  
3ffff7a0:  00000008 000b1d85 3fff6ba4 4024d9b2  
3ffff7b0:  0000006b 00000005 79b6f5bf 00000000  
3ffff7c0:  402708e0 4027088c 3fff7f4c 4027046a  
3ffff7d0:  3fff844c 3fff6634 3fff7f4c 402a51b4  
3ffff7e0:  3fff6ba4 00000001 3fff6ba4 4024da64  
3ffff7f0:  3fff6ba4 3fffa3cc 00000000 4024dc19  
3ffff800:  3fff6634 00000000 00000014 3ffffb35  
3ffff810:  000001d1 3fff6634 3fff6ba4 3ffffb35  
3ffff820:  000001d1 3fff6634 3fff6ba4 4024dcdd  
3ffff830:  4025f310 6d7fb16c 4025f310 6d7fb16c  
3ffff840:  00000001 3fff6ba4 3fff65f4 4023714f  
3ffff850:  0000000b 00000400 000000a0 3ffffb00  
3ffff860:  00000001 0000000b 3ffff8b0 4025922c  
3ffff870:  3ffff8c8 3ffff8bc 3ffff8b0 40259268  
[... 1170 lines cut ...]
#define USE_SCRIPT
#endif

#ifndef USE_SCRIPT_WEB_DISPLAY
#define USE_SCRIPT_WEB_DISPLAY
#endif

#ifndef USE_GOOGLE_CHARTS
#define USE_GOOGLE_CHARTS
#endif

#ifndef USE_SENDMAIL
#define USE_SENDMAIL
#endif

#ifndef USE_UFILESYS 
#define USE_UFILESYS
#endif 

#ifdef USE_DOMOTICZ
#undef USE_DOMOTICZ                              
#endif

Setting Debug in mail ino does not give differen output nor result.
**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-02-13:
You have syntax errors in the sendmail command, lose those `<>`

barbudor, 2022-02-13:
@sfromis those `< >` are required on ESP8266
https://github.com/arendst/Tasmota/issues/14404#issuecomment-1024884991

PHPJJ, 2022-02-13:
I used same command in project with version 8.5.1 also in a script. Even if syntaxis is incorrect it schould not generatie wdt/reboot. I copy/paste the command line from the script to console and it sends the mail. See the first few lines of the console log dump.

gemu2015, 2022-02-14:
could not reproduce the issue. however ESP8266 sendmail in scripts is very critical due to possible memory issues. check your free ram memory. may also be a stack problem.
email driver itself has not changed between 8.5 and 11.0

PHPJJ, 2022-02-14:
I have been investigating problem a little further and found a way to recreate. I found a way to recreate the problen that might help investigation.

This is what I did:

1 – Erased flach in Wemos D1 Mini Pro 
2 - Took a fresh copy of tasmota 11.0.0.1 from the zip I downloaded.
3 - Verified compile with no change
4 – Recopiled with following in user_config_override.h

#ifndef USE_SENDMAIL
#define USE_SENDMAIL
#endif

#ifdef USE_RULES
#undef USE_RULES
#define USE_SCRIPT
#endif

5 - I tested the sendmail line from console. Works OK.

6 - Use the following script, simplified from script I want to use.

Basically the script looks for a contact on pin 13 to go low. 
As long as the pin is high an LED at pin 14 is on for 1 second every 10 seconds.
Variable altm is used to prevent sending multiple mails.
--------------------------------------------------------------
>D
cnt=0
th=0
tm=0
t:altm=0
punct=":"
tmp=0
alrm=1
alrmdur=0
tt=0
hm=0

; mail bdy
>m
Email raport van: %tstamp%\n
#

>W
Tijd %0th%%punct%%0tm%

>B
spinm(13 2)
spinm(14 1)
spin(14 0)
=>print Restarting

>T
>F
>S
th=int(time/60)
tm=time-(th*60)
if tm<10 {
punct=":0"
}
else{punct=":"
}

alrm=pin[13]
if ((upsecs%10==0)or(alrm==0))
{
spin(14 1)
}
else
{
spin(14 0)
}

if ((alrm==0)and(altm==0)){
=>sendmail [smtp.gmail.com:465:nnnnnnnnn:nnnnnnnnn:<tasmota2@gmail.com>:<my.gmail@gmail.com>:stookkelder]water detected.
altm=100
alrmdur=alrmdur+1
}
=>print %upsecs% Alarm:%0alrm% Atimer:%0altm% Duration:%0alrmdur%

--------------------------------------------------------------------

The script above, e-mail is send. So that **does not recreate the problem**. However I want to use the sendmail from a subroutine.

7 - Changed the script to:
.
.
.
if ((alrm==0)and(altm==0)){
=#sendreport(stookkelder)
altm=100
alrmdur=alrmdur+1
}
=>print %upsecs% Alarm:%0alrm% Atimer:%0altm% Duration:%0alrmdur%

#sendreport(subj)
=>sendmail [smtp.gmail.com:465:nnnnnnnnn:nnnnnnnnn:<tasmota2@gmail.com>:<my.gmail@gmail.com>:stookkelder]water detected.

---------------------------------------------------------------
Now sendmail fails in 
3:57:43.063 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["401076f4","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40250872","40250872","40245567","40266254","40266200","40265dde","40299b14","40245680","40245835","402458f9","40254d50","40254d50","402310df","4024f5dc","4024f618","40231507","401018ef","402a21ab","4024944a","40257400","40231582","40293548","40293548","4021ed88","4029352c","402934fc","401011e0","40227983","40295e9a","40295d34","40217656"]},"BootCount":18}}

Note: I do not (yet) use the (subj) in this test, nor do I use the .m mail body option. If I do the error persists, so keep it the simples way to recreate.

This means, moving the sendmail command to the subroutine causes the problem to show. 

Before reporting the original problem I’m sure I tested the direct sendmail as in the first script. This lets me believe that there is some memory of timing critical flaw that gets wors if more options are added such as support for SHT3X etc.
Hope this helps to recreate the problem.

gemu2015, 2022-02-14:
as i said memory is very limited and mail uses TLS which also consumes memory.
if you put sendmail in a subroutine you use an additional portion of stack which could lead to a stack overflow. it may well be that stack usage has changed between 8.5 and 11.0
to reliably send emails better use an ESP32

arendst, 2022-02-16:
As explained in the above answer it's all about memory.

closed 2022-02-16 by arendst

PHPJJ, 2022-02-16:
I have been experimenting and ofcause removed the sendmail from subroutine. Standard tasmota with only script and sendmail added will execute sendmail. However if I add i2c and sht3x etc, even outside subroutine sendmail is unreliable (sometimes error "wrong" parameters.)
I will take an other route and start from minimal and add just what I need.
Thanks for your efforts.