arendst/Tasmota #14845: Tasmota RULES potentially unreliable

rubenkr, 2022-02-14:
### PROBLEM DESCRIPTION
Hi everyone,

I can hardly express how grateful I am over Tasmota and its project. All possible devices at my home are running this awesome firmware.

In my hobby I am growing mushrooms which requires reliable intervals of fresh air exchange and humidification.

Devices: SONOFF S20
Firmware: Tasmota 10.1.0 by Theo Arends

However this early morning I got a notification that the Humidifier Switch went ON over 15Minutes and ignored my Pulsetime rules to turn it off - leaving me with excess water on my basement floor. Why is that happening?

### REQUESTED INFORMATION
REQUESTED INFORMATION
I wanted to run the intervals on Tasmota as it should work autonomously even if Node-Red or Home-Assistant are off.
For this I am using the following rules

FAE Switch: Every 38Minutes turn ON for 5Min

Humidity Switch: Every 18Minutes turn ON for 3Min

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
09:47:21.375 CMD: Backlog Rule1; Rule2; Rule3
09:47:21.458 MQT: stat/tasmota_79C223/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":76,"Free":435,"Rules":"ON Power1#state=0 DO ruletimer1 900 ENDON ON rules#timer=1 DO Power1 1 ENDON"}}
09:47:21.680 MQT: stat/tasmota_79C223/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:47:21.930 MQT: stat/tasmota_79C223/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:47:46.486 CMD: Status 0
09:47:46.493 MQT: stat/tasmota_79C223/STATUS = {"Status":{"Module":1,"DeviceName":"SONOFF-S20-B-HUMIDIFIER","FriendlyName":["SONOFF-S20-B-HUMIDIFIER"],"Topic":"tasmota_79C223","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:47:46.507 MQT: stat/tasmota_79C223/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"3T18:00:42","StartupUTC":"2022-02-10T14:47:04","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2021-07-27T17:34:32","SaveCount":6977,"SaveAddress":"F8000"}}
09:47:46.518 MQT: stat/tasmota_79C223/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"379/699"}}
09:47:46.528 MQT: stat/tasmota_79C223/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vespa-O",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
09:47:46.547 MQT: stat/tasmota_79C223/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:47:46.564 MQT: stat/tasmota_79C223/STATUS5 = {"StatusNET":{"Hostname":"tasmota-79C223-0547","IPAddress":"10.0.0.223","Gateway":"1xx.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"1xx.0.0.1","DNSServer2":"0.0.0.0","Mac":"5C:CF:7F:79:C2:23","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:47:46.578 MQT: stat/tasmota_79C223/STATUS6 = {"StatusMQT":{"MqttHost":"1xx.0.0.52","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_79C223","MqttUser":"DVES_USER","MqttCount":12,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:47:46.594 MQT: stat/tasmota_79C223/STATUS7 = {"StatusTIM":{"UTC":"2022-02-14T08:47:46","Local":"2022-02-14T09:47:46","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:00","Sunset":"18:08"}}
09:47:46.608 MQT: stat/tasmota_79C223/STATUS10 = {"StatusSNS":{"Time":"2022-02-14T09:47:46"}}
09:47:46.622 MQT: stat/tasmota_79C223/STATUS11 = {"StatusSTS":{"Time":"2022-02-14T09:47:46","Uptime":"3T18:00:42","UptimeSec":324042,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":12,"POWER":"ON","Wifi":{"AP":1,"SSId":"VespXXXX","BSSId":"26:5A:4C:xxxxx","Channel":6,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":4,"Downtime":"0T00:00:24"}}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Intervals: ![image](https://user-images.githubusercontent.com/1043739/153829662-26426143-e8de-45d2-ad92-5f95feb73254.png)

Wrong Intervals: ![image](https://user-images.githubusercontent.com/1043739/153829102-26564b31-8f78-4571-af75-b77e63e72f97.png)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-02-14:
Your wifi signal is weak. Maybe running in connection problems which results in unexpected behaviour.

arendst, 2022-02-14:
That would be the case in a perfect world. 

Alas, at least in versions before 10.1.0.4 the TCP stack could result in a buffer overflow for the MQTT driver when poor wifi is observed resulting in data corruption and most likely an exception restart. This restart will probably causing your rule hick-ups.

For the short term I suggest you upgrade to the latest release version and monitor if you observe your issue again.

rubenkr, 2022-02-14:
Thank you @arendst @Jason2866 so much ! I will do so.

closed 2022-02-14 by rubenkr