arendst/Tasmota #14857: ESP32 BLE versions past 9.4 do not publish autodiscover MQTT sensor messages for Home Assistant

znanev, 2022-02-15:
### PROBLEM DESCRIPTION
Versions of tasmota32-bluetooth, past release 9.4 (and possibly 9.5) do not publish autodiscover MQTT sensor messages for Home Assistant.
I'm using tasmota32-bluetooth pre-compiled release 9.4 on a ESP32-Devkit as a BLE to MQTT bridge. I have a number of Xiaomi BLE sensors (mostly LYWSD03MMC and MHO_C401, flashed with custom firmware - ATC). This setup was working for me for more than a year. Several months ago I decided to upgrade the Tasmota version and as result I "lost" all BLE sensors in Home Assistant (HA). I noticed then, that option `SetOption19 1` did nothing, but thought that it was a bug and reflashed the working version (9.4 at the time). All the BLE sensors appeared again in HA so I was again a happy camper. Yesterday I decided to give a go of the new release - 11.0 and problems started again. All BLE sensors disappeared again from HA and `SetOption19 1` still appears to be doing nothing... So just to be able to test simultaneously, I left the "working" version 9.4 on my "main" ESP32 devkit and flashed the most recent one - 11.0 onto a second ESP32 devkit. The behaviour of the new version is persistent - it doens't send autoconfiguration MQTT message for every BLE sensor found, no matter the state of `SetOption19` option.
What I noticed on the "working" version 9.4 is, that on each `mi32period`, autoconfiguration messages are published to `homeassistant/sensor/<sensorId>/<sensorEntity>/config topics:

```
15:06:34.818 MQT: homeassistant/sensor/ATC5056c0/Temperature/config = {"availability":[],"device":{"identifiers":["BLEATC5056c0"],"name":"ATC5056c0","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"temperature","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC5056c0","name":"ATC5056c0_Temperature","state_topic":"tele/tasmota_ble/ATC5056c0","uniq_id":"ATC5056c0_Temperature","unit_of_meas":"°C","val_tpl":"{{ value_json.Temperature }}"}
15:06:34.858 MQT: homeassistant/sensor/ATC5056c0/Humidity/config = {"availability":[],"device":{"identifiers":["BLEATC5056c0"],"name":"ATC5056c0","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"humidity","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC5056c0","name":"ATC5056c0_Humidity","state_topic":"tele/tasmota_ble/ATC5056c0","uniq_id":"ATC5056c0_Humidity","unit_of_meas":"%","val_tpl":"{{ value_json.Humidity }}"}
15:06:34.900 MQT: homeassistant/sensor/ATC5056c0/DewPoint/config = {"availability":[],"device":{"identifiers":["BLEATC5056c0"],"name":"ATC5056c0","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"temperature","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC5056c0","name":"ATC5056c0_DewPoint","state_topic":"tele/tasmota_ble/ATC5056c0","uniq_id":"ATC5056c0_DewPoint","unit_of_meas":"°C","val_tpl":"{{ value_json.DewPoint }}"}
15:06:34.940 MQT: homeassistant/sensor/ATC5056c0/Battery/config = {"availability":[],"device":{"identifiers":["BLEATC5056c0"],"name":"ATC5056c0","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"battery","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC5056c0","name":"ATC5056c0_Battery","state_topic":"tele/tasmota_ble/ATC5056c0","uniq_id":"ATC5056c0_Battery","unit_of_meas":"%","val_tpl":"{{ value_json.Battery }}"}
15:06:34.976 MQT: homeassistant/sensor/ATC5056c0/RSSI/config = {"availability":[],"device":{"identifiers":["BLEATC5056c0"],"name":"ATC5056c0","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"signal_strength","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC5056c0","name":"ATC5056c0_RSSI","state_topic":"tele/tasmota_ble/ATC5056c0","uniq_id":"ATC5056c0_RSSI","unit_of_meas":"dB","val_tpl":"{{ value_json.RSSI }}"}
```

On newer versions (10.1 and 11.0), these messages are no longer sent and as a result Home Assistant is "losing" all previously discovered "tas" devices. 

I'd be grateful if someone can look into this, I'm more than happy to provide more information. Thanks.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-Devkit
- [x] Tasmota binary firmware version number used: 11.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64.exe
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:27:03.364 CMD: Backlog Template; Module; GPIO 255
15:27:03.426 MQT: stat/tasmota_D4AD7C/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:27:03.662 MQT: stat/tasmota_D4AD7C/RESULT = {"Module":{"1":"ESP32-DevKit"}}
15:27:03.925 MQT: stat/tasmota_D4AD7C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
15:29:56.376 CMD: Backlog Rule1; Rule2; Rule3
15:29:56.430 MQT: stat/tasmota_D4AD7C/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":46,"Free":465,"Rules":"on System#Boot do backlog mi32option2 1; mi32option6 1; endon"}}
15:29:56.669 MQT: stat/tasmota_D4AD7C/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:29:56.867 MQT: stat/tasmota_D4AD7C/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```

- [x] Provide the output of this command: `Status 0`:
```lua
15:31:59.808 CMD: status 0
15:31:59.821 MQT: stat/tasmota_D4AD7C/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D4AD7C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:31:59.872 MQT: stat/tasmota_D4AD7C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:37:31","StartupUTC":"2022-02-15T13:54:28","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2022-02-14T15:11:34","SaveCount":10}}
15:31:59.920 MQT: stat/tasmota_D4AD7C/STATUS2 = {"StatusFWR":{"Version":"11.0.0(bluetooth)","BuildDateTime":"2022-02-12T14:14:01","Core":"2_0_2_2","SDK":"v4.4-3-g6afb23d90a","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"431/699"}}
15:31:59.944 MQT: stat/tasmota_D4AD7C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<MySSID>",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004002"]}}
15:31:59.991 MQT: stat/tasmota_D4AD7C/STATUS4 = {"StatusMEM":{"ProgramSize":1492,"Free":1856,"Heap":91,"StackLowMark":3,"PsrMax":4093,"PsrFree":4083,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","8F9AC7C7","00148001","000000CF","010 [... cut ...]
15:06:43.002 MQT: homeassistant/sensor/ATC3f5836/RSSI/config = {"availability":[],"device":{"identifiers":["BLEATC3f5836"],"name":"ATC3f5836","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"signal_strength","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC3f5836","name":"ATC3f5836_RSSI","state_topic":"tele/tasmota_ble/ATC3f5836","uniq_id":"ATC3f5836_RSSI","unit_of_meas":"dB","val_tpl":"{{ value_json.RSSI }}"}
```

### SCREENSHOTS
Not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-15:
There have been quite some changes in BLE handling since 9.4
Are you aware of MI32Option6 ? Is that what you are using ?
Please provide more details on how you configured the your tasmota32-bluetooth

The default behavior is to send all sensor data on a SENSOR message as per Tasmota standard
Unless you activate Mi32Option6 where it would send 1 message per sensor on dedicated topic `tele/tasmota_ble/XXXXX` where `tasmota_ble` is an hardcoded topic, and XXXX is either the MAC address of the device or its alias if BLEAlias have been used.

Please confirm how you are using BLE and exact configuration
Note that Mi32Option and BLEAlias are not peristent by default and you should use an `autoexec.bat` file to reconfigure at each restart.

znanev, 2022-02-15:
Hi @barbudor 

Indeed, I'm using `MI32Option6 1` on startup, that way every sensor reading is reported into its separate topic under tasmota_ble, which works fine in all versions (9.4, 10.1, 11). I also have option `MI32Option2 1` set, as it reports sensor readings immediately, without aggregating them.

What misses in newer versions (v10.1, v11.0) are the messages written to `homeassistant/sensor/<sensorId>/Temperature/config`, .e.g:

```
15:06:40.831 MQT: homeassistant/sensor/ATC50c365/Temperature/config = {"availability":[],"device":{"identifiers":["BLEATC50c365"],"name":"ATC50c365","manufacturer":"tas","model":"ATC","via_device":"tasmota_265B84-7044"},"dev_cla":"temperature","expire_after":600,"json_attr_t":"tele/tasmota_ble/ATC50c365","name":"ATC50c365_Temperature","state_topic":"tele/tasmota_ble/ATC50c365","uniq_id":"ATC50c365_Temperature","unit_of_meas":"°C","val_tpl":"{{ value_json.Temperature }}"}
````

The effect is, that when I upgrade the ESP32 to newer versions, I loose all BLE sensors in Home Assistant. Dongrading to v9.4 brings them automatically back.

So yes, I confirm that MI32Options that I use are configured as noted (`MI32Option6 1` and `MI32Option2 1` ) on startup, I'm aware that setting them is not persistent.

Thanks for looking into this issue.

znanev, 2022-02-18:
OK,

So to help myself a bit, I looked at the code and found that the tasmota32-bluetooth.bin variant is built with the following defines:

```cpp
#define USE_TASMOTA_DISCOVERY
#undef USE_HOME_ASSISTANT
```

This explains the bit why `SetOption19 1` has no effect in recent releases. So I took the opportunity to build the firmware myself with the defines above flipped:

```cpp
#undef USE_TASMOTA_DISCOVERY
#define USE_HOME_ASSISTANT
```

and wow and behold! Setting `SetOption19 1` now correctly publishes BLE sensors autoconfiguration data to topics like
 
`homeassistant/sensor/<sensorId>/Temperature/config`
 
As far as I remember, HomeAssistant autoconfiguration and Tasmota autoconfiguration previously co-existed in the same firmware and could be switched during runtime with `SetOption19 1` / `SetOption19 0`. So for current tasmota32-bluetooth firmware at least it seems that the pre-built versions opt for Tasmota autodiscovery only and HomeAssistant autodiscovery is not available, until one compiles the firmware himself.

Which leads me to another question - if HomeAssistant autodiscovery is excluded by default, shouldn't Tasmota autodiscovery take its place and publish all BLE sensors autoconfigurations, so those can be picked up by Home Assistant? Because as it stands now, if one uses a pre-compiled tasmota32-bluetooth firmware, there is no way to have the BLE sensors autodisovered by Home Assistant.

closed 2022-03-20 by github-actions[bot]