arendst/Tasmota #14864: SPM Main Issue with 20 relays

fegauthier, 2022-02-15:
### PROBLEM DESCRIPTION
Everything was working fine with 16 relays. I just plugged another relay today to get 20 relays and there is some issues. Relay seems to be detected by Tasmota but toggling ON/OFF not working and line status doesn't seems to work too.

I'm supposed to have 9 Line ON but it only shows 8.

[image]

If you look at the video below, you will find a weird behaviour with relays.

https://drive.google.com/file/d/1FlqipDFfyegvrP9xRA4SkciTuTsTdQzl/view?usp=sharing

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): SPM Main and SPM Relays
- [ ] Tasmota binary firmware version number used: Latest
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: Tasmota Web GUI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

20:22:17.864 CMD: Backlog Template; Module; GPIO 255
20:22:17.867 SRC: WebConsole from 192.168.1.31
20:22:17.871 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
20:22:17.936 SRC: Backlog
20:22:17.939 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
20:22:17.949 RSL: RESULT = {"NAME":"Sonoff SPM (POC2)","GPIO":[1,0,1,0,3200,5536,0,0,1,1,1,0,3232,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,544,1,1,32,1,0,0,1],"FLAG":0,"BASE":1}
20:22:18.159 SRC: Backlog
20:22:18.163 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
20:22:18.167 RSL: RESULT = {"Module":{"0":"Sonoff SPM (POC2)"}}
20:22:18.384 SRC: Backlog
20:22:18.387 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
20:22:18.404 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"3200":"Serial Tx"},"GPIO5":{"5536":"ETH POWER"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"3232":"Serial Rx"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"544":"LedLink"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"32":"Button1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

20:22:45.630 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
20:22:45.637 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","","","","","","",""],"Topic":"tasmota_C781CC","ButtonTopic":"0","Power":3900,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:22:45.648 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota32-ethernet.bin","RestartReason":"Software reset CPU","Uptime":"0T00:54:32","StartupUTC":"2022-02-15T19:28:13","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-12-18T21:29:03","SaveCount":12345}}
20:22:45.658 RSL: STATUS2 = {"StatusFWR":{"Version":"2022.1.4(tasmota)","BuildDateTime":"2022-02-04T07:19:41","Core":"2_0_2_2","SDK":"v4.4-3-g6afb23d90a","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"411/699"}}
20:22:45.663 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garage",""],"TelePeriod":300,"Resolution":"548B00C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
20:22:45.685 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1341,"Free":1856,"Heap":135,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7C7","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","0034482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
20:22:45.693 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-C781CC-0460","IPAddress":"192.168.10.5","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"1.1.1.1","DNSServer2":"8.8.8.8","Mac":"94:3C:C6:C7:81:CC","Ethernet":{"Hostname":"tasmota-C781CC-0460-eth","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"94:3C:C6:C7:81:CF"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:22:45.699 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C781CC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:22:45.707 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-02-15T20:22:45","Local":"2022-02-15T20:22:45","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+00:00","Sunrise":"06:58","Sunset":"17:10"}}
20:22:45.758 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-02-15T20:22:45","ESP32":{"Temperature":40.0},"SPM":{"Energy":[23.7,12.1,193.4,151.3,122.5,0.0,0.0,0.0,0.0,156.3,167.5,128.7,11.2,21.1,11.5,22.4,0.0,0.0,0.0,0.0],"Yesterday":[0.7,0.6,24.2,18.8,15.0,0.0,0.0,0.0,0.0,19.7,20.9,18.2,5.0,4.9,5.2,5.4,0.0,0.0,0.0,0.0],"Today":[0.0,0.0,24.8,19.3,15.4,0.0,0.0,0.0,0.0,20.2,21.4,18.7,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"ActivePower":[0.0,0.0,1212.0,958.1,747.6,0.0,0.0,0.0,0.0,991.0,1057.1,896.6,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"ApparentPower":[0.0,0.0,1321.5,946.4,790.4,0.0,0.0,0.0,0.0,1001.1,1055.9,905.6,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"ReactivePower":[0.0,0.0,45.2,52.1,47.9,0.0,0.0,0.0,0.0,45.8,23.3,12.9,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Factor":[0.00,0.00,0.92,1.00,0.95,0.00,0.00,0.00,0.00,0.99,1.00,0.99,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00],"Voltage":[0.0,0.0,233.3,233.8,234.4,234.1,0.0,0.0,233.9,233.9,233.1,233.1,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Current":[0.00,0.00,5.62,4.15,3.38,0.00,0.00,0.00,0.00,4.27,4.56,3.90,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00]},"TempUnit":"C"}}
20:22:45.773 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-02-15T20:22:45","Uptime":"0T00:54:32","UptimeSec":3272,"Heap":132,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"ON","POWER5":"ON","POWER6":"ON","POWER7":"OFF","POWER8":"OFF","POWER9":"ON","POWER10":"ON","POWER11":"ON","POWER12":"ON","POWER13":"OFF","POWER14":"OFF","POWER15":"OFF","POWER16":"OFF","POWER17":"OFF","POWER18":"OFF","POWER19":"OFF","POWER20":"OFF","Wifi":{"AP":1,"SSId":"Garage","BSSId":"B0:A7:B9:8D:18:D0","Channel":6,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

20:24:00.627 SPM: ESP AA55010000000000000000000000000016000D5164464E363536173536323600234A08
20:24:00.714 SPM: ARM AA55010000000000000000000000008016017E005164464E36353617353632360007E6020F07E508130001000000000501045F01130000000000...
[... 54 lines cut ...]
20:24:27.407 SPM: ESP AA550100000000000000000000000080060001001331BA
20:24:27.658 SPM: ESP AA5501000000000000000000000000001800105B64464E36353617353632360101003C33E809
20:24:27.702 SPM: ARM AA5501000000000000000000000000801800010033EB13
[... cut ...]
20:24:29.401 SPM: ESP AA5501000000000000000000000000001800105B64464E36353617353632360102003C346E48
20:24:29.538 SPM: ARM AA55010000000000000000000000008018000100342952
20:24:29.713 HTP: Console
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Plug 20 relays

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-02-15:
Pls enable weblog 3 and provide output of command sspmscan

arendst, 2022-02-15:
You also might want to upgrade to latest Dev v11.0.0.1 which has some fixes too

fegauthier, 2022-02-15:
I just compiled the latest pull from master branch.

There is the Console logs

```
00:00:00.002 HDW: ESP32-D0WD-V3 
00:00:00.003 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.082 UFS: FlashFS mounted with 292 kB free
00:00:00.176 CFG: Loaded from File, Count 12358
00:00:00.193 QPC: Count 1
00:00:00.195 CFG: CR 404/699, Busy 0
00:00:00.204 ROT: Mode 1
00:00:00.492 CFG: No '*.autoconf' file found
00:00:00.498 BRY: Berry initialized, RAM used=4089 bytes
00:00:00.522 BRY: No 'preinit.be'
00:00:00.526 SRC: Restart
00:00:00.531 Project tasmota - Tasmota Version 11.0.0.1(tasmota)-2_0_2_2(2022-02-15T16:17:07)
00:00:04.365 ETH: Attempting connection...
00:00:04.419 CFG: SPM Use defaults
00:00:04.594 CFG: SPM loaded from file
00:00:04.835 BRY: No 'autoexec.be'
00:00:05.135 WIF: Attempting connection...
00:00:05.621 WIF: Connecting to AP1 Garage Channel 6 BSSId B0:A7:B9:8D:18:D0 in mode 11n as tasmota-C781CC-0460...
00:00:05.682 WIF: Attempting connection...
00:00:07.476 SPM: Main version 1.0.0 found
00:00:07.484 SPM: Start relay scan...
00:00:07.633 WIF: Connected
00:00:08.013 RTC: UTC 2022-02-15T21:27:45, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
21:27:45.030 HTP: Web server active on tasmota-C781CC-0460 with IP address 192.168.10.5
21:27:45.726 SPM: Command 15 result 2
21:27:45.907 RSL: INFO1 = {"Info1":{"Module":"Sonoff SPM","Version":"11.0.0.1(tasmota)","FallbackTopic":"cmnd/DVES_C781CC_fb/","GroupTopic":"cmnd/tasmotas/"}}
21:27:45.923 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-C781CC-0460","IPAddress":"192.168.10.5"}}
21:27:45.927 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":33}}
21:27:45.950 TFS: File 'autoexec.bat' not found
21:27:46.283 HTP: Main Menu
21:27:48.901 QPC: Reset
21:27:50.856 APP: Boot Count 33
21:27:50.866 RSL: STATE = {"Time":"2022-02-15T21:27:50","Uptime":"0T00:00:09","UptimeSec":9,"Heap":137,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":44},"Wifi":{"AP":1,"SSId":"Garage","BSSId":"B0:A7:B9:8D:18:D0","Channel":6,"Mode":"11n","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:27:50.898 RSL: SENSOR = {"Time":"2022-02-15T21:27:50","ESP32":{"Temperature":38.9},"TempUnit":"C"}
21:27:51.751 CFG: Saved, Count 12359, Bytes 4096
21:28:05.975 SPM: 4Relay 1 (mapped to 0) type 130 version 1.0.0 found with id 5162464E3635361735363236
21:28:05.981 SPM: 4Relay 2 (mapped to 1) type 130 version 1.0.0 found with id 2018464E3635361435363236
21:28:05.996 SPM: 4Relay 3 (mapped to 2) type 130 version 1.0.0 found with id 2218464E3635361435363236
21:28:06.001 SPM: 4Relay 4 (mapped to 3) type 130 version 1.0.0 found with id 5B64464E3635361735363236
21:28:06.016 SPM: 4Relay 5 (mapped to 4) type 130 version 1.0.0 found with id 5164464E3635361735363236
21:28:06.020 SPM: WARNING 4Relay mapping possibly changed
21:28:07.025 CFG: Saved, Count 12360, Bytes 4096
21:28:07.087 SPM: Relay scan done
21:29:01.262 HTP: Consoles
21:29:02.495 HTP: Console
```

fegauthier, 2022-02-15:
I think I found something interesting.

SspmMap returns two time 1 in the array

```
21:33:24.309 CMD: Grp 0, Cmnd 'SSPMMAP', Idx 1, Len 0, Data ''
21:33:24.317 RSL: RESULT = {"SSPMMap":[1,1,2,3,4]}
```

arendst, 2022-02-16:
Yes that's the issue here. I had troiubles correctly detecting a new connected module as it can pop up in any order. So the message stating WARNING mapping changed is supposed to trigger you to remap the modules in the correct physical order.

Once you do that your issue should be fixed.

arendst, 2022-02-16:
Great,

in the meantime I think I solved the double 1 issue in your mapping. Try latest dev release and let's see what the future brings.

I'll close this issue as it is solved.

closed 2022-02-16 by arendst