arendst/Tasmota #14885: Servo Motor do not work with latest release.

mikep1998, 2022-02-17:
### PROBLEM DESCRIPTION
On ESP32 Servo motor worked with code version 2022.01.2. But with the latest code servo motor does not work. No error reported. It just does not move. using pwm1 on pin14 with setoption15 0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _ESP32 DOIT
- [x] Tasmota binary firmware version number used: Latest as of 2/16/2022
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
enter console commands:
PWM 40
PWM 70 
servo does not move.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Servo to move.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
May have something to do with the changes to esp32 pwm support.

**(Please, remember to close the issue when the problem has been addressed)**

mikep1998, 2022-02-17:
Now it is working??? Don't know why.

ascillato2, 2022-02-17:
May be some wiring issue?

closed 2022-02-17 by ascillato2

mikep1998, 2022-02-17:
Not sure what is going on. Right after I closed this issue it stopped working again. I have tried 2 esp32s, 2 servos and rewired the whole thing. 
I already have separate 7vdc power for the servo motor and the ESP32. The grounds are tied together. 
1 thing I do know is it will not run at 100 pwmfrequency like some have suggested. It must be 50 pwmfrequency. 
The only thing it could be is the control signal. I do not have an oscilloscope to look at the signal.